bdring/FluidNC #1066: Problem:

bwenquan, 2023-11-11:
### Wiki Search Terms

- esp32-s2
- esp32-s2 ipc

### Controller Board

[image]
[image]

### Machine Description

-

### Configuration file

```yaml
-
```

### Startup Messages

```text
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffESP-ROM:esp32s2-rc4-20191025
Build:Oct 25 2019
rst:0x7 (TG0WDT_SYS_RST),boot:0x8 (SPI_FAST_FLASH_BOOT)
Saved PC:0x4001276a
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
```

### What happened?

This error occurred when I compiled esp32-s2 `default_envs = wifi_s2`

[image]

bdring, 2023-11-11:
Why do you need to compile. 

Can you just install?

http://wiki.fluidnc.com/en/installation

bwenquan, 2023-11-11:
Because the firmware in the release was for the ESP32 (more accurately, it should be called the ESP32-S), the ESP32-S2 had a lot of differences from the ESP32, and the firmware of the two is not universal.

The key difference is that the esp32-wroom has two cores( Xtensa® 32-bit LX6 ), while the esp32-s2-wroom has only one core (Xtensa® 32-bit LX7 ):
[image]
[image]

So, I need to recompile the firmware using the chip environment of the esp32-s2.

bwenquan, 2023-11-11:
I followed the documentation to produce this result:

[image]

Now, I modified part of the code to fit ESP32-S2, and I will try to make a PR after test on my machine.
[image]

bdring, 2023-11-11:
I am familiar with that ESP32 chip and the other versions.

MitchBradley, 2023-11-11:
The S2 code is not complete and not supported.  We started working on it but have not had time to finish, because of other priorities.  If you can help us get it working that would be much appreciated.  In addition to getting the code working, the release scripts will need to be modified to include s2 versions of the binary files, and to list them in the manifest.json file so the web installer can see them.

MitchBradley, 2023-11-11:
Is the Kirizu Laser board a commercial product or something that you built yourself?

bwenquan, 2023-11-11:
Thanks, Kirizu Laser board is built by myself. 
I tried to modify the code, but I found that it runs on 2 core, I think it is not suitable to port to the single core ESP32-S2, I will re-modify my board to ESP32-WROOM.

This is my change:
1. Enable USB-CDC over USB Pin (Not Test)
```
-DARDUINO_USB_CDC_ON_BOOT=1
```
2.  Uart.cpp (line near 40)
```
        // We init UARTs on core 0 so the interrupt handler runs there,
        // thus avoiding conflict with the StepTimer interrupt
        esp_ipc_call_blocking(0, uart_driver_n_install, (void*)_uart_num);
```
changed to
```
    #ifdef CONFIG_IDF_TARGET_ESP32S2
        // ESP32-S2
        // single core
        
        uart_driver_n_install((void*)_uart_num);
    #elif
        // For: Esp32 / Esp32-S3
        // We init UARTs on core 0 so the interrupt handler runs there,
        // thus avoiding conflict with the StepTimer interrupt
        esp_ipc_call_blocking(0, uart_driver_n_install, (void*)_uart_num);
    #endif
```
3. MachineConfig.cpp (line near 40)
```
        handler.section("uart1", _uarts[1], 1);
        handler.section("uart2", _uarts[2], 2);
```
changed to (because esp32-s2 on has uart 0/1)
```
        for (size_t i = 1; i < MAX_N_UARTS; i++)
        {
            handler.section(("uart" + String(i)).c_str(), _uarts[i], i);
            handler.section(("uart_channel" +  String(i)).c_str(), _uart_channels[i]);
        }
```

Now, the firmware is initialized and running normally, but I guess there is a problem with the handler, there is a WiFi AP (it can be connected) but I can't access the WEB through it, and sending commands to the mcu through serial is not responding. Telnet also fails to connect.

bwenquan, 2023-11-25:
Hi, I spent some time to make this firmware to run at ESP32-S3. It's working now, but I haven't tested it on my machine yet.

There are my changes:

/FluidNC/esp32/PwmPin.cpp
```cpp
     // Increase the precision (bits) until it exceeds the frequency
     // The hardware maximum precision is 20 bits
     const uint8_t  ledcMaxBits = 20;
-    const uint32_t apbFreq     = 80000000;
+    const uint32_t apbFreq     = 40000000;
     const uint32_t maxCount    = apbFreq / frequency;
     for (uint8_t bits = 2; bits <= ledcMaxBits; ++bits) {
```
---
FluidNC/esp32/gpio.cpp
```cpp
-std::vector<int> avail_gpios = { 0, 1, 3, 4, 5, 12, 13, 14, 15, 16, 17, 18, 19, 21, 22, 23, 25, 26, 27, 32, 33, 34, 35, 36, 39 };
+std::vector<int> avail_gpios = { 0, 1, 3, 4, 5, 12, 13, 14, 15, 16, 17, 18, 19, 21, 22, 23, 25, 26, 27, 32, 33, 34, 35, 36, 39 ,40,41,42,43,44,45,46,47,48};

 struct pin_mux {
     int         pinnum;
@@ -240,6 +240,16 @@ struct pin_mux {
     { 37, "SENSOR_CAPP", { "GPIO37", "-", "GPIO37", "-", "-", "-" } },
     { 38, "SENSOR_CAPN", { "GPIO38", "-", "GPIO38", "-", "-", "-" } },
     { 39, "SENSOR_VN", { "GPIO39", "-", "GPIO39", "-", "-", "-" } },
+    { 40, "GPIO40", { "GPIO25", "-", "GPIO25", "-", "-", "EMAC_RXD0" } },
+    { 41, "GPIO41", { "GPIO25", "-", "GPIO25", "-", "-", "EMAC_RXD0" } },
+    { 42, "GPIO25", { "GPIO25", "-", "GPIO25", "-", "-", "EMAC_RXD0" } },
+    { 43, "GPIO25", { "GPIO25", "-", "GPIO25", "-", "-", "EMAC_RXD0" } },
+    { 44, "GPIO25", { "GPIO25", "-", "GPIO25", "-", "-", "EMAC_RXD0" } },
+    { 45, "GPIO25", { "GPIO25", "-", "GPIO25", "-", "-", "EMAC_RXD0" } },
+    { 46, "GPIO25", { "GPIO25", "-", "GPIO25", "-", "-", "EMAC_RXD0" } },
+    { 47, "GPIO25", { "GPIO25", "-", "GPIO25", "-", "-", "EMAC_RXD0" } },
+    { 48, "GPIO25", { "GPIO25", "-", "GPIO25", "-", "-", "EMAC_RXD0" } },
+
     { -1, "", { "" } },
 };
```
---
FluidNC/src/Pins/GPIOPinDetail.cpp
```cpp
     PinCapabilities GPIOPinDetail::GetDefaultCapabilities(pinnum_t index) {
         // See https://randomnerdtutorials.com/esp32-pinout-reference-gpios/ for an overview:
 #ifdef CONFIG_IDF_TARGET_ESP32S3
            switch(index){
                case 0:  // Outputs PWM signal at boot
                    return PinCapabilities::Native | PinCapabilities::Input | PinCapabilities::Output | PinCapabilities::PullUp |
                       PinCapabilities::PullDown | PinCapabilities::PWM | PinCapabilities::ISR |
                       PinCapabilities::UART;
                case 1:
                case 2:
                case 3:
                case 4:
                case 5:
                case 6:
                case 7:
                case 8:
                case 9:
                case 10:
                case 11:
                case 12:
                case 13:
                case 14:
                case 15:
                case 16:
                case 17:
                case 18:
                case 19:
                case 20:
[... 122 lines cut ...]
[MSG:INFO: X Axis driver test passed]
[MSG:INFO: Y Axis driver test passed]
[MSG:INFO: Kinematic system: Cartesian]
[MSG:INFO: Laser Ena:gpio.2:low:pu Out:gpio.38 Freq:25000Hz Period:1023]
[MSG:INFO: Using spindle Laser]
[MSG:INFO: Connecting to STA SSID:1703]
[MSG:INFO: Connecting.]
[MSG:INFO: Connecting..]
[MSG:INFO: Connected - IP is 192.168.36.64]
[MSG:INFO: WiFi on]
[MSG:INFO: Start mDNS with hostname:http://fluidnc.local/]
[MSG:INFO: SSDP Started]
[MSG:INFO: HTTP started on port 80]
[MSG:INFO: Telnet started on port 23]

Grbl 3.0 [FluidNC v3.0.x (noGit) (wifi) '$' for help]
[MSG:INFO: ALARM: Unhomed]
ALARM:14
[MSG:INFO: WebUI: Request from 192.168.36.211]
[MSG:INFO: WebUI: Request from 192.168.36.211]
```
[image]

Can I submit a PR? Is there anything to pay attention to here and could you help me check where I made a mistake?
Thanks.

MitchBradley, 2023-11-25:
You will need to find some way to make the changes conditional so the same code base can be used for both CPU variants.  Also the release scripts will need to change to build both variants.