bdring/FluidNC #1067: Update MKS DLC32 v2.1

FMD14, 2023-11-11:
### Wiki Search Terms

Update

### Controller Board

MKS DLC32 v2.1

### Machine Description

Laser Neje corexy

### Configuration file

```yaml
board: MKS-DLC32 V2.1
name: CoreXY Laser NJE 30W
meta: 02/11/2022 FMD14

kinematics:
  corexy:

stepping:
  engine: I2S_STATIC
  idle_ms: 0
  pulse_us: 4
  dir_delay_us: 1
  disable_delay_us: 0

axes:
  shared_stepper_disable_pin: I2SO.0
  x:
    steps_per_mm: 80
    max_rate_mm_per_min: 25000.000
    acceleration_mm_per_sec2: 3200.000
    max_travel_mm: 515.00
    soft_limits: true
    homing:
      cycle: 1
      positive_direction: false
      mpos_mm: 0.000
      feed_mm_per_min: 300.000
      seek_mm_per_min: 7000.000
      settle_ms: 525.00
      seek_scaler: 1.100
      feed_scaler: 1.100

    motor0:
      limit_neg_pin: gpio.35:low
      hard_limits: true
      pulloff_mm: 2.000
      stepstick:
        step_pin: I2SO.1
        direction_pin: I2SO.2:low

  y:
    steps_per_mm: 80
    max_rate_mm_per_min: 25000.000
    acceleration_mm_per_sec2: 3200.000
    max_travel_mm: 555.00
    soft_limits: true
    homing:
      cycle: 2
      positive_direction: false
      mpos_mm: 0.000
      feed_mm_per_min: 300.000
      seek_mm_per_min: 7000.000
      settle_ms: 500
      seek_scaler: 1.100
      feed_scaler: 1.100

    motor0:
      limit_neg_pin: gpio.36:low
      hard_limits: true
      pulloff_mm: 2.000
      stepstick:
        step_pin: I2SO.5
        direction_pin: I2SO.6:low

  z:
    steps_per_mm: 157.750
    max_rate_mm_per_min: 1200.000
    acceleration_mm_per_sec2: 500.000
    max_travel_mm: 80.000
    soft_limits: true
    homing:
      cycle: 0
      positive_direction: false
      mpos_mm: 0.000
      feed_mm_per_min: 300.000
      seek_mm_per_min: 1000.000
      settle_ms: 500
      seek_scaler: 1.100
      feed_scaler: 1.100

    motor0:
      limit_neg_pin: gpio.34:low
      hard_limits: false
      pulloff_mm: 1.000
      stepstick:
        step_pin: I2SO.3
        direction_pin: I2SO.4:low

i2so:
  bck_pin: gpio.16
  data_pin: gpio.21
  ws_pin: gpio.17

spi:
  miso_pin: gpio.12
  mosi_pin: gpio.13
  sck_pin: gpio.14

sdcard:
  cs_pin: gpio.15
  card_detect_pin: NO_PIN

control:
  safety_door_pin: NO_PIN
  reset_pin: NO_PIN
  feed_hold_pin: NO_PIN
  cycle_start_pin: NO_PIN
  macro0_pin: gpio.33:low:pu
  macro1_pin: NO_PIN
  macro2_pin: NO_PIN
  macro3_pin: NO_PIN

macros:
  startup_line0: 
  startup_line1: 
  macro0: $SD/Run=lasertest.gcode
  macro1: $SD/Run=home.gcode
  macro2:
  macro3:

coolant:
  flood_pin: NO_PIN
  mist_pin: gpio.04
  delay_ms: 0

probe:
  pin: gpio.22
  check_mode_start: true

Laser:
  pwm_hz: 5000
  #L on Beeper / IN on TTL
  output_pin: gpio.32
  enable_pin: I2SO.7
  disable_with_s0: false
  s0_with_disable: false
  tool_num: 0
  speed_map: 0=0.000%  1000=100.000%
# 135=0mA 270=5mA 400=10mA 700=16mA

user_outputs:
  analog0_pin: NO_PIN
  analog1_pin: NO_PIN
  analog2_pin: NO_PIN
  analog3_pin: NO_PIN
  analog0_hz: 5000
  analog1_hz: 5000
  analog2_hz: 5000
  analog3_hz: 5000
  digital0_pin: NO_PIN
  digital1_pin: NO_PIN
  digital2_pin: NO_PIN
  digital3_pin: NO_PIN

start:
  must_home: false

# 5,18,19,22,23,25,26,27,32,33,39,I2SO.7
# SDA 0 / SCL 4
```

### Startup Messages

```text
Unable to update
```

### What happened?

Unable to boot
[image]
choice 2
[image]
the message
how to do

bdring, 2023-11-11:
Try using the installer in the release files.
See this to about the boot button.
http://wiki.fluidnc.com/en/support/faq#when-loading-firmware-i-get-a-message-like-connecting-_______-and-it-eventually-times-out

FMD14, 2023-11-11:
You need to connect a button to the J5 reset connector
[image]

MitchBradley, 2023-11-11:
The problem is with the DLC32 board, not with the FluidNC installer software.  See this thread for a workaround.  https://github.com/makerbase-mks/MKS-DLC32/issues/19#issuecomment-978785658

closed 2023-11-11 by MitchBradley

ernst200, 2024-12-10:
Hi all
It is an old post but maybe it helps someone.

I just flashed fluidnc on a DLC32 V2.1 board.
I also got a warning "Booloader not active"

You have to put a jumper on de **middle 2**  I2C pins (SDA & GND) aka GPIO 0 & GND.
See picture above from FMD14
Then you can flash fluidnc on it. Works great.
Cheap and great little board

Uraharadono, 2026-02-06:
Posting this as well in hopes it helps someone in future. I am using MKS TinyBee.

 Connect wires as so: 

[image]

[image]

 So left pins on rows 2 and 5.
Hold reset button while connecting - or if that doesnt work release it just before installing FluidNC and it should work.

Uraharadono, 2026-02-07:
> It is extremely unlikely that someone who is having problems with Tinybee will find your post in an issue titled DLC32, especially after the issue is eventually closed. The place to put information so it might help someone is in the wiki, on the Tinybee page.

I completely agree with what you are saying, and it makes sense until you google this issue - even with MKS TinyBee in search string.
First result is this one: https://github.com/makerbase-mks/MKS-DLC32/issues/19
Second result is this very thread. 

I can delete this comment with no issues. Only thing I have from spending my time to comment here is to help someone in future.