bdring/FluidNC #1070: Feature: refined Homing Pull-Off for rotary axes

FrankNz, 2023-11-15:
### Machine Context

We a building a link trainer using FluidNC to control the axes. 
For our purposes xyz -> roll/pitch/yaw. Yaw being 'direction/heading'. The heading can rotate through 360 degrees.
Homing x/y no problem.
The Z axis has a sensor which is activated for 180 degrees. (approximately)

### Feature Description

Using the existing Homing, Pull-Off features and settings. 
My version of pseudo code below.
This would ensure that no matter where the direction (rotary) axis starts it will always find the limit switch transition being sort.

Set Pull-Off to 190degrees (assume steps/mm is read as Steps/Degree)
If Limit is not activated - 'normal' Homing.
If Limit is activated. 
 begin
   While Limit is Activated
        Case 
              Limit not Activated : Exit While
              Pull-Off range exceeded : Error
        end Case
        Continue Pul-Off
   end While
   'Normal' Homing
 End

Hope the above makes sense!
   
        

### Other Approaches

Currently using external code to achieve the above.

### How I Can Help

Sadly my programming skills would be more destructive than helpfull at present. 
Have bought a Jackpot control card from V1 Engineering. :)

MitchBradley, 2023-11-15:
Do you have a budget for this work?

FrankNz, 2023-11-15:
Sadly no, we are a bunch of retired people who volunteer at the Museum. 
I do realise the above is most likely a fringe case, so if it can be done it may be useful to others who are using rotary axes, if not, ohh well I did ask. :)

MitchBradley, 2023-11-15:
Only a tiny fraction of users have rotary axes.  4 and 5 axis CAM is rather expensive.

Have you tried the existing homing code with the index switch declared as limit_neg ?  I suspect that it might work.  The existing code will do an initial pulloff in the positive direction if there is a limit_neg switch (or in the negative direction if it is limit_pos).

FrankNz, 2023-11-15:
HI Mitch,
Thanks for taking the time to reply to me.

I agree with your first line comment.

I have tried your suggestion below. 
What seems to happen...
Pull-Off gets executed even if a limited switch is not activated (on reflection, not certain about this! Will test this weekend.)
Pull-off does not check for having cleared the limit switch during the process of pulling off, only at the end of Pull-Off. This would be the only change
In my case the Pull-Off can travel past the zone where the limit switch is not activated!  The switch is activated for half a circle and allowing for sensitivity errors the switch may be active for more than 180Deg and a Pull-off of 180 to try get into a non-active zone could fail.

That is fundamentally the change I was suggesting in my post. 

As an advantage it would not affect the current homing process either.

I do understand it is a low priority suggestion, so may never get addressed.

Frank

MitchBradley, 2023-11-15:
The initial pulloff only happens if the switch is activated at the beginning of homing.  The machine model is that the switch activation distance is a small fraction of the total travel.  We are unlikely to change that given that, as far as we know, you are the only person with a setup that violates that assumption.  We have more work to do than we have time for, by a large margin.

MitchBradley, 2023-11-15:
> The initial pulloff only happens if the switch is activated at the beginning of homing. The machine model is that the switch activation distance is a small fraction of the total travel. We are [...]

Every change has a high probability of breaking something.  It is the nature of complex software used by many people with widely varying setups.

FrankNz, 2023-11-15:
Hi Mitch,Thanks for the reply and thanks for your time.I understand your situation.FrankSent from my iPhone

RDS356, 2023-11-24:
Hi..
What kind of sensor are you using ?
I have a machine with rotary axis with homig, but in this axis the sensor is optical. Works perfectly.
Is this your problem ?

[image]
[image]

FrankNz, 2023-11-24:
HI,

Briefly, we as building a replica of a Link Trainer (AKA Blue Box Trainer) using xyz axis as Roll(Bank) - Pitch(climb/dive) - Yaw (Direction).
Using Proximity switches on all axes.
Have the steps/mm set such that it equates to steps/deg
The XY axes have +/- limits switches to stop the worm drives breaking things! Homing/Limiting on these is working just fine. :)
The worm drive for the z axis comes from a ships radar! (100:1)
The z has a cam, welded to the axis, that causes the limit switch to be active for about 180deg.  
This was (is) a mistake. If it was active for a couple of degrees the system would (should?) work. 
Having read and thought about your setup from the image. The 'main guy(John)’ and I will see if there is a way we can put a much shorter ‘active zone’ sensor in place. We can access the other end of the yaw drive.
Thank you for triggering the thought.

Frank

￼

MitchBradley, 2023-11-24:
In principle it should be easy to attach something to the z axis that would interrupt a photosensor.  3d print something, or clamp on a stick, or ...

FrankNz, 2023-11-24:
You are right, we are going to see if we have sufficient access to the lower shaft.

FrankNz, 2023-11-25:
Right, a suitable narrow tab to trigger the proximity sensor works. So homing now works.
Thanks for the clues.
Please consider this matter closed.

Frank

closed 2023-11-25 by MitchBradley