bdring/FluidNC #1073: Problem: 0-10V Spindle and dust control trigger for a few seconds at startup

ColPaulR, 2023-12-02:
### Wiki Search Terms

I2SO, spindle, mist

### Controller Board

MKS Tinybee with external interface circuitry 

### Machine Description

Gantry router/laser. Queenbee Pro 1500x1500 with 2.2kw water cooled spindle, diode laser, and dust collection system. 

### Input Circuits

```yaml
Problem is with output circuitry.
```

### Configuration file

```yaml
board: MKS TinyBee V1.0_001
name: CNC Router XYYZA 0-10V spindle and laser
meta: (01.02.2022) by Skorpi

kinematics:
  Cartesian:

stepping:
  engine: I2S_STATIC
  idle_ms: 255
  pulse_us: 4
  dir_delay_us: 1
  disable_delay_us: 0

axes:
  x:
    # X
    steps_per_mm: 250
    max_rate_mm_per_min: 7500.000
    acceleration_mm_per_sec2: 200.000
    max_travel_mm: 1300.000
    soft_limits: true
    homing:
      cycle: 3
      positive_direction: true
      mpos_mm: 1265.00
      feed_mm_per_min: 300.000
      seek_mm_per_min: 2000.000
      settle_ms: 250
      seek_scaler: 1.100
      feed_scaler: 1.100

    motor0:
      limit_neg_pin: GPIO.33:low
      hard_limits: true
      pulloff_mm: 2.500
      standard_stepper:
        step_pin: I2SO.1
        direction_pin: I2SO.2:low
        disable_pin: I2SO.0

  y:
    # Y
    steps_per_mm: 250
    max_rate_mm_per_min: 7500.000
    acceleration_mm_per_sec2: 200.000
    max_travel_mm: 1300.000
    soft_limits: true
    homing:
[... 552 lines cut ...]
$WiFi/Mode=STA>AP
$Sta/SSID=RounsavallWPA2
$Sta/Password=******
$Sta/MinSecurity=WPA2-PSK
$WiFi/FastScan=OFF
$Sta/IPMode=Static
$Sta/IP=192.168.0.235
$Sta/Gateway=192.168.0.1
$Sta/Netmask=255.255.255.0
$AP/Country=01
$AP/SSID=FluidNC
$AP/Password=******
$AP/IP=192.168.0.1
$AP/Channel=1
$Hostname=fluidnc
$HTTP/BlockDuringMotion=ON
$HTTP/Enable=ON
$HTTP/Port=80
$Telnet/Enable=ON
$Telnet/Port=23
$Notification/Type=NONE
$Notification/T1=
$Notification/T2=
$Notification/TS=
ok

bdring, 2023-12-02:
I2S chips have an indeterminate start state.

http://wiki.fluiydnc.com/en/support/controller_design_guidelines#i2so-chips

MitchBradley, 2023-12-02:
This is the latest Tinybee schematic from the MKS Github page at https://github.com/makerbase-mks/MKS-TinyBee/blob/main/hardware/MKS%20TinyBee%20V1.0_003/MKS%20TinyBee%20V1.0_003%20SCH.pdf
[image]
As you can see, the OE_ pin is hardwired to GND, so there is no high-Z behavior at startup.  Furthermore, the chip reset SRCLR_ is pulled high so there is no hardware reset, unlike on Bart's boards where we added a capacitor to hold the reset low (active) briefly when the power first comes on, thus forcing the I2SO outputs to be initially low, at least on a power on reset.  Here is the 6 pack circuit

[image]

FluidNC tries to set all the I2SO outputs to 0 as soon as it has discovered, via the config file, that I2SO is present on the board.  There is not a lot that the software can do when the Espressif bootloader is running, as FluidNC software does not have control of the machine then.

Perhaps you might be able to find some Tinybee pin that is at a predictable state during the problem period, and add external logic circuitry to use it disable the problem outputs then.  This is a situation over which the FluidNC firmware has very little control.  Any tricks that we might try to do to mitigate your specific situation would risk having the opposite effect on someone else.

ColPaulR, 2023-12-02:
Thank you so much for the rapid response and the detailed explanation that my choice of control is the issue, based on their poor design choices. I'm considering using the 2nd channel of my external dual relay module as an enable signal for the dust collection solid state relay and the spindle FWD relay. I'll drive this "enable relay" in the opposite polarity as the FWD signal. If the I2SO starts up all high as it appears to do until the ESP32 can assert control then "enable relay" will prevent the signals from activating the hardware. 

Feel free to close this issue. However, I'll try to post if my new configuration mitigates the reset issue.

closed 2023-12-02 by ColPaulR

ColPaulR, 2023-12-26:
I overcame this (or at least greatly reduced the frequency of occurrence) by using both channels of my dual relay module. I wired both channels of the dual relay module in series with the FWD and DCM signals (e.g. VFD FWD to COM1, NO1 to COM2 and NO2 to VFD DCM) and activate the two channels with opposite polarity outputs from the two I2SO pins (I used forward_pin: I2SO.21 and enable_pin: I2SO.22). As long as all the output pins initial to either high or low, the spindle will not spin up during ESP32 startup.