bdring/FluidNC #1075: No Movement

kuito, 2023-12-06:
### Wiki Search Terms

motors not moving
faq motors not moving
wiring correctly motors not moving

### Controller Board

https://amzn.to/3uS7FIF -- i realize you guys do not like these types of boards, but I have gotten it to work with other firmware so I know the board isn't it.. plus i've thoroughly tested all three boards

### Machine Description

I am using an X-Carve Pro as the actual machine (4ftx4ft--XYYZ), but I am using the above esp32, dev board, dm542t (drivers), and nema32 motors, XYYZ.  48v power supply powering only the drivers/motors, and the esp32 is being powered via usb-c. 

I don't have photos right this second but I can definitely get some if need be.

### Configuration file

```yaml
board: ESP32 TERM ADAPTER
name: GRFN_PRJX_ESP32_TERM_ADTR_XYYZ-PWM-neu
meta: Customized Config
stepping:
  engine: RMT
  idle_ms: 255
  pulse_us: 10
  dir_delay_us: 1
  disable_delay_us: 0

axes:
  shared_stepper_disable_pin: gpio.17:low
  x:
    steps_per_mm: 42.353
    max_rate_mm_per_min: 7620.000
    acceleration_mm_per_sec2: 250.000
    max_travel_mm: 1219.000
    soft_limits: false
    homing:
      cycle: 2
      feed_mm_per_min: 25.000
      seek_mm_per_min: 750.000
      settle_ms: 500
      seek_scaler: 1.100
      feed_scaler: 1.100

    motor0:
      limit_neg_pin: gpio.35
      limit_pos_pin: NO_PIN
      limit_all_pin: NO_PIN
      hard_limits: false
      pulloff_mm: 2.000
      standard_stepper:
        step_pin: gpio.12
        direction_pin: gpio.14

  y:
    steps_per_mm: 42.353
    max_rate_mm_per_min: 7620.000
    acceleration_mm_per_sec2: 350.000
    max_travel_mm: 1219.000
    soft_limits: false
    homing:
      cycle: 2
      feed_mm_per_min: 25.000
      seek_mm_per_min: 750.000
      settle_ms: 500
      seek_scaler: 1.100
      feed_scaler: 1.100

    motor0:
      limit_neg_pin: gpio.34
      limit_pos_pin: NO_PIN
      limit_all_pin: NO_PIN
      hard_limits: false
      pulloff_mm: 1.000
      standard_stepper:
        step_pin: gpio.27
        direction_pin: gpio.26

    motor1:
      limit_neg_pin: gpio.36
      limit_pos_pin: NO_PIN
      limit_all_pin: NO_PIN
      hard_limits: false
      pulloff_mm: 1.000
      standard_stepper:
        step_pin: gpio.33
        direction_pin: gpio.32

  z:
    steps_per_mm: 200.000
    max_rate_mm_per_min: 300.000
    acceleration_mm_per_sec2: 100.000
    max_travel_mm: 50.000
    soft_limits: false
    homing:
      cycle: 1
      feed_mm_per_min: 100.000
      seek_mm_per_min: 200.000
      settle_ms: 500
      seek_scaler: 1.100
      feed_scaler: 1.100

    motor0:
      limit_neg_pin: gpio.39
      limit_pos_pin: NO_PIN
      limit_all_pin: NO_PIN
      hard_limits: false
      pulloff_mm: 1.000
      standard_stepper:
        step_pin: gpio.15
        direction_pin: gpio.2

spi:
  miso_pin: gpio.19
  mosi_pin: gpio.23
  sck_pin: gpio.18

# sdcard:
#   card_detect_pin: NO_PIN
#   cs_pin: gpio.5

# probe:
#   pin: gpio.22:low
#   check_mode_start: true

control:
  safety_door_pin: NO_PIN
  reset_pin: 
  feed_hold_pin: 
  cycle_start_pin:
  estop_pin:

macros:
  startup_line0:
  startup_line1:
  macro0:
  macro1:
  macro2:
  macro3:

start:
  must_home: false

coolant:
  mist_pin: gpio.21
  delay_ms: 0

PWM:
  pwm_hz: 5000
  output_pin: gpio.25
  enable_pin: gpio.13
  direction_pin: gpio.4
  disable_with_s0: false
  s0_with_disable: true
  spinup_ms: 30000
  spindown_ms: 10000
  tool_num: 0
  speed_map: 0=0% 0=25% 6000=25% 24000=100%
```

### Startup Messages

```text
rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:1184
load:0x40078000,len:13220
ho 0 tail 12 room 4
load:0x40080400,len:3028
entry 0x400805e4

[MSG:INFO: FluidNC 3.7.10 https://github.com/bdring/FluidNC]
[MSG:INFO: Compiled with ESP32 SDK:v4.4.4]
[MSG:INFO: Local filesystem type is littlefs]
[MSG:INFO: Configuration file:esp32_term_adapter_xyyz-neu.yml]
[MSG:INFO: Machine GRFN_PRJX_ESP32_TERM_ADTR_XYYZ-PWM-neu]
[MSG:INFO: Board ESP32 TERM ADAPTER]
[MSG:INFO: SPI SCK:gpio.18 MOSI:gpio.23 MISO:gpio.19]
[MSG:INFO: Stepping:RMT Pulse:10us Dsbl Delay:0us Dir Delay:1us Idle Delay:255ms]
[MSG:INFO: Axis count 3]
[MSG:INFO: Shared stepper disable gpio.17:low]
[MSG:INFO: Axis X (-1219.000,0.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     standard_stepper Step:gpio.12 Dir:gpio.14 Disable:NO_PIN]
[MSG:INFO:  X Neg Limit gpio.35]
[MSG:INFO: Axis Y (-1219.000,0.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     standard_stepper Step:gpio.27 Dir:gpio.26 Disable:NO_PIN]
[MSG:INFO:  Y Neg Limit gpio.34]
[MSG:INFO:   Motor1]
[MSG:INFO:     standard_stepper Step:gpio.33 Dir:gpio.32 Disable:NO_PIN]
[MSG:INFO:  Y2 Neg Limit gpio.36]
[MSG:INFO: Axis Z (-50.000,0.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     standard_stepper Step:gpio.15 Dir:gpio.2 Disable:NO_PIN]
[MSG:INFO:  Z Neg Limit gpio.39]
[MSG:INFO: Kinematic system: Cartesian]
[MSG:INFO: PWM Spindle Ena:gpio.13 Out:gpio.25 Dir:gpio.4 Freq:5000Hz Period:8191]
[MSG:INFO: Using spindle PWM]
[MSG:INFO: Mist coolant gpio.21]
[MSG:INFO: Connecting to STA SSID:icee]
[MSG:INFO: Connecting.]
[MSG:INFO: Connecting..]
[MSG:INFO: Connecting...]
[MSG:INFO: Connected - IP is 192.168.1.189]
[MSG:INFO: WiFi on]
[MSG:INFO: Start mDNS with hostname:http://fluidnc.local/]
[MSG:INFO: SSDP Started]
[MSG:INFO: HTTP started on port 80]
[MSG:INFO: Telnet started on port 23]

Grbl 3.7 [FluidNC 3.7.10 (wifi) '$' for help]
```

### User Interface Software

FluidTerm, WebUI

### What happened?

I have it rigged up for only 1 motor (x/motor0)
- I plug the esp32 on
- power on the power supply
- all the lights look good
- and then I run FluidTerm
- select 2 to connect via the usb/serial
- I see/read the Start Messages which look find to me.  
- Then I try:  $md (which locks the motors) 
- Then $me (which allows you to move the motor with your fingers)
- Then I attempt X10 or X-10
- I then take a multimeter to see if the dir pin changes based on X10 vs X-10 which it does.

I've re-wired, I've tested continuity to make sure the wires are ok and connected.

I have then loaded grbl on a separate card and swap the wires for the same motor, connect this card, and then use OB CONTROL to connect and am able to do X10 without an issue.

Again, I have tried the exact setup except with different firmware and with an arduino uno with old school grbl and everything works as expected.

### GCode File

NA

### Other Information

I've done so much troubleshooting, and have really tried to figure this out on my own, but it is quite disheartening with how much time i've spent and then throw together basic grbl and it works np.  So I am probably missing something somewhere, and I am hoping you guys can help me out, but if you are too busy, no worries.

Ultimately, I would love to do a youtube video on switching the X-Carve Pro from their proprietary mega uno with built in drivers to a controller->drivers->motors system preferably with fluidnc cuz I want to be able to add a lot more to this machine eventually.

Thank you.

kuito, 2023-12-06:
Here is the XCP grbl settings for the machine as provided by inventables (if it helps):

> $132=130.000
> $131=1219.000
> $130=1219.000
> $122=500.000
> $121=350.000
> $120=250.000
> $112=2540.000
> $111=7620.000
> $110=7620.000
> $102=160.000
> $101=42.353
> $100=42.353
> $40=0
> $32=0
> $31=0
> $30=24000
> $27=2.000
> $26=250
> $25=750.000
> $24=25.000
> $23=3
> $22=1
> $21=0
> $20=0
> $13=0
> $12=0.002
> $11=0.020
> $10=3
> $6=0
> $5=0
> $4=0
> $3=14
> $2=0
> $1=255
> $0=10

bdring, 2023-12-06:
It sounds like you are doing a decent job at troubleshooting, but I did not hear anything about verifying the step signals. That might be the problem.

We (the developers) generally don't like to help with DIY hardware, because it always ends up being a hardware issue and not the firmware.

If you are using external drivers, I recommend using a controller specifically designed for that. (5V vs the 3.3V of the ESP32)

http://wiki.fluidnc.com/en/hardware/existing_hardware

If you get it working, you should make the video. If I made it, it would be quite awkward ;-)

kuito, 2023-12-06:
Thanks for the quick response.

I did check the step pin (like the dir pin) with continuity which is one level and then when typing X100 in fluidterm it does increase then return to the prior value upon the command finishing.  

I am fairly new to electronics troubleshooting, but it does seem to be doing *something* as far as the multimeter shows.

originally I was just going to buy the openbuilds blackbox, but then thought “this would probably be better for replacing a burnt out driver” and little did I know I’d be in for a world of humbling lol.  That was the problem I kept having with the inventables card was a driver would fail and then I’d have to replace the whole damn controller box. 

At any rate, I saw you might be close to Chicago?  Got any extra fluidnc cards for external drivers laying around wherever you are at?  Or wanna come hang out at my shop in Chicago? :)

MitchBradley, 2023-12-06:
If MD locks the motors and ME unlocks them, you have the logic sense of the enable pin backwards.  MD is Motor *DIS*able; ME is Motor *EN*able

MitchBradley, 2023-12-06:
Furthermore, it is important to realize that the input marked ENA on DM542 drivers is really "disable", in the sense that, if you leave the input unconnected, the driver is enabled.  Conversely, if you apply a voltage from ENA+ to ENA-, thus making current flow through the optocoupler inside the driver, that will disable the driver.  Yes, this is very confusing labeling.  Blame China.

kuito, 2023-12-06:
So is the solution to add remove :low from shared_stepper_disable_pin?

kuito, 2023-12-06:
Also adding the exact drivers I’m using is:  https://www.omc-stepperonline.com/digital-stepper-driver-1-0-4-2a-20-50vdc-for-nema-17-23-24-stepper-motor-dm542t

MitchBradley, 2023-12-06:
That is the first thing I would try.  I cannot say for sure that it *is* the solution because you have told us nothing at all about how you have wired your system.  You provided links to the components but did not tell us how you hooked things up.

MitchBradley, 2023-12-06:
SteppersOnline is generally reputable, but with unbranded stuff from China, there is no way to tell what is inside.  There have been reports of stuff that is labeled "DM542" that actually has TB6600 chips inside, or even worse.

When we help people with custom hardware, we are usually shooting in the dark, and our effort helps exactly one person, and they rarely contribute to the project by donating or helping in other ways.

kuito, 2023-12-06:
Always up for helping out any way I can with the project.  I am going to eventually make a video on switching the proprietary xcarve pro all-in-one control board with what I’ve described.

I’ve hooked up all the components with Amazon-quality wire (https://amzn.to/46EX80Q), I pig tail the grounds on the driver to one ground pin.  I’ve used both twist wire caps and now Waco connectors.

Honestly, I’ve re-wired this more than I care to admit in nearly every possible way I can think of.  Like I mentioned, I have used the same wires along with driver, power supply, etc with old school grbl (1.1h) and using arduino uno.

I think maybe one of the issues could be something Bart alluded to about the current 3.3v signal output vs 5v — so I ordered logic level converters (https://amzn.to/48mFikH) and this octocoupler (https://amzn.to/481IHVz).

as you might be able to tell, I’m a glutton for punishment and pretty new to this stuff, but I rarely give up. :)

MitchBradley, 2023-12-06:
Those level converters are useful for high-impedance voltage signaling between CMOS chips on a board.  They are worse than useless for driving optocouplers.

Just disconnect the ENA lines entirely and see what happens.

MitchBradley, 2023-12-06:
Your GRBL setting for the enable is backwards from your FluidNC setting.  $4=0 means active high (do not invert), whereas shared_stepper_disable_pin: gpio.17:low means active low (do invert).

Photos would have helped.  That is why we asked for them.  You said that you could get them, but we have spent a lot of time in the interim talking about stuff that would have been answered immediately by them.

kuito, 2023-12-06:
Sorry about the photos, I posted when I got home from my workshop so I did not/do not have access to take them right now.  I will be back there early tomorrow am to try removing the :low and see how that goes.

I am always reticent to post asking for help before exhausting all options — I’ll cancel my order on the logic level converter for now — I appreciate your help, and will report back tomorrow.

kuito, 2023-12-18:
Hey there,
I want to provide an update — the :low on the “ shared_stepper_disable_pin” seemed to be the problem because after removing it or setting to :high this happened:

https://github.com/bdring/FluidNC/assets/5720347/32ff6ca1-f3e2-4602-ae79-7173842ba777

im going to add a couple photos of things to for others that have a X-Carve Pro and find this issue to show how I’ve retrofit things.
[image]
[image]
[image]
[image]
[image]

I have a lot more photos and videos because I plan on uploading a YouTube video on how to convert the xcarve pro to a esp32+drivers, etc.

I’m going to tackle the spindle which looks like a H100.  Not sure what the SCHRACK SR4 is really used for but will also figure that out at some point.

I also want to make their HMI + estop work too.

closed 2023-12-18 by MitchBradley