bdring/FluidNC #1081: Add configuration for Atomstack Maker A5 V2

darfij, 2023-12-10:
### Machine Context

This is a Atomstack Maker A5 V2 machine that uses a ESP32 controller
It has x and y endstops and a non motorised Z axis
The motherboard is a LS ESP32 V1.0 
(similar to LASERBOX ESP32 V1.0 and LASER 32-A5 V1.0 but the x and y axis drivers are soldered in, there is a plug space for a Z axis driver, and it also has a RJ11 display connector)

### Feature Description

Need a example configuration .yaml for this engraver,

I uploaded fluidnc v3.7.11 without problems and can use the command line without problems.

Tested in LaserGRBL and the fluidnc webGUI the engraver will not move or home the axis, but the laser works and the X home switch also ($limits).
I tested all (3) MKS_DLC32 available example configurations with the same result:
$H
<Home|MPos:-944.500,-784.500,0.000|FS:0,0>
[MSG:INFO: ALARM: Homing Fail Approach]
ALARM:9
Homing fail. Could not find limit switch within search distances. Try increasing max travel, decreasing pull-off distance, or check wiring.
ok
<Alarm|MPos:-944.500,-784.500,0.000|FS:0,0>

attached:
-console boot report:
[console_at_boot.txt](https://github.com/bdring/FluidNC/files/13627189/console_at_boot.txt)

-currently used config.yaml (adapted MKS_DLC32)
[config.yaml.txt](https://github.com/bdring/FluidNC/files/13627195/config.yaml.txt)

### Other Approaches

I tested all (3) MKS_DLC32 available example configurations but none work with same or similar results

### How I Can Help

I could make and test the config for this machine if someone can give me the pinouts and maybe some help

darfij, 2023-12-22:
I was able to reverse engineer the pins of the board and I now have a fully functional FluidNC configuration.
I will open a pull request.

board: LS ESP32 V1.1
name: Atomstack Maker A5 V2
meta: by D.F.
stepping:
  engine: RMT
  idle_ms: 255
  pulse_us: 2
  dir_delay_us: 1
  disable_delay_us: 0

axes:
  shared_stepper_disable_pin: gpio.33
  x:
    steps_per_mm: 80.000
    max_rate_mm_per_min: 10000.000
    acceleration_mm_per_sec2: 500.000
    max_travel_mm: 300.000
    soft_limits: false
    homing:
      cycle: 2
      feed_mm_per_min: 1000.000
      seek_mm_per_min: 5000.000
      positive_direction: false
      settle_ms: 500
      seek_scaler: 1.10036
      feed_scaler: 1.100

    motor0:
      limit_neg_pin: gpio.36:low
      limit_pos_pin: NO_PIN
      limit_all_pin: NO_PIN
      hard_limits: false
      pulloff_mm: 2.000
      standard_stepper:
        step_pin: gpio.25
        direction_pin: gpio.26:low

  y:
    steps_per_mm: 80.000
    max_rate_mm_per_min: 10000.000
    acceleration_mm_per_sec2: 500.000
    max_travel_mm: 300.000
    soft_limits: false
    homing:
      cycle: 2
      feed_mm_per_min: 1000.000
      seek_mm_per_min: 5000.000
      positive_direction: false
      settle_ms: 500
      seek_scaler: 1.100
      feed_scaler: 1.100

    motor0:
      limit_neg_pin: gpio.39:low
      limit_pos_pin: NO_PIN
      limit_all_pin: NO_PIN
      hard_limits: false
      pulloff_mm: 2.000
      standard_stepper:
        step_pin: gpio.27
        direction_pin: gpio.05

#  z:
#    steps_per_mm: 80.000
#    max_rate_mm_per_min: 10000.000
#    acceleration_mm_per_sec2: 500.000
#    max_travel_mm: 300.000
#    soft_limits: false
#    homing:
#      cycle: 2
#      feed_mm_per_min: 1000.000
#      seek_mm_per_min: 5000.000
#      positive_direction: false
#      settle_ms: 500
#      seek_scaler: 1.100
#      feed_scaler: 1.100
#
#    motor0:
#      limit_neg_pin: NO_PIN
#      limit_pos_pin: NO_PIN
#      limit_all_pin: NO_PIN
#      hard_limits: false
#      pulloff_mm: 1.000
#      standard_stepper:
#        step_pin: gpio.18
#        direction_pin: gpio.19

spi:
  miso_pin: gpio.12
  mosi_pin: gpio.13
  sck_pin: gpio.14

sdcard:
  card_detect_pin: gpio.35
  cs_pin: gpio.15

probe:
  pin: gpio.22:low
  check_mode_start: true

control:
  safety_door_pin: NO_PIN

macros:
  startup_line0:
  startup_line1:
  macro0:
  macro1:
  macro2:
  macro3:

start:
  must_home: false

coolant:
  flood_pin: NO_PIN
  mist_pin: NO_PIN
  delay_ms: 0

Laser:
  pwm_hz: 5000
  #L on Beeper / IN on TTL
  output_pin: gpio.32
  enable_pin: I2SO.7
  disable_with_s0: false
  s0_with_disable: false
  tool_num: 0
  speed_map: 0=0.000% 0=12.500% 1700=100.000%
#  speed_map: 0=0.000% 1000=100.000%
# 135=0mA 270=5mA 400=10mA 700=16mA
user_outputs:
  analog0_pin: NO_PIN
  analog1_pin: NO_PIN
  analog2_pin: NO_PIN
  analog3_pin: NO_PIN
  analog0_hz: 5000
  analog1_hz: 5000
  analog2_hz: 5000
  analog3_hz: 5000
  digital0_pin: NO_PIN
  digital1_pin: NO_PIN
  digital2_pin: NO_PIN
  digital3_pin: NO_PIN

MitchBradley, 2023-12-22:
Please note that user contributed configs have been moved to a separate repo https://github.com/bdring/fluidnc-config-files, so your PR should target that.

MitchBradley, 2023-12-22:
In particular, you should use the contributed/ folder therein.

darfij, 2023-12-22:
posted a PR

bdring, 2024-02-03:
PR Merged

closed 2024-02-03 by bdring

killthekid, 2024-04-11:
@darfij - i successfully installed fluidnc 3.7.16 on the LS ESP32 V1.1.
I exabled Z-Axis, added a TMC2208 V3 and successfully enabled a additional stepper Motor for Z Axis.

What i now try do archive is connect a limitswitch to the Z-Axis limit switch port.
Somehow this does not work, even i physically bridged both pins (not a hardware issue so far)

Do you have any clue, how i can enable the limit switch port for Z Axis, or which configuration should be considered?

Thanks in advance!

darfij, 2024-04-11:
Hi killthekid
Sorry I can't be of much help.
I did not use the Z axis or it's end stop.
I do seem to recall that I could not trace the end stop pin to a esp32 pin, but did not try hard.

darfij, 2024-04-11:
Perhaps (depending on the software you use) you could set-up a soft limit for the Z?

killthekid, 2024-04-11:
Hej @darfij,
this means to set `soft_limits: false` in Z in the config of fluidnc?
Or do i have to set this up, i.e. in LightBurn?

What's behind soft_limit ? i'm pretty new to cnc... :/

darfij, 2024-04-12:
Yes Have not tried it my self

Yahoo Mail: Search, Organize, Conquer

EduardoTesla, 2025-04-08:
Hola @darfij
El archivo YAML sera compatible con  Laser 32-A5 V1.0

[image]

Desconfigure la tarjeta y solo pude volver al firmware original, pero me gustaria usar FLuidNC con esta tarjeta

Saludos