bdring/FluidNC #1082: Problem: Issuing the "disable echo" command trigger error 9 if in alarm state

breiler, 2023-12-10:
### Wiki Search Terms

N/A

### Controller Board

ESP32  CNC Dev Controller v4.1

### Machine Description

Just the controller board without any periphials with default configuration.

### Input Circuits

```yaml
N/A
```

### Configuration file

```yaml
None, using the default one.
```

### Startup Messages

```text
rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:1184
load:0x40078000,len:13220
ho 0 tail 12 room 4
load:0x40080400,len:3028
entry 0x400805e4

[MSG:INFO: FluidNC v3.7.11 https://github.com/bdring/FluidNC]
[MSG:INFO: Compiled with ESP32 SDK:v4.4.4]
[MSG:INFO: Local filesystem type is littlefs]
[MSG:WARN: Cannot open configuration file:config.yaml]
[MSG:INFO: Using default configuration]
[MSG:INFO: Axes: using defaults]
[MSG:INFO: Machine Default (Test Drive)]
[MSG:INFO: Board None]
[MSG:INFO: Stepping:RMT Pulse:4us Dsbl Delay:0us Dir Delay:0us Idle Delay:255ms]
[MSG:INFO: Axis count 3]
[MSG:INFO: Axis X (-1000.000,0.000)]
[MSG:INFO:   Motor0]
[MSG:INFO: Axis Y (-1000.000,0.000)]
[MSG:INFO:   Motor0]
[MSG:INFO: Axis Z (-1000.000,0.000)]
[MSG:INFO:   Motor0]
[MSG:INFO: Kinematic system: Cartesian]
[MSG:INFO: Using spindle NoSpindle]
[MSG:INFO: Connecting to STA SSID:Murkelslottet]
[MSG:INFO: Connecting.]
[MSG:INFO: Connecting..]
[MSG:INFO: Connecting...]
[MSG:INFO: Connecting....]
[MSG:INFO: Connecting.]
[MSG:INFO: Connected - IP is 192.168.1.10]
[MSG:INFO: WiFi on]
[MSG:INFO: Start mDNS with hostname:http://fluidnc.local/]
[MSG:INFO: SSDP Started]
[MSG:INFO: HTTP started on port 80]
[MSG:INFO: Telnet started on port 23]

Grbl 3.7 [FluidNC v3.7.11 (wifi) '$' for help]
[MSG:INFO: ALARM: Unhomed]rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:1184
load:0x40078000,len:13220
ho 0 tail 12 room 4
load:0x40080400,len:3028
entry 0x400805e4

[MSG:INFO: FluidNC v3.7.11 https://github.com/bdring/FluidNC]
[MSG:INFO: Compiled with ESP32 SDK:v4.4.4]
[MSG:INFO: Local filesystem type is littlefs]
[MSG:WARN: Cannot open configuration file:config.yaml]
[MSG:INFO: Using default configuration]
[MSG:INFO: Axes: using defaults]
[MSG:INFO: Machine Default (Test Drive)]
[MSG:INFO: Board None]
[MSG:INFO: Stepping:RMT Pulse:4us Dsbl Delay:0us Dir Delay:0us Idle Delay:255ms]
[MSG:INFO: Axis count 3]
[MSG:INFO: Axis X (-1000.000,0.000)]
[MSG:INFO:   Motor0]
[MSG:INFO: Axis Y (-1000.000,0.000)]
[MSG:INFO:   Motor0]
[MSG:INFO: Axis Z (-1000.000,0.000)]
[MSG:INFO:   Motor0]
[MSG:INFO: Kinematic system: Cartesian]
[MSG:INFO: Using spindle NoSpindle]
[MSG:INFO: Connecting to STA SSID:Murkelslottet]
[MSG:INFO: Connecting.]
[MSG:INFO: Connecting..]
[MSG:INFO: Connecting...]
[MSG:INFO: Connecting....]
[MSG:INFO: Connecting.]
[MSG:INFO: Connected - IP is 192.168.1.10]
[MSG:INFO: WiFi on]
[MSG:INFO: Start mDNS with hostname:http://fluidnc.local/]
[MSG:INFO: SSDP Started]
[MSG:INFO: HTTP started on port 80]
[MSG:INFO: Telnet started on port 23]

Grbl 3.7 [FluidNC v3.7.11 (wifi) '$' for help]
[MSG:INFO: ALARM: Unhomed]
```

### User Interface Software

UGS, FluidNC Web Installer terminal

### What happened?

As a part of the initialization of the controller we disable the echo mode (with `0x0c` byte command) to be able to properly communicate with the controller.

When connecting to the controller using USB serial mode this command works fine.

But if we use a telnet channel this command results in either error `9: GCode cannot be executed in lock or alarm state` if in an alarm state. Or an error `22: Gcode undefined feed rate` if we are in an idle state.

### Other Information

The "disable echo" real time command is sent before the $I and is not visible in the log:

```
*** Connecting to tcp://192.168.1.10:23
*** Fetching device status
> ?
<Idle|MPos:0.000,0.000,0.000|FS:0,0>
Other: ok
*** Fetching device firmware version
> $I
error:Gcode undefined feed rate
*** Expected a 'FluidNC 3.3.0' or later but got 'Unknown 0.0.0'
*** Could not establish connection with the controller
*** Connection closed
```

MitchBradley, 2023-12-12:
Disable echo applies only to channels that can use line editing.  At present, that is only UART channels.  Telnet channels do not implement any line editing and never echo.  Telnet clients on most OSs implement local line editing and send complete lines to the other end.  It is sometimes possible to configure telnet clients for character-at-a-time use with editing and echoing done at the far end, but that is tricky and is not the operating mode for which the TELNET protocol was optimized.  It is a throwback to the teletype era.

MitchBradley, 2023-12-12:
Basically what this means is that you should treat a tcp connection like a classic grbl connection, sending complete lines and not expecting any echoing.

breiler, 2023-12-14:
Ok

I have added a crude detection for if echo mode is enabled by sending an empty line. If the response is `\nok` we will attempt to disable it by sending `0x0c`.

closed 2023-12-14 by breiler

MitchBradley, 2023-12-14:
The connection type is not available to the code in question?

breiler, 2023-12-14:
Technically we could expose the connection to the upper layers and do a type check. But it was hidden intentionally so no one would be tempted to use it directly and ruin the performance of the whole application.

I did consider it - but I dislike having the upper layers knowing implementation details of lower ones, so I decided to try this instead.