bdring/FluidNC #1084: Problem: status_outputs: key issues

skiddykiddie, 2023-12-11:
### Wiki Search Terms

status_outputs:

### Controller Board

6-pack

### Machine Description

3 axis plasma

### Configuration file

```yaml
board: 6 Pack
name: PlasmaBlaster
stepping:
  engine: I2S_STREAM
  idle_ms: 255
  pulse_us: 4
  dir_delay_us: 1
  disable_delay_us: 0

axes:
  shared_stepper_disable_pin: NO_PIN
  x:
    steps_per_mm: 800.000
    max_rate_mm_per_min: 2200.000
    acceleration_mm_per_sec2: 50.000
    max_travel_mm: 690.000
    soft_limits: true
    homing:
      cycle: 2
      positive_direction: false
      mpos_mm: 0.000
      feed_mm_per_min: 400.000
      seek_mm_per_min: 1000.000
      settle_ms: 500
      seek_scaler: 1.100
      feed_scaler: 1.100

    motor0:
      limit_neg_pin: gpio.27: low
      limit_pos_pin: NO_PIN
      limit_all_pin: NO_PIN
      hard_limits: false
      pulloff_mm: 10.000
      standard_stepper:
        step_pin: I2SO.2
        direction_pin: I2SO.1:low
        disable_pin: I2SO.0

  y:
    steps_per_mm: 800.000
    max_rate_mm_per_min: 2000.000
    acceleration_mm_per_sec2: 50.000
    max_travel_mm: 800.000
    soft_limits: true
    homing:
      cycle: 3
      positive_direction: false
      mpos_mm: 0.000
      feed_mm_per_min: 100.000
      seek_mm_per_min: 1000.000
      settle_ms: 500
      seek_scaler: 1.100
      feed_scaler: 1.100

    motor0:
      limit_neg_pin: gpio.16:low
      limit_pos_pin: NO_PIN
      limit_all_pin: NO_PIN
      hard_limits: false
      pulloff_mm: 1.000
      standard_stepper:
        step_pin: I2SO.5
        direction_pin: I2SO.4:low
        disable_pin: I2SO.7

  z:
    steps_per_mm: 200.000
    max_rate_mm_per_min: 3000.000
    acceleration_mm_per_sec2: 100.000
    max_travel_mm: 81.000
    soft_limits: false
    homing:
      cycle: 1
      positive_direction: true
      mpos_mm: 0.000
      feed_mm_per_min: 100.000
      seek_mm_per_min: 200.000
      settle_ms: 500
      seek_scaler: 1.100
      feed_scaler: 1.100

    motor0:
      limit_neg_pin: NO_PIN
      limit_pos_pin: gpio.4:low
      limit_all_pin: NO_PIN
      hard_limits: false
      pulloff_mm: 1.000
      standard_stepper:
        step_pin: I2SO.10
        direction_pin: I2SO.9:low
        disable_pin: I2SO.8
          
probe:
  pin: gpio.26:low
  check_mode_start: true
  
i2so:
  bck_pin: gpio.22
  data_pin: gpio.21
  ws_pin: gpio.17

spi:
  miso_pin: gpio.19
  mosi_pin: gpio.23
  sck_pin: gpio.18

sdcard:
  card_detect_pin: NO_PIN
  cs_pin: gpio.5

user_outputs:
  analog0_pin: gpio.15
  analog1_pin: gpio.12
  analog2_pin: NO_PIN
  analog3_pin: NO_PIN
  analog0_hz: 5000
  analog1_hz: 5000
  analog2_hz: 5000
  analog3_hz: 5000
  digital0_pin: gpio.14
  digital1_pin: gpio.13
  digital2_pin: NO_PIN
  digital3_pin: NO_PIN

start:
  must_home: false

relay:
  output_pin: i2so.24
  enable_pin: NO_PIN
  direction_pin: NO_PIN
  disable_with_s0: false
  s0_with_disable: true
  spinup_ms: 0
  spindown_ms: 0
  tool_num: 100
  speed_map: 0=0.000% 0=100.000% 1=100%

status_outputs:
  report_interval_ms: 500
  idle_pin: NO_PIN
  run_pin: NO_PIN
  hold_pin: NO_PIN
  alarm_pin: gpio.25

control:
  safety_door_pin: NO_PIN
  reset_pin: NO_PIN
  feed_hold_pin: gpio.34:low

macros:
  startup_line0:
  startup_line1:
  macro0: 
  macro1: 
  macro2:
  macro3:
  after_homing: G10 L20 P0 X0 Y0 Z0 & G0 X10 Y750
  after_reset: 
  after_unlock:
```

### Startup Messages

```text
[MSG:WARN: Ignored key status_outputs]
[MSG:ERR: Skipping key report_interval_ms indent 2 this indent 0]
[MSG:ERR: Skipping key idle_pin indent 2 this indent 0]
[MSG:ERR: Skipping key run_pin indent 2 this indent 0]
[MSG:ERR: Skipping key hold_pin indent 2 this indent 0]
[MSG:ERR: Skipping key alarm_pin indent 2 this indent 0]
```

### User Interface Software

All interfaces

### What happened?

It appears to not like either the placement of my status outputs code, or the format?  Ive searched through the wiki and all the yaml examples and cant seem to identify what the problem is.

MitchBradley, 2023-12-12:
You only showed us an excerpt from the Startup Messages.  Without the full set of messages I cannot determine enough context to solve your problem.

MitchBradley, 2023-12-12:
The instructions in the Problem Report template says, with emphasis "Do not excerpt just the messages you think are relevant."

skiddykiddie, 2023-12-12:
$SS
[MSG:INFO: FluidNC v3.7.8]
[MSG:INFO: Compiled with ESP32 SDK:v4.4.4]
[MSG:INFO: Local filesystem type is spiffs]
[MSG:INFO: Configuration file:6_Pext_PlasmaV6.yaml]
[MSG:WARN: Ignored key status_outputs]
[MSG:ERR: Skipping key report_interval_ms indent 2 this indent 0]
[MSG:ERR: Skipping key idle_pin indent 2 this indent 0]
[MSG:ERR: Skipping key run_pin indent 2 this indent 0]
[MSG:ERR: Skipping key hold_pin indent 2 this indent 0]
[MSG:ERR: Skipping key alarm_pin indent 2 this indent 0]
[MSG:INFO: Machine PlasmaBlaster]
[MSG:INFO: Board 6 Pack]
[MSG:INFO: I2SO BCK:gpio.22 WS:gpio.17 DATA:gpio.21]
[MSG:INFO: SPI SCK:gpio.18 MOSI:gpio.23 MISO:gpio.19]
[MSG:INFO: SD Card cs_pin:gpio.5 detect:NO_PIN freq:8000000]
[MSG:INFO: Stepping:I2S_stream Pulse:4us Dsbl Delay:0us Dir Delay:1us Idle Delay:255ms]
[MSG:INFO: User Digital Output:0 on Pin:gpio.14]
[MSG:INFO: User Digital Output:1 on Pin:gpio.13]
[MSG:INFO: User Analog Output 0 on Pin:gpio.15 Freq:5000Hz]
[MSG:INFO: User Analog Output 1 on Pin:gpio.12 Freq:5000Hz]
[MSG:INFO: Axis count 3]
[MSG:INFO: Axis X (0.000,690.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     standard_stepper Step:I2SO.2 Dir:I2SO.1:low Disable:I2SO.0]
[MSG:INFO:  X Neg Limit gpio.27:low]
[MSG:INFO: Axis Y (0.000,800.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     standard_stepper Step:I2SO.5 Dir:I2SO.4:low Disable:I2SO.7]
[MSG:INFO:  Y Neg Limit gpio.16:low]
[MSG:INFO: Axis Z (-81.000,0.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     standard_stepper Step:I2SO.10 Dir:I2SO.9:low Disable:I2SO.8]
[MSG:INFO:  Z Pos Limit gpio.4:low]
[MSG:INFO: feed_hold_pin gpio.34:low]
[MSG:INFO: Kinematic system: Cartesian]
[MSG:INFO: Relay Spindle Ena:NO_PIN Out:I2SO.24 Dir:NO_PIN]
[MSG:INFO: Using spindle Relay]
[MSG:INFO: Probe Pin: gpio.26:low]
[MSG:INFO: AP SSID FluidNC IP 192.168.0.1 mask 255.255.255.0 channel 1]
[MSG:INFO: AP started]
[MSG:INFO: WiFi on]
[MSG:INFO: Captive Portal Started]
[MSG:INFO: HTTP started on port 80]
[MSG:INFO: Telnet started on port 23]
ok

MitchBradley, 2023-12-12:
`[MSG:INFO: FluidNC v3.7.8]`

You are two revs behind.  The status_outputs feature is recent

skiddykiddie, 2023-12-12:
Darn.  Didnt realize that was new.  Thanks for the info.

skiddykiddie, 2023-12-12:
Update fixed the problem. Additional problem was a wrong pinout.

The GPIO outputs are mislabeled on this one
http://wiki.fluidnc.com/6ext_pinout.png

They are correct on the following one
http://wiki.fluidnc.com/hardware/6pex_switches.png

I should have checked that earlier, but was going strictly by the pinout posted.

closed 2023-12-12 by MitchBradley