bdring/FluidNC #1085: Best practices for controlling plasma cutter relay with MKS-DLC32

cvjensen, 2023-12-13:
### Wiki Search Terms

Found nothing on plasma cutter

### Controller Board

MKS-DLC32

### Machine Description

Small 600x700mm gantry 
MKS-DLC32
nema17 and TMC2209
12V PSU
Herocut 55X CNC plasma cutter

### Configuration file

```yaml
board: MKS-DLC32
name: DeathStar

kinematics:
  Cartesian:

stepping:
  engine: I2S_STATIC
  idle_ms: 0
  pulse_us: 6
  dir_delay_us: 10
  disable_delay_us: 0

axes:
  shared_stepper_disable_pin: I2SO.0
  #Left to right
  x:
    steps_per_mm: 80
    max_rate_mm_per_min: 10000.000
    acceleration_mm_per_sec2: 1000.000
    max_travel_mm: 580.000
    soft_limits: false
    homing:
      cycle: 0
      positive_direction: false
      mpos_mm: 0.000
      feed_mm_per_min: 300.000
      seek_mm_per_min: 6000.000
      settle_ms: 500
      seek_scaler: 1.100
      feed_scaler: 1.100
    motor0:
      limit_neg_pin: gpio.36
      hard_limits: false
      pulloff_mm: 1.000
      stepstick:
        step_pin: I2SO.1
        direction_pin: I2SO.2:low
  #Back and forth
  y:
    steps_per_mm: 80
    max_rate_mm_per_min: 10000.000
    acceleration_mm_per_sec2: 1000.000
    max_travel_mm: 670.000
    soft_limits: false
    homing:
      cycle: 0
      positive_direction: false
      mpos_mm: 0.000
      feed_mm_per_min: 300.000
      seek_mm_per_min: 6000.000
      settle_ms: 500
      seek_scaler: 1.100
      feed_scaler: 1.100
    motor0:
      limit_neg_pin: gpio.35
      hard_limits: false
      pulloff_mm: 1.000
      stepstick:
        step_pin: I2SO.5
        direction_pin: I2SO.6:low

  z:
    steps_per_mm: 1600
    max_rate_mm_per_min: 1000.000
    acceleration_mm_per_sec2: 100.000
    max_travel_mm: 80.000
    soft_limits: false
    homing:
      cycle: 1
      positive_direction: false
      mpos_mm: -15.000
      feed_mm_per_min: 300.000
      seek_mm_per_min: 300.000
      settle_ms: 500
      seek_scaler: 1.100
      feed_scaler: 1.100

    motor0:
      limit_neg_pin: gpio.34
      hard_limits: false
      pulloff_mm: 15.000
      stepstick:
        step_pin: I2SO.3
        direction_pin: I2SO.4

i2so:
  bck_pin: gpio.16
  data_pin: gpio.21
  ws_pin: gpio.17

spi:
  miso_pin: gpio.12
  mosi_pin: gpio.13
  sck_pin: gpio.14

sdcard:
  cs_pin: gpio.15
  card_detect_pin: gpio.39

control:
  safety_door_pin: NO_PIN
  reset_pin: NO_PIN
  feed_hold_pin: NO_PIN
  cycle_start_pin: NO_PIN
  macro0_pin: gpio.33:low:pu
  macro1_pin: NO_PIN
  macro2_pin: NO_PIN
  macro3_pin: NO_PIN

Laser:
  pwm_hz: 5000
  output_pin: gpio.32
  enable_pin: NO_PIN
  disable_with_s0: true
  s0_with_disable: false
  tool_num: 0
  speed_map: 0=0.000% 1000=100.000%

user_outputs:
  analog0_pin: NO_PIN
  analog1_pin: NO_PIN
  analog2_pin: NO_PIN
  analog3_pin: NO_PIN
  analog0_hz: 5000
  analog1_hz: 5000
  analog2_hz: 5000
  analog3_hz: 5000
  digital0_pin: NO_PIN
  digital1_pin: NO_PIN
  digital2_pin: NO_PIN
  digital3_pin: NO_PIN

start:
  must_home: false

coolant:
  flood_pin: gpio.4
  mist_pin: NO_PIN
  delay_ms: 300
```

### Startup Messages

```text
$SS
[MSG:INFO: FluidNC v3.7.9 https://github.com/bdring/FluidNC]
[MSG:INFO: Compiled with ESP32 SDK:v4.4.4]
[MSG:INFO: Local filesystem type is littlefs]
[MSG:INFO: Configuration file:config.yaml]
[MSG:INFO: Machine DeathStar]
[MSG:INFO: Board MKS-DLC32]
[MSG:INFO: I2SO BCK:gpio.16 WS:gpio.17 DATA:gpio.21]
[MSG:INFO: SPI SCK:gpio.14 MOSI:gpio.13 MISO:gpio.12]
[MSG:INFO: SD Card cs_pin:gpio.15 detect:gpio.39 freq:8000000]
[MSG:INFO: Stepping:I2S_static Pulse:6us Dsbl Delay:0us Dir Delay:10us Idle Delay:0ms]
[MSG:INFO: Axis count 3]
[MSG:INFO: Shared stepper disable I2SO.0]
[MSG:INFO: Axis X (0.000,580.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     stepstick Step:I2SO.1 Dir:I2SO.2:low Disable:NO_PIN]
[MSG:INFO:  X Neg Limit gpio.36]
[MSG:INFO: Axis Y (0.000,670.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     stepstick Step:I2SO.5 Dir:I2SO.6:low Disable:NO_PIN]
[MSG:INFO:  Y Neg Limit gpio.35]
[MSG:INFO: Axis Z (-15.000,65.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     stepstick Step:I2SO.3 Dir:I2SO.4 Disable:NO_PIN]
[MSG:INFO:  Z Neg Limit gpio.34]
[MSG:INFO: macro0_pin gpio.33:low:pu]
[MSG:INFO: Kinematic system: Cartesian]
[MSG:INFO: Laser Ena:NO_PIN Out:gpio.32 Freq:5000Hz Period:8191]
[MSG:INFO: Using spindle Laser]
[MSG:INFO: Flood coolant gpio.4]
[MSG:INFO: Connecting to STA SSID:WiFi]
[MSG:INFO: Connecting.]
[MSG:INFO: Connecting..]
[MSG:INFO: Connected - IP is 192.168.0.70]
[MSG:INFO: WiFi on]
[MSG:INFO: Start mDNS with hostname:http://fluidnc.local/]
[MSG:INFO: SSDP Started]
[MSG:INFO: HTTP started on port 80]
[MSG:INFO: Telnet started on port 23]
ok
<Idle|MPos:0.000,0.000,0.000|FS:0,0|Pn:XY|Ov:100,100,100|A:S>
```

### User Interface Software

Lightburn for now

### What happened?

I just attached the plasma to the same relay i used for laser air assist (aka. coolant). Air assist has been working as I hoped.

I wanted to see if I could just make a "quick and dirty" solution by using the air option in lightburn whenever i switched to the plasma. But its like the plasma arc is dragged across the cutting area whenever the gantry moves to a new area. 
Maybe because coolant isnt supposed to stop under movement ?
or because of the delay?
or lightburn isnt suited for generating plasma gcode?
idk? 

**Anyway I want to focuse on what the best practice is when it comes to controlling the relay for a plasma with the board I have?**

When I have the hardware and config in place I can move on and see if a more suited software is needed.

### Other Information

[image]
See how it drags the plasma arc across the workpiece.

bdring, 2023-12-13:
Is the move with the unintended cut a G0 or G1 move? Read the wiki on lasers and how that works.

cvjensen, 2023-12-13:
Seems like its a G0

[image]

But I just still feel like its a bit wrong to use the "air" option in lightburn which translates to coolant .. isnt it ?
Wouldn't it be better if it used the spindle or laser? Or let me just ask how would you attach it .. you know , hypothetically.. if it was your board and plasma ?

bdring, 2023-12-13:
You might be able to use the laser `enable_pin` signal. 

If it were mine, I would create a new plasma spindle class and add torch height control. It would probably many weeks to do.

cvjensen, 2023-12-13:
So can I use the gpio.0 I2C_SDA as enable pin 
like I use the gpio.4 I2C_SCL as "coolant" ?

bdring, 2023-12-13:
I assume so.

cvjensen, 2023-12-13:
whats the difference between enable and output pin then?

bdring, 2023-12-13:
Read the wiki about spindle enables.

The enable_pin is an on/off signal. The output_pin controls the power.

You should consider donating to this project.

cvjensen, 2023-12-13:
Thanks I will test it out

cvjensen, 2023-12-15:
I guess the enable pin and relay are working . Now I'm only having problems making a delay at the start and end of the cut .

closed 2023-12-15 by cvjensen

bdring, 2023-12-15:
I don't know how you would do that with the firmware automatically. Delays could be added to the gcode.

You could create a new spindle class. It would take some decent programming skills, but the firmware is designed to add new features easily. 

We are not interested in doing it right now. We would do a comprehensive plasma solution with torch height control if we did it. We have a full workload for months already.

We have a plasma channel on our discord with other users, but the developers don't do much there yet.

cvjensen, 2023-12-15:
Thanks for the quick response . 

Regarding the g-code .. maybe you could give me a hint where things go wrong.

I made these cuts (its supposed to be holes stacked on top of eachother)

[image]

2 and 3 had this option added:

[image]

which made, as you can see, no difference.

**1 - without start and end pause time**

```
G00 G17 G40 G21 G54
G91
M4
; Cut @ 30 mm/sec, 100% power
M9
G0 X3.685Y1.092
; Layer Plasma
G1 X-0.173Y0.231S1000F1800
G1 X-0.15Y0.247
G1 X-0.126Y0.262
G1 X-0.101Y0.276
```

**2 and 3 with start and end pause time** 

```
G00 G17 G40 G21 G54
G91
M4
; Cut @ 30 mm/sec, 100% power
M9
G0 X3.685Y1.092
G4 P2
; Layer Plasma
G1 X-0.173Y0.231S1000F1800
G1 X-0.15Y0.247
G1 X-0.126Y0.262
G1 X-0.101Y0.276
```

So my question is why? shouldn't the G4 P2 make it stay still for 2 hole seconds before moving?
or is it because of: [Note: Some spindles, like lasers may disable during this move](http://wiki.fluidnc.com/en/features/supported_gcodes#g0-rapid-motion)
and there is a G0 after the M4

bdring, 2023-12-15:
Try editing the gcode to verify your guesses.