bdring/FluidNC #1086: Problem: with configurations in ultimate bee, updated to FluidNC, only one y axis moter working

Bigrigins, 2023-12-13:
### Wiki Search Terms

axis, homing, spindle, gpio , config yaml  

### Controller Board

Spark Concepts CNC XPRO V5

### Machine Description

 Bulkman 3D  1500x1500 Ultimate Bee , Duel y machine , soft limit switch's on Y+ ,X+, and Z+ Huanyang VFD and 2.2 kW water cooled spindle. high torque NEMA 23 stepper moters. XPRO V5 controller

### Input Circuits

```yaml
set pins up to what was recommended on FluidNC Wikipage
```

### Configuration file

```yaml
board: tmc_5160_xpro v5
name: ULTIMATE BEE
stepping:
  engine: RMT
  idle_ms: 255
  pulse_us: 2
  dir_delay_us: 0
  disable_delay_us: 0

axes:
  shared_stepper_disable_pin: NO_PIN
  x:
    steps_per_mm: 400.00
    max_rate_mm_per_min: 300.00
    acceleration_mm_per_sec2: 100.00
    max_travel_mm: 1240.00
    soft_limits: true
    homing:
      cycle: 2
      positive_direction: true
      mpos_mm: 150.000
      feed_mm_per_min: 100.000
      seek_mm_per_min: 1500.00
      seek_scaler: 1.100
      feed_scaler: 1.100

    motor0:
      limit_neg_pin: gpio.35
      limit_pos_pin: NO_PIN
      limit_all_pin: NO_PIN
      hard_limits: false
      pulloff_mm: 3.00
      tmc_5160:
        cs_pin: gpio.17
        spi_index: 1
        r_sense_ohms: 0.075
        run_amps: 2.5
        hold_amps: 0.50
        microsteps: 16
        stallguard: 0
        stallguard_debug: false
        toff_disable: 0
        toff_stealthchop: 5
        toff_coolstep: 3
        run_mode: CoolStep
        homing_mode: CoolStep
        use_enable: true
        step_pin: gpio.12
        direction_pin: gpio.14
[... 468 lines cut ...]
[MSG:INFO:   Motor1]
[MSG:INFO:     tmc_5160 Step:gpio.33 Dir:gpio.32 CS:NO_PIN Disable:NO_PIN Index:4 R:0.075]
[MSG:INFO: Axis Z (-10.000,150.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     tmc_5160 Step:gpio.15 Dir:gpio.2 CS:NO_PIN Disable:NO_PIN Index:3 R:0.075]
[MSG:INFO:  Z Neg Limit gpio.39]
[MSG:INFO: X Axis driver test passed]
[MSG:INFO: Y2 Axis driver test passed]
[MSG:INFO: Z Axis driver test passed]
[MSG:INFO: safety_door_pin gpio.16:low]
[MSG:INFO: macro0_pin gpio.13:low]
[MSG:INFO: macro1_pin gpio.0:low]
[MSG:INFO: Kinematic system: Cartesian]
[MSG:INFO: Using spindle NoSpindle]
[MSG:INFO: Probe Pin: gpio.22:low]
[MSG:INFO: BT Started with FluidNC]

Grbl 3.7 [FluidNC 3.7.10 (bt) '$' for help]
[MSG:INFO: ALARM: Unhomed]
ALARM:14
```

### What happened?

Bought this machine in December of 2022 worked on the machine when I had free time throughout the year had it all put together in May started to configure the XPRO V5 software (not FluidNC) had everything configured was ready to surface the spoil board sent everything home and that's when it started binding again.  Apparently i didn't have it  all worked out.  YA2 worked as it should, along with x and z. When Y would move it would shutter. Then Y moter would just stop completely ,  YA2 would continue moving causing the Gantry to bind,  forcing me to hit E stop. I would have to realign everything and try again,   even trying to jog would shutter and bind. Problem was it was intermitting,  sometimes it would work sometimes it wouldn't which led me to believe that it was a bad drive. So was able to get in contact with spark Concepts they sent me a new controller out back in  November, the new controller of course was loaded with the new FluidNC.  So after researching and studying I started to try and configure FluidNC,  so this is where I'm at.  It seems to be the same issue that I was having with the old software  in XPro V5. I should also mention that I put a new stepper motor on thinking possibly that the motor was bad , it wasn't, new motor is still having the same issue. The only other thing I haven't done is run new wire,  the new wire will be here today. I can't possibly imagine that I have a bad driver in a brand new controller, what are the odds lol. So any help would be greatly appreciated , any ideas,  things I can try. I'm at a loss like I said I've had this thing for almost a year I'm getting very  frustrated, want to be able to start using this thing and start making some chips!! thanks again for any advice or help.

bdring, 2023-12-13:
Your indentation is off in many places.

Look at motor0 under Y. It should have the indentation level as the other motors.

Also, it is best to have motor0 defined before motor1. Also motor0 has A/Y2 after it, You cannot do that. If you want a comment follow the [comment rules](http://wiki.fluidnc.com/en/config/overview#file-specifications).

Your spindle indentation is also off. The PWM should not be indented and the sub items should.

Try using a [free yaml file editor](http://wiki.fluidnc.com/en/support/troubleshooting_config_files#editing-the-file). It can help you spot errors and will visually format the file.

You might not get it right the first time, but the startup messages will give you some clues. Do not try using the machine until the startup messages are free from errors and warning.

Bigrigins, 2023-12-13:
> Your indentation is off in many places.
>
> Look at motor0 under Y. It should have the indentation level as the other motors.
>
> Also, it is best to have motor0 defined before motor1. Also motor0 has [...]

I understand everything you said and I will correct everything thst you suggested when I get home in front of my machine laptop, but I'm not understanding what you mean by indentation.  could you elaborate? I looked at the different Motors and they all seem to be the same other than your gpio pins Direction pins etc etc I'm confused on the indentation or what I'm looking for.

bdring, 2023-12-13:
[image]

Bigrigins, 2023-12-13:
> ![image](https://private-user-images.githubusercontent.com/189677/290229767-2856855a-84ba-4882-be2f-61feac828be9.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTEiLCJleHAiOjE3MDI0Nzk0MTcsIm5iZiI6MTcwMjQ3OTExNywicGF0aCI6Ii8xODk2NzcvMjkwMjI5NzY3LTI4NTY4NTVhLTg0YmEtNDg4Mi1iZTJmLTYxZmVhYzgyOGJlOS5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBSVdOSllBWDRDU1ZFSDUzQSUyRjIwMjMxMjEzJTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDIzMTIxM1QxNDUxNTdaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT02Yzg4ODQxZDY3NDdkNGMzOGYwM2IwOGFlODQ2NDFjNzg2ZjZiODRlMzA0OWY5NWYzZWU4NzEyMjRhYzFmYWFhJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.gBvMXcxHZFcLXjQ-lnVEfFLsLz_Te6T1FM9pJ-Jk0uA)

Yeah I see that and plan  to correct that when I get in front of my computer. Please excuse my ignorance but that is a indentation that your referring to?

Bigrigins, 2023-12-13:
> Yeah I see that and plan to correct that when I get in front of my computer. Please excuse my ignorance but that is a indentation that your referring to?

bdring, 2023-12-13:
Yes, YAML is a hierarchy.  Things like  `limit_neg_pin:` belong to a specific motor. The file tells describes that using indenting.

Motor0 has to be indented at the exact place so we know what axis letter it belongs to. The stuff the belongs to motor0 have to be indented under motor0.

Right now it looks like motor0 is indented under tmc_5160 and FluidNC is rejecting that.

Please consider donating to this project.

Bigrigins, 2023-12-13:
> Yes, YAML is a hierarchy. Things like `limit_neg_pin:` belong to a specific motor. The file tells describes that using indenting.
>
> Motor0 has to be indented at the exact place so we know what axis [...]

I will for sure donate after I figure this out 👍, see that's the problem motor 0 A/Y2 is the only moter working on that axis, the other moter is the one not working , motor one is not working that's what I'm trying to achieve. I appreciate all your help this far.

Bigrigins, 2023-12-13:
> Yes, YAML is a hierarchy. Things like `limit_neg_pin:` belong to a specific motor. The file tells describes that using indenting.
>
> Motor0 has to be indented at the exact place so we know what axis [...]

Donated to your cause 👍

So what do you suggest that I put in there? how do you suggest a fix other than changing 0 and 1?

MitchBradley, 2023-12-13:
Fix the indentation by making sure that each section name is preceded by the correct number of spaces.

For the motors that work, the "    motorN" section name is preceded by 4 spaces and no junk after it.  The bad line is the "        motor0: A/Y2" line that starts with 8 spaces and has extra junk after it (the junk being "A/Y2"

Fix that and the motors will probably work.

You will also need to fix the mis-indented coolant and PWM sections.  You have them as
```
  coolant:
    mist_pin: gpio.21
    delay_ms: 0  

    PWM:
```
but they should be
```
coolant:
  mist_pin: gpio.21
  delay_ms: 0  

PWM:
```
In other words, "coolant:" and "PWM" should both be at the left margin, with no preceding spaces.

bdring, 2023-12-13:
I tried cleaning up your file, but you have a lot of reused pins. I just removed those uses. You will have to add missing features back after you spend some time on the wiki.

Try this. I don't know that controller, so some motors like Y0 and Y1 might be reversed. Just change the wiring. You might contact the seller for additional help.

```yaml
board: tmc_5160_xpro v5
name: ULTIMATE BEE
stepping:
  engine: RMT
  idle_ms: 255
  pulse_us: 2
  dir_delay_us: 0
  disable_delay_us: 0

axes:
  shared_stepper_disable_pin: NO_PIN
  x:
    steps_per_mm: 400.00
    max_rate_mm_per_min: 300.00
    acceleration_mm_per_sec2: 100.00
    max_travel_mm: 1240.00
    soft_limits: true
    homing:
      cycle: 2
      positive_direction: true
      mpos_mm: 150.000
      feed_mm_per_min: 100.000
      seek_mm_per_min: 1500.00
      seek_scaler: 1.100
      feed_scaler: 1.100

    motor0:
      limit_neg_pin: gpio.35
      limit_pos_pin: NO_PIN
      limit_all_pin: NO_PIN
      hard_limits: false
      pulloff_mm: 3.00
      tmc_5160:
        cs_pin: gpio.17
        spi_index: 1
        r_sense_ohms: 0.075
        run_amps: 2.5
        hold_amps: 0.50
        microsteps: 16
        stallguard: 0
        stallguard_debug: false
        toff_disable: 0
        toff_stealthchop: 5
        toff_coolstep: 3
        run_mode: CoolStep
        homing_mode: CoolStep
        use_enable: true
        step_pin: gpio.12
        direction_pin: gpio.14
        tpfd: 4
       

  y:
    steps_per_mm: 400.00
    max_rate_mm_per_min: 300.00
    acceleration_mm_per_sec2: 100.00
    max_travel_mm: 1240.00
    soft_limits: true
    homing:
      cycle: 2
      positive_direction: true
      mpos_mm: 150.000
      feed_mm_per_min: 100.000
      seek_mm_per_min: 1500.00
      seek_scaler: 1.100
      feed_scaler: 1.100
      
    motor0:
      limit_neg_pin: NO_PIN
      limit_pos_pin: NO_PIN
      limit_all_pin: NO_PIN
      hard_limits: false
      pulloff_mm: 3.00
      tmc_5160:
        cs_pin: NO_PIN
        spi_index: 4
        r_sense_ohms: 0.075
        run_amps: 2.5
        hold_amps: 0.50
        microsteps: 16
        stallguard: 0
        stallguard_debug: false
        toff_disable: 0
        toff_stealthchop: 5
        toff_coolstep: 3
        run_mode: CoolStep
        homing_mode: CoolStep
        use_enable: true
        step_pin: gpio.33
        direction_pin: gpio.32
        tpfd: 4
       

    motor1:
      limit_neg_pin: gpio.34
      limit_pos_pin: NO_PIN
      limit_all_pin: NO_PIN
      hard_limits: false
      pulloff_mm: 3.00
      tmc_5160:
        cs_pin: NO_PIN
        spi_index: 2
        r_sense_ohms: 0.075
        run_amps: 2.5
        hold_amps: 0.50
        microsteps: 16
        stallguard: 0
        stallguard_debug: false
        toff_disable: 0
        toff_stealthchop: 5
        toff_coolstep: 3
        run_mode: CoolStep
        homing_mode: CoolStep
        use_enable: true
        step_pin: gpio.27
        direction_pin: gpio.26
        tpfd: 4
       
  z:
    steps_per_mm: 400.00
    max_rate_mm_per_min: 300.00
    acceleration_mm_per_sec2: 100.00
    max_travel_mm: 160.00
    soft_limits: true
    homing:
      cycle: 1
      positive_direction: true
      mpos_mm: 150.000
      feed_mm_per_min: 100.000
      seek_mm_per_min: 1500.00
      seek_scaler: 1.100
      feed_scaler: 1.100

    motor0:
      limit_neg_pin: gpio.39
      limit_pos_pin: NO_PIN
      limit_all_pin: NO_PIN
      hard_limits: false
      pulloff_mm: 3.00
      tmc_5160:
        cs_pin: NO_PIN
        spi_index: 3
        r_sense_ohms: 0.075
        run_amps: 2.5
        hold_amps: 0.50
        microsteps: 16
        stallguard: 0
        stallguard_debug: false
        toff_disable: 0
        toff_stealthchop: 5
        toff_coolstep: 3
        run_mode: CoolStep
        homing_mode: CoolStep
        use_enable: true
        step_pin: gpio.15
        direction_pin: gpio.2
        tpfd: 4

spi:
  miso_pin: gpio.19
  mosi_pin: gpio.23
  sck_pin: gpio.18

sdcard:
  card_detect_pin: NO_PIN
  cs_pin: gpio.5

macros:
  startup_line0:
  startup_line1:
  macro0:
  macro1:
  macro2:
  macro3:

user_outputs:
  analog0_pin: NO_PIN
  analog1_pin: NO_PIN
  analog2_pin: NO_PIN
  analog3_pin: NO_PIN
  analog0_hz: 5000
  analog1_hz: 5000
  analog2_hz: 5000
  analog3_hz: 5000
  digital0_pin: NO_PIN
  digital1_pin: NO_PIN
  digital2_pin: NO_PIN
  digital3_pin: NO_PIN

start:
  must_home: true

coolant:
  mist_pin: gpio.21
  delay_ms: 0  

PWM:
  pwm_hz: 5000
  output_pin: gpio.25
  enable_pin: gpio.4
  direction_pin: NO_PIN
  disable_with_s0: false
  s0_with_disable: true
  spinup_ms: 24000
  spindown_ms: 6000
  tool_num: 0
  speed_map: 0=0.000% 24000=100.000%
```

Bigrigins, 2023-12-13:
> Fix the indentation by making sure that each section name is preceded by the correct number of spaces.
>
> For the motors that work, the " motorN" section name is preceded by 4 spaces and no junk after [...]

Ahhhh now I see it plane as day, I had no idea what I was looking for, I'm learning as I go LOL as you probably can tell . was not aware about spacing, I will for sure be mindful of it now thank you

Bigrigins, 2023-12-13:
> I tried cleaning up your file, but you have a lot of reused pins. I just removed those uses. You will have to add missing features back after you spend some time on the wiki.
>
> Try this. I don't know [...]

Thank you so much bdring I will go through this tonight and get back to you in a day or two let you know how I made out thank you guys so much for your help.

Bigrigins, 2023-12-13:
> Please don't quote everything I write in your replies unless specifically needed. It makes the thread too long.

Tried the your settings and still getting the same messages as above. Same Y steeper Moter  still not turning.  Switched plugs at controller and moter that wasn't working started working , switched plugs back to original positions in controller and moter stopped working again ( original moter, moter 1 Y2A  that was working  works again after switching plugs back) so it almost seems that I have a dead driver  for one of the y axis moters. This is a new controller,  is it possible that I got another bad controller? I'm at a lost now more then ever . Not sure what to do now 😕

bdring, 2023-12-13:
I doubt you are really using my file. I tested it for errors.

Send $CD in fluidterm and paste the response in a code block.

http://wiki.fluidnc.com/en/support/requesting_help#pasting-the-config-file-on-github

MitchBradley, 2023-12-13:
Do you still get the warning and error messages?  If so there is absolutely no point in swapping motors or anything else having to do with hardware.

If you are getting the warning messages, then you have probably not succeeded in uploading Bart's file or choosing it.

Bigrigins, 2023-12-13:
Well then I'm an idiot then because I went through and compared line for line before I uploaded into the controller, I highlighted and pasted it to my example file  with all the settings and then  ctrl U, $bye . I'm out of time today to try it again , I will  go through everything again in the morning and see what I come up with. It very well could be something I'm not doing right. ( most likely lol) let me be clear I am by no means saying that your file is the problem,  I'm just saying when I tried it I couldn't get it working. IM  missing something I'm sure ,I'm not doing something right.  I will try again tomorrow morning. Thanks again for all your guys help !

MitchBradley, 2023-12-13:
What editor program are you using for your config file?  It needs to be either a plain text editor like Notepad, or a YAML-aware editor like Notepad++.  If you use a fancy-text editor like Word, it will probably mess up the formatting by adding extra spaces.  Having the correct number of spaces at the beginning of each line is critical.

When I load the config file that you posted above into Notepad++, with the filename ending in .yaml, it looks like this:
[image]
It needs to look like this:
[image]

MitchBradley, 2023-12-13:
You should take that file that Bart gave you VERBATIM, get it working, then add stuff to it as necessary.  Since your file is messed up, trying to go the other way and merge Barts stuff into your file is just propagating the errors.

bdring, 2023-12-14:
1. Get Notepad++
2. Paste my version into a new file.
3. Save the file as `bart.yaml`
4. Upload that file.
5. Send `$Config/Filename=bart.yaml`
6. Restart

If it works you can try creating your own file, but leave my file on the ESP32, so you can always go back to it.

You seems to be avoiding spending time to understand YAML. Our simplified use of YAML is just about the simplest data markup format there is.

Bigrigins, 2023-12-14:
> You should take that file that Bart gave you VERBATIM, get it working, then add stuff to it as necessary. Since your file is messed up, trying to go the other way and merge Barts stuff into your file is just propagating the errors.

Yes plan to use and  add to his file as soon as I csn get back outbto the shop, tomorrow afternoon.

Bigrigins, 2023-12-14:
> 1. Get Notepad++
> 2. Paste my version into a new file.
> 3. Save the file as `bart.yaml`
> 4. Upload that file.
> 5. Send `$Config/Filename=bart.yaml`
> 6. Restart
>
> If it works you can try creating your own [...]

I wouldn't say I'm avoid spending time understanding YAML. I've read and read and researched and researched FluidNC Wiki pages,  ( I'm seeing it in my sleep) yes it seems to be a lot easier to understand then the previous GRBL version that I was using in XPRO V5 . I just downloaded Notepad ++ today  so hopefully that will help . Was using  visual studio , maybe that is part of the reason I'm having issues.  I am  going try and upload your file tomorrow afternoon. Will report back in the next day or two how I made out .thanks again for your help.

MitchBradley, 2023-12-14:
Talk to us when you are near the machine and can try what we suggest.  Answering questions burns a lot of our precious development time and when the user cannot try our suggestions immediately, but can chat, that just uses up more unproductive time.

Bigrigins, 2023-12-16:
I've tried over and over to get the old configuration out originally named ultimatebee, and pit bars in, for what ever reason I cannot get the original config out. I've done what Bart has asked $cd pasted his into the fluidterm screen liked he asked, tried the original way ctrl-U click on his file, hit enter , says its loading, then $bye , then ctrl-r . When it loads everything up it still says ultimatebee config. Not sure what to do now .

bdring, 2023-12-16:
Sent `$Config/Filename` at the console.

Bigrigins, 2023-12-16:
> Sent `$Config/Filename` at the console.

Awesome that did it, thank you

bdring, 2023-12-16:
> 1. Get Notepad++
> 2. Paste my version into a new file.
> 3. Save the file as `bart.yaml`
> 4. Upload that file.
> 5. Send `$Config/Filename=bart.yaml`
> 6. Restart

You even quoted my instructions before ;-)

Bigrigins, 2023-12-16:
The y-axis is still going in the wrong direction, just changed it from True to false and now flew a term says cannot open serial Port press any key to continue, now what?

Bigrigins, 2023-12-16:
> > 1. Get Notepad++
> > 2. Paste my version into a new file.
> > 3. Save the file as `bart.yaml`
> > 4. Upload that file.
> > 5. Send `$Config/Filename=bart.yaml`
> > 6. Restart
> 
> You even quoted my instructions before ;-)

LOL you are right everything you put in is exactly the way I was supposed to do it unfortunately I'm an idiot LOL sometimes I find it hard to follow directions, and again this is all new to me so I'm learning lol

bdring, 2023-12-16:
Slow down, read, follow instructions, learn.

If you can't open a serial port is is usually open somewhere else. Find the program that has it open or restart your computer.

Follow these instructions for help

http://wiki.fluidnc.com/en/support/help_with_switch_problems#testing

Bigrigins, 2023-12-16:
So trying to do the axis test following the link you posted, x axis was traveling too far made a change in notepad ++  went  to reload conf following your directions ( it worked once and allowed me to get bart.yaml in)  checked limit switches like it says that checked out good then moved on to the axis, now when I do $J=G91 X10 F100 it's given me an error:8 what am I not doing  correctly?

bdring, 2023-12-16:
What is error 8?

Bigrigins, 2023-12-17:
All axis are moving in the correct direction except y, y+ goes in a - direction,  switched the direction configuration in Notepad++  made no difference. Weather I have true or false put in,  the direction of the machine is not corresponding with the controller buttons, still moving in opposite directions X Plus moves In A minus Direction X - moves in a positive direction . Any ideas?

MitchBradley, 2023-12-17:
There is no such thing as "true" or "false" pins.  Pins are either ":low" or ":high", and if you do not specify, that is the same as ":high". http://wiki.fluidnc.com/config/config_IO#output-pin-attributes

"true" and "false" are for things like homing .. direction_positive.

MitchBradley, 2023-12-17:
When you ask a question, do not speak generally, but rather tell us exactly what you did.  Do not say "switching the direction configuration".  Instead say "I changed to " and then give us the exact text of what you changed, like ```yaml
      tmc_5160:
        ...
        step_pin: gpio.15
        direction_pin: gpio.2:low
```

closed 2023-12-31 by Bigrigins