bdring/FluidNC #1089: Problem: Motors are not locked when make home - 3.7.11

ahmetbozaci, 2023-12-17:
### Wiki Search Terms

NA

### Controller Board

MKS DLC 32

### Machine Description

nema17, tmc2209, dual y motors, with end stops

### Configuration file

```yaml
board: MKS-DLC32 V2.1
name: XProCraft
meta: 25-11-2023

kinematics:
  Cartesian:

stepping:
  engine: I2S_STATIC
  idle_ms: 255
  pulse_us: 6
  dir_delay_us: 10
  disable_delay_us: 0
axes:
  shared_stepper_disable_pin: I2SO.0
  x:
    steps_per_mm: 80
    max_rate_mm_per_min: 6000.000
    acceleration_mm_per_sec2: 1000
    max_travel_mm: 720
    soft_limits: false
    homing:
      cycle: 1
      positive_direction: false
      mpos_mm: 0.000
      feed_mm_per_min: 500.000
      seek_mm_per_min: 1000.000
      settle_ms: 500
      seek_scaler: 1.100
      feed_scaler: 1.100

    motor0:
      limit_neg_pin: gpio.36:low
      hard_limits: false
      pulloff_mm: 1.000
      stepstick:
        step_pin: I2SO.1
        direction_pin: I2SO.2:low

  y:
    steps_per_mm: 80
    max_rate_mm_per_min: 6000.000
    acceleration_mm_per_sec2: 500.000
    max_travel_mm: 900
    soft_limits: false
    homing:
      cycle: 1
      positive_direction: false
      mpos_mm: 0.000
      feed_mm_per_min: 500.000
      seek_mm_per_min: 1000.000
      settle_ms: 500
      seek_scaler: 1.100
      feed_scaler: 1.100

    motor0:
      limit_neg_pin: gpio.35:low
      hard_limits: false
      pulloff_mm: 1.000
      stepstick:
        step_pin: I2SO.5
        direction_pin: I2SO.6:high
i2so:
  bck_pin: gpio.16
  data_pin: gpio.21
  ws_pin: gpio.17

spi:
  miso_pin: gpio.12
  mosi_pin: gpio.13
  sck_pin: gpio.14

sdcard:
  cs_pin: gpio.15
  card_detect_pin: gpio.39

Laser:
  pwm_hz: 5000
  output_pin: gpio.32
  enable_pin: NO_PIN
  disable_with_s0: false
  s0_with_disable: false
  tool_num: 0
  speed_map: 0=0.000% 1000=100.000%
```

### Startup Messages

```text
NA
```

### What happened?

Typically, I go through the process of homing my setup, during which the motors are expected to lock as intended. However, in version 3.7.11, I've encountered an issue where the motors remain unlocked after homing. without homing and not locked motors, I am getting an error on the Lightburn as expected. To address this, I've resorted to using the command $X to unlock, moving the axis, at which point the motors finally lock, allowing me to proceed with the homing process.
It is no problem in the 3.7.9 version.

bdring, 2023-12-17:
I don't understand why you have to unlock. Unlock is needed before homing. After homing you should be in idle mode.

Try doing everything in FluidTerm. Before and after each command send the `?` to get status. Then paste everything here in a response, including the startup messages.

ahmetbozaci, 2023-12-17:
- I believe the motors should remain locked when the machine starts.
[image]
[image]

- However, despite a successful homing process, the motors remain unlocked, resulting in errors when attempting to move any axis.

[image]
- To resolve this, I use the command $X to clear the alarm, followed by moving the axis, homing again, and finally achieving the intended motor lock:
[image]

MitchBradley, 2023-12-17:
It is showing an Alarm state after homing, so something did not home correctly.  Send $Message/Level=Debug and rerun the $H, then show us the message log from that.

MitchBradley, 2023-12-17:
The "motors not locked" is just a side effect.  The deeper problem is the fact that the system is still in Alarm state, and we need to find out what is causing that.

ahmetbozaci, 2023-12-17:
[image]

MitchBradley, 2023-12-17:
It is probably related to a recent change in the way that we track whether individual axes are homed.  Stop focusing about motor locking; that is just a symptom of a deeper root cause.  I am looking at the problem now.  I will tell you when I need you to try something.

MitchBradley, 2023-12-17:
Do you have a build setup so you can try code changes without me making a release?

ahmetbozaci, 2023-12-17:
It can build with platformiO in the visual code?

MitchBradley, 2023-12-17:
Yes.

MitchBradley, 2023-12-17:
Send $A just after homing completes and tell me the result

ahmetbozaci, 2023-12-17:
[image]

MitchBradley, 2023-12-17:
I have reproduced you problem on a machine here, so I should be able to find the problem soon.

ahmetbozaci, 2023-12-17:
How can I move the axis on the fluidterm?

MitchBradley, 2023-12-17:
Send any gcode command that moves, like G0 or G1.

MitchBradley, 2023-12-17:
You can test the fix by installing the "Test1090" release with [WebInstaller ](https://installer.fluidnc.com/)

ahmetbozaci, 2023-12-17:
[image]
It worked. Thanks

PR #1090 (2023-12-17): Inferred _homing sections were treated as home-able
Fixes #1089
closed 2023-12-17 by MitchBradley