bdring/FluidNC #1100: Problem: additional UART doesn't work

michalelektryk, 2023-12-30:
### Wiki Search Terms

UART Sections, Top Level Items, Inputs

### Controller Board

MKS DLC32

### Machine Description

3 axis CNC, based on proxxon MK70

### Input Circuits

```yaml
rp2040 powered from 5V on EXP1 header.
gpio.25 / LCD_CS used as TX also taken from EXP1.
gpio.4 / I2C_SCL used as RX taken from I2C connector.
```

### Configuration file

```yaml
board: MKS-DLC32
name: proxxon MF70

kinematics:
  Cartesian:

#motor info:
#max current 1.8A/phase
#holding torque: 0.5Nm
#step angle 1.8 degree
#mass: 0.34kg

#probe and limit switches have pullups on PCB

#jog uart
uart2:
  txd_pin: gpio.25
  rxd_pin: gpio.4
  #rts_pin: NO_PIN
  baud: 115200
  mode: 8N1
#jog setup
uart_channel1:
   uart_num: 2
   report_interval_ms: 100
   all_messages: true

stepping:
  engine: I2S_STATIC
  idle_ms: 254
  pulse_us: 6
  dir_delay_us: 10
  disable_delay_us: 0
axes:
  shared_stepper_disable_pin: I2SO.0
  x:
    #1 microstepping * 200 steps per rotation * 1 rotation per mm
    steps_per_mm: 400.000
    max_rate_mm_per_min: 100.000
    acceleration_mm_per_sec2: 50.000
    max_travel_mm: 150.000
    soft_limits: true
    homing:
      #hack, change to 2 after switch works
      cycle: -1
      positive_direction: false
      mpos_mm: 0.000
      feed_mm_per_min: 30.000
      seek_mm_per_min: 300.000
      settle_ms: 500
      seek_scaler: 1.100
      feed_scaler: 1.100

    motor0:
      limit_neg_pin: gpio.36:low
      hard_limits: false
      pulloff_mm: 1.000
      stepstick:
        step_pin: I2SO.1:low
        direction_pin: I2SO.2:low

  y:
    steps_per_mm: 400.000
    max_rate_mm_per_min: 100.000
    acceleration_mm_per_sec2: 50.000
    max_travel_mm: 70.000
    soft_limits: true
    homing:
      cycle: 2
      positive_direction: false
      mpos_mm: 0.000
      feed_mm_per_min: 300.000
      seek_mm_per_min: 6000.000
      settle_ms: 500
      seek_scaler: 1.100
      feed_scaler: 1.100

    motor0:
      limit_neg_pin: gpio.35:low
      hard_limits: false
      pulloff_mm: 1.000
      stepstick:
        step_pin: I2SO.5
        direction_pin: I2SO.6:low

  z:
    steps_per_mm: 400.000
    max_rate_mm_per_min: 100.000
    acceleration_mm_per_sec2: 25.000
    max_travel_mm: 80.000
    soft_limits: true
    homing:
      cycle: 1
      positive_direction: true
      mpos_mm: 0.000
      feed_mm_per_min: 300.000
      seek_mm_per_min: 1000.000
      settle_ms: 500
      seek_scaler: 1.100
      feed_scaler: 1.100

    motor0:
      limit_pos_pin: gpio.34:low
      hard_limits: false
      pulloff_mm: 1.000
      stepstick:
        step_pin: I2SO.3
        direction_pin: I2SO.4:low

i2so:
  bck_pin: gpio.16
  data_pin: gpio.21
  ws_pin: gpio.17

spi:
  miso_pin: gpio.12
  mosi_pin: gpio.13
  sck_pin: gpio.14

sdcard:
  cs_pin: gpio.15
  card_detect_pin: gpio.39

control:
  safety_door_pin: NO_PIN
  reset_pin: NO_PIN
  feed_hold_pin: NO_PIN
  cycle_start_pin: NO_PIN
  macro0_pin: NO_PIN
  macro1_pin: NO_PIN
  macro2_pin: NO_PIN
  macro3_pin: NO_PIN

macros:
  startup_line0:
  startup_line1:
  macro0: $SD/Run=lasertest.gcode
  macro1: $SD/Run=home.gcode
  #These are examples
  macro2:
  macro3:

coolant:
  flood_pin: NO_PIN
  mist_pin: NO_PIN
  delay_ms: 0

probe:
  pin: gpio.22:low
  check_mode_start: true

relay:
  direction_pin: NO_PIN
  output_pin: gpio.32
  disable_with_s0: false
  s0_with_disable: true
  spinup_ms: 1000
  spindown_ms: 1000
  tool_num: 0
  speed_map: 0=0.000% 0=100.000% 1=100.000%
  off_on_alarm: false

#TODO: set back
start:
  must_home: false

# 5,18,19,22,23,25,26,27,32,33,39,I2SO.7
# SDA 0 / SCL 4
```

### Startup Messages

```text
[MSG:INFO: FluidNC v3.7.12 https://github.com/bdring/FluidNC]
[MSG:INFO: Compiled with ESP32 SDK:v4.4.4]
[MSG:DBG: Spiffs mount failed: ESP_FAIL]
[MSG:INFO: Local filesystem type is littlefs]
[MSG:INFO: Configuration file:config.yaml]
[MSG:DBG: Running after-parse tasks]
[MSG:DBG: Checking configuration]
[MSG:INFO: Machine proxxon MF70]
[MSG:INFO: Board MKS-DLC32]
[MSG:INFO: UART2 Tx:gpio.25 Rx:gpio.4 RTS:NO_PIN Baud:115200]
[MSG:INFO: I2SO BCK:gpio.16 WS:gpio.17 DATA:gpio.21]
[MSG:INFO: SPI SCK:gpio.14 MOSI:gpio.13 MISO:gpio.12]
[MSG:INFO: SD Card cs_pin:gpio.15 detect:gpio.39 freq:8000000]
[MSG:INFO: Stepping:I2S_static Pulse:6us Dsbl Delay:0us Dir Delay:10us Idle Delay:254ms]
[MSG:INFO: Axis count 3]
[MSG:INFO: Shared stepper disable I2SO.0]
[MSG:INFO: Axis X (0.000,150.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     stepstick Step:I2SO.1:low Dir:I2SO.2:low Disable:NO_PIN]
[MSG:INFO:  X Neg Limit gpio.36:low]
[MSG:DBG:  X Neg Limit 0]
[MSG:INFO: Axis Y (0.000,70.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     stepstick Step:I2SO.5 Dir:I2SO.6:low Disable:NO_PIN]
[MSG:INFO:  Y Neg Limit gpio.35:low]
[MSG:DBG:  Y Neg Limit 0]
[MSG:INFO: Axis Z (-80.000,0.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     stepstick Step:I2SO.3 Dir:I2SO.4:low Disable:NO_PIN]
[MSG:INFO:  Z Pos Limit gpio.34:low]
[MSG:DBG:  Z Pos Limit 0]
[MSG:INFO: Kinematic system: Cartesian]
[MSG:INFO: Relay Spindle Ena:NO_PIN Out:gpio.32 Dir:NO_PIN]
[MSG:INFO: Using spindle Relay]
[MSG:INFO: Probe Pin: gpio.22:low]
[MSG:INFO: Connecting to STA SSID:a1]
[MSG:INFO: Connecting.]
[MSG:INFO: Connecting..]
[MSG:INFO: Connected - IP is 192.168.1.151]
[MSG:INFO: WiFi on]
[MSG:INFO: Start mDNS with hostname:http://fluidnc.local/]
[MSG:INFO: SSDP Started]
[MSG:INFO: HTTP started on port 80]
[MSG:DBG: config.yaml hash "3CF45EFC96EB00DF3BF27398B896CF10FD27230A45F70EF308852B740C40E03B"]
[MSG:DBG: favicon.ico hash "B8B2871A343CA0F9A7A130213226962AD2BE4DFD28D7D8B756C3557569CB876D"]
[MSG:DBG: index.html.gz hash "9A6C71930393EB2638A427F1C1C8956685993A47308FD39381658D494C2024C1"]
[MSG:INFO: Telnet started on port 23]
```

### User Interface Software

WebUI, picocom, custom pendant

### What happened?

Hi,
I have fully working CNC machine and decided to add pedant today. I assembled pedant to G-code converter on rp2040 and verified it's working and sending correct G-code on correct baud rate ($J=G91 F999 X0.00 Y0.00 Z0.00\n @ 115200 every 300ms). I also verified that it's connected to the correct pin (pin 4) of ESP32 by shorting given pin to 0V or 3V3 and issuing $GD to test state of input pin. Not only that, but I also checked continuity from one microcontroller to the other. I also connected TX pin (25) of ESP32 to logic analyzer and verified I'm receiving reports when I issued jog using web interface.

The problem is that it seems as if the controller is not reacting to commands from external UART. The requests are not getting acknowledged (no 'ok', 'ERROR' or any other response) and position of machine in web interface isn't changing (tried both auto and poll). 

What has been tried:
changing UART1<->UART2
different TX and RX pins, swapping RX and TX. I'm aware that gpio.25, currently used is going through buffer. 
sending the same commands, by hand using USB serial.

### GCode File

$J=G91 F999 X0.00 Y0.00 Z0.00

### Other Information

The machine has steppers disconnected and PSU temporary changed for linear, lab bench PSU to eliminate all interference issues.
All signal verification was done both with logic analyzer (to decode data exactly, check baud rates, etc) and oscilloscope (to make sure that levels are proper digital levels).

bdring, 2024-01-04:
```yaml
#jog uart
uart2:
  txd_pin: gpio.25
  rxd_pin: gpio.4
  #rts_pin: NO_PIN
  baud: 115200
  mode: 8N1
#jog setup
uart_channel1:
   uart_num: 2
   report_interval_ms: 100
   all_messages: true
   ```

I have never tried mixing UART numbers. It should work, but I do not have time to test. Try changing `uart_channel1:` to `uart_channel2:`

MitchBradley, 2024-01-04:
I just tried uart_channel2: referring to uart_num: 1  and it worked.  The board is Fysetc E4 talking to an M5 Dial Pendant.

MitchBradley, 2024-01-04:
I tried the other way too - uart_channel1: referring to uart_num: 2 .  That worked as well.

MitchBradley, 2024-01-04:
I would try connecting a USB-serial dongle to the UART you want to use as a pendant and try typing characters manually.

closed 2024-02-02 by bdring