bdring/FluidNC #1102: Problem: Custom XyyZ RMT setup problem

MarcinanProfesjonalista, 2024-01-03:
### Wiki Search Terms

Setup configuration problem for custom board

### Controller Board

Custom board 
Schematics
[image]
Testing setup
[image]

### Machine Description

XYYZ machine. Runned on basic HSS86 that need STEP, DIR, ENABLE, ALARM,( PENDING) singals. 

### Input Circuits

```yaml
I'm testing Output for X axis. Does ESP32 put any pulses on outputs.
```

### Configuration file

```yaml
board: Marcin_nowa
name: 4 External XYYZ
meta: 2024-01-01 Marcin
stepping:
  engine: RMT
  idle_ms: 250
  pulse_us: 4
  dir_delay_us: 1
  disable_delay_us: 0

axes:
  shared_stepper_disable_pin: gpio.13
  x:
    steps_per_mm: 100.000
    max_rate_mm_per_min: 5000.000
    acceleration_mm_per_sec2: 100.000
    max_travel_mm: 300.000
    soft_limits: false
    homing:
      cycle: 2
      positive_direction: true
      mpos_mm: 0.000
      feed_mm_per_min: 100.000
      seek_mm_per_min: 200.000
      settle_ms: 500
      seek_scaler: 1.100
      feed_scaler: 1.100

    motor0:
      limit_neg_pin: gpio.39
      limit_pos_pin: gpio.36
      limit_all_pin: NO_PIN
      hard_limits: false
      pulloff_mm: 1.000
      standard_stepper:
        step_pin: gpio.18
        direction_pin: gpio.5
        disable_pin: NO_PIN

  y:
    steps_per_mm: 100.000
    max_rate_mm_per_min: 5000.000
    acceleration_mm_per_sec2: 100.000
    max_travel_mm: 300.000
    soft_limits: false
    homing:
      cycle: 3
      positive_direction: false
      mpos_mm: 0.000
      feed_mm_per_min: 100.000
      seek_mm_per_min: 200.000
      settle_ms: 500
      seek_scaler: 1.100
      feed_scaler: 1.100

    motor0:
      limit_neg_pin: gpio.35
      limit_pos_pin: gpio.34
      limit_all_pin: NO_PIN
      hard_limits: false
      pulloff_mm: 1.000
      standard_stepper:
        step_pin: gpio.17
        direction_pin: gpio.16
        disable_pin: NO_PIN
    
    motor1:
      limit_neg_pin: gpio.32
      limit_pos_pin: gpio.33
      limit_all_pin: NO_PIN
      hard_limits: false
      pulloff_mm: 1.000
      standard_stepper:
        step_pin: gpio.4
        direction_pin: gpio.0
        disable_pin: NO_PIN

  z:
    steps_per_mm: 100.000
    max_rate_mm_per_min: 5000.000
    acceleration_mm_per_sec2: 100.000
    max_travel_mm: 150.000
    soft_limits: false
    homing:
      cycle: 1
      positive_direction: true
      mpos_mm: 150.000
      feed_mm_per_min: 100.000
      seek_mm_per_min: 800.000
      settle_ms: 500
      seek_scaler: 1.100
      feed_scaler: 1.100

    motor0:
      limit_neg_pin: gpio.27
      limit_pos_pin: gpio.25
      limit_all_pin: NO_PIN
      hard_limits: false
      pulloff_mm: 1.000
      standard_stepper:
        step_pin: gpio.2
        direction_pin: gpio.15
        disable_pin: NO_PIN

probe:
  pin: gpio.26
  check_mode_start: false
  
control:
  safety_door_pin: NO_PIN
  reset_pin: NO_PIN
  feed_hold_pin: gpio.23
  cycle_start_pin: NO_PIN
  macro0_pin: NO_PIN
  macro1_pin: NO_PIN
  macro2_pin: NO_PIN
  macro3_pin: NO_PIN
  
        
i2so:
  bck_pin: NO_PIN
  data_pin: NO_PIN
  ws_pin: NO_PIN

spi:
  miso_pin: NO_PIN
  mosi_pin: NO_PIN
  sck_pin: NO_PIN

sdcard:
  card_detect_pin: NO_PIN
  cs_pin: NO_PIN

start:
  must_home: false
```

### Startup Messages

```text
Grbl 3.7 [FluidNC v3.7.12 (wifi) '$' for help]
ets Jul 29 2019 12:21:46

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:1184
load:0x40078000,len:13220
ho 0 tail 12 room 4
load:0x40080400,len:3028
entry 0x400805e4

[MSG:INFO: FluidNC v3.7.12 https://github.com/bdring/FluidNC]
[MSG:INFO: Compiled with ESP32 SDK:v4.4.4]
[MSG:INFO: Local filesystem type is littlefs]
[MSG:INFO: Configuration file:marcin_nowa.yaml]
[MSG:INFO: Machine 4 External XYYZ]
[MSG:INFO: Board Marcin_nowa]
[MSG:INFO: I2SO BCK:NO_PIN WS:NO_PIN DATA:NO_PIN]
[MSG:INFO: Not setting up I2SO: WS pin has incorrect capabilities]
[MSG:INFO: Stepping:RMT Pulse:4us Dsbl Delay:0us Dir Delay:1us Idle Delay:250ms]
[MSG:INFO: Axis count 3]
[MSG:INFO: Shared stepper disable gpio.13]
[MSG:INFO: Axis X (-300.000,0.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     standard_stepper Step:gpio.18 Dir:gpio.5 Disable:NO_PIN]
[MSG:INFO:  X Neg Limit gpio.39]
[MSG:INFO:  X Pos Limit gpio.36]
[MSG:INFO: Axis Y (0.000,300.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     standard_stepper Step:gpio.17 Dir:gpio.16 Disable:NO_PIN]
[MSG:INFO:  Y Neg Limit gpio.35]
[MSG:INFO:  Y Pos Limit gpio.34]
[MSG:INFO:   Motor1]
[MSG:INFO:     standard_stepper Step:gpio.4 Dir:gpio.0 Disable:NO_PIN]
[MSG:INFO:  Y2 Neg Limit gpio.32]
[MSG:INFO:  Y2 Pos Limit gpio.33]
[MSG:INFO: Axis Z (0.000,150.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     standard_stepper Step:gpio.2 Dir:gpio.15 Disable:NO_PIN]
[MSG:INFO:  Z Neg Limit gpio.27]
[MSG:INFO:  Z Pos Limit gpio.25]
[MSG:INFO: feed_hold_pin gpio.23]
[MSG:INFO: Kinematic system: Cartesian]
[MSG:INFO: Using spindle NoSpindle]
[MSG:INFO: Probe Pin: gpio.26]
[MSG:INFO: Connecting to STA SSID:DEKARSKA5_LOKAL7]
[MSG:INFO: Connecting.]
[MSG:INFO: Connecting..]
[MSG:INFO: Connected - IP is 172.16.31.250]
[MSG:INFO: WiFi on]
[MSG:INFO: Start mDNS with hostname:http://fluidnc.local/]
[MSG:INFO: SSDP Started]
[MSG:INFO: HTTP started on port 80]
[MSG:INFO: Telnet started on port 23]

Grbl 3.7 [FluidNC v3.7.12 (wifi) '$' for help]
ok
[VER:3.7 FluidNC v3.7.12:]
[OPT:PHS]
[MSG: Machine: 4 External XYYZ]
[MSG: Mode=STA:SSID=DEKARSKA5_LOKAL7:Status=Connected:IP=172.16.31.250:MAC=08-D1-F9-E7-15-9C]
ok

Grbl 3.7 [FluidNC v3.7.12 (wifi) '$' for help]
```

### What happened?

I'm upgrading from old ESP32 GRBL software to that FluidNC becouse it got beter safety functions. But It don't send any singals on outputs.

MitchBradley, 2024-01-03:
What command are you using to try to make it send step pulses?

bdring, 2024-01-03:
In the photo, what are you trying to measure and exactly what commands are you sending.

MarcinanProfesjonalista, 2024-01-03:
Im pressing +X on website
$J=G91 G21 F1000 X10
$
$=ON
ok
<Idle|MPos:0.000,0.000,0.000|FS:0,0|WCO:90.000,30.900,0.000>
$J=G91 G21 F1000 X10
ok
<Jog|MPos:0.290,0.000,0.000|FS:1000,0|Ov:100,100,100>

bdring, 2024-01-03:
Pin 5 is a direction pin

MarcinanProfesjonalista, 2024-01-03:
Yes. Im mesuring both pins. Step pin and Dir pin on osciloscope

bdring, 2024-01-03:
Thank you for answering my question from my earlier post

bdring, 2024-01-03:
Your startup messages look fine. It is most likely a hardware or config issue. I know you say that it worked before, but, FluidNC works for everyone else.

I suggest you start with tests using a meter rather than the o'scope on the disable pin (gpio.13). Follow these instructions.

http://wiki.fluidnc.com/en/support/faq#motors-do-not-move

http://wiki.fluidnc.com/en/support/controller_design_guidelines#developer-policy-on-user-created-hardware

MitchBradley, 2024-01-03:
It works for me. I loaded your config file into a bare ESP32, connected a scope to gpio18, and ran jog X+10 from webui.

Here is a single step pulse
[image]

Here is a group of pulses from the deceleration phase of the jog
[image]

Maybe your scope is not triggering

MitchBradley, 2024-01-03:
[image]

MarcinanProfesjonalista, 2024-01-03:
Yes. It's working correctly. Can I don't close this issue? I want to create 1 function that works after an alarm from feed_hold_pin: gpio.23. 
[image]

I want to create custom function, that when feed_hold_pin is active, the work can be resumed by sending global enable pulse on shared_stepper_disable_pin: gpio.13

All becouse of Alarm feature of HSS86, that sends alarm when motor cant keep up (or stall) on Alarm output. It can be restarted by just sending pulse on enable input of HSS86. 

Algoritm:
• Motor stall: HSS86 gives HIGH to GPIO13. Feed Cycle stopped. 
○ User checks machine
• If Everything looks ok tries to resume feed. Presses macro button
• ESP32 Resets alarm by sending HIGH pulse on gpio.23. 
• Checks if alarm is still on. If not, resume work.

MitchBradley, 2024-01-04:
To clear an alarm you must send $X.  I do not think that will work from a macro because macros cannot run in Alarm state, only in Idle state.

I use fault_pin in the control section to handle alarms from closed loop steppers.  When it happens, the machine position is almost never correct, so resuming is inappropriate.  If you want to resume in this situation, you would have to change the FluidNC code.

MarcinanProfesjonalista, 2024-01-05:
I will test the Macros later. Now i need to get going with that shield. At the moment I'm sure it sends pulses.

closed 2024-01-05 by MarcinanProfesjonalista