bdring/FluidNC #1106: Problem: ERROR:20 when reversing pwm spindle

BeatrootRissole, 2024-01-06:
### Wiki Search Terms

spindle
pwm
reverse_pin
forward_pin

### Controller Board

mks tinybee

### Machine Description

3080 gantry router with TMC2209 steppers, single motors on all axis, Spindle is a genmitsu 24V 20K RPM.

### Input Circuits

```yaml
nothing custom to show here, its just pins to devices straight from the mks tinybee with nothing in between.
```

### Configuration file

```yaml
board: MKS TinyBee V1.0_001
name: K40 Laser
meta: (01.02.2022) by Skorpi

kinematics:
  Cartesian:

stepping:
  engine: I2S_STATIC
  idle_ms: 0
  pulse_us: 4
  dir_delay_us: 1
  disable_delay_us: 0

axes:
  x:
    # X
    steps_per_mm: 800
    max_rate_mm_per_min: 1000
    acceleration_mm_per_sec2: 100
    max_travel_mm: 320
    homing:
      cycle: 2
      positive_direction: false
      mpos_mm: 0.000
      feed_mm_per_min: 300.000
      seek_mm_per_min: 1200
      settle_ms: 500
      seek_scaler: 1.100
      feed_scaler: 1.100

    motor0:
      limit_neg_pin: gpio.33
      hard_limits: true
      pulloff_mm: 5
      stepstick:
        step_pin: I2SO.1
        direction_pin: I2SO.2
        disable_pin: I2SO.0

  y:
    # Y
    steps_per_mm: 800
    max_rate_mm_per_min: 1000
    acceleration_mm_per_sec2: 100.000
    max_travel_mm: 200.000
    homing:
      cycle: 2
      positive_direction: false
      mpos_mm: 0.000
      feed_mm_per_min: 300.000
      seek_mm_per_min: 1200.000
      settle_ms: 500
      seek_scaler: 1.100
      feed_scaler: 1.100

    motor0:
      limit_neg_pin: gpio.32
      hard_limits: true
      pulloff_mm: 15.000
      stepstick:
        step_pin: I2SO.4
        direction_pin: I2SO.5:low
        disable_pin: I2SO.3

  z:
    # Z
    steps_per_mm: 800
    max_rate_mm_per_min: 600
    acceleration_mm_per_sec2: 40
    max_travel_mm: 71.020
    homing:
      cycle: 1
      positive_direction: true
      mpos_mm: 0.000
      feed_mm_per_min: 300.000
      seek_mm_per_min: 1000.000
      settle_ms: 500
      seek_scaler: 1.100
      feed_scaler: 1.100

    motor0:
      limit_neg_pin: gpio.22
      hard_limits: true
      pulloff_mm: 5
      stepstick:
        step_pin: I2SO.7
        direction_pin: I2SO.8
        disable_pin: I2SO.6

#  a:
#    # E0
#    steps_per_mm: 157.750
#    max_rate_mm_per_min: 18000.000
#    acceleration_mm_per_sec2: 1500.000
#    max_travel_mm: 325.000
#    soft_limits: true
#    homing:
#      cycle: 1
#      positive_direction: false
#      mpos_mm: 0.000
#      feed_mm_per_min: 300.000
#      seek_mm_per_min: 5000.000
#      settle_ms: 500
#      seek_scaler: 1.100
#      feed_scaler: 1.100
#
#    motor0:
#      limit_neg_pin: gpio.35
#      hard_limits: false
#      pulloff_mm: 2.000
#      stepstick:
#        step_pin: I2SO.10
#        direction_pin: I2SO.11:low
#        disable_pin: I2SO.9

#  b:
#    # E1
#    steps_per_mm: 157.750
#    max_rate_mm_per_min: 12000.000
#    acceleration_mm_per_sec2: 500.000
#    max_travel_mm: 80.000
#    soft_limits: true
#    homing:
#      cycle: 0
#      positive_direction: false
#      mpos_mm: 0.000
#      feed_mm_per_min: 300.000
#      seek_mm_per_min: 1000.000
#      settle_ms: 500
#      seek_scaler: 1.100
#      feed_scaler: 1.100
#
#    motor0:
#      limit_neg_pin: gpio.2
#      hard_limits: false
#      pulloff_mm: 1.000
#      stepstick:
#        step_pin: I2SO.13
#        direction_pin: I2SO.14
#        disable_pin: I2SO.12

i2so:
  bck_pin: gpio.25
  data_pin: gpio.27
  ws_pin: gpio.26

spi:
  miso_pin: gpio.19
  mosi_pin: gpio.23
  sck_pin: gpio.18

sdcard:
  cs_pin: gpio.5
  card_detect_pin: NO_PIN
  frequency_hz: 5000000

control:
  safety_door_pin: NO_PIN
  reset_pin: NO_PIN
  feed_hold_pin: NO_PIN
  cycle_start_pin: NO_PIN
  # EXP1 BTN_ENC
  # macro0_pin: gpio.4:low:pu
  macro1_pin: NO_PIN
  macro2_pin: NO_PIN
  macro3_pin: NO_PIN

macros:
  startup_line0:
  startup_line1:
  macro0: G38.2 G91 F80 Z-20 P1.63
  macro1: 
  macro2:
  macro3:

coolant:
  flood_pin: NO_PIN
  mist_pin: NO_PIN
  delay_ms: 0

probe:
# 3DTouch pin
  pin: gpio.2
  check_mode_start: true

pwm:
  pwm_hz: 2500
#  direction_pin: NO_PIN
  reverse_pin: gpio.4:high
  forward_pin: gpio.13:high
  # EXP1 BTN_ENC
  output_pin: gpio.17:high
  # FAN1
  # enable_pin: I2SO.16
  # enable_pin: gpio.4
  s0_with_disable: true
  spinup_ms: 4000
  spindown_ms: 4000
  tool_num: 0
  speed_map: 0=0.000% 20000=90.000%
# 135=0mA 270=5mA 400=10mA 700=16mA
user_outputs:
  analog0_pin: NO_PIN
  analog1_pin: NO_PIN
  analog2_pin: NO_PIN
  analog3_pin: NO_PIN
  analog0_hz: 5000
  analog1_hz: 5000
  analog2_hz: 5000
  analog3_hz: 5000
  digital0_pin: NO_PIN
  digital1_pin: NO_PIN
  digital2_pin: NO_PIN
  digital3_pin: NO_PIN

start:
  must_home: true

 # USB  TX0(1)  RX0(3)
 # J3-> TX2(17) RX2(16)
 # J1-> I2SO.23 I2SO.22
 # TH1(36)        TH2(34)        THB(39)
 # HE0(I2SO.17)   HE1(I2SO.18)   HBED(I2SO.16)
 # FAN1(I2SO.19)  FAN2(I2SO.20)
 #                   EXP1                                   EXP2
 #(BEEPER) I2SO.21 | 1  2 | 13 (BTN_ENC)    (SPI MISO) 19 | 1  2 | 18 (SPI SCK)
 #    (LCD_EN)  21 | 3  4 |  4 (LCD_RS)      (BTN_EN1) 14 | 3  4 |  5 (SPI CS)
 #    (LCD_D4)   0 | 5  6   16 (LCD_D5)      (BTN_EN2) 12 | 5  6   23 (SPI MOSI)
 #    (LCD_D6)  15 | 7  8 | 17 (LCD_D7)      (SPI_DET) 34 | 7  8 | RESET
 #             GND | 9 10 | 5V                        GND | 9 10 | 3.3V
```

### Startup Messages

```text
$ss
[MSG:INFO: FluidNC v3.7.8]
[MSG:INFO: Compiled with ESP32 SDK:v4.4.4]
[MSG:INFO: Local filesystem type is spiffs]
[MSG:INFO: Configuration file:config.yaml]
[MSG:WARN: Ignored key reverse_pin]
[MSG:WARN: Ignored key forward_pin]
[MSG:INFO: Machine K40 Laser]
[MSG:INFO: Board MKS TinyBee V1.0_001]
[MSG:INFO: I2SO BCK:gpio.25 WS:gpio.26 DATA:gpio.27]
[MSG:INFO: SPI SCK:gpio.18 MOSI:gpio.23 MISO:gpio.19]
[MSG:INFO: SD Card cs_pin:gpio.5 detect:NO_PIN freq:5000000]
[MSG:INFO: Stepping:I2S_static Pulse:4us Dsbl Delay:0us Dir Delay:1us Idle Delay:0ms]
[MSG:INFO: Axis count 3]
[MSG:INFO: Axis X (0.000,320.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     stepstick Step:I2SO.1 Dir:I2SO.2 Disable:I2SO.0]
[MSG:INFO:  X Neg Limit gpio.33]
[MSG:INFO: Axis Y (0.000,200.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     stepstick Step:I2SO.4 Dir:I2SO.5:low Disable:I2SO.3]
[MSG:INFO:  Y Neg Limit gpio.32]
[MSG:INFO: Axis Z (-71.020,0.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     stepstick Step:I2SO.7 Dir:I2SO.8 Disable:I2SO.6]
[MSG:INFO:  Z Neg Limit gpio.22]
[MSG:INFO: Kinematic system: Cartesian]
[MSG:INFO: PWM Spindle Ena:NO_PIN Out:gpio.17 Dir:NO_PIN Freq:2500Hz Period:16383]
[MSG:INFO: Using spindle PWM]
[MSG:INFO: Probe Pin: gpio.2]
[MSG:INFO: Connecting to STA SSID:LR24]
[MSG:INFO: Connecting.]
[MSG:INFO: Connecting..]
[MSG:INFO: Connected - IP is 192.168.1.229]
[MSG:INFO: WiFi on]
[MSG:INFO: Start mDNS with hostname:http://fluidnc.local/]
[MSG:INFO: SSDP Started]
[MSG:INFO: HTTP started on port 80]
[MSG:INFO: Telnet started on port 23]
ok
```

### User Interface Software

WebUI

### What happened?

received the error ERROR:20
Unsupported or invalid g-code command found in block. 
When trying to reverse the spindle direction. 
I have just configured forward and reverse. I have only ever used forward direction previously which works fine. 
If i swap the forward and reverse pins and activate spindle in forward direction then the spindle runs in reverse so my pwm controller seems fine to me.  

### GCode File

Wasnt running any gcode, just activating forward and reverse spindle direction in the spindle section of the webui

bdring, 2024-01-06:
```
[MSG:INFO: PWM Spindle Ena:NO_PIN Out:gpio.17 Dir:NO_PIN Freq:2500Hz Period:16383]
```

You need a direction pin to use M4

BeatrootRissole, 2024-01-06:
Ok thanks.

closed 2024-01-06 by bdring