bdring/FluidNC #1107: Feature: Separate brain from end device expanding servo control options.

jharvey, 2024-01-08:
### Machine Context

I am CNC-ing a Bridgeport. Nothing too special just x,y,z with RS485 speed control. I want to develop a low cost servo / closed loop with robust electrical immunity. Which can be used by anyone. 

Servos commonly have ground loop or other noise issues. I want a system which is robust against this noise and the problems that this noise brings.

### Feature Description

FliudNC should allow local servo control / DC motor with an encoder like a servocity SKU 638258. 
https://www.servocity.com/624-rpm-premium-planetary-gear-motor/

FluidNC should allow a local servo to take commands from a display / brain like Makerfabs SKU MTESPS37. 
https://www.makerfabs.com/esp32-s3-parallel-tft-with-touch-7-inch.html

Then with coms like ESPNOW the display brain would command a variety of local servos to react. Local servo's would deal with tracking an encoder, processing limit switch's and PIDs. While the display would deal with higher level management. Similar to klipper topology.

Servos benefit from short wires which result in short antennas. Ground loops and a variety of RF issues make implementing remote servo control difficult for the average user. This difficult situation causes costs and complexity to go up. A small resistance in a connection or small signal delays in a servo circuit can cause tank circuits which causes motors to overvoltage internally and burn up the motors internal windings. The complexity of servo's decreases greatly if the wires from the driver to the motor are short.

By separating the lower level PID and similar processing into a servo assembly, FluidNC would be more expandable. Every time a servo is added, the overall bit pumping capacity is also increased. More memory and more processing resources are naturally added to an expanded system.

By segregating the high level management from the low level tasks it minimizes the impacts of bugs. A small change to the web serving code is far less likely to influence time critical events like decoding a motor encoder. 

By segregating like this it also allows FOC servos or other devices to be implemented without impacting the higher level code. 

### Other Approaches

Stepper signals could be sent to devices like makerbase SERVO42D or more traditional off the shelf servo controllers like a Gecko.

However this typically results in the brain having less information to work with. With less information accessable to the brain, it becomes difficult to add features like abnormal friction detection / alarming or object collision detection. 

As well with servo control being under someone else's domain it makes tracking and debugging issues more difficult. It's difficult to get the information from other devices which are often non open devices. 

### How I Can Help

I'm an electro-mechanical engineer by schooling and trade. 

I can read and talk code reasonably well. I can design circuits or similar hardware related tasks. However I'm weak at originating code. 

I can design circuits with KiCAD or preform similar hardware related tasks. I can design equipment and get it into the hands of software developers. 

I can also do regression testing to validate designs or I can provide information to software developers who might not have access to scopes, logic analysers, etc. 

I can help document features. 

Years ago ginger did a bunch of AVR servo stuff. If FluidNC takes servo under its umbrella, that code would likely be a useful reference. I added a magnetic encoder to V3. If that's useful, I could likely find that reference as well.
https://github.com/ginge/OpenServo

MitchBradley, 2024-01-08:
This would be a lot of work for what would be, in my estimation, very little payoff.

jharvey, 2024-01-08:
@MitchBradley do you mean you see servo or features like collision detection and friction warning as low payoff? 

Perhaps you see less effort methods to obtain such features? I know these features are very common in the commercial CNC industry so I don't expect these features to be considered low payoff. Could you elaborate? I'm not sure if you're saying you don't personally see the value so you're not personally interested in developing these features. I'm guessing you are saying that other feature requests are lowered effort and therefore should hold the attention of the developers. 

I'm not sure if I should offer supporting information to show the value of such features to encourage the software development or if you're saying this is a hobby device and these features don't have value like they do in commercial applications.

Let me know if there is something I can do to help develop these features.

bdring, 2024-01-08:
To have FluidNC do all the work of closing the loops with encoders would require a lot of work and use a lot of I/O. To do it over a com channel is not really realistic. The bandwidth and timing is just too much. A lot of people are using closed loop with step and direction inputs. You can setup the motors with RS485, etc and monitor them with alarm inputs. A lot of people are using closed loop with step and direction inputs. You can setup the motors with RS485, etc and monitor them with alarm inputs.

Every feature is compiled into the same firmware so we need to be careful about big features that appeal to a small group. That might be the basic for Mitch's answer.

A lot of people are using closed loop with step and direction inputs. You can setup the motors with RS485, etc and monitor them with alarm inputs.

Note: The newest iteration of the Maslow CNC project uses closed loop motors and FluidNC.

jharvey, 2024-01-08:
@bdring I understand that at this time FluidNC has low interest in doing servo, and using an external servo driver is encouraged. 

I understand that a servo driver will have some kind of processor in it, and instead of FluidNC developing a driver which allows FluidNC to have control over low level control, you prefer that someone else develop that part of the system and FluidNC interfaces via stepper pulses and discrete alarm pulses. 

Do you know of a compatible open DC servo? I've only found closed DC servo drivers so far. I see some ESP32 FOC projects, but I've not found DC servo's yet. The closest I know of is the now defunct AVR based OpenServo. Perhaps I should see if I can port OpenServo to an ESP32 and make an open DC servo. Then once that part is working, then perhaps FluidNC might consider streamlining the overall process. I see the stepper pulses as a slow a problematic interface for servo control. 

If I had a serial stream, it would simplify my efforts to port the AVR based OpenServo code to an ESP32 platform. For me to make my own DC servo, I'd have to develop a pulse decoder, before I'd have a commanded position which I could use for servo control. How hard would it be to get a serial stream out of FluidNC? Would it be hard to get a serial stream formatted something like this "X=1.983Y=-2.26Z=5.56"? The default ESP-NOW bit rate is 1 Mbps. So the serial stream could potentially be that fast. I'm chipping away at learning the FluidNC code now. I'm assuming you currently have those in a variable, and I'm assuming a reasonably small bit of code could put those bytes of data into a USART buffer once ever mS or so.

MitchBradley, 2024-01-08:
FluidNC is solidly in the DIY/Hobbyist space, not the commercial space.  The commercial machines that use servos typically cost many thousands of dollars - often tens of thousands.  We charge nothing, and the donation rate is not enough to pay for the one cup of coffee I drink per day.  So for me, "payoff" means "appeals to something that I am interested in".  If someone were willing to pay the amount of money it would actually cost to do the development and then support the feature over time, that too would constitute payoff.  Lacking that incentive, just addressing a new and probably small market segment is just not interesting.  I know that the commercial market as a whole is large, but to address it, you need to have a company with all that entails.  Starting a company to mix it up in the big machine space is not where we are at.

MitchBradley, 2024-01-08:
The Dynamixel2 Motor class already does something related to what you want.  Maybe it can be used as a starting point for your effort.

bdring, 2024-01-08:
The RC servo and dynamixel use something similar. FluidNC is running a virtual machine and just sends periodic notification to those closed loop device. They are basically chasing the position. The update rates are configurable, but I would worry about going fast than 50Hz. Those are not very accurate actuators, so it is fine. 

https://www.youtube.com/watch?v=wPDP6ckP0sQ

Keep in mind 1 Mbps fills up fast when you are sending big strings of Bytes.

jharvey, 2024-01-09:
@bdring @MitchBradley looks like dynamixel is doing a similar topology to what I'm proposing. As well there are other examples of this similar topology. I'm not purposing it because others do it, I'm purposing it because I feel it's a good approach. If you ever get into safety rated devices you'll need to breakup the  the code like this for a code validation standpoint. With the potential of heavy machinery I'd encourage striving for safety options. 

I don't see dynamixel schematic or hardware source files posted. It appears the code is released for several of the motors and there's an interface document. So it's a semi open platform. 

I'm looking for a fully open platform. I want the freedom to diagnose and fix issues. With dynamixel it appears I'd have to blindly replace servos if something is not acting properly. 

I'd have to look closer, but I understand the 1Mbps is reasonably long range and you can go faster if you trade for a shorter distance. I think it's also only 1 channel of many potential channels. So each device can likely have a dedicated channel. As well yagi PCB antennas can avoid stray signals.

MitchBradley, 2024-01-09:
If you want to use heavy machinery safely, free open source code maintained by unpaid volunteers is probably not the answer.  Safety requires extensive testing, homologation, interfacing with regulatory agencies, and many other factors that end up costing a lot of money.  You get what you pay for, or more apropos to this situation, you don't get what you don't pay for.

jharvey, 2024-01-09:
I'm not saying this should be safety rated. I'm saying it's not a bad idea to follow practices which result in robust code. It's also a good idea to follow MIRSA C. Not required but a good idea.

jharvey, 2024-01-10:
Looks like the ESPNOW servo already exists. https://docs.dagor.dev/contact Hmmm, where to buy.... I'll poke him and see if he's got some he might be willing to sell.

bdring, 2024-02-02:
Can this issue be closed?

closed 2024-02-03 by bdring