bdring/FluidNC #1108: Feature: new laser control board

f2knpw, 2024-01-10:
### Machine Context

HI,
I have finished my Trotec laser retrofit and built a FluidNC control board.
Works very well. Everything is open  Hardware. Details here : [](https://hackaday.io/project/193893-retrofitting-an-old-laser-engraver)

I would be glad if you could add this little board to the "hardware" page. 

Thanks a lot for all the efforts made for this great FluidNC firmware

Regards
JP

### Feature Description

new control board

### Other Approaches

ESP32

### How I Can Help

I can provide all the needed files (schematics, PCB, gerber)

bdring, 2024-01-10:
You add it yourself. There are details here about how to get editor rights.

http://wiki.fluidnc.com/en/wiki_contributions

f2knpw, 2024-01-10:
Ok thanks.
I have created the account. Waiting for activation please.

bdring, 2024-01-10:
Activated.

closed 2024-01-10 by f2knpw