bdring/FluidNC #1110: Problem: Axis direction and control

Toolman19, 2024-01-12:
### Wiki Search Terms

Axis 
steps per mm
max rate
acceleration
max travel
direction gpio state
soft limit state

### Controller Board

XPro V5 XYYZ

### Machine Description

Wookman QueenBee Pro with xPro V5 controller and Nema23 drives and Huanyang 1.5kw air cooled spindle.

### Configuration file

```yaml
$G
[GC:G0 G54 G17 G21 G90 G94 M5 M9 T0 F0 S0]
ok
<Idle|MPos:0.000,0.000,0.000|FS:0,0|WCO:0.000,0.000,0.000>
$LocalFS/Show=config.yaml
board: XPro V5
name: xPro V5 Ondrej 
stepping:
  engine: RMT
  idle_ms: 255
  pulse_us: 4
  dir_delay_us: 1
  disable_delay_us: 0
axes:
  shared_stepper_disable_pin: NO_PIN
  x:
    steps_per_mm: 199.999
    max_rate_mm_per_min: 2500.000
    acceleration_mm_per_sec2: 100.000
    max_travel_mm: 775.000
    soft_limits: true
    homing:
      cycle: 2
      positive_direction: false
      mpos_mm: 150.000
      feed_mm_per_min: 100.000
      seek_mm_per_min: 2000.000
      settle_ms: 500
      seek_scaler: 1.100
      feed_scaler: 1.100
    motor0:
      limit_neg_pin: gpio.35:low
      limit_pos_pin: NO_PIN
      limit_all_pin: NO_PIN
      hard_limits: false
      pulloff_mm: 2.500
      tmc_5160:
        step_pin: gpio.12
        direction_pin: gpio.14:low
        use_enable: true
        cs_pin: gpio.17
        spi_index: 1
        r_sense_ohms: 0.050
        run_amps: 1.800
        hold_amps: 0.250
        microsteps: 8
        toff_disable: 0
        toff_stealthchop: 5
        run_mode: CoolStep
[... 258 lines cut ...]
[MSG:INFO: Y Axis driver test passed]
[MSG:INFO: Y2 Axis driver test passed]
[MSG:INFO: Z Axis driver test passed]
[MSG:INFO: safety_door_pin gpio.16:low]
[MSG:INFO: macro0_pin gpio.13:low]
[MSG:INFO: macro1_pin gpio.0:low]
[MSG:INFO: Kinematic system: Cartesian]
[MSG:INFO: Huanyang Spindle  Tx:gpio.4 Rx:gpio.25 RTS:NO_PIN Baud:9600]
[MSG:INFO: Using spindle Huanyang]
[MSG:INFO: Mist coolant gpio.21]
[MSG:INFO: Probe Pin: gpio.22:low]
[MSG:INFO: STA SSID is not set]
[MSG:INFO: AP SSID FluidNC IP 192.168.0.1 mask 255.255.255.0 channel 1]
[MSG:INFO: AP started]
[MSG:INFO: WiFi on]
[MSG:INFO: Captive Portal Started]
[MSG:INFO: HTTP started on port 80]
[MSG:INFO: Huanyang PD0011, PD005 Freq range (120,400) Hz (7200,24000) RPM]
[MSG:INFO: Telnet started on port 23]

Grbl 3.7 [FluidNC v3.7.12 (wifi) '$' for help]
[MSG:INFO: Huanyang PD144 Rated RPM @ 50Hz:3000]
[MSG:INFO: Huanyang PD143 Poles:2]
[MSG:INFO: Huanyang PD014 Accel:10.000]
[MSG:INFO: Huanyang PD015 Decel:10.000]

closed 2024-01-15 by Toolman19

MitchBradley, 2024-01-15:
What was wrong?

Toolman19, 2024-01-15:
Hi MitchIt was a combination of the hard and soft limits that seem to be causing the issue in the end. I had to go back to the start and rewrite the config. Good learning curve. ChrisSent from my iPhone