bdring/FluidNC #1116: "check_security" function is broken on Linux

weskerfoot, 2024-01-21:
### Wiki Search Terms

N/A

### Controller Board

ESP32-MINI-1

### Machine Description

N/A

### Configuration file

```yaml
N/A
```

### Startup Messages

```text
N/A
```

### What happened?

Tried running ./install-wifi.sh on Arch Linux with esptool.py v3.0 installed in virtualenv

Got the following output (I added set -x to the script)

```
++ Bootloader='0x1000 wifi/bootloader.bin'
++ Bootapp='0xe000 common/boot_app0.bin'
++ Firmware='0x10000 wifi/firmware.bin'
++ Partitions='0x8000 wifi/partitions.bin'
+ install
+ check_security
+ esptool_basic dump_mem 0x3ff5a018 4 SecurityFuses.bin
+ echo esptool.py --chip esp32 --baud 115200 dump_mem 0x3ff5a018 4 SecurityFuses.bin
esptool.py --chip esp32 --baud 115200 dump_mem 0x3ff5a018 4 SecurityFuses.bin
+ esptool.py --chip esp32 --baud 115200 --chip esp32 --baud 115200 dump_mem 0x3ff5a018 4 SecurityFuses.bin
esptool.py v3.0
Found 1 serial ports
Serial port /dev/ttyUSB0
Connecting......
Chip is ESP32-U4WDH (revision 3)
Features: WiFi, BT, Dual Core, 240MHz, Embedded Flash, VRef calibration in efuse, Coding Scheme None
Crystal is 40MHz
MAC: 08:b6:1f:0c:5f:58
Uploading stub...
Running stub...
Stub running...
Traceback (most recent call last):
  File "/home/wes/.local/bin/esptool.py", line 3969, in <module>
    _main()
  File "/home/wes/.local/bin/esptool.py", line 3962, in _main
    main()
  File "/home/wes/.local/bin/esptool.py", line 3604, in main
    operation_func(esp, args)
  File "/home/wes/.local/bin/esptool.py", line 2840, in dump_mem
    print_overwrite("Read %d bytes" % f.tell(), last_line=True)
                                      ^^^^^^^^
ValueError: I/O operation on closed file
+ test 1 '!=' 0
+ echo esptool failed
esptool failed
+ exit 1
```

If I comment out the `check_security` function it works (have not tested the actual software yet though)

Looks to be a bug in esptool.py or something about my particular board.

I tried different cables and usb ports, no change. I tried erasing the flash with esptool, and that didn't change anything either.

MitchBradley, 2024-01-21:
We are not highly motivated to fix it given that the new Web Installer https://installer.fluidnc.com/ is generally more convenient and works better than the install scripts.  It is more likely that we will just drop the install scripts entirely.  Linux has proven to be difficult to support.

weskerfoot, 2024-01-22:
Ok, I will try the web installer. I think it will be useful to have a comment somewhere about this in case anyone else comes across this problem with esptool

MitchBradley, 2024-01-22:
http://wiki.fluidnc.com/en/installation .  "Somewhere" doesn't guarantee that any individual will see it, especially when there are an arbitrary number of old instructions and videos that someone might come across.