bdring/FluidNC #1119: Problem: G2, G3 Arc commands skipped when doing full circle

quirsh, 2024-01-26:
### Wiki Search Terms

G2, G3, Arc, command skipped

### Controller Board

6-pack

### Machine Description

Gantry plasma cutter, dual Y axis motors

### Configuration file

```yaml
board: 6 Pack
name: 6 Pack External XYYZ
meta: 2022-04-14 B. Dring for Lennart Oosterlaar

stepping:
  engine: I2S_STREAM
  idle_ms: 250
  pulse_us: 4
  dir_delay_us: 1
  disable_delay_us: 0

axes:
  shared_stepper_disable_pin: NO_PIN
  x:
    steps_per_mm: 42.666
    max_rate_mm_per_min: 8000
    acceleration_mm_per_sec2: 150
    max_travel_mm: 1180
    soft_limits: true
    homing:
      cycle: 2
      positive_direction: false
      mpos_mm: 0.000
      feed_mm_per_min: 100.000
      seek_mm_per_min: 1500.000
      settle_ms: 500
      seek_scaler: 1.100
      feed_scaler: 1.100

    motor0:
      limit_neg_pin: gpio.33
      limit_pos_pin: NO_PIN
      limit_all_pin: NO_PIN
      hard_limits: true
      pulloff_mm: 3.000
      standard_stepper:
        step_pin: I2SO.10
        direction_pin: I2SO.9
        disable_pin: I2SO.8

  y:
    steps_per_mm: 42.666
    max_rate_mm_per_min: 8000
    acceleration_mm_per_sec2: 150
    max_travel_mm: 2000
    soft_limits: true
    homing:
      cycle: 3
      positive_direction: false
      mpos_mm: 0.000
      feed_mm_per_min: 100.000
      seek_mm_per_min: 1500.000
      settle_ms: 500
      seek_scaler: 1.100
      feed_scaler: 1.100

    motor0:
      limit_neg_pin: gpio.34
      limit_pos_pin: NO_PIN
      limit_all_pin: NO_PIN
      hard_limits: true
      pulloff_mm: 3.000
      standard_stepper:
        step_pin: I2SO.5
        direction_pin: I2SO.4
        disable_pin: I2SO.7
    
    motor1:
      limit_neg_pin: gpio.32
      limit_pos_pin: NO_PIN
      limit_all_pin: NO_PIN
      hard_limits: true
      pulloff_mm: 3.000
      standard_stepper:
        step_pin: I2SO.2
        direction_pin: I2SO.1
        disable_pin: I2SO.0

  z:
    steps_per_mm: 320
    max_rate_mm_per_min: 1000
    acceleration_mm_per_sec2: 100
    max_travel_mm: 190
    soft_limits: true
    homing:
      cycle: 1
      positive_direction: true
      mpos_mm: 0.000
      feed_mm_per_min: 100.000
      seek_mm_per_min: 800.000
      settle_ms: 500
      seek_scaler: 1.100
      feed_scaler: 1.100

    motor0:
      limit_neg_pin: NO_PIN
      limit_pos_pin: NO_PIN
      limit_all_pin: gpio.35
      hard_limits: true
      pulloff_mm: 1.000
      standard_stepper:
        step_pin: I2SO.13
        direction_pin: I2SO.12
        disable_pin: I2SO.15
        
probe:
  pin: NO_PIN
  check_mode_start: true
  
        
i2so:
  bck_pin: gpio.22
  data_pin: gpio.21
  ws_pin: gpio.17

spi:
  miso_pin: gpio.19
  mosi_pin: gpio.23
  sck_pin: gpio.18

sdcard:
  cs_pin: gpio.5
  card_detect_pin: NO_PIN

coolant:
  flood_pin: gpio.4:high
  delay_ms: 0

relay:
  direction_pin: NO_PIN
  output_pin: gpio.26
  enable_pin: NO_PIN
  disable_with_s0: false
  s0_with_disable: true
  spinup_ms: 0
  spindown_ms: 0
  tool_num: 0
  speed_map: 0=0.000% 0=100.000% 1=100.000%
  off_on_alarm: false

start:
  must_home: false

#poss failure mode gpio16
```

### Startup Messages

```text
[MSG:INFO: FluidNC v3.7.12 https://github.com/bdring/FluidNC]
[MSG:INFO: Compiled with ESP32 SDK:v4.4.4]
[MSG:INFO: Local filesystem type is spiffs]
[MSG:INFO: Configuration file:6P_extn_XYYZ.yaml]
[MSG:INFO: Machine 6 Pack External XYYZ]
[MSG:INFO: Board 6 Pack]
[MSG:INFO: I2SO BCK:gpio.22 WS:gpio.17 DATA:gpio.21]
[MSG:INFO: SPI SCK:gpio.18 MOSI:gpio.23 MISO:gpio.19]
[MSG:INFO: SD Card cs_pin:gpio.5 detect:NO_PIN freq:8000000]
[MSG:INFO: Stepping:I2S_stream Pulse:4us Dsbl Delay:0us Dir Delay:1us Idle Delay:250ms]
[MSG:INFO: Axis count 3]
[MSG:INFO: Axis X (0.000,1180.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     standard_stepper Step:I2SO.10 Dir:I2SO.9 Disable:I2SO.8]
[MSG:INFO:  X Neg Limit gpio.33]
[MSG:INFO: Axis Y (0.000,2000.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     standard_stepper Step:I2SO.5 Dir:I2SO.4 Disable:I2SO.7]
[MSG:INFO:  Y Neg Limit gpio.34]
[MSG:INFO:   Motor1]
[MSG:INFO:     standard_stepper Step:I2SO.2 Dir:I2SO.1 Disable:I2SO.0]
[MSG:INFO:  Y2 Neg Limit gpio.32]
[MSG:INFO: Axis Z (-190.000,0.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     standard_stepper Step:I2SO.13 Dir:I2SO.12 Disable:I2SO.15]
[MSG:INFO:  Z All Limit gpio.35]
[MSG:INFO: Kinematic system: Cartesian]
[MSG:INFO: Relay Spindle Ena:NO_PIN Out:gpio.26 Dir:NO_PIN]
[MSG:INFO: Using spindle Relay]
[MSG:INFO: Flood coolant gpio.4]
[MSG:INFO: STA SSID is not set]
[MSG:INFO: AP SSID FluidNC IP 192.168.0.1 mask 255.255.255.0 channel 1]
[MSG:INFO: AP started]
[MSG:INFO: WiFi on]
[MSG:INFO: Captive Portal Started]
[MSG:INFO: HTTP started on port 80]
[MSG:INFO: Telnet started on port 23]
```

### User Interface Software

UGS

### What happened?

Hi.

I had this problem now occur several times and it is still quite unpredictable for me.
It happens when I try to cut full circles on my plasma cutter, sometimes it does as it is supposed to do, lead in, full circle, lead out.
But then, with other circles, it just does the lead in, skips the full circle and continues directly with the lead out.
I have not been able to determine the cause of this behavior, I was just able to rule out size and speed as the cause. Most of the times it happened with small circle (~14mm), but recently it also happened with a quite big one (~150mm), see the attached gcodes. It does not matter if I dial the feed way down or leave at my average cutting speed, it still sometimes (as in, with specific gcodes) skips the full circle.

The only workaround I was able to find is to put the lead in/lead out points at the very top of the circle (in my setup the means furthest most in X direction), this would reliable execute the circle command and do a full circle. I am using Sheetcam as my CAM program in case this is of interest.

My gcode senders (UGS, but I also tried it with CNCjs) always displays the full circle.

Another weird thing I discovered was that [this online gcode viewer](https://ncviewer.com/) also sometimes skips the full circle, however not in the same way as FluidNC does. In this specific case it is exactly opposite of what FluidNC did, but in other case it would skip a seemingly random subset of circles while FluidNC skipped a different subset of  circles, both subsets might be overlapping but neither the same or the opposite.

NC Viewer showing the gcode that works with FluidNC, but skips the full circle in the viewer.
[image]

NC Viewer showing the gcode that does not work with FluidNC, but shows the full circle in the viewer.
[image]

### GCode File

Gcode that skips the full circle:
```circle_issue.gcode```
```
 G21
 M08
 G04 P2
 
 G00 Z30.0000
 X823.7873 Y1754.9609
 Z3.0000
 M03 S1000
 G04 P0.3
 G02 X822.3727 Y1754.4722 Z1.5000 I-2.7825 J5.7635 F100.0
 X815.2413 Y1757.9418 I-1.3679 J6.2521 F1050.0
 X815.2413 Y1757.9418 I48.5229 J23.4259
 X816.7318 Y1762.2148 I2.8817 J1.3912
 M05
 G00 Z30.0000
 X797.1831 Y1734.0817
 Z3.0000
 M03 S1000
 G04 P0.3
 G02 X798.4359 Y1734.9004 Z1.5000 I4.1033 J-4.9115 F100.0
 X806.1979 Y1733.2736 I2.8505 J-5.7301 F1050.0
 G03 X806.1979 Y1733.2736 I57.5663 J48.0942
 G02 X805.7939 Y1728.7662 I-2.4557 J-2.0517
 M05
 G00 Z30.0000
 M05 M30
 M09
 G53 G0 Z-20
```

Gcode that executes the full circle:
```circle_correct.gcode```
```
 G21
 M08
 G04 P2
 
 G00 Z30.0000
 X823.7873 Y1754.9609
 Z3.0000
 M03 S1000
 G04 P0.3
 G02 X822.3727 Y1754.4722 Z1.5000 I-2.7825 J5.7635 F100.0
 X815.2413 Y1757.9418 I-1.3679 J6.2521 F1050.0
 X815.2413 Y1757.9418 I48.5229 J23.4259
 X816.7318 Y1762.2148 I2.8817 J1.3912
 M05
 G00 Z30.0000
 X870.1642 Y1862.7806
 Z3.0000
 M03 S1000
 G04 P0.3
 G02 X869.9892 Y1861.2943 Z1.5000 I-6.4000 J0.0000 F100.0
 X863.7642 Y1856.3806 I-6.2250 J1.4863 F1050.0
 G03 X863.7642 Y1856.3806 I0.0000 J-75.0128
 G02 X860.5642 Y1859.5806 I0.0000 J3.2000
 M05
 G00 Z30.0000
 M05 M30
 M09
 G53 G0 Z-20
```

terjeio, 2024-01-26:
Perhaps the same underlying problem as [this issue](https://github.com/grbl/grbl/issues/1823)?

MitchBradley, 2024-01-26:
@terjeio 's suggestion is a good one.  Full-circle arcs are known to be troublesome in GCode due to numerical precision issues.  Some industrial gcode controls don't allow them.  It might be possible to configure your CAM program to avoid generating them, instead breaking up full circles into semicircles or quarter circles.  Breaking up the circle won't cause performance problems because the planner breaks up arcs into segments anyway.

MitchBradley, 2024-01-26:
I haven't run the details of the math for the exact circles you are trying to cut, but here is a plausibility argument that suggests numerical problems:  IEEE single precision floating point format has 24 bits in the significant-digits part of the number.  With 24 bits, you can represent integers from 0 to 1677215, and the exponent part of the floating point number lets you scale that by various powers of two.  The Y component of your starting position is 1757.9418.  Scaled up to an integer, that would be 17579418 which is larger than the largest 24-bit number.  Therefore there is guaranteed to be roundoff error, even without considering the discrepancies introduced by power-of-two scaling vs power-of-ten scaling.  It gets even worse when you factor in the X component, which makes the magnitude of the vector larger, and the roundoff errors resulting from sqrt(x-squared + y-squared) calculations and trig functions that are needed for calculating angles.

quirsh, 2024-01-27:
Thank you very much for your comments.

@MitchBradley Without me being able to understand the deeper workings of the gcode interpreter, your assumption sounds plausible. However, your mentioned example is presented in both codes as it is the inner circle, which is cut in both cases. For less confusion, also of myself, I edited my issue and marked the gcode lines which is skipped or rightfully executed. In both cases the y-component of the starting positions of the circle are well above the described limit of the 24 bit integer. So I can't see how this suddenly makes the command being skipped.
The main difference of these commands is the that the I-component of the circle is being 0 for the one variant. Not sure how this affects the execution. 
As well, those errors here happened on the very far side of my table, causing X and Y to be quite high. As said in my issue post, I had had these problems before and I'm quite certain to have had them as well close to the origin of my table. In case that might be interesting I would also test this again.

quirsh, 2024-01-27:
@MitchBradley 

> @terjeio 's suggestion is a good one. Full-circle arcs are known to be troublesome in GCode due to numerical precision issues. Some industrial gcode controls don't allow them. It might be possible [...]

Thank you for the hint. I now set up my Sheetcam post processor in a way that it splits up full circles in 4 arcs. I hope this will solve the problem. For the curious, I added ```post.SetOptions(post.ARC_SEGMENTS)``` to the post processor.

Regarding the issue, if it is not solvable on the FluidNC side due to the mentioned numerical issues, might be an idea to add a comment to the wiki pointing users to the source of the problem and possible workarounds?

MitchBradley, 2024-01-27:
> Regarding the issue, if it is not solvable on the FluidNC side due to the mentioned numerical issues, might be an idea to add a comment to the wiki pointing users to the source of the problem and possible workarounds?

Go for it  - http://wiki.fluidnc.com/en/wiki_contributions