bdring/FluidNC #1120: Problem: Need support but can not get access to the Discord server

PST75, 2024-01-27:
### Wiki Search Terms

TMC5160 pro configuration in *.yaml

### Controller Board

MKS DLC32 v.2.1

### Machine Description

QueenBee 1500 with added laser module

### Configuration file

```yaml
board: MKS-DLC32 V2.1
name: QueenBee Laser
meta: 2024-01-27 PST

kinematics:
  Cartesian:

stepping:
  engine: I2S_STATIC
  idle_ms: 255
  pulse_us: 4
  dir_delay_us: 1
  disable_delay_us: 0
axes:
  shared_stepper_disable_pin: I2SO.0
  x:
    steps_per_mm: 157.750
    max_rate_mm_per_min: 2500.000
    acceleration_mm_per_sec2: 400.000
    max_travel_mm: 325.000
    soft_limits: true
    homing:
      cycle: 1
      positive_direction: false
      mpos_mm: 0.000
      feed_mm_per_min: 300.000
      seek_mm_per_min: 5000.000
      settle_ms: 500
      seek_scaler: 1.100
      feed_scaler: 1.100

    motor0:
      limit_neg_pin: gpio.36:low
      hard_limits: false
      pulloff_mm: 2.500
      stepstick:
        step_pin: I2SO.1
        direction_pin: I2SO.2

  y:
    steps_per_mm: 157.750
    max_rate_mm_per_min: 2500.000
    acceleration_mm_per_sec2: 400.000
    max_travel_mm: 220.000
    soft_limits: true
    homing:
      cycle: 1
      positive_direction: false
      mpos_mm: 0.000
      feed_mm_per_min: 300.000
      seek_mm_per_min: 5000.000
      settle_ms: 500
      seek_scaler: 1.100
      feed_scaler: 1.100

    motor0:
      limit_neg_pin: gpio.35:low
      hard_limits: false
      pulloff_mm: 2.500
      stepstick:
        step_pin: I2SO.5
        direction_pin: I2SO.6:low

  z:
    steps_per_mm: 157.750
    max_rate_mm_per_min: 2500.000
    acceleration_mm_per_sec2: 400.000
    max_travel_mm: 80.000
    soft_limits: true
    homing:
      cycle: 1
      positive_direction: false
      mpos_mm: 0.000
      feed_mm_per_min: 300.000
      seek_mm_per_min: 1000.000
      settle_ms: 500
      seek_scaler: 1.100
      feed_scaler: 1.100

    motor0:
      limit_neg_pin: gpio.34:low
      hard_limits: false
      pulloff_mm: 2.500
      stepstick:
        step_pin: I2SO.3
        direction_pin: I2SO.4

i2so:
  bck_pin: gpio.16
  data_pin: gpio.21
  ws_pin: gpio.17

spi:
  miso_pin: gpio.12
  mosi_pin: gpio.13
  sck_pin: gpio.14

sdcard:
  cs_pin: gpio.15
  card_detect_pin: NO_PIN

control:
  safety_door_pin: NO_PIN
  reset_pin: NO_PIN
  feed_hold_pin: NO_PIN
  cycle_start_pin: NO_PIN
  macro0_pin: gpio.33:low:pu
  macro1_pin: NO_PIN
  macro2_pin: NO_PIN
  macro3_pin: NO_PIN

macros:
  startup_line0:
  startup_line1:
  macro0: $SD/Run=lasertest.gcode
  macro1: $SD/Run=home.gcode
  macro2:
  macro3:

coolant:
  flood_pin: NO_PIN
  mist_pin: NO_PIN
  delay_ms: 0

probe:
  pin: gpio.22:low
  check_mode_start: true

Laser:
  pwm_hz: 5000
  #L on Beeper / IN on TTL
  output_pin: gpio.32
  enable_pin: I2SO.7
  disable_with_s0: false
  s0_with_disable: false
  tool_num: 0
  speed_map: 0=0.000% 100=100.000%
# 135=0mA 270=5mA 400=10mA 700=16mA
user_outputs:
  analog0_pin: NO_PIN
  analog1_pin: NO_PIN
  analog2_pin: NO_PIN
  analog3_pin: NO_PIN
  analog0_hz: 5000
  analog1_hz: 5000
  analog2_hz: 5000
  analog3_hz: 5000
  digital0_pin: NO_PIN
  digital1_pin: NO_PIN
  digital2_pin: NO_PIN
  digital3_pin: NO_PIN

start:
  must_home: false

# 5,18,19,22,23,25,26,27,32,33,39,I2SO.7
# SDA 0 / SCL 4
```

### Startup Messages

```text
rst:0x1 (POWERON_RESET),boot:0x1b (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:1184
load:0x40078000,len:13220
ho 0 tail 12 room 4
load:0x40080400,len:3028
entry 0x400805e4
[MSG:INFO: uart_channel0 created]
[MSG:RST]
[MSG:INFO: FluidNC 3.7.13 https://github.com/bdring/FluidNC]
[MSG:INFO: Compiled with ESP32 SDK:v4.4.4]
[MSG:INFO: Local filesystem type is spiffs]
[MSG:INFO: Configuration file:MKS_DLC32_21_XYZ_QBLaser.yaml]
[MSG:INFO: Machine QueenBee Laser]
[MSG:INFO: Board MKS-DLC32 V2.1]
[MSG:INFO: I2SO BCK:gpio.16 WS:gpio.17 DATA:gpio.21]
[MSG:INFO: SPI SCK:gpio.14 MOSI:gpio.13 MISO:gpio.12]
[MSG:INFO: SD Card cs_pin:gpio.15 detect:NO_PIN freq:8000000]
[MSG:INFO: Stepping:I2S_static Pulse:4us Dsbl Delay:0us Dir Delay:1us Idle Delay:255ms]
[MSG:INFO: Axis count 3]
[MSG:INFO: Shared stepper disable I2SO.0]
[MSG:INFO: Axis X (0.000,325.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     stepstick Step:I2SO.1 Dir:I2SO.2 Disable:NO_PIN]
[MSG:INFO:  X Neg Limit gpio.36:low]
[MSG:INFO: Axis Y (0.000,220.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     stepstick Step:I2SO.5 Dir:I2SO.6:low Disable:NO_PIN]
[MSG:INFO:  Y Neg Limit gpio.35:low]
[MSG:INFO: Axis Z (0.000,80.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     stepstick Step:I2SO.3 Dir:I2SO.4 Disable:NO_PIN]
[MSG:INFO:  Z Neg Limit gpio.34:low]
[MSG:INFO: macro0_pin gpio.33:low:pu]
[MSG:INFO: Kinematic system: Cartesian]
[MSG:INFO: Laser Ena:I2SO.7 Out:gpio.32 Freq:5000Hz Period:8191]
[MSG:INFO: Using spindle Laser]
[MSG:INFO: Probe Pin: gpio.22:low]
[MSG:INFO: Connecting to STA SSID:TPAC]
[MSG:INFO: Connecting.]
[MSG:INFO: Connecting..]
[MSG:INFO: Connected - IP is 192.168.1.215]
[MSG:INFO: WiFi on]
[MSG:INFO: Start mDNS with hostname:http://fluidnc.local/]
[MSG:INFO: SSDP Started]
[MSG:INFO: HTTP started on port 80]
[MSG:INFO: Telnet started on port 23]

Grbl 3.7 [FluidNC 3.7.13 (wifi) '$' for help]
```

### What happened?

I need support on how to correctly add the TMC5160 pro in to my *.yaml file for this board.

bdring, 2024-01-27:
http://wiki.fluidnc.com/en/config/axes#tmc_5160pro-tmc_2160pro-motors-expert-mode

https://discord.gg/WaAD23eN

PST75, 2024-01-28:
I'm doing this to add a laser module on to my QueenBee CNC. To run it for milling I have it configured (and working) with an XproV5 controller but I have read that it is difficult to switch that controller to be used with either a VFD/Spindle and in Laser mode communicating with the Laser Module.....

bdring, 2024-01-28:
How are you wiring the MKS DLC32? Does that controller normally support SPI drivers.

PST75, 2024-01-28:
[image]
[image]
[image]
[image]

Hi.

Dont know how "correct" these are connected🙄 And I think that the MKS DLC32 2.1 support SPI but I'm not sure. Could I maybe use the 5160 stepper drivers in standalone mode and configure microstep and current on the jumper.. Maybe if I use them in standalone the tested "standard yaml" can work as it is? I do not need SPI functionality etc. as I only bought the 5160 pro to cope with the Nema23..

bdring, 2024-01-28:
It won't work in SPI mode. Use standard stepper and use the jumpers to set your current, etc.

PST75, 2024-01-28:
Thank you very, very much for your support. Can I ask you how the jumpers should be set to get 32 microsteps and IRUN=16...sorry about all this questions but as you probably realized I'm very new on this...
[image]
[image]
[image]

If you could "draw" or similar in the picture how to move the jumper it would be gr8.

Again thank you very much for your support..sorry to take up this amount of your time. Is there a way for me to donate etc. I will

PST75, 2024-01-28:
[image]

PST75, 2024-01-28:
[image]
[image]

PST75, 2024-01-28:
Could thi be the correct jumper setting to achieve :

32 microsteps
Irun 16
Spread cycle
Ihold=1/2 Irun
[image]
[image]

bdring, 2024-01-28:
I don't know anything about those drivers. I think you are on the right track. I would experiment and test until you get what you want.

MitchBradley, 2024-01-28:
You could ask the people who sold you the product.

PST75, 2024-01-28:
I did but was on Aliexpress and It does not seem like they know a lot unfortunately 😐 But I thank you for all your help anyway👍😊 Will test and see if I can get it to work. If not can you recommend any hardware were I can control the three axis cnc with a vfd spindle and a laser module attached using FluidNC? Thanks again for taking the time👍

MitchBradley, 2024-01-28:
Let me try to explain the situation to you.  It is sort of complicated.  Convention stepper drivers (not TMCs) are configurated with jumpers and potentiometers to set things like microstepping and run current.  The microprocessor in the controller board is not required to anything about the driver configuration, except that you have to tell the firmware the resulting steps_per_mm value, which depends on both the microstepping and also the machine characteristics like screw pitches.

TMC drivers, on the other hand, can be configured either with jumpers and pots, and also with software, by sending commands over a serial bus.  Depending on the TMC model, the serial bus is either SPI or UART.  The software configuration method gives you precise control over many aspects of the driver's operation - so many that it can be very difficult to figure out good settings from all the possibilities.  In order for the software method to work at all, the controller board must have special wiring to the stepper sockets for the particular flavor of serial bus that the driver uses.  TMC5160 use the SPI serial bus.  There are some FluidNC controller boards that support SPI configuration of TMC chips - notably Bart's 6-pack boards (but not the external driver flavors) support it - but most FluidNC controller boards do not.  MKS DLC32 does not support TMC5160 drivers in SPI mode, because the stepper socket pins that would need to be connected to an SPI bus are instead connected to DIP switches that are for configuring conventional jumper-configured drivers.

That means that, with an MKS DLC32 controller, you must use the TMC5160-PRO board in standalone mode, not in SPI mode.

When you are using standalone mode, in the FluidNC config file you would not declare the motor type as "tmc5160", but rather "standard_stepper".  standard_stepper is for drivers where all of the configuration is done with jumpers or switches, so FluidNC does not have to do (and cannot do) anything to configure them.

If the MKS DLC32 can control both your VFD spindle and your laser, then it should be suitable for you needs, with the TMC5160-PRO in standalone mode and "standard_stepper" in the config.yaml file.  As to its suitability for VFD + laser, that would depend on whether there are enough output pins left, after using any others that you might need for other purposes.  It also depends on the level of control you need for the VFD - on/off, 0-10V speed control, PWM speed control, or RS485 smart control.  Those different possibilities require pins and possibly external hardware.  It is a complicated topic because there are so many possibilities, and different VFDs have different details as to exactly how they are wired and controlled.

The general problem with most controller boards is not enough pins to do everything that you might want.  Most low-end, and some high-end, controllers suffer from this restriction.  The controller that is the most likely to do everything is Bart's 6-pack universal controller, but it is out of stock at the moment.  This one might suit your needs, but you would have to do some wiring to connect the TMC5160-PRO to the connectors.

MitchBradley, 2024-01-28:
As an alternative to the TMC5160-PRO, you could use an external stepper driver like DM542 - but beware of super-cheap ones costing less than $20, as they are probably fakes with obsolete boards inside.  There is nothing wrong with TMC5160-PRO, apart from it being harder to set up and wire to random boards.

PST75, 2024-01-31:
Hi.

I just saw the 6x CNC Controller and got interested.

How can I purchase it if i live outside of the US (Sweden) and whats the
cost incl. shipping?

Thanks,
/Pär

bdring, 2024-01-31:
https://www.elecrow.com/6x-cnc-controller-for-fluidnc.html

PST75, 2024-02-01:
Hi

Thanks. From where would this board be shipped out if I order it on elecrow?

I'm asking because of possible VAT, import duties etc. as I'm living within
EU.

Br,
/Pär

bdring, 2024-02-01:
I am pretty sure it ships from China

PST75, 2024-02-01:
@BlueOrangeLive 

Thank you for feedback

@bdring 

With this board is it possible to use the TMC 5160 in SPI mode or only as standalone?

PST75, 2024-02-01:
@bdring 
Could I use this board to both use it with a spindle (Hyanyang + VFD) and a laser module (shift in G code)?

bdring, 2024-02-01:
You should be using the standard stepper motor type.

http://wiki.fluidnc.com/en/config/axes#standard-stepper

There is no easy way to measure current on a stepper motor with a meter. You should be set it through the jumpers.

PST75, 2024-02-01:
@bdring
Could I use this board to both use it with a spindle (Hyanyang + VFD) and a laser module (shift in G code)? Have the Hyanyang and VFD been tested with the board?

bdring, 2024-02-01:
I don't know. FluidNC can of course handle that, but that is pushing the capabilities of the I/O on that. We have a wiki page with some info and our Discord server has many users of that controller.

http://wiki.fluidnc.com/en/hardware/3rd-party/MKS_DLC32

Note: We prefer not to add unrelated items to Github issues, so they can be closed. The subject of this issue is already pretty unhelpful.

PST75, 2024-02-02:
@bdring 

Sorry but my question related to if the 6x CNC board could handle both a spindle+vfd and a laser module and if they could be controlled/switched between using a g code command

bdring, 2024-02-02:
Yes, I'll add a config example soon.

bdring, 2024-02-02:
The example on this page has a laser section. If you uncomment that (remove #) section.

http://wiki.fluidnc.com/en/hardware/official/6x_CNC_Controller#example-config-file

The Huanyang section is already uncommented.

Can we close this issue?

closed 2024-02-02 by PST75