bdring/FluidNC #1121: Problem: Limited speed (not reaching the required travel speed)

souhail-abdallah, 2024-01-29:
### Wiki Search Terms

I have a problem with the MKS DLC32 card, which is that I cannot reach high speeds with it. Is there a software solution or perhaps an electronic solution, such as adding an electronic element or card to amplify or accelerate the signal? Thank you.

### Controller Board

MKS DLC 32 V2.1 with MKS TS35-R 

### Machine Description

They are only hand-crafted 2D CNC custom lettering

### Configuration file

```yaml
Handmade only
```

### Startup Messages

```text
there is no problem about startup
```

### What happened?

In the beginning, I use the ArtCam drawing program, and I output the files in .nc format, and I choose a walking speed of 400mm/sec per second or more, but I find that the machine does not respond. When I create several identical files in the drawing, but at varying speeds, that is, more than 300 mm per second, I find that they are the same. Speed when executed on the machine, That is, it completes the drawing perfectly, except for the speed. When I reach higher speeds, it remains fast. Perhaps this is its highest speed that the card can give to the machine.

MitchBradley, 2024-01-29:
No way to know because you did not supply the information that the ticket asked for.  We need the configuration file and the startup messages.

bdring, 2025-01-21:
Unresponsive OP