bdring/FluidNC #1122: Abnormal operating speed

zhuyuwu110, 2024-01-31:
### Wiki Search Terms

----

### Controller Board

4-pack

### Machine Description

XYZ 3-axis moving gantry  router machine

### Configuration file

```yaml
name: New configuration
board: 4 Pack
start:
  must_home: false
  deactivate_parking: false
stepping:
  engine: I2S_STATIC
  idle_ms: 250
  pulse_us: 4
  dir_delay_us: 0
  disable_delay_us: 0
axes:
  x:
    steps_per_mm: 1280
    max_rate_mm_per_min: 3000
    acceleration_mm_per_sec2: 300
    max_travel_mm: 700
    motor0:
      limit_all_pin: NO_PIN
      limit_neg_pin: NO_PIN
      limit_pos_pin: gpio.36
      hard_limits: false
      pulloff_mm: 1
      standard_stepper:
        step_pin: i2so.6
        direction_pin: i2so.7
        disable_pin: NO_PIN
    soft_limits: false
  y:
    steps_per_mm: 1280
    max_rate_mm_per_min: 3000
    acceleration_mm_per_sec2: 300
    max_travel_mm: 500
    soft_limits: false
    motor0:
      limit_all_pin: NO_PIN
      limit_neg_pin: NO_PIN
      limit_pos_pin: gpio.39
      hard_limits: false
      pulloff_mm: 1
      standard_stepper:
        step_pin: i2so.4
        direction_pin: i2so.5
        disable_pin: NO_PIN
  z:
    steps_per_mm: 1280
    max_rate_mm_per_min: 3000
    acceleration_mm_per_sec2: 300
    max_travel_mm: 150
    soft_limits: false
    motor0:
      limit_all_pin: NO_PIN
      limit_neg_pin: NO_PIN
      limit_pos_pin: gpio.34
      hard_limits: false
      pulloff_mm: 1
      standard_stepper:
        step_pin: i2so.2
        direction_pin: i2so.3
        disable_pin: NO_PIN
  a:
    steps_per_mm: 1280
    max_rate_mm_per_min: 3000
    acceleration_mm_per_sec2: 300
    max_travel_mm: 360
    soft_limits: false
    motor0:
      limit_all_pin: NO_PIN
      limit_neg_pin: NO_PIN
      limit_pos_pin: gpio.35
      hard_limits: false
      pulloff_mm: 1
      standard_stepper:
        step_pin: i2so.15
        direction_pin: i2so.1
        disable_pin: NO_PIN
  shared_stepper_disable_pin: i2so.0
  shared_stepper_reset_pin: NO_PIN
control:
  safety_door_pin: gpio.14:pu:low
  feed_hold_pin: gpio.26:pu:low
  cycle_start_pin: gpio.25:pu:low
  reset_pin: gpio.32:pu:low
  macro0_pin: gpio.16
  macro1_pin: NO_PIN
  macro2_pin: NO_PIN
  macro3_pin: NO_PIN
  fault_pin: NO_PIN
  estop_pin: gpio.33:pu:low

macros:
  startup_line0:
  startup_line1:
  macro0: $SD/Run=00.nc
  macro1: 
  macro2:
  macro3:

coolant:
  flood_pin: i2so.11
  mist_pin: NO_PIN
  delay_ms: 0
probe:
  pin: gpio.27:pu:low
  check_mode_start: true
Laser:
  output_pin: gpio.2
  enable_pin: NO_PIN
  tool_num: 2
spi:
  miso_pin: gpio.19:pu
  mosi_pin: gpio.23:pu
  sck_pin: gpio.18
sdcard:
  card_detect_pin: NO_PIN
  cs_pin: gpio.5:pu
i2so:
  bck_pin: gpio.22
  data_pin: gpio.21
  ws_pin: gpio.17
10V:
  forward_pin: i2so.9
  reverse_pin: i2so.10
  output_pin: gpio.15:pu
  enable_pin: gpio.4:pu
  direction_pin: NO_PIN
  tool_num: 1
  speed_map: 0=0.000% 2400=10.000%12000=50.000% 24000=100.000%
status_outputs:
  report_interval_ms: 500
  idle_pin: NO_PIN
  run_pin: i2so.12
  hold_pin: i2so.13
  alarm_pin: i2so.14
```

### Startup Messages

```text
[CTRL+X] < Grbl 3.7 [FluidNC v3.7.12 (wifi) '$' for help]
S999 < ok
$SS < [MSG:INFO: FluidNC v3.7.12 https://github.com/bdring/FluidNC]
[MSG:INFO: Compiled with ESP32 SDK:v4.4.4]
[MSG:INFO: Local filesystem type is littlefs]
[MSG:INFO: Configuration file:ESP32 4axis XYZA .yaml]
[MSG:INFO: Machine New configuration]
[MSG:INFO: Board 4 Pack]
[MSG:INFO: I2SO BCK:gpio.22 WS:gpio.17 DATA:gpio.21]
[MSG:INFO: SPI SCK:gpio.18 MOSI:gpio.23:pu MISO:gpio.19:pu]
[MSG:INFO: SD Card cs_pin:gpio.5:pu detect:NO_PIN freq:8000000]
[MSG:INFO: Status outputs Interval:500 Idle:NO_PIN Cycle:I2SO.12 Hold:I2SO.13 Alarm:I2SO.14]
[MSG:INFO: Stepping:I2S_stream Pulse:4us Dsbl Delay:0us Dir Delay:0us Idle Delay:250ms]
[MSG:INFO: Axis count 4]
[MSG:INFO: Shared stepper disable I2SO.0]
[MSG:INFO: Axis X (-700.000,0.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     standard_stepper Step:I2SO.6 Dir:I2SO.7 Disable:NO_PIN]
[MSG:INFO:  X Pos Limit gpio.36]
[MSG:INFO: Axis Y (-500.000,0.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     standard_stepper Step:I2SO.4 Dir:I2SO.5 Disable:NO_PIN]
[MSG:INFO:  Y Pos Limit gpio.39]
[MSG:INFO: Axis Z (-150.000,0.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     standard_stepper Step:I2SO.2 Dir:I2SO.3 Disable:NO_PIN]
[MSG:INFO:  Z Pos Limit gpio.34]
[MSG:INFO: Axis A (-360.000,0.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     standard_stepper Step:I2SO.15 Dir:I2SO.1 Disable:NO_PIN]
[MSG:INFO:  A Pos Limit gpio.35]
[MSG:INFO: safety_door_pin gpio.14:low:pu]
[MSG:INFO: reset_pin gpio.32:low:pu]
[MSG:INFO: feed_hold_pin gpio.26:low:pu]
[MSG:INFO: cycle_start_pin gpio.25:low:pu]
[MSG:INFO: macro0_pin gpio.16]
[MSG:INFO: estop_pin gpio.33:low:pu]
[MSG:INFO: Kinematic system: Cartesian]
[MSG:INFO: Laser Ena:NO_PIN Out:gpio.2 Freq:5000Hz Period:8191]
[MSG:INFO: 10V Spindle Ena:gpio.4:pu Out:gpio.15:pu Dir:NO_PIN Fwd:I2SO.9 Rev:I2SO.10 Freq:5000Hz Period:8191]
[MSG:INFO: Using spindle Laser]
[MSG:INFO: Flood coolant I2SO.11]
[MSG:INFO: Probe Pin: gpio.27:low:pu]
[MSG:INFO: STA SSID is not set]
[MSG:INFO: AP SSID FluidNC IP 192.168.0.1 mask 255.255.255.0 channel 1]
[MSG:INFO: AP started]
[MSG:INFO: WiFi on]
[MSG:INFO: Captive Portal Started]
[MSG:INFO: HTTP started on port 80]
[MSG:INFO: Telnet started on port 23]
ok
```

### User Interface Software

candle

### What happened?

Jogging or running G code both run at a very slow speed, and sometimes it is normal for ESP32 to power on again.

### Other Information

https://github.com/bdring/FluidNC/assets/158246784/fa9d166f-1ed1-489e-9b64-50de66e96484

MitchBradley, 2024-01-31:
We do not know enough about your situation to solve the problem.  What is the control board?  What kind of stepper drivers?  What motors?  What is the Gcode program you are trying to run?  Have you tried running at lower speeds or lower values of max_rate_mm_per_min ?  Have you tried lower values for acceleration_mm_per_sec2 ?  Have you tried larger values for pulse_us ?  Have you tried lower values for max_rate_mm_per_min ?

MitchBradley, 2024-02-01:
I loaded your config file onto a 6 pack controller with FluidNC version 3.7.12 and connected to it with Candle.  On my system the jogging behaves as expected.

Can you try using FluidTerm to send jogging commands manually?

zhuyuwu110, 2024-02-01:
Thank you again for your patient reply. A few minutes ago, I connected to Candle again and everything was running normally. I didn't even modify any settings, and the manual sending of jog commands by fluidterm was also normal. I am very confused and don't know if the same problem will occur again next time,

bdring, 2024-02-02:
Can this issue be closed?

closed 2024-02-03 by zhuyuwu110