bdring/FluidNC #1123: Problem: Cannot load file to sdcard using Webui with FluidNC 3.7.13 and 14

SnowBeaver, 2024-02-01:
### Wiki Search Terms

SDCARD

### Controller Board

6-pack

### Machine Description

Gantry router

### Configuration file

```yaml
board: 6 Pack
name: 3 axis std_step X--ZYY v1.5 05Sept23
stepping:
  engine: I2S_STATIC
  idle_ms: 255
  pulse_us: 4
  dir_delay_us: 1
  disable_delay_us: 0

axes:
  shared_stepper_disable_pin: NO_PIN
  x:
    steps_per_mm: 53.400
    max_rate_mm_per_min: 5000.000
    acceleration_mm_per_sec2: 100.000
    max_travel_mm: 1050.000
    soft_limits: false
    homing:
      cycle: 2
      positive_direction: false
      mpos_mm: -3.000
      feed_mm_per_min: 100.000
      seek_mm_per_min: 400.000
      settle_ms: 500
      seek_scaler: 1.100
      feed_scaler: 1.100

    motor0:
      limit_neg_pin: NO_PIN
      limit_pos_pin: NO_PIN
      limit_all_pin: gpio.33:low:pu
      hard_limits: false
      pulloff_mm: 3.000
      tmc_2130:
        cs_pin: i2so.3
        r_sense_ohms: 0.110
        run_amps: 0.650
        hold_amps: 0.500
        microsteps: 16
        stallguard: 0
        stallguard_debug: false
        toff_disable: 0
        toff_stealthchop: 5
        toff_coolstep: 3
        run_mode: StealthChop
        homing_mode: StealthChop
        use_enable: false
        step_pin: I2SO.2
        direction_pin: I2SO.1
        disable_pin: I2SO.0

  y:
    steps_per_mm: 53.400
    max_rate_mm_per_min: 5000.000
    acceleration_mm_per_sec2: 100.000
    max_travel_mm: 960.000
    soft_limits: false
    homing:
      cycle: 2
      positive_direction: false
      mpos_mm: -3.000
      feed_mm_per_min: 100.000
      seek_mm_per_min: 400.000
      settle_ms: 500
      seek_scaler: 1.100
      feed_scaler: 1.100

    motor0:
      limit_neg_pin: NO_PIN
      limit_pos_pin: NO_PIN
      limit_all_pin: gpio.36:low
      hard_limits: false
      pulloff_mm: 3.000
      tmc_2130:
        cs_pin: i2so.19
        r_sense_ohms: 0.110
        run_amps: 0.650
        hold_amps: 0.500
        microsteps: 16
        stallguard: 0
        stallguard_debug: false
        toff_disable: 0
        toff_stealthchop: 5
        toff_coolstep: 3
        run_mode: StealthChop
        homing_mode: StealthChop
        use_enable: false
        step_pin: I2SO.18
        direction_pin: I2SO.17
        disable_pin: I2SO.16

    motor1:
      limit_neg_pin: NO_PIN
      limit_pos_pin: NO_PIN
      limit_all_pin: gpio.39:low
      hard_limits: false
      pulloff_mm: 3.000
      tmc_2130:
        cs_pin: i2so.22
        r_sense_ohms: 0.110
        run_amps: 0.650
        hold_amps: 0.500
        microsteps: 16
        stallguard: 0
        stallguard_debug: false
        toff_disable: 0
        toff_stealthchop: 5
        toff_coolstep: 3
        run_mode: StealthChop
        homing_mode: StealthChop
        use_enable: false
        step_pin: I2SO.21
        direction_pin: I2SO.20
        disable_pin: I2SO.23

  z:
    steps_per_mm: 808.000
    max_rate_mm_per_min: 2000.000
    acceleration_mm_per_sec2: 20.000
    max_travel_mm: 360.000
    soft_limits: false
    homing:
      cycle: 0
      positive_direction: false
      mpos_mm: -3.000
      feed_mm_per_min: 100.000
      seek_mm_per_min: 400.000
      settle_ms: 500
      seek_scaler: 1.100
      feed_scaler: 1.100

    motor0:
      limit_neg_pin: NO_PIN
      limit_pos_pin: NO_PIN
      limit_all_pin: gpio.35:low
      hard_limits: false
      pulloff_mm: 3.000
      tmc_2130:
        cs_pin: i2so.14
        r_sense_ohms: 0.110
        run_amps: 0.6500
        hold_amps: 0.6500
        microsteps: 16
        stallguard: 0
        stallguard_debug: false
        toff_disable: 0
        toff_stealthchop: 5
        toff_coolstep: 3
        run_mode: StealthChop
        homing_mode: StealthChop
        use_enable: false
        step_pin: I2SO.13
        direction_pin: I2SO.12
        disable_pin: I2SO.15

i2so:
  bck_pin: gpio.22
  data_pin: gpio.21
  ws_pin: gpio.17

spi:
  miso_pin: gpio.19
  mosi_pin: gpio.23
  sck_pin: gpio.18

sdcard:
  card_detect_pin: NO_PIN
  cs_pin: gpio.5

control:
  safety_door_pin: NO_PIN
  reset_pin: NO_PIN
  feed_hold_pin: NO_PIN
  cycle_start_pin: NO_PIN
  macro0_pin: NO_PIN
  macro1_pin: NO_PIN
  macro2_pin: NO_PIN
  macro3_pin: NO_PIN

coolant:
  flood_pin: NO_PIN
  mist_pin: NO_PIN
  delay_ms: 0

probe:
  pin: gpio.2:high:pu
  check_mode_start: true

macros:
  startup_line0:
  startup_line1:
  macro0:
  macro1:
  macro2:
  macro3:

user_outputs:
  analog0_pin: NO_PIN
  analog1_pin: NO_PIN
  analog2_pin: NO_PIN
  analog3_pin: NO_PIN
  analog0_hz: 5000
  analog1_hz: 5000
  analog2_hz: 5000
  analog3_hz: 5000
  digital0_pin: NO_PIN
  digital1_pin: NO_PIN
  digital2_pin: NO_PIN
  digital3_pin: NO_PIN

start:
  must_home: false
  
relay:
  output_pin: gpio.26
  enable_pin: NO_PIN
  direction_pin: NO_PIN
  disable_with_s0: false
  s0_with_disable: true
  spinup_ms: 0
  spindown_ms: 0
  tool_num: 0
  speed_map: 0=0.000% 1=100.000%

laser:
  pwm_hz: 2000
  output_pin: gpio.13:high
  enable_pin: NO_PIN
  disable_with_s0: false
  s0_with_disable: true
  tool_num: 100
  speed_map: 0=0.000% 1000=100.000%
  off_on_alarm: true
```

### Startup Messages

```text
Grbl 3.7 [FluidNC 3.7.13 (wifi) '$' for help]
[MSG:INFO: uart_channel0 created]
[MSG:RST]
[MSG:INFO: FluidNC 3.7.13 https://github.com/bdring/FluidNC]
[MSG:INFO: Compiled with ESP32 SDK:v4.4.4]
[MSG:INFO: Local filesystem type is littlefs]
[MSG:INFO: Configuration file:config.yaml]
[MSG:INFO: Machine 3 axis std_step X--ZYY v1.5 05Sept23]
[MSG:INFO: Board 6 Pack]
[MSG:INFO: I2SO BCK:gpio.22 WS:gpio.17 DATA:gpio.21]
[MSG:INFO: SPI SCK:gpio.18 MOSI:gpio.23 MISO:gpio.19]
[MSG:INFO: SD Card cs_pin:gpio.5 detect:NO_PIN freq:8000000]
[MSG:INFO: Stepping:I2S_static Pulse:4us Dsbl Delay:0us Dir Delay:1us Idle Delay:255ms]
[MSG:INFO: Axis count 3]
[MSG:INFO: Axis X (-3.000,1047.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     tmc_2130 Step:I2SO.2 Dir:I2SO.1 CS:I2SO.3 Disable:I2SO.0 Index:-1 R:0.110]
[MSG:INFO:  X All Limit gpio.33:low:pu]
[MSG:INFO: Axis Y (-3.000,957.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     tmc_2130 Step:I2SO.18 Dir:I2SO.17 CS:I2SO.19 Disable:I2SO.16 Index:-1 R:0.110]
[MSG:INFO:  Y All Limit gpio.36:low]
[MSG:INFO:   Motor1]
[MSG:INFO:     tmc_2130 Step:I2SO.21 Dir:I2SO.20 CS:I2SO.22 Disable:I2SO.23 Index:-1 R:0.110]
[MSG:INFO:  Y2 All Limit gpio.39:low]
[MSG:INFO: Axis Z (-3.000,357.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     tmc_2130 Step:I2SO.13 Dir:I2SO.12 CS:I2SO.14 Disable:I2SO.15 Index:-1 R:0.110]
[MSG:INFO:  Z All Limit gpio.35:low]
[MSG:INFO: X Axis driver test passed]
[MSG:INFO: Y Axis driver test passed]
[MSG:INFO: Y2 Axis driver test passed]
[MSG:INFO: Z Axis driver test passed]
[MSG:INFO: Kinematic system: Cartesian]
[MSG:INFO: Relay Spindle Ena:NO_PIN Out:gpio.26 Dir:NO_PIN]
[MSG:INFO: Laser Ena:NO_PIN Out:gpio.13 Freq:2000Hz Period:32767]
[MSG:INFO: Using spindle Relay]
[MSG:INFO: Probe Pin: gpio.2:pu]
[MSG:INFO: Connecting to STA SSID:wifi_terrain]
[MSG:INFO: Connecting.]
[MSG:INFO: Connecting..]
[MSG:INFO: Connected - IP is 10.0.0.201]
[MSG:INFO: WiFi on]
[MSG:INFO: Start mDNS with hostname:http://fluidnc.local/]
[MSG:INFO: SSDP Started]
[MSG:INFO: HTTP started on port 80]
[MSG:INFO: Telnet started on port 23]

Grbl 3.7 [FluidNC 3.7.13 (wifi) '$' for help]
ok
[VER:3.7 FluidNC 3.7.13:]
[OPT:PHS]
[MSG:Machine: 3 axis std_step X--ZYY v1.5 05Sept23]
[MSG:Mode=STA:SSID=wifi_terrain:Status=Connected:IP=10.0.0.201:MAC=78-E3-6D-CC-F7-D8]
ok
```

### User Interface Software

I was using Webui version github.com/MitchBradley/ESP3D-WEBUI@4c7780c / FW: fluidnc 3.7.13

### What happened?

Needed to load a nc file into the sdcard. Used the Upload button from the main tab and got an error message while loading.

[MSG:ERR: sdmmc_card_init failed code 0x107]
No SD card
ERROR:60
SD failed to mount

Then right after I received the error message I went to terminal and did $SD/LIST it also fails:

[MSG:ERR: sdmmc_card_init failed code 0x107]
No SD card
ERROR:60
SD failed to mount

However If I do $SD/LIST after a reset I can read the SDCARD ok. 

Grbl 3.7 [FluidNC 3.7.13 (wifi) '$' for help]
$SD/LIST
[DIR:System Volume Information]
[FILE:  IndexerVolumeGuid|SIZE:76]
[FILE: poulie.nc|SIZE:450513]
[FILE: toolpath-temujin-T01.nc|SIZE:332823]
[DIR:localfs]
[FILE:  macrocfg.json|SIZE:1008]
[FILE:  macro2.g|SIZE:23]
[FILE:  macro3.g|SIZE:27]
[FILE:  preferences.json|SIZE:1087]
[FILE:  index.html.gz|SIZE:126709]
[FILE:  macro1.g|SIZE:29]
[FILE:  favicon.ico|SIZE:1150]
[FILE: cercle_20.nc|SIZE:490]
[FILE: trou-2.875.nc|SIZE:29763]
[/sd/ Free:14.81 GB Used:1.31 MB Total:14.81 GB]

Then I tried with the following firmware 3.17.10, 11 and 12 and can load or delete file from WEBUI without any error messages.

BlueOrangeLive, 2024-02-01:
The error occurs for me too. With WEBUI@4c7780c with 3.7.13/14 I cannot upload files to the SD card.    

Grbl 3.7 [FluidNC 3.7.13 (wifi) '$' for help]
[MSG:INFO: WebUI: Request from 192.168.188.22]
E (143209) sdmmc_common: sdmmc_init_ocr: send_op_cond (1) returned 0x107
[MSG:ERR: sdmmc_card_init failed code 0x107]
Guru Meditation Error: Core  0 panic'ed (IllegalInstruction). Exception was unhandled.

Core  0 register dump:
PC      : 0x78302065  PS      : 0x00060a30  A0      : 0x80157d1a  A1      : 0x3ffe92f0  
A2      : 0x3fff19b8  A3      : 0x3ffe9310  A4      : 0x00000028  A5      : 0x3ffe9340  
A6      : 0x00060a20  A7      : 0x00000002  A8      : 0x801efa11  A9      : 0x3ffe9300  
A10     : 0x30363a72  A11     : 0x3ffe9310  A12     : 0x3ffbf0e8  A13     : 0x3ffcdef4  
A14     : 0x007c0978  A15     : 0x003fffff  SAR     : 0x00000020  EXCCAUSE: 0x00000000  
EXCVADDR: 0x00000000  LBEG    : 0x4008b74c  LEND    : 0x4008b757  LCOUNT  : 0x00000000  

Backtrace: 0x78302062:0x3ffe92f0 |<-CORRUPTED

ELF file SHA256: 080db31bcef4ea05

Rebooting...
ets Jun  8 2016 00:22:57

BlueOrangeLive, 2024-02-01:
This happens with the 6X CNC controller and on my test ESP32, where it otherwise always works.

SnowBeaver, 2024-02-01:
I have been using this sdcard without problem for two years. I switch to 4000000 but that did not help.

MitchBradley, 2024-02-01:
Turn on Developer Tools in your browser and tell us the last command before the crash

SnowBeaver, 2024-02-01:
here is the list of message in Chrome after a file upload. It is the first time I copy/paste from developper mode, so could be garbage info

[MSG:ERR: sdmmc_card_init failed code 0x107]
(index):42 No SD card
(index):42 error:60
(index):42 
                
 POST http://fluidnc.local/upload net::ERR_CONNECTION_RESET
ProcessFileHttp @ (index):42
process_cmd @ (index):42
http_resultfn @ (index):42
ProcessGetHttp.a.onreadystatechange @ (index):42
XMLHttpRequest.send (async)
ProcessGetHttp @ (index):42
process_cmd @ (index):42
SendGetHttp @ (index):42
SendPrinterCommand @ (index):42
files_check_if_upload @ (index):42
onchange @ (index):41
(index):42 Disconnected
(index):42 Socket is 10.0.0.201:81
(index):42 Connected
(index):42 connection id = 0

MitchBradley, 2024-02-01:
I have reproduced the problem here.  Working on it.

MitchBradley, 2024-02-01:
Try this webui https://github.com/MitchBradley/ESP3D-WEBUI/commit/3e9cc8cd8bcdd05cbd7640b2eb080dbd63bb7e7f

MitchBradley, 2024-02-01:
The problem was caused by a recent change that attempted to clarify the distinction between synchronous and asynchronous WebUI commands.  It resulted in a pair of SD commands - one to test the SD card presence and the other to start an upload - being executed at the same time in different threads on different CPU cores.  That caused the SD card driver to be reentered, resulting in mayhem and a crash.

SnowBeaver, 2024-02-02:
I confirm that latest webui corrects the File uploading problem. Thanks Mitch for the quick fix.

BlueOrangeLive, 2024-02-02:
It works for me again too. Thank you Mitch for your quick action.

closed 2024-02-02 by SnowBeaver