bdring/FluidNC #1125: Lost probe with 3.7.13

massiveradiation, 2024-02-05:
### Wiki Search Terms

lost probe, probe fail

### Controller Board

6 Pack External Driver CNC Controller

### Machine Description

3 axis gantry mill, dual limit Y. Probe is NC so high.

### Input Circuits

```yaml
Standard connection of probe to gpio 34 on the board. Wire checked good. Was working on v3.7.10 and confirmed that it stops working with latest firmware.
```

### Configuration file

```yaml
board: 6 Pack
name: 6 Pack External XYYZ
meta: 2023-11-08 B. Dring for Tyler Cartner
stepping:
  engine: I2S_STREAM
  idle_ms: 250
  pulse_us: 4
  dir_delay_us: 1
  disable_delay_us: 0

axes:
  shared_stepper_disable_pin: NO_PIN
  x:
    steps_per_mm: 19.973
    max_rate_mm_per_min: 12000.000
    acceleration_mm_per_sec2: 600.000
    max_travel_mm: 548.000
    soft_limits: false
    homing:
      cycle: 2
      positive_direction: true
      mpos_mm: 0.000
      feed_mm_per_min: 25.000
      seek_mm_per_min: 500.000
      settle_ms: 250
      seek_scaler: 1.100
      feed_scaler: 1.100

    motor0:
      limit_neg_pin: gpio.12:low
      hard_limits: false
      pulloff_mm: 1.000
      standard_stepper:
        step_pin: I2SO.2
        direction_pin: I2SO.1
        disable_pin: I2SO.0:low

  y:
    steps_per_mm: 19.993

    max_rate_mm_per_min: 12000.000
    acceleration_mm_per_sec2: 600.000
    max_travel_mm: 4000.000
    soft_limits: false
    homing:
      cycle: 3
      positive_direction: false
      mpos_mm: 0.000
      feed_mm_per_min: 500.000
      seek_mm_per_min: 25.000
      settle_ms: 250
      seek_scaler: 1.100
      feed_scaler: 1.100

    motor0:
      limit_neg_pin: gpio.15:low
      hard_limits: false
      pulloff_mm: 1.000
      standard_stepper:
        step_pin: I2SO.5
        direction_pin: I2SO.4
        disable_pin: I2SO.7:low
    
    motor1:
      limit_neg_pin: gpio.13:low
      hard_limits: false
      pulloff_mm: 1.000
      standard_stepper:
        step_pin: I2SO.10
        direction_pin: I2SO.9
        disable_pin: I2SO.8:low

  z:
    steps_per_mm: 100.000
    max_rate_mm_per_min: 2400.000
    acceleration_mm_per_sec2: 350.000
    max_travel_mm: 103.000
    soft_limits: false
    homing:
      cycle: 1
      positive_direction: true
      mpos_mm: 150.000
      feed_mm_per_min: 25.000
      seek_mm_per_min: 500.000
      settle_ms: 250
      seek_scaler: 1.100
      feed_scaler: 1.100

    motor0:
      limit_pos_pin: gpio.14:low
      hard_limits: false
      pulloff_mm: 1.000
      standard_stepper:
        step_pin: I2SO.13
        direction_pin: I2SO.12
        disable_pin: I2SO.15:low

probe:
  pin: gpio.34:low
  check_mode_start: true

coolant:
  flood_pin: gpio.2
  mist_pin: NO_PIN
  delay_ms: 0

i2so:
  bck_pin: gpio.22
  data_pin: gpio.21
  ws_pin: gpio.17

spi:
  miso_pin: gpio.19
  mosi_pin: gpio.23
  sck_pin: gpio.18

sdcard:
  card_detect_pin: NO_PIN
  cs_pin: gpio.5

start:
  must_home: false
  
10V:
  forward_pin: gpio.4
  reverse_pin: gpio.16
  pwm_hz: 5000
  output_pin: gpio.26
  enable_pin: NO_PIN
  direction_pin: NO_PIN
  disable_with_s0: false
  s0_with_disable: true
  spinup_ms: 1000
  spindown_ms: 1000
  tool_num: 0
  speed_map: 0=0.000% 1000=0.000% 24000=100.000%
  off_on_alarm: false
```

### Startup Messages

```text
$ss
[MSG:INFO: FluidNC 3.7.10 https://github.com/bdring/FluidNC]
[MSG:INFO: Compiled with ESP32 SDK:v4.4.4]
[MSG:INFO: Local filesystem type is spiffs]
[MSG:INFO: Configuration file:config.yaml]
[MSG:WARN: Increasing stepping/pulse_us to the IS2 minimum value 4]
[MSG:INFO: Machine 6 Pack External XYYZ]
[MSG:INFO: Board 6 Pack]
[MSG:INFO: I2SO BCK:gpio.22 WS:gpio.17 DATA:gpio.21]
[MSG:INFO: SPI SCK:gpio.18 MOSI:gpio.23 MISO:gpio.19]
[MSG:INFO: SD Card cs_pin:gpio.5 detect:NO_PIN freq:8000000]
[MSG:INFO: Stepping:I2S_stream Pulse:4us Dsbl Delay:0us Dir Delay:1us Idle Delay:250ms]
[MSG:INFO: Axis count 3]
[MSG:INFO: Axis X (0.000,840.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     standard_stepper Step:I2SO.2 Dir:I2SO.1:low Disable:I2SO.0:low]
[MSG:INFO:  X Neg Limit gpio.27:low]
[MSG:INFO: Axis Y (0.000,830.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     standard_stepper Step:I2SO.5 Dir:I2SO.4:low Disable:I2SO.7:low]
[MSG:INFO:  Y Neg Limit gpio.16:low]
[MSG:INFO:   Motor1]
[MSG:INFO:     standard_stepper Step:I2SO.10 Dir:I2SO.9:low Disable:I2SO.8:low]
[MSG:INFO:  Y2 Neg Limit gpio.4:low]
[MSG:INFO: Axis Z (0.000,150.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     standard_stepper Step:I2SO.13 Dir:I2SO.12:low Disable:I2SO.15:low]
[MSG:INFO:  Z Pos Limit gpio.26:low]
[MSG:INFO: Kinematic system: Cartesian]
[MSG:INFO: 10V Spindle Ena:NO_PIN Out:gpio.14 Dir:NO_PIN Fwd:gpio.13 Rev:gpio.15 Freq:5000Hz Period:8191]
[MSG:INFO: Using spindle 10V]
[MSG:INFO: Flood coolant gpio.2]
[MSG:INFO: Probe Pin: gpio.34]
[MSG:INFO: STA SSID is not set]
[MSG:INFO: AP SSID FluidNC IP 192.168.0.1 mask 255.255.255.0 channel 1]
[MSG:INFO: AP started]
[MSG:INFO: WiFi on]
[MSG:INFO: Captive Portal Started]
[MSG:INFO: HTTP started on port 80]
[MSG:INFO: Telnet started on port 23]
ok
```

### User Interface Software

bCNC

### What happened?

Was playing with new firmware and hadn't realized that I lost probe until the following day.

bdring, 2024-02-05:
Attach a photo of your controller so I can see the wiring.

Send the `?` command with the probe circuit inactive and active and show the response.

massiveradiation, 2024-02-05:
gpio 34 is now set to high, not low, but working in v3.7.10.

massiveradiation, 2024-02-05:
[image]

massiveradiation, 2024-02-05:
`
[image]

bdring, 2024-02-05:
You had to change it to make it work in 3.7.10?

massiveradiation, 2024-02-05:
Yes, once I re-installed 3.7.10, the probe works fine. The added strange thing is that coolant is assigned to gpio 2 in config, but it's connected to 25, my mistake. However, M8 activates gpio 25 on the board. That's a head scratcher. I'm going to now change the gpio to 25 for coolant in config and see if M8 still works to trigger my relay.

MitchBradley, 2024-02-05:
This line
`[MSG:WARN: Increasing stepping/pulse_us to the IS2 minimum value 4]`

in conjunction with this line
`   pulse_us: 4`

makes me wonder if you might not be using the config file that you think you are using.

bdring, 2024-02-05:
"Yes, once I re-installed 3.7.10, the probe works fine."

What does the "yes" refer to?

massiveradiation, 2024-02-05:
Ok, I just changed config.yaml to gpio 25 for coolant and m8 no longer works. There is most definitely something going on with pin mapping. 3.7.10 somehow has assigned hardware gpio 2 to 25 pinout?! That is according to the pinout sheet that I downloaded from the wiki.

bdring, 2024-02-05:
I think you broke your config when you upgraded. You returned to the old firmware and things still did not work. You fixed some things in the config and now some stuff works.

massiveradiation, 2024-02-05:
"Yes" was in reference to your question, "you had to change to..."

bdring, 2024-02-05:
Did you try fixing that in the new firmware?

massiveradiation, 2024-02-05:
Sorry, fixing what?

massiveradiation, 2024-02-05:
I was never able to get the probe to work in the new firmware, with the config.yaml I uploaded.

bdring, 2024-02-05:
gpio 34 is now set to high, not low.

What was it in the new firmware?

massiveradiation, 2024-02-05:
Yes, I did set that correctly with the new firmware. I tried both high and low, but there was no response either way when manipulating the probe.

massiveradiation, 2024-02-05:
OK,  i figured out the pin swap issue! The pinout sheet on wiki is wrong. That pin I'm connected to is actually gpio 2.

bdring, 2024-02-05:
Look at the bottom of the controller for the pinout

bdring, 2024-02-05:
I'll fix the wiki

massiveradiation, 2024-02-05:
I am editing and uploading the correct config.yaml file. I just swapped the coolant gpio back to 2 and it's working again.

massiveradiation, 2024-02-05:
So, for the moment, I'll stick with 3.7.10. I had swapped to 3.7.13 and found the installer to work better, but I need the probe, so... Thanks for your help Bart! I haven't yet played with the new board.

massiveradiation, 2024-02-06:
Oh, I somehow missed this! "Send the ? command with the probe circuit inactive and active and show the response."
Unfortunately, I had to do some work so I switched back to 3.7.10. When I get a moment, I'll reinstall the new firmware and check this for you.

cbrassel, 2024-03-15:
Hi It seem's I have the same problem since upgrading to 3.7.13 I have lost the probe.
I'm using the xPro V5 controller, the probe output is at 5v. Everything other is working fine.
I will provide the ? output tomorrow.
Regard's
Claude

cbrassel, 2024-03-17:
here the normal status :
< Idle MPos: -733.962,507.194,-1.003|FS:0,0|Pn:Py>
and went the probe is triggered :
‹ Idle MPos: -733-962,507.194, -1.003 55:8,0|Pn: PY|Ov: 100, 100,100>
Which is very surprising that the Override appears ! And I confirm whiteout triggering the probe I have no Ov in the status
Any Idea ?

MitchBradley, 2024-03-17:
The Ov field is issued on every tenth or every twentieth report depending on the state.  It is unrelated to the probe trigger state.  This behavior is inherited from classic Grbl.

MitchBradley, 2024-03-17:
Excerpt from the official Grbl documentation at https://github.com/gnea/grbl/blob/master/doc/markdown/interface.md :
```
 - **Override Values:**

      - `Ov:100,100,100` indicates current override values in percent of programmed values for feed, rapids, and spindle speed, respectively.

      - Override maximum, minimum, and increment sizes are all configurable within config.h. Assume that a user or OEM will alter these based on customized use-cases. Recommend not hard-coding these values into a GUI, but rather just show the actual override values and generic increment buttons.

      - Override values don't change often during a job once set and only requires intermittent refreshing. This data field appears:

        - After 10 or 20 (configurable 1-255) status reports, depending on is in a motion state or not.
        - If an override value has changed, this data field will appear immediately in the next report. However, if `WCO:` is present, this data field will be delayed one report.
        - In the second report after a reset/power-cycle.
```
The part about it being configurable is true only if you are willing to recompile the code.

cbrassel, 2024-03-18:
Hello,

Ok for the override, but how can I debug/check the none working probe ? 

Regard's
Claude

bdring, 2024-03-18:
Send $probe at the console, so you can show me the config.

cbrassel, 2024-03-19:
client : Sprobe 
/probe:
pin: gpio.22:low:pu
toolsetter _pin: NO_PIN check _node_start: false ok

bdring, 2024-03-19:
The gpio.22 pin is used for the i2so section. You probably have errors in your startup messages about this. You can only use these pins as inputs.

[image]

After any change to your config file, check for errors in the messages.

cbrassel, 2024-03-19:
I'm sorry, I'm not @massiveradiation :) 
I have not the same board and not the same configuration, maybe I should have open a new request ! Sorry for that.
As mentionned earlier I have a Xpro V5 board 
Dp you want me to open another request ?

massiveradiation, 2024-03-19:
Hey folks,
I submitted this issue and then never tried again to update to the latest
firmware and check to see why the probe stopped working to provide
feedback. I simply haven't had the time as I use the machine every day for
work.
Cheers,
Tyler

bdring, 2024-03-19:
This issue is getting confusing.

cbrassel, 2024-03-19:
no problem, I open a new one.

bdring, 2024-12-12:
I think this is resolved. Let us know on a new issue if the problem exists on the latest firmware.

closed 2024-12-12 by bdring