bdring/FluidNC #1126: Problem: I am not sure what the exact problem is other than wifi.   It only stays connected in ap mode and only on my cheap old tablet. my computer contantly disconnects.   fluiidterm says its connected but i never stays connected.  The board manufacturer and i have exhausted all avenues and he seems to think it is a fluidnc thing.  I can run a project throught the we gui and it starts but I  rarely have any controls after that, no pause or it pauses and wont let me restart or it disconnects and does not show the file running when i reconnect.  I have erased and reloaded  twice but no luck.  after the job finishes the gcode is still running in the command line as if it is still going.

jayjojeans, 2024-02-06:
### Wiki Search Terms

wifi disconnect.

### Controller Board

root controller

### Machine Description

tb6600 driver nema 23 dual y 

### Configuration file

```yaml
board: Root Controller ISO
name: Rev 4 demo
stepping:
  engine: I2S_STREAM
  idle_ms: 255
  pulse_us: 4
  dir_delay_us: 1
  disable_delay_us: 3

axes:
  shared_stepper_disable_p
  x:
    steps_per_mm: 320.000
    max_rate_mm_per_min: 3
    acceleration_mm_per_se
    max_travel_mm: 901.7
    soft_limits: false
    homing:
      cycle: 2
      positive_direction: 
      mpos_mm: 100
      feed_mm_per_min: 100
      seek_mm_per_min: 800
      settle_ms: 500
      seek_scaler: 1.100
      feed_scaler: 1.100

    motor0:
      limit_neg_pin: gpio.
      limit_pos_pin: gpio.
      limit_all_pin: NO_PI
      hard_limits: false
      pulloff_mm: 5.000
      standard_stepper:
        step_pin: I2SO.7:l
        direction_pin: I2S
        disable_pin: I2SO.

  y:
    steps_per_mm: 320.000
    max_rate_mm_per_min: 3
    acceleration_mm_per_se
    max_travel_mm: 882.65
    soft_limits: false
    homing:
      cycle: 2
      positive_direction: 
      mpos_mm: 100.000
      feed_mm_per_min: 100
      seek_mm_per_min: 800
      settle_ms: 500
      seek_scaler: 1.100
      feed_scaler: 1.100

    motor0:
      limit_neg_pin: gpio.
      limit_pos_pin: gpio.
      limit_all_pin: NO_PI
      hard_limits: false
      pulloff_mm: 5.000
      standard_stepper:
        step_pin: I2SO.12:
        direction_pin: I2S
        disable_pin: I2SO.
      
    motor1:
      limit_neg_pin: gpio.
      limit_pos_pin: gpio.
      limit_all_pin: NO_PI
      hard_limits: false
      pulloff_mm: 5.000
      standard_stepper:
        step_pin: I2SO.13:
        direction_pin: I2S
        disable_pin: I2SO.

  z: 
    steps_per_mm: 199.166
    max_rate_mm_per_min: 5
    acceleration_mm_per_se
    max_travel_mm: 292.1
    soft_limits: false
    homing:
      cycle: 1
      positive_direction: 
      mpos_mm: 100
      feed_mm_per_min: 100
      seek_mm_per_min: 800
      settle_ms: 500
      seek_scaler: 1.100
      feed_scaler: 1.100

    motor0:
      limit_neg_pin: gpio.
      limit_pos_pin: NO_PI
      limit_all_pin: NO_PI
      hard_limits: false
      pulloff_mm:5.000
      standard_stepper:
        step_pin: I2SO.18:
        direction_pin: I2S
        disable_pin: I2SO.
        
i2so:
  bck_pin: gpio.22
  data_pin: gpio.12
  ws_pin: gpio.21

spi:
  miso_pin: gpio.19
  mosi_pin: gpio.23
  sck_pin: gpio.18

sdcard:
  card_detect_pin: NO_PIN
  cs_pin: gpio.5

control:
  safety_door_pin: NO_PIN
  reset_pin: NO_PIN
  feed_hold_pin: NO_PIN
  cycle_start_pin: NO_PIN
  macro0_pin: NO_PIN
  macro1_pin: NO_PIN
  macro2_pin: NO_PIN
  macro3_pin: NO_PIN

coolant:
  flood_pin: NO_PIN
  mist_pin: NO_PIN
  delay_ms: 0

probe:
  pin: gpio.14
  check_mode_start: true

macros:
  startup_line0:
  startup_line1:
  macro0: G90&G53G0Z-1&G0X
  macro1: 
  macro2: 
  macro3:
  
user_outputs:
  analog0_pin: NO_PIN
  analog1_pin: NO_PIN
  analog2_pin: NO_PIN
  analog3_pin: NO_PIN
  analog0_hz: 5000
  analog1_hz: 5000
  analog2_hz: 5000
  analog3_hz: 5000
  digital0_pin: NO_PIN
  digital1_pin: NO_PIN
  digital2_pin: NO_PIN
  digital3_pin: i2so.15

start:
  must_home: true
```

### Startup Messages

```text
$G
[GC:G0 G54 G17 G21 G90 G94 M5 M9 T0 F0 S0]
ok
<Idle|MPos:78.700,0.000,0.000|FS:0,0>
$ss
[MSG:INFO: FluidNC 3.7.13 https://github.com/bdring/FluidNC]]
[MSG:INFO: Compiled with ESP32 SDK:v4.4.4]]
[MSG:INFO: Local filesystem type is littlefs]]
[MSG:INFO: Configuration file:Root_4_test probe z 2.yaml]]
[MSG:INFO: Machine Rev 4 demo]]
[MSG:INFO: Board Root Controller ISO]]
[MSG:INFO: I2SO BCK:gpio.22 WS:gpio.21 DATA:gpio.12]]
[MSG:INFO: SPI SCK:gpio.18 MOSI:gpio.23 MISO:gpio.19]]
[MSG:INFO: SD Card cs_pin:gpio.5 detect:NO_PIN freq:8000000]]
[MSG:INFO: Stepping:I2S_stream Pulse:4us Dsbl Delay:3us Dir Delay:1us Idle Delay:255ms]]
[MSG:INFO: User Digital Output:3 on Pin:I2SO.15]]
[MSG:INFO: Axis count 3]]
[MSG:INFO: Axis X (-801.700,100.000)]]
[MSG:INFO:   Motor0]]
[MSG:INFO:     standard_stepper Step:I2SO.7:low Dir:I2SO.5:low Disable:I2SO.3]]
[MSG:INFO:  X Neg Limit gpio.34]]
[MSG:INFO:  X Pos Limit gpio.26]]
[MSG:INFO: Axis Y (-782.650,100.000)]]
[MSG:INFO:   Motor0]]
[MSG:INFO:     standard_stepper Step:I2SO.12:low Dir:I2SO.10:low Disable:I2SO.8]]
[MSG:INFO:  Y Neg Limit gpio.32]]
[MSG:INFO:  Y Pos Limit gpio.35]]
[MSG:INFO:   Motor1]]
[MSG:INFO:     standard_stepper Step:I2SO.13:low Dir:I2SO.11:low Disable:I2SO.9]]
[MSG:INFO:  Y2 Neg Limit gpio.2]]
[MSG:INFO:  Y2 Pos Limit gpio.15]]
[MSG:INFO: Axis Z (-192.100,100.000)]]
[MSG:INFO:   Motor0]]
[MSG:INFO:     standard_stepper Step:I2SO.18:low Dir:I2SO.16 Disable:I2SO.14]]
[MSG:INFO:  Z Neg Limit gpio.27]]
[MSG:INFO: Kinematic system: Cartesian]]
[MSG:INFO: Using spindle NoSpindle]]
[MSG:INFO: Probe Pin: gpio.14]]
[MSG:INFO: Connecting to STA SSID:Dewy123]]
[MSG:INFO: Connecting.]]
[MSG:INFO: Connecting..]]
```

### User Interface Software

webui ugs and openbuilds

### What happened?

Problem: I am not sure what the exact problem is other than wifi.   It only stays connected in ap mode and only on my cheap old tablet. my computer contantly disconnects.   fluiidterm says its connected but i never stays connected.  The board manufacturer and i have exhausted all avenues and he seems to think it is a fluidnc thing.  I can run a project throught the we gui and it starts but I  rarely have any controls after that, no pause or it pauses and wont let me restart or it disconnects and does not show the file running when i reconnect.  I have erased and reloaded  twice but no luck.  after the job finishes the gcode is still running in the command line as if it is still going.

MitchBradley, 2024-02-06:
Does the problem only happen when running a program?  If so, the problem could be caused by EMI from the spindle or other motors.

jayjojeans, 2024-02-06:
rann $ss again and this 
$ss
[MSG:INFO: FluidNC 3.7.13 https://github.com/bdring/FluidNC]]
[MSG:INFO: Compiled with ESP32 SDK:v4.4.4]]
[MSG:INFO: Local filesystem type is littlefs]]
[MSG:INFO: Configuration file:Root_4_test probe z 2.yaml]]
[MSG:INFO: Machine Rev 4 demo]]
[MSG:INFO: Board Root Controller ISO]]
[MSG:INFO: I2SO BCK:gpio.22 WS:gpio.21 DATA:gpio.12]]
[MSG:INFO: SPI SCK:gpio.18 MOSI:gpio.23 MISO:gpio.19]]
[MSG:INFO: SD Card cs_pin:gpio.5 detect:NO_PIN freq:8000000]]
[MSG:INFO: Stepping:I2S_stream Pulse:4us Dsbl Delay:3us Dir Delay:1us Idle Delay:255ms]]
[MSG:INFO: User Digital Output:3 on Pin:I2SO.15]]
[MSG:INFO: Axis count 3]]
[MSG:INFO: Axis X (-801.700,100.000)]]
[MSG:INFO:   Motor0]]
[MSG:INFO:     standard_stepper Step:I2SO.7:low Dir:I2SO.5:low Disable:I2SO.3]]
[MSG:INFO:  X Neg Limit gpio.34]]
[MSG:INFO:  X Pos Limit gpio.26]]
[MSG:INFO: Axis Y (-782.650,100.000)]]
[MSG:INFO:   Motor0]]
[MSG:INFO:     standard_stepper Step:I2SO.12:low Dir:I2SO.10:low Disable:I2SO.8]]
[MSG:INFO:  Y Neg Limit gpio.32]]
[MSG:INFO:  Y Pos Limit gpio.35]]
[MSG:INFO:   Motor1]]
[MSG:INFO:     standard_stepper Step:I2SO.13:low Dir:I2SO.11:low Disable:I2SO.9]]
[MSG:INFO:  Y2 Neg Limit gpio.2]]
[MSG:INFO:  Y2 Pos Limit gpio.15]]
[MSG:INFO: Axis Z (-192.100,100.000)]]
[MSG:INFO:   Motor0]]
[MSG:INFO: WiFi on]]
[MSG:INFO: Start mDNS with hostname:http://fluidnc.local/]]
[MSG:INFO: SSDP Started]]
[MSG:INFO: HTTP started on port 80]]
[MSG:INFO: Telnet started on port 23]]
ok

and the third time
$ss
[MSG:INFO: FluidNC 3.7.13 https://github.com/bdring/FluidNC]]
[MSG:INFO: Compiled with ESP32 SDK:v4.4.4]]
[MSG:INFO: Local filesystem type is littlefs]]
[MSG:INFO: Configuration file:Root_4_test probe z 2.yaml]]
[MSG:INFO: Machine Rev 4 demo]]
[MSG:INFO: Board Root Controller ISO]]
[MSG:INFO: I2SO BCK:gpio.22 WS:gpio.21 DATA:gpio.12]]
[MSG:INFO: SPI SCK:gpio.18 MOSI:gpio.23 MISO:gpio.19]]
[MSG:INFO: SD Card cs_pin:gpio.5 detect:NO_PIN freq:8000000]]
[MSG:INFO: Stepping:I2S_stream Pulse:4us Dsbl Delay:3us Dir Delay:1us Idle Delay:255ms]]
[MSG:INFO: User Digital Output:3 on Pin:I2SO.15]]
[MSG:INFO: Axis count 3]]
[MSG:INFO: Axis X (-801.700,100.000)]]
[MSG:INFO:   Motor0]]
[MSG:INFO:     standard_stepper Step:I2SO.7:low Dir:I2SO.5:low Disable:I2SO.3]]
[MSG:INFO:  X Neg Limit gpio.34]]
[MSG:INFO:  X Pos Limit gpio.26]]
[MSG:INFO: Axis Y (-782.650,100.000)]]
[MSG:INFO:   Motor0]]
[MSG:INFO:     standard_stepper Step:I2SO.12:low Dir:I2SO.10:low Disable:I2SO.8]]
[MSG:INFO:  Y Neg Limit gpio.32]]
[MSG:INFO:  Y Pos Limit gpio.35]]
[MSG:INFO: Connecting...]]
[MSG:INFO: Connected - IP is 192.168.1.177]]
[MSG:INFO: WiFi on]]
[MSG:INFO: Start mDNS with hostname:http://fluidnc.local/]]
[MSG:INFO: SSDP Started]]
[MSG:INFO: HTTP started on port 80]]
[MSG:INFO: Telnet started on port 23]]
ok

and it fails to transfer gcodes to sd card 75% of the time

MitchBradley, 2024-02-06:
Try with version 3.7.10

jayjojeans, 2024-02-06:
Yes, it happens all the time. I will try the other version. I do feel like it was working better before the update.

Sent from my Verizon, Samsung Galaxy smartphone