bdring/FluidNC #1127: I'm getting soft limit errors within the prescribed distances but can jog normally

twoofknaves, 2024-02-07:
### Wiki Search Terms

Soft limit error. ALARM:2 Soft limit alarm. 

### Controller Board

Xpro V5

### Machine Description

Ultimate Bee 1515 220v 2.2kw water cooled spindle. 

### Configuration file

```yaml
board: tmc_5160_xpro v5
name: ULTIMATE BEE
stepping:
  engine: RMT
  idle_ms: 255
  pulse_us: 2
  dir_delay_us: 0
  disable_delay_us: 0

axes:
  shared_stepper_disable_pin: NO_PIN
  x:
    steps_per_mm: 320.300
    max_rate_mm_per_min: 2500.00
    acceleration_mm_per_sec2: 250.00
    max_travel_mm: 1240.00
    soft_limits: true
    homing:
      cycle: 2
      positive_direction: true
      mpos_mm: 1240.000
      feed_mm_per_min: 200.000
      seek_mm_per_min: 1500.00
      seek_scaler: 1.100
      feed_scaler: 1.100

    motor0:
      limit_neg_pin: NO_PIN
      limit_pos_pin: gpio.35
      limit_all_pin: NO_PIN
      hard_limits: false
      pulloff_mm: 3.00
      tmc_5160:
        cs_pin: gpio.17
        spi_index: 1
        r_sense_ohms: 0.075
        run_amps: 2.5
        hold_amps: 0.50
        microsteps: 16
        stallguard: 0
        stallguard_debug: false
        toff_disable: 0
        toff_stealthchop: 5
        toff_coolstep: 3
        run_mode: CoolStep
        homing_mode: CoolStep
        use_enable: true
        step_pin: gpio.12
        direction_pin: gpio.14
        tpfd: 4
        
       
  y:
    steps_per_mm: 320.3
    max_rate_mm_per_min: 2500.00
    acceleration_mm_per_sec2: 200.00
    max_travel_mm: 1240.00
    soft_limits: true
    homing:
      cycle: 2
      positive_direction: true
      mpos_mm: 1240.000
      feed_mm_per_min: 100.000
      seek_mm_per_min: 1500.00
      seek_scaler: 1.100
      feed_scaler: 1.100
      
    motor0:
      limit_neg_pin: NO_PIN
      limit_pos_pin: NO_PIN
      limit_all_pin: NO_PIN
      hard_limits: false
      pulloff_mm: 3.00
      tmc_5160:
        cs_pin: NO_PIN
        spi_index: 4
        r_sense_ohms: 0.075
        run_amps: 2.5
        hold_amps: 0.50
        microsteps: 16
        stallguard: 0
        stallguard_debug: false
        toff_disable: 0
        toff_stealthchop: 5
        toff_coolstep: 3
        run_mode: CoolStep
        homing_mode: CoolStep
        use_enable: true
        step_pin: gpio.33
        direction_pin: gpio.32:low
        tpfd: 4
        
        
    motor1:
      limit_neg_pin: NO_PIN
      limit_pos_pin: gpio.34
      limit_all_pin: NO_PIN
      hard_limits: false
      pulloff_mm: 3.00
      tmc_5160:
        cs_pin: NO_PIN
        spi_index: 2
        r_sense_ohms: 0.075
        run_amps: 2.5
        hold_amps: 0.50
        microsteps: 16
        stallguard: 0
        stallguard_debug: false
        toff_disable: 0
        toff_stealthchop: 5
        toff_coolstep: 3
        run_mode: CoolStep
        homing_mode: CoolStep
        use_enable: true
        step_pin: gpio.27
        direction_pin: gpio.26:low
        tpfd: 4
       
  z:
    steps_per_mm: 799.430
    max_rate_mm_per_min: 2000.00
    acceleration_mm_per_sec2: 200.00
    max_travel_mm: 160.00
    soft_limits: true
    homing:
      cycle: 1
      positive_direction: true
      mpos_mm: 150.000
      feed_mm_per_min: 100.000
      seek_mm_per_min: 1500.00
      seek_scaler: 1.100
      feed_scaler: 1.100

    motor0:
      limit_neg_pin: NO_PIN
      limit_pos_pin: gpio.39
      limit_all_pin: NO_PIN
      hard_limits: false
      pulloff_mm: 3.00
      tmc_5160:
        cs_pin: NO_PIN
        spi_index: 3
        r_sense_ohms: 0.075
        run_amps: 2.5
        hold_amps: 0.50
        microsteps: 16
        stallguard: 0
        stallguard_debug: false
        toff_disable: 0
        toff_stealthchop: 5
        toff_coolstep: 3
        run_mode: CoolStep
        homing_mode: CoolStep
        use_enable: true
        step_pin: gpio.15
        direction_pin: gpio.2
        tpfd: 4

spi:
  miso_pin: gpio.19
  mosi_pin: gpio.23
  sck_pin: gpio.18

sdcard:
  card_detect_pin: NO_PIN
  cs_pin: gpio.5

control:
  safety_door_pin: gpio.16
  reset_pin: NO_PIN
  feed_hold_pin: NO_PIN
  cycle_start_pin: NO_PIN
  macro0_pin: gpio.13: low
  macro1_pin: gpio.0: low
  macro2_pin: NO_PIN
  macro3_pin: NO_PIN

probe:
  pin: gpio.22: low
  check_mode_start: true

macros:
  startup_line0:
  startup_line1:
  macro0:
  macro1:
  macro2:
  macro3:

user_outputs:
  analog0_pin: NO_PIN
  analog1_pin: NO_PIN
  analog2_pin: NO_PIN
  analog3_pin: NO_PIN
  analog0_hz: 5000
  analog1_hz: 5000
  analog2_hz: 5000
  analog3_hz: 5000
  digital0_pin: NO_PIN
  digital1_pin: NO_PIN
  digital2_pin: NO_PIN
  digital3_pin: NO_PIN
  
start:
  must_home: true

coolant:
  mist_pin: gpio.21
  delay_ms: 0
  
Huanyang:
 uart:
   txd_pin: gpio.4
   rxd_pin: gpio.25
   rts_pin: NO_PIN
   baud: 9600
   mode: 8N1
 modbus_id: 1
 tool_num: 0
 speed_map: 0=0% 6000=25% 24000=100%
 off_on_alarm: false
```

### Startup Messages

```text
Grbl 3.7 [FluidNC 3.7.13 (wifi) '$' for help]
```

### User Interface Software

Fluidwifi

### What happened?

I've had success with this yaml file in homing and giving g91 codes but all of a sudden its stopped. I get a soft limit error even when I ask it to move 5mm and the spindle is in the middle of the table.  The work position is often incorrect. The machine jogs fine after initializing and going through homing on all axes. 

This file has worked fine before and I've attempted to remove soft limits with false but it still gives the same error. This occurs even if the VFD is unplugged from the unit although in previous files the vfd functioned well without interrupting the machine. It didn't like how my speed % was set up so I fixed that but I'm still not able to position the tool head correctly.

### Other Information

ALARM:2
Soft limit alarm. G-code motion target exceeds machine travel. Machine position retained. Alarm may be safely unlocked, click the Reset Button.
[MSG:ERR: Reset to continue]
g90x-30
[MSG:INFO: Soft limit on Z target:150.001]
ok
[MSG:INFO: ALARM: Soft Limit]
ALARM:2
Soft limit alarm. G-code motion target exceeds machine travel. Machine position retained. Alarm may be safely unlocked, click the Reset Button.
[MSG:ERR: Reset to continue]
g90y-30
[MSG:INFO: Soft limit on Z target:150.001]
ok
[MSG:INFO: ALARM: Soft Limit]
ALARM:2
Soft limit alarm. G-code motion target exceeds machine travel. Machine position retained. Alarm may be safely unlocked, click the Reset Button.
[MSG:ERR: Reset to continue]
g90z-20
ok
g90z-40
ok

All of those "ok's" are ignored by the mahine.

MitchBradley, 2024-02-07:
Please supply all the startup message as shown by $SS, not just the one signon line.

MitchBradley, 2024-02-07:
Also what is "fluidwifi"?  Do you mean WebUI?   Did you home the machine first?  We need complete information including everything that you did after starting the machine.

twoofknaves, 2024-02-07:
Thanks Mitch. I do home the machine after every startup and firmware change because otherwise it wont function.  Yes I meant webui using wifi (tried to designate that it wasnt bluetooth)

I simply home the machine, enter a g91 code and get the error. Reset, and it jogs perfectly. Tell it another code like g91x-50 while its at the 1240mm positive home postion, or 1200 etc. and the soft limit error pops up.

This is new after I fixed the speed_map because it had 0% and 0-25%. But this occurs without powering up the VFD

MitchBradley, 2024-02-07:
I am going to guess that you have a work coordinate offset that is outside the machine range.  But I can't tell for sure because you have still not provided enough data for me to evaluate the situation.  I would like to see the complete startup messages, and a complete list of exactly what you did.

twoofknaves, 2024-02-07:
$G
[GC:G0 G54 G17 G21 G90 G94 M5 M9 T0 F0 S0]
ok
<Alarm|MPos:0.000,0.000,0.000|FS:0,0|WCO:0.000,0.000,0.001>
$SS
[MSG:INFO: FluidNC 3.7.13 https://github.com/bdring/FluidNC]]
[MSG:INFO: Compiled with ESP32 SDK:v4.4.4]]
[MSG:INFO: Local filesystem type is littlefs]]
[MSG:INFO: Configuration file:Feb6-24mk2.yml]]
[MSG:INFO: Machine ULTIMATE BEE]]
[MSG:INFO: Board tmc_5160_xpro v5]]
[MSG:INFO: SPI SCK:gpio.18 MOSI:gpio.23 MISO:gpio.19]]
[MSG:INFO: SD Card cs_pin:gpio.5 detect:NO_PIN freq:8000000]]
[MSG:INFO: Stepping:RMT Pulse:2us Dsbl Delay:0us Dir Delay:0us Idle Delay:255ms]]
[MSG:INFO: Axis count 3]]
[MSG:INFO: Axis X (0.000,1240.000)]]
[MSG:INFO:   Motor0]]
[MSG:INFO:     tmc_5160 Step:gpio.12 Dir:gpio.14 CS:gpio.17 Disable:NO_PIN Index:1 R:0.075]]
[MSG:INFO:  X Pos Limit gpio.35]]
[MSG:INFO: Axis Y (0.000,1240.000)]]
[MSG:INFO:   Motor0]]
[MSG:INFO:     tmc_5160 Step:gpio.33 Dir:gpio.32:low CS:NO_PIN Disable:NO_PIN Index:4 R:0.075]]
[MSG:INFO:   Motor1]]
[MSG:INFO:     tmc_5160 Step:gpio.27 Dir:gpio.26:low CS:NO_PIN Disable:NO_PIN Index:2 R:0.075]]
[MSG:INFO:  Y2 Pos Limit gpio.34]]
[MSG:INFO: Mist coolant gpio.21]]
[MSG:INFO: Probe Pin: gpio.22:low]]

twoofknaves, 2024-02-07:
$X
[MSG:INFO: Caution: Unlocked]
ok
$H
ok
<Idle|MPos:1240.000,1240.000,150.001|FS:0,0|Ov:100,100,100>
g91x-500
[MSG:INFO: Soft limit on Z target:150.001]
ok
[MSG:INFO: ALARM: Soft Limit]
ALARM:2
Soft limit alarm. G-code motion target exceeds machine travel. Machine position retained. Alarm may be safely unlocked, click the Reset Button.
[MSG:ERR: Reset to continue]

Heres what happens when I try to move along the x axis

MitchBradley, 2024-02-07:
Thanks, now type ? several times until you see a status report with a WCO: field and tell me what is in that field.

twoofknaves, 2024-02-07:
Can you explain how to do that? I'm pressing "?" and enter over and over and I've tried pressing "?????????" and enter. lol. I hope i'm not the king of amateur hour over here

MitchBradley, 2024-02-07:
Hit enter or send after each one if you are in the WebUI console

MitchBradley, 2024-02-07:
Do you have verbose enabled in the WebUI console?

Support is much easier with FluidTerm because there is less in the way.

twoofknaves, 2024-02-07:
oh.. finally got it <Alarm|MPos:1240.000,1240.000,150.001|FS:0,0|WCO:0.000,0.000,0.001>

MitchBradley, 2024-02-07:
Looks like there is a 0.001 offset on the Z axis

twoofknaves, 2024-02-07:
Is there a way to fix that? I'm not sure why that would happen

MitchBradley, 2024-02-07:
We might be seeing a rounding error problem.    Internally, positions are held in integer units of steps.  Your Z axis steps_per_mm is 799.43.  Multiply that by 150 mm and you get 11914.5 , which is not an exact integer so when it gets converted to an integer it is rounded up to 11915 .  Convert that back to floating point by dividing by 799.43 and that gives 150.000625 which is closer to 150.001 .

twoofknaves, 2024-02-07:
I really feel like I botched the file for the Z stepspermm. I used a calculator to correct the movement but I feel like it shouldnt be so far off the 320 of the other two. I'm pretty sure its the same ball drive and pitch on the ultimate bee 1515

MitchBradley, 2024-02-07:
I suspect that it would work if the z steps_per_mm were 800.  The closest value to 799.43 that doesn't cause rounding problems at 150 mm is 799.42

twoofknaves, 2024-02-07:
Mitch, youre a god amongst plebs with hobby interests outside of their skill level. Thank you!

I'm ready to test the VFD for EMI now (which hasnt been a problem from before this started) I haven't changed the z steps_per_mm for days so I would have never caught that and I'm not sure why it started becoming an issue. Maybe running the VFD started the error?  Its on a 220 circuit and the steppers are on a separate 110. I've heard people having less issues when theyre both on the same 220 circuit so if it crops up again i might try that

Thanks again!

MitchBradley, 2024-02-07:
It seems unlikely that VFD interference would cause this particular problem.  If it were me I would jog off the Z and set the work coordinate zero somewhere away from the limits of travel.

twoofknaves, 2024-02-07:
Do you mean to initially jog it and then set zero with the zero button?

I'm having issues with seeing the work position to the updated current value. I incrementally moved X axis over the entire 1240mm but it only showed it to be 640 away. Once I home the z, it updated the x position

MitchBradley, 2024-02-07:
I'm sorry, I don't want to think about this anymore right now.  Got other stuff on my mind.

twoofknaves, 2024-02-07:
No problem, Mitch. Appreciate the speedy help. I'll do some research. Thanks again