bdring/FluidNC #1132: FluidNC apresenta a menssagem "SetConsoleColor Failed" na instalação.

ArildoKamimura, 2024-02-11:
### Wiki Search Terms

Mensagens de Erro, 
Erro instalação FluidTerm, 
Mensagem Set Console Color Failed,
Console color failed,
Instalar FluidTerm

### Controller Board

ESP 32 Wemos D1 R32,
ESP32 Devkit V1,
ESP32 Wroom 38 pinos

### Machine Description

CNC 3 eixos com 3 motores Nema 17,
3 Drivers DRV 8825
Módulo Laser 10W
Não foi conectado nenhuma placa ao equipamento, pois não foi possivel instalar o FluidNC nas ESPs

### Input Circuits

```yaml
Nada conectado ainda
```

### Configuration file

```yaml
name: "ESP32 Dev Controller V4"
board: "ESP32 Dev Controller V4"

stepping:
  engine: RMT
  idle_ms: 250
  dir_delay_us: 1
  pulse_us: 2
  disable_delay_us: 0

axes:
  shared_stepper_disable_pin: gpio.13:low
  
  x:
    steps_per_mm: 800
    max_rate_mm_per_min: 2000
    acceleration_mm_per_sec2: 25
    max_travel_mm: 1000
    homing:
      cycle: 2
      mpos_mm: 10
      positive_direction: false
    
    motor0:
      limit_all_pin: gpio.17:low:pu
      stepstick:
        direction_pin: gpio.14
        step_pin: gpio.12
    motor1:
      null_motor:

  y:
    steps_per_mm: 800
    max_rate_mm_per_min: 2000
    acceleration_mm_per_sec2: 25
    max_travel_mm: 1000
    homing:
      cycle: 2
      mpos_mm: 10
      positive_direction: false

    motor0:
      limit_all_pin: gpio.4:low:pu
      stepstick:
        direction_pin: gpio.15
        step_pin: gpio.26
    motor1:
      null_motor:

  z:
    steps_per_mm: 800
    max_rate_mm_per_min: 2000
    acceleration_mm_per_sec2: 25
    max_travel_mm: 1000
    homing:
      cycle: 1
      mpos_mm: 10
      positive_direction: true

    motor0:
      limit_all_pin: gpio.16:low:pu
      stepstick:
        direction_pin: gpio.33
        step_pin: gpio.27
    motor1:
      null_motor:

spi:
  miso_pin: gpio.19
  mosi_pin: gpio.23
  sck_pin: gpio.18

sdcard:
  cs_pin: gpio.5
  card_detect_pin: NO_PIN

coolant:
  flood_pin: gpio.25
  mist_pin:  gpio.21

        
probe:
  pin: gpio.32:low:pu

PWM:
  pwm_hz: 5000
  output_pin: gpio.2
  enable_pin: gpio.22
  direction_pin: NO_PIN
  disable_with_s0: false
  s0_with_disable: true
  spinup_ms: 0
  spindown_ms: 0
  tool_num: 0
  speed_map: 0=0% 10000=100%
```

### Startup Messages

```text
Não existem mensagens, não carrega para poder iniciar.
```

### User Interface Software

Somente tentando instalar o FluidTerm, sem sucesso

### What happened?

Tentei instalar o FluidNC tanto pelo arquivo Install-wifi, depois o install-bt
tentei pelo Web Installer FluidNC Wifi e BT
limpei os ESP32 com o arquivo Erase e tentei reinstalar todos
tentei instalar o fluidterm após a instalação do fluidNC

### GCode File

ainda não consegui chegar nesta etapa

### Other Information

[image]

MitchBradley, 2024-02-12:
That usually means that you are using a really old version of Windows, which we do not support

DmiZar, 2024-02-12:
I tried running the program on different computers. On Win 11 with an i3-8130 processor with a graphics core it does not work with the same error. It works fine on a Win 11 virtual machine (Virtual Box) and on Win 11 with an Nvidia graphics card.

MitchBradley, 2024-02-12:
A failure of FluidTerm does not mean that the install failed.  The installer just runs FluidTerm at the end as a convenience.  You can interact with FluidNC later, using any terminal emulator program like PuTTY or TeraTerm, or with WebUI or FluidNC WebInstaller in a browser, or the console in a sender like UGS, CNCjs, lightburn, gSender, or many others.  You also might be able to run FluidTerm from a PowerShell terminal instead of a cmd.exe terminal.

Since there are so many other options for interacting with FluidNC, and since I cannot reproduce your problem on my machine, I am going to close this issue as "will not fix".

closed 2024-02-12 by MitchBradley