bdring/FluidNC #1133: Homing Cycle Fails for Axes with Pre-Engaged Limit Switches in Positive Direction

040medien, 2024-02-12:
### Wiki Search Terms

max_travel_mm pulloff_mm positive_direction
Homing fail. Pull off travel failed to clear limit switch

### Controller Board

TinyBee

### Machine Description

Lowrider V3 (gantry router with XYYZZ configuration and dual endstops on Y and Z) with TMC2209 (standalone) drivers

### Input Circuits

```yaml
regular NC endstops are connected to signal and ground pins of the negative endstops on the board via simple two-wire cables. Also using TB, TH1 MT_DET as additional endstop pins.
```

### Configuration file

```yaml
board: MKS TinyBee V1.0_003
name: MPCNC Lowrider v3 XYYZZ
meta: (08.02.2024) by Frederik

kinematics:
  Cartesian:

i2so:
  bck_pin: gpio.25
  data_pin: gpio.27
  ws_pin: gpio.26

spi:
  miso_pin: gpio.19
  mosi_pin: gpio.23
  sck_pin: gpio.18

stepping:
  engine: I2S_STATIC
 # idle_ms: 0
 # pulse_us: 4
 # dir_delay_us: 1
 # disable_delay_us: 0
  idle_ms: 255
  pulse_us: 4
  dir_delay_us: 1
  disable_delay_us: 2

sdcard:
  cs_pin: gpio.5
  # uses TH2 IO34 active low - MAKE SURE jumper J2 is set to SDDET!!!
  card_detect_pin: NO_PIN
  #gpio.34:low

axes:
  x:
    # X
    steps_per_mm: 100
    max_rate_mm_per_min: 9000
    acceleration_mm_per_sec2: 230
    max_travel_mm: 672
    soft_limits: true
    homing:
      cycle: 2
      positive_direction: false
      mpos_mm: 0.000
      feed_mm_per_min: 300.000
      seek_mm_per_min: 3600.000
      settle_ms: 500
      seek_scaler: 1.100
      feed_scaler: 1.100

    motor0:
      limit_neg_pin: gpio.33:high:pu
      hard_limits: true
      pulloff_mm: 1.000
      stepstick:
        # X
        step_pin: I2SO.1
        direction_pin: I2SO.2:low
        disable_pin: I2SO.0

  y:
    # Y
    steps_per_mm: 100
    max_rate_mm_per_min: 9000
    acceleration_mm_per_sec2: 230
    max_travel_mm: 1683
    soft_limits: true
    homing:
      cycle: 3
      positive_direction: false
      mpos_mm: 0.000
      feed_mm_per_min: 300.000
      seek_mm_per_min: 3600.000
      settle_ms: 500
      seek_scaler: 1.100
      feed_scaler: 1.100

    motor0:
        limit_neg_pin: gpio.32:high:pu
        hard_limits: true
        pulloff_mm: 1.000
        stepstick:
            # Y
            step_pin: I2SO.4
            direction_pin: I2SO.5
            disable_pin: I2SO.3

    motor1:
        # MT_DET
        limit_neg_pin: gpio.35:high
        hard_limits: true
        pulloff_mm: 1.000
        stepstick:
            # Z
            step_pin: I2SO.7
            direction_pin: I2SO.8:low
            disable_pin: I2SO.6

  z:
    # Z
    steps_per_mm: 400
    max_rate_mm_per_min: 800
    acceleration_mm_per_sec2: 80
    max_travel_mm: 73
    soft_limits: true
    homing:
      cycle: 1
      positive_direction: true
      mpos_mm: 73.000
      feed_mm_per_min: 100.000
      seek_mm_per_min: 500.000
      settle_ms: 500
      seek_scaler: 1.100
      feed_scaler: 1.100

    motor0:
      limit_neg_pin: gpio.22:high:pu
      hard_limits: true
      pulloff_mm: 1.000
      stepstick:
          # E0
          step_pin: I2SO.10
          direction_pin: I2SO.11:low
          disable_pin: I2SO.9

    motor1:
        # TH1
        limit_neg_pin: gpio.36:high
        hard_limits: true
        pulloff_mm: 1.000
        stepstick:
            # E1
            step_pin: I2SO.13
            direction_pin: I2SO.14:low
            disable_pin: I2SO.12

control:
  safety_door_pin: NO_PIN
  reset_pin: NO_PIN
  feed_hold_pin: NO_PIN
  cycle_start_pin: NO_PIN
  macro0_pin: NO_PIN
  macro1_pin: NO_PIN
  macro2_pin: NO_PIN
  macro3_pin: NO_PIN

macros:
  # Start the case fan (FAN1) on startup
  startup_line0: M62 P0
  startup_line1:
  macro0:
  macro1:
  macro2:
  macro3:

user_outputs:
  analog0_pin: NO_PIN
  analog1_pin: NO_PIN
  analog2_pin: NO_PIN
  analog3_pin: NO_PIN
  analog0_hz: 5000
  analog1_hz: 5000
  analog2_hz: 5000
  analog3_hz: 5000
  # FAN1
  digital0_pin: i2so.19
  # FAN2
  digital1_pin: i2so.20
  # BEEPER
  digital2_pin: i2so.21
  # H-BED
  digital3_pin: i2so.16
  # H-E0
  digital4_pin: i2so.17
  # H-E1
  digital5_pin: i2so.18
start:
  must_home: false

relay:
    direction_pin: NO_PIN
    # BTN_EN2 on EXP2
    output_pin: gpio.12
    enable_pin: NO_PIN
    disable_with_s0: false
    s0_with_disable: true
    spinup_ms: 3000
    spindown_ms: 5000
    tool_num: 0
    speed_map: 0=0.000% 0=100.000% 1=100.000%
    off_on_alarm: true

probe:
  # TB
  pin: gpio.39:low
```

### Startup Messages

```text
[MSG:INFO: FluidNC 3.7.13 https://github.com/bdring/FluidNC]]
[MSG:INFO: Compiled with ESP32 SDK:v4.4.4]]
[MSG:DBG: Spiffs mount failed: ESP_FAIL]]
[MSG:INFO: Local filesystem type is littlefs]]
[MSG:INFO: Configuration file:config.yaml]]
[MSG:DBG: Running after-parse tasks]]
[MSG:DBG: Checking configuration]]
[MSG:INFO: Machine MPCNC Lowrider v3 XYYZZ]]
[MSG:INFO: Board MKS TinyBee V1.0_003]]
[MSG:INFO: I2SO BCK:gpio.25 WS:gpio.26 DATA:gpio.27]]
[MSG:INFO: SPI SCK:gpio.18 MOSI:gpio.23 MISO:gpio.19]]
[MSG:INFO: SD Card cs_pin:gpio.5 detect:NO_PIN freq:8000000]]
[MSG:INFO: Stepping:I2S_static Pulse:4us Dsbl Delay:2us Dir Delay:1us Idle Delay:255ms]]
[MSG:INFO: User Digital Output:0 on Pin:I2SO.19]]
[MSG:INFO: User Digital Output:1 on Pin:I2SO.20]]
[MSG:INFO: User Digital Output:2 on Pin:I2SO.21]]
[MSG:INFO: User Digital Output:3 on Pin:I2SO.16]]
[MSG:INFO: User Digital Output:4 on Pin:I2SO.17]]
[MSG:INFO: User Digital Output:5 on Pin:I2SO.18]]
[MSG:INFO: Axis count 3]]
[MSG:INFO: Axis X (0.000,672.000)]]
[MSG:INFO:   Motor0]]
[MSG:INFO:     stepstick Step:I2SO.1 Dir:I2SO.2:low Disable:I2SO.0]]
[MSG:INFO:  X Neg Limit gpio.33:pu]]
[MSG:INFO: Axis Y (0.000,1683.000)]]
[MSG:INFO:   Motor0]]
[MSG:INFO:     stepstick Step:I2SO.4 Dir:I2SO.5 Disable:I2SO.3]]
[MSG:INFO:  Y Neg Limit gpio.32:pu]]
[MSG:INFO:   Motor1]]
[MSG:INFO:     stepstick Step:I2SO.7 Dir:I2SO.8:low Disable:I2SO.6]]
[MSG:INFO:  Y2 Neg Limit gpio.35]]
[MSG:INFO: Axis Z (0.000,73.000)]]
[MSG:INFO:   Motor0]]
[MSG:INFO:     stepstick Step:I2SO.10 Dir:I2SO.11:low Disable:I2SO.9]]
[MSG:INFO:  Z Neg Limit gpio.22:pu]]
[MSG:INFO:   Motor1]]
[MSG:INFO:     stepstick Step:I2SO.13 Dir:I2SO.14:low Disable:I2SO.12]]
[MSG:INFO:  Z2 Neg Limit gpio.36]]
[MSG:INFO: Kinematic system: Cartesian]]
[MSG:INFO: Relay Spindle Ena:NO_PIN Out:gpio.12 Dir:NO_PIN]]
[MSG:INFO: Using spindle Relay]]
[MSG:INFO: Probe Pin: gpio.39:low]]
[MSG:INFO: Connecting to STA SSID:fnet]]
[MSG:INFO: Connecting.]]
[MSG:INFO: Connecting..]]
[MSG:INFO: Connected - IP is 192.168.10.95]]
[MSG:INFO: WiFi on]]
[MSG:INFO: Start mDNS with hostname:http://lowrider.local/]]
[MSG:INFO: SSDP Started]]
[MSG:INFO: HTTP started on port 80]]
[MSG:INFO: Telnet started on port 23]]
```

### User Interface Software

WebUI

### What happened?

My Z axis is set to home in the positive direction. If its limit switches are already engaged at the beginning of the cycle, the pull-off procedure fails to disengage the switch before proceeding with the homing sequence. This issue results in the homing cycle not completing as expected for the affected axis.
I actually tested the same behavior after flipping the direction of the stepper motors in the config and setting it to home to the "negative" direction with the same endstops. It worked fine, pulling off the axis from the endstops.

I get this error:
```[MSG:DBG: Planned move to 0.000,0.000,1.000 @ 100.000]
[MSG:DBG: enable delay:2] [MSG:DBG: enable delay:2]
[MSG:DBG: CycleStop PrePulloff]
[MSG:DBG: enable delay:2]
[MSG:DBG: enable delay:2]
[MSG:INFO: ALARM: Homing Fail Pulloff]
ALARM:8
Homing fail. Pull off travel failed to clear limit switch. Try increasing pull-off setting or check wiring.
[MSG:ERR: Macro can only be used in idle state] ok
<Alarm|MPos:0.000,0.000,1.000|FS:0,0|Pn:Z|WC0:0.000,0.000,0.000> 
```

Steps to Reproduce
Set up an axis to home to its max position.
Engage the positive limit switch(es) of the axis manually (e.g., by moving the axis to its positive limit).
Initiate the homing cycle for the machine or the specific axis.
Observe that the axis fails to move away from the limit switch and complete the homing cycle correctly.

Expected Behavior
The axis should detect that its limit switch is already engaged at the start of the homing cycle and perform an initial pull-off motion in the opposite direction to disengage the switch before proceeding with the standard homing procedure.

Current Behavior
The axis does not correctly perform the initial pull-off motion to disengage the engaged limit switch when set to home in the positive direction, leading to a failure in completing the homing cycle.

MitchBradley, 2024-02-12:
Shouldn't you have limit_pos_pin instead of limit_neg_pin on positive direction axes?

040medien, 2024-02-12:
D'oh! That fixed it. Thank you @MitchBradley!

closed 2024-02-12 by 040medien