bdring/FluidNC #1135: Problem: Z-AXIS not responsive

tomerinbar1, 2024-02-17:
### Wiki Search Terms

Motor Z-Axis not responsive to commands

### Controller Board

MKS DLC32 2.1
[image]

### Machine Description

3018 Pro Max CNC Machine, Spindle motor 24V 200W 12000 rpm

### Configuration file

```yaml
board: MKS-DLC32 V2.1
name: 3018 Pro Max Spindle
meta: (17.02.2024) by Tomer

kinematics:
  Cartesian:

stepping:
  engine: I2S_STATIC
  idle_ms: 0
  pulse_us: 4
  dir_delay_us: 1
  disable_delay_us: 0
axes:
  shared_stepper_disable_pin: I2SO.0
  x:
    steps_per_mm: 157.750
    max_rate_mm_per_min: 18000.000
    acceleration_mm_per_sec2: 1500.000
    max_travel_mm: 325.000
    soft_limits: true
    homing:
      cycle: 1
      positive_direction: false
      mpos_mm: 0.000
      feed_mm_per_min: 300.000
      seek_mm_per_min: 5000.000
      settle_ms: 500
      seek_scaler: 1.100
      feed_scaler: 1.100

    motor0:
      limit_neg_pin: gpio.36
      hard_limits: false
      pulloff_mm: 2.000
      stepstick:
        step_pin: I2SO.1
        direction_pin: I2SO.2

  y:
    steps_per_mm: 157.750
    max_rate_mm_per_min: 12000.000
    acceleration_mm_per_sec2: 300.000
    max_travel_mm: 220.000
    soft_limits: true
    homing:
      cycle: 1
      positive_direction: false
      mpos_mm: 0.000
      feed_mm_per_min: 300.000
      seek_mm_per_min: 5000.000
      settle_ms: 500
      seek_scaler: 1.100
      feed_scaler: 1.100

    motor0:
      limit_neg_pin: gpio.35
      hard_limits: false
      pulloff_mm: 2.000
      stepstick:
        step_pin: I2SO.5
        direction_pin: I2SO.6:low

  z:
    steps_per_mm: 157.750
    max_rate_mm_per_min: 12000.000
    acceleration_mm_per_sec2: 500.000
    max_travel_mm: 80.000
    soft_limits: true
    homing:
      cycle: 0
      positive_direction: false
      mpos_mm: 0.000
      feed_mm_per_min: 300.000
      seek_mm_per_min: 1000.000
      settle_ms: 500
      seek_scaler: 1.100
      feed_scaler: 1.100

    motor0:
      limit_neg_pin: gpio.34
      hard_limits: false
      pulloff_mm: 1.000
      stepstick:
        step_pin: I2SO.3
        direction_pin: I2SO.4

i2so:
  bck_pin: gpio.16
  data_pin: gpio.21
  ws_pin: gpio.17

spi:
  miso_pin: gpio.12
  mosi_pin: gpio.13
  sck_pin: gpio.14

sdcard:
  cs_pin: gpio.15
  card_detect_pin: NO_PIN

control:
  safety_door_pin: NO_PIN
  reset_pin: NO_PIN
  feed_hold_pin: NO_PIN
  cycle_start_pin: NO_PIN
  macro0_pin: gpio.33:low:pu
  macro1_pin: NO_PIN
  macro2_pin: NO_PIN
  macro3_pin: NO_PIN

macros:
  startup_line0:
  startup_line1:
  macro0: $SD/Run=lasertest.gcode
  macro1: $SD/Run=home.gcode
  macro2:
  macro3:

coolant:
  flood_pin: NO_PIN
  mist_pin: NO_PIN
  delay_ms: 0

probe:
  pin: gpio.22
  check_mode_start: true

PWM:
  pwm_hz: 5000
  output_pin: gpio.32
  direction_pin: NO_PIN
  disable_with_s0: false
  s0_with_disable: true
  spinup_ms: 1000
  spindown_ms: 1000
  tool_num: 0
  speed_map: 0=0.000% 12000=100.000%

start:
  must_home: false

# 5,18,19,22,23,25,26,27,32,33,39,I2SO.7
# SDA 0 / SCL 4
```

### Startup Messages

```text
[MSG:INFO: FluidNC 3.7.14-webui3-test https://github.com/bdring/FluidNC]]
[MSG:INFO: Compiled with ESP32 SDK:v4.4.4]]
[MSG:INFO: Local filesystem type is spiffs]]
[MSG:INFO: Configuration file:MKS_DLC32_21_XYZ_MOD.yaml]]
[MSG:INFO: Machine 3018 Pro Max Spindle]]
[MSG:INFO: Board MKS-DLC32 V2.1]]
[MSG:INFO: I2SO BCK:gpio.16 WS:gpio.17 DATA:gpio.21]]
[MSG:INFO: SPI SCK:gpio.14 MOSI:gpio.13 MISO:gpio.12]]
[MSG:INFO: SD Card cs_pin:gpio.15 detect:NO_PIN freq:8000000]]
[MSG:INFO: Stepping:I2S_static Pulse:4us Dsbl Delay:0us Dir Delay:1us Idle Delay:0ms]]
[MSG:INFO: Axis count 3]]
[MSG:INFO: Shared stepper disable I2SO.0]]
[MSG:INFO: Axis X (0.000,325.000)]]
[MSG:INFO:   Motor0]]
[MSG:INFO:     stepstick Step:I2SO.1 Dir:I2SO.2 Disable:NO_PIN]]
[MSG:INFO:  X Neg Limit gpio.36]]
[MSG:INFO: Axis Y (0.000,220.000)]]
[MSG:INFO:   Motor0]]
[MSG:INFO:     stepstick Step:I2SO.5 Dir:I2SO.6:low Disable:NO_PIN]]
[MSG:INFO:  Y Neg Limit gpio.35]]
[MSG:INFO: Axis Z (0.000,80.000)]]
[MSG:INFO:   Motor0]]
[MSG:INFO:     stepstick Step:I2SO.3 Dir:I2SO.4 Disable:NO_PIN]]
[MSG:INFO:  Z Neg Limit gpio.34]]
[MSG:INFO: macro0_pin gpio.33:low:pu]]
[MSG:INFO: Kinematic system: Cartesian]]
[MSG:INFO: PWM Spindle Ena:NO_PIN Out:gpio.32 Dir:NO_PIN Freq:5000Hz Period:8191]]
[MSG:INFO: Using spindle PWM]]
[MSG:INFO: Probe Pin: gpio.22]]
[MSG:INFO: Connecting to STA SSID:Inbar]]
[MSG:INFO: Connecting.]]
[MSG:INFO: Connecting..]]
[MSG:INFO: Connecting...]]
[MSG:INFO: Connecting....]]
[MSG:INFO: Connecting.]]
[MSG:INFO: Connecting..]]
[MSG:INFO: Connecting...]]
[MSG:INFO: Connecting....]]
[MSG:INFO: Connecting.]]
[MSG:INFO: Connected - IP is 192.168.1.161]]
[MSG:INFO: WiFi on]]
[MSG:INFO: Start mDNS with hostname:http://fluidnc.local/]]
[MSG:INFO: SSDP Started]]
[MSG:INFO: HTTP started on port 80]]
[MSG:INFO: Telnet started on port 23]]
ok
<Idle|MPos:110.003,110.003,0.000|FS:0,0|Pn:PXYZ>
```

### User Interface Software

WebUI

### What happened?

First time setup a CNC Machine with a 3rd party board, I install the firmware as insructed and upload the configuration file to the machine via the terminal. Everything works apart from Z axis. It should configured correctly based on the diagram on FluidNC site, and the connection to the board are at the right port. When I command the Z axis to move up or down, the command is sent and nothing happaned on the machine side.

bdring, 2024-02-17:
Can you lock and unlock the x and y motors with $MD and $ME?

Does z ever lock?

bdring, 2024-02-17:
The motors share a disable pin. If the Z does not lock and unlock like Xand Y, you probably have a hardware or wiring problem.

tomerinbar1, 2024-02-17:
I'm able to lock and unlock both X and Y axis. I can't verify the Z axis as it doesn't have a manual control. The Z axis used to work just fine with a board that came with the machine.

bdring, 2024-02-17:
$MD and $ME should lock and unlock all motors.

tomerinbar1, 2024-02-17:
> $MD and $ME should lock and unlock all motors.

I know it should, but I cannot check if the Z axis is locked or not because I don't have a manual selector like the other axis

bdring, 2024-02-18:
Please see the instructions here on how to verify if the motor is locked. No "manual selector" is needed.

http://wiki.fluidnc.com/en/support/faq#motors-do-not-move

MitchBradley, 2024-02-18:
Basically what you want to do to test for motor lock is try to turn the z axis leadscrew with your fingers.  It might be a tight fit to reach in between the supports.  If you can't get your fingers in there, you could try wrapping a small rubber belt around the screw and trying to turn the shaft by pulling on the ends of the belt.  Or a piece of insulated wire, or anything flexible and grabby.

The next test is to swap the connectors so the Z motor is plugged into the Y output, and the Y motor is plugged into the Z output.  If that makes the Z motor start working (when you ask for a Y move) and the Y motor no longer works (when you ask for a Z move), then the problem could be with the stepstick driver modules.  But if the Z motor still won't move when plugged into the Y socket, the wiring has probably failed.

You can also swap the stepstick modules to see if the problem follows the module. If the problem follows the stepstick module, the driver chip could be burned out, or perhaps the driver current is set too low.  You should always check that setting to ensure that it is suitable for your motors.  The internet has many resources explaining how to do that.

This is all basic CNC debugging that is not specific to FluidNC.

tomerinbar1, 2024-02-18:
Test no 1: With the command $ME all motors locked include the Z motor.
Test no 2: Swap between Z and Y motor connectors: Z-Axis is responding to Y commands and Y-Axis isn't responsible to Z command. Wiring is OK
Test no 3: Put back the motor connectors and swap the driver modules: Y Axis responding to commands and Z-Axis isn't responsive. Driver modules OK.

So it is something with the configuration or the firmware. 
Please let me know you thoughts.

bdring, 2024-02-18:
Thank you. That is the type of information I needed.

I can't think of a reason why it is not working. The config file cannot change the lock/unlock for just the Z motor. As you can see on the schematic. The same signal controls all lock/unlock signals. something is wrong with the hardware for that socket. Thousands of people are probably using FluidNC with that controller.

[image]

What type of motor drivers are being used?
Are the jumpers setup the same for all motors?

tomerinbar1, 2024-02-18:
I'm using TMC2209, all the jumpers are the same.
I hope I didn't short anything while try to flash it.
I try to get it into bootloader first by jump between GPIO0 and GRD, and later tried to jumper the RST based on information I found online. 
I'll try to operate it with GBRL via USB and let's see if anything change.

tomerinbar1, 2024-02-18:
I tried again to connect the Y motor to the Z port and command Z Axis to move and it also just clicking.
Now I sure that the Z Axis port has something, maybe it defective.

MitchBradley, 2024-02-18:
Try reducing the max rate and acceleration.  Those values are very high, especially for a Z axis that has to work against gravity.  I have a lightweight laser gantry (with no Z axis) that can run that fast, but none of my spindle-based machines can run anywhere near that fast.  Clicking often means that the motor is stalling because it does not have enough torque to run at the rate you are requesting.

tomerinbar1, 2024-02-18:
Its working!!! experience is something you cannot buy :-)

closed 2024-02-18 by tomerinbar1

bdring, 2024-02-18:
Please share what you think fixed the problem

tomerinbar1, 2024-02-18:
As @MitchBradley said, the rate and acceleration was too high for this poor stepper motor, reducing the speeds makes the stepper motor great again

bdring, 2024-02-18:
...must have been some serious clicking

MitchBradley, 2024-02-18:
When cutting, the loads on the machine are higher, and as the machine ages, the screws might not work as smoothly, or might get dirty, or need lubrication.  Any of those factors can reduce the maximum speed.  Do not make the mistake of setting the machine to go as fast as possible right now.  Find the max speed, while adding some extra load, then back off from that to give yourself some margin.

tomerinbar1, 2024-02-18:
Thank you for your tips!
The machine is brand new and make only a simple text with 30 degrees bit just for testing.
I'm learning step by step the CNC world, I'm sure it will be with amazing results.

Thank you for your help