bdring/FluidNC #1136: Could not download the release asset TypeError: Failed to fetch

zhuyuwu110, 2024-02-18:
### Wiki Search Terms

-------

### Controller Board

6-pack

### Machine Description

----------

### Configuration file

```yaml
---------
```

### Startup Messages

```text
---------
```

### What happened?

Could not download the release asset TypeError: Failed to fetch
[image]

breiler, 2024-02-18:
I just tried it, it worked fine for me.

Can you right click somewhere on the page and select "Inspect". Then click on the network tab and console and try again. Then take a screenshot if it fails. It should look something like this:

[image]

zhuyuwu110, 2024-02-18:
[image]
[image]

closed 2024-02-18 by zhuyuwu110

breiler, 2024-02-18:
You closed the issue, does this mean that you managed to resolve it?

I am interested in the full URL here, can you right click and choose copy link?
[image]