bdring/FluidNC #1137: Problem: $ESP400 behavior

ColPaulR, 2024-02-18:
### Wiki Search Terms

ESP400 EEPROM

### Controller Board

MKS Tinybee

### Machine Description

4-axis external drivers gantry CNC with laser

### Configuration file

```yaml
board: MKS TinyBee V1.0_001
name: CNC Router XYYZA 0-10V spindle and laser
meta: (01.02.2022) by Skorpi

kinematics:
  Cartesian:

stepping:
  engine: I2S_STATIC
  idle_ms: 255
  pulse_us: 4
  dir_delay_us: 1
  disable_delay_us: 0

axes:
  x:
    # X
    steps_per_mm: 250
    max_rate_mm_per_min: 10000.000
    acceleration_mm_per_sec2: 200.000
    max_travel_mm: 1300.000
    soft_limits: true
    homing:
      cycle: 2
      positive_direction: true
      mpos_mm: 1265.00
      feed_mm_per_min: 300.000
      seek_mm_per_min: 2000.000
      settle_ms: 250
      seek_scaler: 1.100
      feed_scaler: 1.100

    motor0:
      limit_neg_pin: GPIO.33:low
      hard_limits: true
      pulloff_mm: 2.500
      standard_stepper:
        step_pin: I2SO.1
        direction_pin: I2SO.2:low
        disable_pin: I2SO.0

  y:
    # Y
    steps_per_mm: 250
    max_rate_mm_per_min: 10000.000
    acceleration_mm_per_sec2: 200.000
    max_travel_mm: 1300.000
    soft_limits: true
    homing:
      cycle: 2
      positive_direction: false
      mpos_mm: 0.000  
      feed_mm_per_min: 300.000
      seek_mm_per_min: 2000.000
      settle_ms: 250
[... 507 lines cut ...]
[MSG:JSON:R) \\H]","S":"40","M":"0"},{"F":"nvs","P":"Firmware/Build","H":"Firmware/Build","T":"S","V":"","S":"]
[MSG:ERR: Device not responding]
[MSG:JSON:20","M":"0"},{"F":"nvs","P":"SD/FallbackCS","H":"SD/FallbackCS","T":"I","V":"-1","S":"40","M":"-1"},]
[MSG:ERR: Device not responding]
[MSG:JSON:{"F":"nvs","P":"Report/Status","H":"Report/Status","T":"I","V":"1","S":"3","M":"0"},{"F":"nvs","P":"]
[MSG:ERR: Device not responding]
[MSG:JSON:Config/Filename","H":"Config/Filename","T":"S","V":"XYZAB_laser_5Jan24v2.yaml","S":"50","M":"1"},{"F]
[MSG:ERR: Device not responding]
[MSG:JSON:":"nvs","P":"Message/Level","H":"Message/Level","T":"B","V":"3","O":[{"Debug":"4"},{"Error":"1"},{"I]
[MSG:ERR: Device not responding]
[MSG:JSON:nfo":"3"},{"None":"0"},{"Verbose":"5"},{"Warning":"2"}]},{"F":"nvs","P":"WiFi/Mode","H":"WiFi/Mode",]
[MSG:ERR: Device not responding]
[MSG:JSON:"T":"B","V":"3","O":[{"AP":"2"},{"Off":"0"},{"STA":"1"},{"STA>AP":"3"}]},{"F":"nvs","P":"Sta/SSID","]
[MSG:ERR: Device not responding]
[MSG:JSON:H":"Sta/SSID","T":"S","V":"RounsavallWPA2","S":"32","M":"0"},{"F":"nvs","P":"Sta/Password","H":"Sta/]
[MSG:ERR: Device not responding]
[MSG:JSON:Password","T":"S","V":"********","S":"64","M":"8"},{"F":"nvs","P":"Sta/MinSecurity","H":"Sta/MinSecu]
[MSG:ERR: Device not responding]
[MSG:JSON:rity","T":"B","V":"3","O":[{"OPEN":"0"},{"WEP":"1"},{"WPA-PSK":"2"},{"WPA-WPA2-PSK":"4"},{"WPA2-ENTE]
[MSG:ERR: Device not responding]
[MSG:JSON:RPRISE":"5"},{"WPA2-PSK":"3"}]},{"F":"nvs","P":"WiFi/FastScan","H":"WiFi/FastScan","T":"B","V":"0","]
[MSG:ERR: Device not responding]
[MSG:JSON:O":[{"OFF":"0"},{"ON":"1"}]},{"F":"nvs","P":"Sta/IPMode","H":"Sta/IPMode","T":"B","V":"0","O":[{"DHC]
[MSG:ERR: Device not responding]
```

bdring, 2024-02-18:
What do the commander people suggest?

ColPaulR, 2024-02-18:
The standard answer from Commander was this is not a Commander problem and that FluidNC is not supported. They did create a repo for prior builds.  I rolled back to Commander Build 300 (from 301). Same response so I’m not sure how I had this working perviously. I’m focused primarily on what $ESP400 should return and if I have somethjng corrupted, as well as why my development TinyBee board (vs one in the CNC) is returning the error message to the $ESP400 command.

ColPaulR, 2024-02-18:
Commander's response is a combination of Facebook and their forum. The forum response is the most complete at [https://www.cnc3d.com.au/forum/cnc3d-commander-users/roll-back-to-build-300](https://www.cnc3d.com.au/forum/cnc3d-commander-users/roll-back-to-build-300). There was a comment in the response relative to expected $ESP400 behavior, specifically:

> Looking at the data you have posted here, there is a LOT of additional data that [ESP400] is returning compared to a standard GRBL ESP32 controller. A lot of this is not needed by Commander but, the controller itself is returning all of it when a standard [ESP400] command is sent.
>
>This is one of the reasons why we do not support FluidNC, there is simply too many changes to existing established commands, this is not good practice from a development standpoint.
>
>It would have been wiser keep the existing [ESP400] command the same output as it always has been since production releases were made and if needed, created a [ESP401] command that did a more advanced EEPROM dump.
>
>This of course is in the hands of the FluidNC development team.
>
>I hope this helps.
>
>After this post was made, we have setup a repo of previous Commander builds here. You are welcome to try them to compare your results:

MitchBradley, 2024-02-18:
The guy who wrote Commander has a narrow view of how programming interfaces work.  He tends to look at specific responses from specific instances of software and assume "what I see now is all there can ever be".  I have tried to work with him to get it right but we were unable to see eye to eye.  Commander has some nice features but we don't have time to deal with the limitations of every one of the dozens of different senders that are out there.  In some cases we are able to work with the authors of senders to resolve problems, but we were not able to do that with Commander, despite some attempts.

The response to ESP400 is in a format that ESP3D-WebUI will accept and render into a settings web page.  The "spec" is what WebUI expects and accepts, not the specific list of things that any particular instance of ESP3D firmware generates on a particular machine.  The WebUI spec allows a broad and open-ended list of configuration choices, whereas the "closed-ended list of stuff I know about" approach is guaranteed to break every time a new configurable item has to be invented.

ColPaulR, 2024-02-18:
Understood. Any idea on why the “ [MSG:ERR: Device not responding]” messages are happening on my development board?

MitchBradley, 2024-02-18:
Yeah, I think I know what is happening there.  It will take me a few to dig into the details.

ColPaulR, 2024-02-18:
Copy.  I’m in no rush. Let me know if I can help

MitchBradley, 2024-02-18:
Okay, I see the problem.  We are trying to handle four different situations here - interactive inspection, WebUI2, WebUI3, and the new pendant stuff.  Historically there were just the first two.  But then WebuI3 came along and changed the format of some ESPxxx things.  And the new pendant stuff needs to have encapsulation so it can direct messages to the right listener, plus flow control so  it doesn't lose messages.  The attempt to satisfy all those use cases resulted in a problem when you use ESP400 from serial or FluidTerm - which was historically only for debugging, although perhaps Commander depends on that.

I am trying to satisfy too many people going off in different directions at the moment.

ColPaulR, 2024-02-18:
Tracking.  Is there a version I should revert to in order to avoid the issue until it gets resolved?

MitchBradley, 2024-02-18:
You could try going back one by one.  I don't have time to give you a better answer.  I am going crazy.

ColPaulR, 2024-02-19:
No worries. Sorry to add to your workload. v.3.7.12 responded without the error message. I'll stick with that on my development board until the error messages get sorted out,

PR #1147 (2024-02-28): Only encapsulate JSON for special pendant functions
This fixes #1137 without breaking FluidDial
closed 2024-02-28 by MitchBradley