bdring/FluidNC #1138: Problem: SSID that includes a space get cut of from webUI

woutervddn, 2024-02-19:
### Wiki Search Terms

SSID space

### Controller Board

Barebones NodeMCU-ESP32S (Got one of the old CNC Dev Controller V3.5 boards to put it on, but that is irrelevant at this stage)

### Machine Description

Non yet, still doing initial setup

### Input Circuits

```yaml
/
```

### Configuration file

```yaml
/
```

### Startup Messages

```text
/
```

### User Interface Software

WebUI3

### What happened?

I entered my SSID that includes a space in the STA SSID config field. 
Lets say the SSID name is `Flying Pinguin`.

When I set it in the UI and reload it, it shows up as `Flying`.
Placing the SSID name between `"` (so typing `"Flying Pinguin"`) also didn't fix it.
It showed up when reloading the UI as `"Flying`

Going to the Terminal, I can see that it tries to connect to either `Flying` or `"Flying`
Entering the SSID in the Terminal using the `$Sta/SSID=Flying Pinguin` command did solve the issue.

### GCode File

/

### Other Information

/

bdring, 2024-02-19:
See if this helps your problem.

http://wiki.fluidnc.com/en/support/faq#international-characters-in-wifi-ssids

boboxx, 2024-02-19:
replace the space by encoder character: &nbsp;
Flying&nbsp;Pinguin

Any special carather should be encoded: https://mateam.net/html-escape-characters/

woutervddn, 2024-02-19:
> See if this helps your problem.
> 
> http://wiki.fluidnc.com/en/support/faq#international-characters-in-wifi-ssids

A space is a US ASCII character, the docs don't mention anything about spaces. So according to the docs it should just work. That being said, I can confirm that entering `%20` instead of a space works in the UI.

I honestly do think that this should be autotranslated by wrapping the inputfield value in encodeURI() before sending it off the server. If not, then the API docs should need an explicit mention for it.

> Replace the space by encoder character:   Flying Pinguin
> 
> Any special carather should be encoded: https://mateam.net/html-escape-characters/

This does not work. It shows the html escape characters (which translates space to `&#32;` instead of the URL encoded version `%20` that is needed for this to work.

MitchBradley, 2024-02-19:
Feel free to offer patches to fix the problem according to how you think it should work.  FluidNC is developed and maintained by volunteers and there is more to do than we have have time for, by a large margin.   This page http://wiki.fluidnc.com/en/wiki_contributions tells you can contribute to the documentation.

bdring, 2024-02-22:
Are you on any other Discord servers? I don't think I can help with initial discord approval.

Note: We are moving to a new version of the WebUI and actively discussing it on Discord.

bdring, 2024-02-22:
Wiki user approved

woutervddn, 2024-02-23:
I made the change on: http://wiki.fluidnc.com/en/support/faq

I also referenced the space character on: http://wiki.fluidnc.com/en/features/wifi_bt

closed 2024-02-23 by woutervddn