bdring/FluidNC #1140: Feature: Hardware development

amirslz74, 2024-02-20:
### Machine Context

I'm always thinking about hardware solutions for CNC machines that can adapt to different setups. My goal is to implement reliable circuitry to protect against interference in noisy environments. I've noticed some complex PCB designs in the community, but there's a need for improvements, especially in terms of sturdy connectors. I'm passionate about contributing to this field and have started developing boards suited for CNC applications.

### Feature Description

I believe there are several aspects to consider adding to the FluidNC project to enhance its versatility. Firstly, I see potential for integration with 3D printers and robotic arms, requiring adjustments to both hardware components and software documentation. Additionally, I view hardware enhancements as crucial for optimizing CNC machine functionality. By implementing a great hardware foundation and refining the puzzle-based design, we can introduce a host of new features to enhance the project's capabilities.

### Other Approaches

I've encountered compatibility issues with various Arduino modules not being supported by FluidNC. Therefore, I believe a more effective approach is to integrate essential Arduino modules directly into FluidNC such as : LCDs, Temp, gyroscope , enhancing CNC control capabilities. Additionally, developing a wireless controller with ESP32 compatibility and a high-quality LCD interface could greatly improve the management of large CNC machines which is very important for them. 

### How I Can Help

I'm willing to assist the hardware team or make a hardware fields and manage it to improving board designs and conducting thorough testing of new features to ensure seamless integration with FluidNC because I think the software have many potentials which are hardwares can't support them. My goal is to enhance the user experience for all FluidNC users.

MitchBradley, 2024-02-20:
Adding features like you suggest would increase the RAM footprint which is already a problem. Expanding into the robotic arm and 3D printing spaces would greatly increase the support and testing burden which is almost at crisis level already.
I am all for hardware reliability but most people don’t want to pay for it. Instead they buy the cheapest chinese controller available - typically MKS DLC32 at <$20 - then ecpect the FluidNC developers to teach them CNC, debug their hardware, and provide a config file for poorly documented hardware and use cases. The few who then donate to the cause typically contribute an amount that would pay for at most 10 minutes of programmer time.

The hardware that Bart designs is already as reliable as he can make it at a price point that some people will buy. It is open source so anyone woo wants to use better connectors can swap.

With any hardware endeavor, you face the difficult problems of parts procurement, stocking, and shipping logistics. If you have a way to make demonstrably more reliable hardware that appeals to a lot of people at a price they will pay, and can manage the ongoing logistics, while providing support for your customers (as opposed to them just asking us to support it for free which is what usually happens), then go wild.

amirslz74, 2024-02-22:
Thanks for your response,

I am not aware of the exact RAM limitations, as the ESP32 comes in various types, some of which offer more RAM. We could consider utilizing ESP32 models with higher RAM capacity for boards that require it.

Your points are valid. However, I am migrating to another country soon, which may impact my ability to produce and market these boards. When I can do that I communicate with you.

I have another question, can I add my boards to the hardware documentation like others? (It seems straightforward today, but I add some modifications.)

MitchBradley, 2024-02-22:
Please do not speculate; that wastes our time.  Do your homework. ESP32 has 520K of RAM; no variant has more.  ESP32-S3 is the closest with 512K, and has the additional advantage that the portion of RAM that is on the instruction bus can be added to the heap if it is not otherwise used.  It is possible to add external RAM as PSRAM but that uses up precious GPIOs.

We sort of support ESP32-S3 in that some people are using it successfully, but it is not in the official release stream yet.  The primary developers are overwhelmed with other work.

You are welcome to add your boards to the list; just request wiki edit access per the instructions in the wiki.

closed 2024-02-23 by amirslz74