bdring/FluidNC #1141: m6 script

megahacker91, 2024-02-20:
### Machine Context

it will be great to add ability to run script when m6 command receved

### Feature Description

when work with cnc it is great to have auto tool change
so when m6 command receved it will run script called m6Macro and we can select this macro from config file

### Other Approaches

i do it with the last version of fluidnc 
because i am using universal gcode sender and it dos not have m6 implement 
and it was very easy to do it 

### How I Can Help

i can write this code but i have to implement it in every new version

MitchBradley, 2024-02-20:
> i can write this code but i have to implement it in every new version

That is what a Pull Request is for.

bdring, 2024-02-21:
FluidNC has some pretty robust tool change capability. It  relies on custom spindle classes rather than macros. The problem with macros is that they are not interactive. You cannot do something different based on the tool number. It is also difficult to do something like move to a specific location and pause for a tool change. There is no way to return to the old position unless you play games with G28 or G30. It can be dangerous to have constantly changing values for those.

We are currently working on implementing a better base class for tool changing to make it easier to create custom classes. You can follow it on Discord.

https://discord.com/channels/780079161460916227/1206568071196713023

After that is done we could look at a simple macro. We would want to make sure that it plays well with any class based stuff.

Here is a video of an ATC. The video was done using Grbl_ESP32, but the code was ported FluidNC.  

https://www.youtube.com/watch?v=3ikLcC5NidU

bdring, 2024-12-11:
Closing old issue M6 macros are now supported.

http://wiki.fluidnc.com/en/features/atc

closed 2024-12-11 by bdring