bdring/FluidNC #1153: Backlash compensation

jeedee23, 2024-03-04:
### Machine Context

Integrating Fluidcnc into an old lathe, 5 axis CNC and a laser. The DRO gives the real position of the toolhead, since everything is measured with rotating or linear encoders. 
I see the mentioning of "install backlash compensation here" in the code. 

### Feature Description

I was wandering if e.g. a "backlash test.gcode" could then run and store the compensation values for each axis for the time the user whishes, so whenever he runs his testgcode again, those values are adapted.

### Other Approaches

The data should come via serial or a fixed protocol and saved on the main ESP32

### How I Can Help

 I could incorporate it myself, but it will take time and if there is already someone doing something similar. This would save time

bdring, 2025-01-21:
This is one our features and issues roadmap, but unlikely to see any work for a while.

kutukvpavel, 2025-02-22:
@jeedee23 Hi! I am also retrofitting an old lathe, so I'm facing the same issue. Did you manage to implement backlash compensation for fluidnc?

jeedee23, 2025-02-23:
Hi,

I developed custom code and built a closed-loop system using linear
encoders from affordable Chinese calipers, with the hyperfast ESP32
industrial version (one per spindle), and stepper motors on all spindles.
This setup achieves an accuracy of 0.0005"—about half a thousandth of an
inch (approximately 1.5 µ)—with zero shaft play and perfect centering.
Spindle backlash is corrected on the fly, ensuring smooth and stable
operation.

Met vriendelijke groeten, meileures salutations,mit freundlichen Grüßen,Best
regards, ودي الخاص,  此致

*Johan Degraeve*

Admin data:

Company:-----------------------*AAVDS BV*
Address:-----------------------Veenstraat 28 | 8470 Gistel | Belgium
Phone (main+mob)---------------+32489947935
BTW/TVA/VAT/MSWT: ------------ BE0841892791
RC:----------------------------2.205.300.245
IBAN/Swift---------------------BE41 7380 3449 2010 / KREDBEBB