bdring/FluidNC #1154: Add dual end stop support

alphaonex86, 2024-03-08:
### Machine Context

I am builing machine 3 AXIS with dual end stop for each AXIS

### Feature Description

I wish support min end stop and max end stop, if min trigger do x++ to be safe, if max trigger do x-- to be safe.

### Other Approaches

No other way

### How I Can Help

I can reply to ask here

bdring, 2024-03-08:
Do what when?

If you hit a limit switch and have hard limits enabled the machine does a hard stop. This means the step pulses are stopped immediately without any deceleration. You have likely lost position. You should rehome the machine. If a limit switch is active before homing FluidNC will know how to safely back off the switch before homing.

alphaonex86, 2024-03-08:
Example: Machine start, Limit for X is active, how know if need do X- or X+? How can know how homing?

If homing doing X- util limit enabled:

- it's well if min X limit is reach, because will start doing directly X+
- if max X limit it reached will broke the machine, because will start doing directly X+

Then with 1 limit entry, need manual homing at machine startup.

bdring, 2024-03-08:
Show your config file. I do not know how your limit switches are configured.

alphaonex86, 2024-03-08:
For now I'm playing with https://github.com/bdring/fluidnc-config-files/blob/main/official/3axis_v4.yaml

And I only see limit_all_pin, I don't see any limit_min_pin/limit_max_pin way

MitchBradley, 2024-03-08:
If you have separate limit switches on both ends, remove the limit_all_pin entry and add limit_neg_pin and limit_pos_pin entries.

The config file that you are using is just an example.  If it does not match your hardware, you have to make a config file that does match.

bdring, 2024-03-08:
Send ***$axes/x*** to see your config for the x axis. It will show you the full config with defaults for anything your file is missing.

Also read the wiki.

http://wiki.fluidnc.com/en/config/axes#motor-settings

MitchBradley, 2024-03-08:
Here is the documentation for limit switches http://wiki.fluidnc.com/en/config/homing_and_limit_switches

alphaonex86, 2024-03-08:
Thanks, my misktake then, sorry

closed 2024-03-08 by alphaonex86