bdring/FluidNC #1156: Feature: I also design a board for fluidnc, but I don't know if someone like this designed style?

abcpibot, 2024-03-09:
### Machine Context

fluidnc board designed

### Feature Description

fluidnc board designed

### Other Approaches

I also design a board for fluidnc, but I don't know if someone like this designed style? https://www.pibot.com/image/catalog/blog/rev46-1/_20230219022906.jpg
and I tested it in modbus 485 function, https://www.pibot.com/blog/rev46-fluidnc

### How I Can Help

Can you add to this list: http://wiki.fluidnc.com/en/hardware/existing_hardware

bdring, 2024-03-09:
Feel free to add it yourself.

http://wiki.fluidnc.com/en/wiki_contributions

abcpibot, 2024-03-10:
thanks， but I can not get into to discord， can you help check？maybe the invite code expired？ thanks again

closed 2024-04-25 by bdring