bdring/FluidNC #1157: Problem: Cannot configure an ssid which contains an exclamation mark

crasu, 2024-03-09:
### Wiki Search Terms

Exclamation mark $! ! 

### Controller Board

6-pack

### Machine Description

MPCNC

### Configuration file

```yaml
-
```

### Startup Messages

```text
[MSG:INFO: Connecting....]
[MSG:INFO: Connecting.]
[MSG:INFO: Connecting..]
[MSG:INFO: Connecting...]
[MSG:INFO: Connecting....]
[MSG:INFO: Connecting.]
[MSG:INFO: Connecting..]
```

### User Interface Software

Serial

### What happened?

When entering the ssid the serial connection dies and the terminal stops responding.

MitchBradley, 2024-03-09:
The answer is in the Frequently Asked Questions page in the wiki

closed 2024-03-09 by MitchBradley

crasu, 2024-03-09:
Thank you in case someone new ends up here as well. ! enables feedhold. You need to escape with %21.

http://wiki.fluidnc.com/en/support/faq#special-character-issue