bdring/FluidNC #1158: Machine is not homing X

alphaonex86, 2024-03-10:
### Wiki Search Terms

Machine homing

### Controller Board

ESP32 DEVKIT V1

### Machine Description

just ESP32 DEVKIT V1, 1 AXIS, 2 optical end stop, stepper from DVD

### Configuration file

```yaml
name: Test X
board: ESP32 DEVKIT V1
meta: cfg
start:
  must_home: false
  check_limits: true
  deactivate_parking: true
axes:
  x:
    steps_per_mm: 80
    max_rate_mm_per_min: 1000
    acceleration_mm_per_sec2: 25
    max_travel_mm: 1000
    homing:
      cycle: -1
    motor0:
      limit_all_pin: NO_PIN
      limit_neg_pin: gpio.26:pu
      limit_pos_pin: gpio.27:pu
      hard_limits: true
      pulloff_mm: 1
      standard_stepper:
        step_pin: gpio.32
        direction_pin: gpio.33
        disable_pin: NO_PIN
stepping:
  engine: RMT
  idle_ms: 250
  pulse_us: 4
  dir_delay_us: 0
  disable_delay_us: 0
```

### Startup Messages

```text
<Idle|MPos:0.000,0.000,0.000|FS:0,0|WCO:0.000,0.000,0.000>
```

### What happened?

Do nothing wuth $HX, no move

### GCode File

$Message/Level=Debug
$HX
ok

### Other Information

I can without problem move, stop correctly when reach end stop

MitchBradley, 2024-03-10:
[image]

alphaonex86, 2024-03-10:
Then maybe return error and not OK, especially in debug mode will help

MitchBradley, 2024-03-10:
It is not an error. -1 has a specific and useful meaning.

closed 2024-03-10 by MitchBradley

alphaonex86, 2024-03-10:
Maybe inform $Message/Level=Debug will be useful

MitchBradley, 2024-03-10:
We can't have information messages about every possible misunderstanding; there are almost an infinite number of them.  If you wish to add some to a private fork, feel free to do so.  You could even publish that fork and support it, collecting donations that compensate you at the rate of about $0.10 / hour for the support time.

alphaonex86, 2024-03-10:
In web interface, create file with UI, Axes, X, Homing, I'm unable to enter Position configuration, all my input are writen into Cycle option.
If I keep position to 0, I have infinity loop:
[MSG:DBG:  X Pos Limit 0]
[MSG:DBG:  X Pos Limit 1]

My actual configuration due to this is:
    homing:
      cycle: 2
      positive_direction: false

I try do 2 homing cycle and stop at 1mm in direction of X-

MitchBradley, 2024-03-10:
Start a new issue and provide your latest complete config file.  I do not understand the details of the file from the description above.

MitchBradley, 2024-03-10:
Try increasing pulloff_mm to 3.  Sometimes endstops, especially optical ones, need large pulloff values.  I do not think that the mpos_mm value is likely to be causing the problem. 

Until we can get a fix for WebInstaller, edit the config file with a plain text editor.

alphaonex86, 2024-03-11:
Right, with plain editor work, the firmware into ESP32 should not default to value >0 like 1.0mm, and if value <=0.0 then set 1.0?