bdring/FluidNC #1163: Problem: Failed to communicate with FW!

Mr-Quignon, 2024-03-14:
### Wiki Search Terms

Failed to communicate with FW!, WEBUI,

### Controller Board

ROOT CONTROLLER ISO REV 3

### Machine Description

No machine wired yet

### Configuration file

```yaml
board: Root Controller ISO
name: Root 4 V20
stepping:
  engine: I2S_STREAM
  idle_ms: 255
  pulse_us: 4
  dir_delay_us: 1
  disable_delay_us: 3

axes:
  shared_stepper_disable_pin: NO_PIN
  x:
    steps_per_mm: 160.000
    max_rate_mm_per_min: 3700.000
    acceleration_mm_per_sec2: 250.000
    max_travel_mm: 1198.000
    soft_limits: true
    homing:
      cycle: 2
      positive_direction: true
      mpos_mm: 1197.000
      feed_mm_per_min: 100.000
      seek_mm_per_min: 2000.000
      settle_ms: 1000
      seek_scaler: 10.00
      feed_scaler: 10.00

    motor0:
      limit_neg_pin: NO_PIN
      limit_pos_pin: NO_PIN
      limit_all_pin: gpio.26:low
      hard_limits: true
      pulloff_mm:2.000
      standard_stepper:
        step_pin: I2SO.2
        direction_pin: I2SO.1:high 
        disable_pin: I2SO.0:low

  y:
    steps_per_mm: 160.000
    max_rate_mm_per_min: 3700.000
    acceleration_mm_per_sec2: 250.000
    max_travel_mm: 1185.000
    soft_limits: true
    homing:
      cycle: 3
      positive_direction: true
      mpos_mm: 1185.000
      feed_mm_per_min: 100.000
      seek_mm_per_min: 2000.000
      settle_ms: 1000
      seek_scaler: 1.100
      feed_scaler: 5.500

    motor0:
      limit_neg_pin: NO_PIN
      limit_pos_pin: gpio.32:low
      limit_all_pin: NO_PIN
      hard_limits: true
      pulloff_mm:1.000
      standard_stepper:
        step_pin: I2SO.6
        direction_pin: I2SO.5:low
        disable_pin: I2SO.4:low
        
    motor1:
      limit_neg_pin: NO_PIN
      limit_pos_pin: gpio.35:low
      limit_all_pin: NO_PIN
      hard_limits: true
      pulloff_mm: 3.120
      standard_stepper:
        step_pin: I2SO.22
        direction_pin: I2SO.21:low
        disable_pin: I2SO.20:low
  z:
    steps_per_mm: 160.000
    max_rate_mm_per_min: 4500.000
    acceleration_mm_per_sec2: 500.000
    max_travel_mm: 159.000
    soft_limits: true
    homing:
      cycle: 1
      positive_direction: true
      mpos_mm: 0.500
      feed_mm_per_min: 100.000
      seek_mm_per_min: 1000.000
      settle_ms: 500
      seek_scaler: 1.100
      feed_scaler: 1.500

    motor0:
      limit_neg_pin: NO_PIN
      limit_pos_pin: gpio.15:low
      limit_all_pin: NO_PIN
      hard_limits: true
      pulloff_mm:1.000
      standard_stepper:
        step_pin: I2SO.10
        direction_pin: I2SO.9:high 
        disable_pin: I2SO.8:low
        

i2so:
  bck_pin: gpio.22
  data_pin: gpio.12
  ws_pin: gpio.21

spi:
  miso_pin: gpio.19
  mosi_pin: gpio.23
  sck_pin: gpio.18

sdcard:
  card_detect_pin: NO_PIN
  cs_pin: gpio.5

control:
  safety_door_pin: NO_PIN
  reset_pin: NO_PIN
  feed_hold_pin: gpio.34
  cycle_start_pin: gpio.36
  macro0_pin: gpio.39
  macro1_pin: NO_PIN
  macro2_pin: NO_PIN
  macro3_pin: NO_PIN

coolant:
  flood_pin: gpio.33
  mist_pin: gpio.13
  delay_ms: 0

probe:
  pin: gpio.27
  check_mode_start: true

macros:
  startup_line0:
  startup_line1:
  macro0: 
  macro1: 
  macro2: 
  macro3:
  
user_outputs:
  analog0_pin: NO_PIN
  analog1_pin: NO_PIN
  analog2_pin: NO_PIN
  analog3_pin: NO_PIN
  analog0_hz: 5000
  analog1_hz: 5000
  analog2_hz: 5000
  analog3_hz: 5000
  digital0_pin: i2so.12
  digital1_pin: i2so.13
  digital2_pin: i2so.14
  digital3_pin: i2so.15

start:
  must_home: true
  
SiemensV20:
  uart:
    txd_pin: gpio.17
    rxd_pin: gpio.16
    rts_pin: gpio.4
    baud: 9600
    mode: 8E1
  modbus_id: 1
  tool_num: 0
  speed_map: 0=0% 24000=100%
```

### Startup Messages

```text
$ss
[MSG:INFO: FluidNC 3.7.13 https://github.com/bdring/FluidNC]]
[MSG:INFO: Compiled with ESP32 SDK:v4.4.4]]
[MSG:INFO: Local filesystem type is spiffs]]
[MSG:INFO: Configuration file:Root_4_Base.yaml]]
[MSG:WARN: Ignored key SiemensV20]]
[MSG:ERR: Skipping key uart indent 2 this indent 0]]
[MSG:ERR: Skipping key txd_pin indent 4 this indent 0]]
[MSG:ERR: Skipping key rxd_pin indent 4 this indent 0]]
[MSG:ERR: Skipping key rts_pin indent 4 this indent 0]]
[MSG:ERR: Skipping key baud indent 4 this indent 0]]
[MSG:ERR: Skipping key mode indent 4 this indent 0]]
[MSG:ERR: Skipping key modbus_id indent 2 this indent 0]]
[MSG:ERR: Skipping key tool_num indent 2 this indent 0]]
[MSG:ERR: Skipping key speed_map indent 2 this indent 0]]
[MSG:INFO: Machine Root 4 V20]]
[MSG:INFO: Board Root Controller ISO]]
[MSG:INFO: I2SO BCK:gpio.22 WS:gpio.21 DATA:gpio.12]]
[MSG:INFO: SPI SCK:gpio.18 MOSI:gpio.23 MISO:gpio.19]]
[MSG:INFO: SD Card cs_pin:gpio.5 detect:NO_PIN freq:8000000]]
[MSG:INFO: Stepping:I2S_stream Pulse:4us Dsbl Delay:3us Dir Delay:1us Idle Delay:255ms]]
[MSG:INFO: User Digital Output:0 on Pin:I2SO.12]]
[MSG:INFO: User Digital Output:1 on Pin:I2SO.13]]
[MSG:INFO: User Digital Output:2 on Pin:I2SO.14]]
[MSG:INFO: User Digital Output:3 on Pin:I2SO.15]]
[MSG:INFO: Axis count 3]]
[MSG:INFO: Axis X (-1.000,1197.000)]]
[MSG:INFO:   Motor0]]
[MSG:INFO:     standard_stepper Step:I2SO.2 Dir:I2SO.1 Disable:I2SO.0:low]]
[MSG:INFO:  X All Limit gpio.26:low]]
[MSG:INFO: Axis Y (0.000,1185.000)]]
[MSG:INFO:   Motor0]]
[MSG:INFO:     standard_stepper Step:I2SO.6 Dir:I2SO.5:low Disable:I2SO.4:low]]
[MSG:INFO:  Y Pos Limit gpio.32:low]]
[MSG:INFO:   Motor1]]
[MSG:INFO:     standard_stepper Step:I2SO.22 Dir:I2SO.21:low Disable:I2SO.20:low]]
[MSG:INFO:  Y2 Pos Limit gpio.35:low]]
[MSG:INFO: Axis Z (-158.500,0.500)]]
[MSG:INFO:   Motor0]]
[MSG:INFO:     standard_stepper Step:I2SO.10 Dir:I2SO.9 Disable:I2SO.8:low]]
[MSG:INFO:  Z Pos Limit gpio.15:low]]
[MSG:INFO: feed_hold_pin gpio.34]]
[MSG:INFO: cycle_start_pin gpio.36]]
[MSG:INFO: macro0_pin gpio.39]]
[MSG:INFO: Kinematic system: Cartesian]]
[MSG:INFO: Using spindle NoSpindle]]
[MSG:INFO: Flood coolant gpio.33]]
[MSG:INFO: Mist coolant gpio.13]]
[MSG:INFO: Probe Pin: gpio.27]]
[MSG:INFO: Connecting to STA SSID:Car&D]]
[MSG:INFO: Connecting.]]
[MSG:INFO: Connecting..]]
[MSG:INFO: Connected - IP is 192.168.1.33]]
[MSG:INFO: WiFi on]]
[MSG:INFO: Start mDNS with hostname:http://fluidnc.local/]]
[MSG:INFO: SSDP Started]]
[MSG:INFO: HTTP started on port 80]]
[MSG:INFO: Telnet started on port 23]]
ok
```

### User Interface Software

WebUI, FluidTerm

### What happened?

Hi everyone, 
I just received the hardware and started to set it up : 

Downloaded [3.7.13] released, then installed install-wifi.bat
I opened FluidTerm
Import a "basic" yaml (provided on the Git of the ROOT CONTROLLER ISO : https://github.com/RootCNC/Root-Controller-ISO), code above, name : Root_4_Base.yaml
Connected to local network by wifi SSID:Car&D, connection seems ok
When trying to use WebUI, following message appears : "Failed to communicate with FW!"
[image]
When I try to open the ESP IP in the browser , the WebUI request is written in the FluidTerm console
[image]

I already try to install the [3.7.14-webui3-test] version, same problem

Can you help please?

MitchBradley, 2024-03-14:
I tried to reproduce your problem and everything worked.  Here are the steps I followed:

1. Ran erase.bat in the installer package by double-clicking (this step is probably not necessary; I did it just to be sure I started from a clean slate)
2. Ran install-wifi.bat then closed the FluidTerm window
3. Ran install-fs.bat
4. Used FluidTerm Ctrl-U to upload Root_4_Base.yaml
5. Typed "$config/filename=Root_4_Base.yaml" in FluidTerm
6. Typed Ctrl-R in FluidTerm to reboot the ESP32 with the new YAML file
7. Found the IP address in the startup messages and browsed to that IP address.  WebUI ran as expected

The only thing I can think of is that maybe your hardware already had a filesystem installed and it already had an old copy of index.html.gz that is incompatible with newer versions of FluidNC.  You did not mention running install-fs.bat or otherwise uploading index.html.gz, so that hypothesis is plausible.  I have no idea what Root controller hardware has preinstalled.

Did you try asking the people who took your money?  Presumably the purchase entitles you to some support.

MitchBradley, 2024-03-14:
The configuration file that you used refers to a spindle type "SiemensV20" that official FluidNC releases do not support.  Perhaps the Root Controller folks maintain a special version that does support that spindle.  That unsupported spindle is the cause of the "MSG:ERR" lines in the startup log.

The other config files in the Root Controller repo seem to use supported spindles like Huanyang and pwm.

RootCNC, 2024-03-18:
Support is offered through email (root@rootcnc.com) or via discord (https://discord.gg/93Ue5SwthW)

alimnette, 2024-03-18:
You must upload index.tar.gz into motherboard

closed 2025-01-21 by bdring