bdring/FluidNC #1165: OTA update to v3.7.15 failed at 36%, threw an error in the WebUI. Rebooting then failed.

shampeon, 2024-03-16:
### Wiki Search Terms

OTA
update failed

### Controller Board

MKS TinyBee

### Machine Description

Workbee 1500x1500 CNC router, dual Y, G-Penny 1.5kw spindle, Huanyang VFD connected by RS485

### Configuration file

```yaml
~~~
board: MKS TinyBee V1.0 XXYYZ
name:

kinematics:
  Cartesian:

i2so:
  bck_pin: gpio.25
  data_pin: gpio.27
  ws_pin: gpio.26

spi:
  miso_pin: gpio.19
  mosi_pin: gpio.23
  sck_pin: gpio.18

sdcard:
  cs_pin: gpio.5
  # uses TH2 IO34 active low - MAKE SURE jumper J2 is set to SDDET!!!
  card_detect_pin: gpio.34:low

stepping:
  engine: I2S_STATIC
  idle_ms: 255
  pulse_us: 5
  dir_delay_us: 1
  disable_delay_us: 2

axes:
  x:
    steps_per_mm: 31.965
    max_rate_mm_per_min: 8000.000
    acceleration_mm_per_sec2: 250.000
    max_travel_mm: 1300.000
    soft_limits: false
    homing:
      cycle: 2
      positive_direction: false
      mpos_mm: 0.000
      feed_mm_per_min: 300.000
      seek_mm_per_min: 1500.000
      settle_ms: 500
      seek_scaler: 1.100
      feed_scaler: 1.100

    motor0:
      limit_all_pin: gpio.33:high:pd
      hard_limits: true
      pulloff_mm: 4.000
      standard_stepper:
        step_pin: I2SO.1
        direction_pin: I2SO.2
        disable_pin: I2SO.0

    # use E0 driver for 2nd X axis motor
    motor1:
      hard_limits: true
      pulloff_mm: 4.000
      standard_stepper:
        step_pin: I2SO.10
        direction_pin: I2SO.11
        disable_pin: I2SO.9

  y:
    steps_per_mm: 32.046
    max_rate_mm_per_min: 8000.000
    acceleration_mm_per_sec2: 250.000
    max_travel_mm: 1320.000
    soft_limits: false
    homing:
      cycle: 3
      positive_direction: false
      mpos_mm: 0.000
      feed_mm_per_min: 300.000
      seek_mm_per_min: 2000.000
      settle_ms: 500
      seek_scaler: 1.100
      feed_scaler: 1.100

    motor0:
      limit_all_pin: gpio.32:high:pd
      hard_limits: false
      pulloff_mm: 4.000
      standard_stepper:
        step_pin: I2SO.4
        direction_pin: I2SO.5
        disable_pin: I2SO.3

    # use E1 driver for 2nd Y axis motor
    motor1:
      limit_neg_pin: NO_PIN
      hard_limits: false
      pulloff_mm: 4.000
      standard_stepper:
        step_pin: I2SO.13
        direction_pin: I2SO.14
        disable_pin: I2SO.12

  z:
    steps_per_mm: 200.000
    max_rate_mm_per_min: 3500.000
    acceleration_mm_per_sec2: 250.000
    max_travel_mm: 110.000
    soft_limits: false
    homing:
      cycle: 1
      positive_direction: true
      mpos_mm: 0.000
      feed_mm_per_min: 300.000
      seek_mm_per_min: 500.000
      settle_ms: 500
      seek_scaler: 1.100
      feed_scaler: 1.100

    motor0:
      limit_all_pin: gpio.22:high:pd
      hard_limits: true
      pulloff_mm: 3.000
      standard_stepper:
        step_pin: I2SO.7
        direction_pin: I2SO.8
        disable_pin: I2SO.6

control:
  safety_door_pin: gpio.16:low:pu
  # on MT_DET connector
  # reset_pin: gpio.35:low
  reset_pin: gpio.17:low:pu
  # on TH1 connector
  feed_hold_pin: gpio.36:low
  # on TB connector
  cycle_start_pin: gpio.39:low
  macro0_pin: NO_PIN
  macro1_pin: NO_PIN
  macro2_pin: NO_PIN
  macro3_pin: NO_PIN

# coolant:
#   # Heated Bed Terminal Block
#   flood_pin: i2so.16
#   # HE0 Terminal Block
#   mist_pin: i2so.17
#   delay_ms: 0

# spindle PWM signal
PWM:
  pwm_hz: 2500
  # on EXP1 IO15 connector.
  # note IO15 will give some short pulses on boot, which may activate spindle
  # use IO17 on EXP1 to avoid
  output_pin: gpio.15:high
  s0_with_disable: true
  tool_num: 100
  spinup_ms: 4000
  spindown_ms: 4000
  speed_map: 8000=0.000% 18000=100.000%

Laser:
  pwm_hz: 5000
  # on 3D Touch connector (has pdwn + PWM)
  output_pin: gpio.2:high:pd
  s0_with_disable: true
  tool_num: 1
  speed_map: 0=0.000% 1000=100.000%

probe:
  pin: gpio.35:pu:low
  check_mode_start: true

start:
  must_home: false

uart2:
  txd_pin: gpio.4
  rxd_pin: gpio.13
  rts_pin: NO_PIN
  baud: 9600
  mode: 8N1

Huanyang:
  uart_num: 2
  modbus_id: 1
  tool_num: 0
  speed_map: 0=0% 0=25% 7200=40% 18000=100%
  off_on_alarm: false
~~~
```

### Startup Messages

```text
Couldn't access it, board was unresponsive
```

### What happened?

I attempted to update to v3.7.15 OTA using the web UI. My machine is standalone, so OTA is the only reasonable way for me to update.

The update progress went in bursts, first to 15% before stalling for a bit, then up to 35% before another stall, then 36%. At that point, it stalled for a long time. An error modal popped up, which I clicked OK on to close.

Thinking it was potentially caused by low memory as it had been running since yesterday, I clicked Restart FluidNC within the web UI. The WebUI showed the board rebooting, but stopped responding. After a few minutes, the reboot modal closed on its own, but the board was unresponsive. I had to hard reboot the board by cutting and restoring power to get it back.

Updating the firmware a second time succeeded.

But there was clearly some state that FluidNC was in before the first firmware update that caused the firmware and reboot failure. If my suspicion is correct that there was low memory, maybe there should be some check before performing a firmware update.

### Other Information

~~~
[MSG:INFO: Update Firmware]
[MSG:INFO: Update 0%]
[MSG:INFO: Update 1%]
[MSG:INFO: Update 2%]
[MSG:INFO: Update 3%]
[MSG:INFO: Update 4%]
[MSG:INFO: Update 5%]
[MSG:INFO: Update 6%]
[MSG:INFO: Update 7%]
[MSG:INFO: Update 8%]
[MSG:INFO: Update 9%]
[MSG:INFO: Update 10%]
[MSG:INFO: Update 11%]
[MSG:INFO: Update 12%]
[MSG:INFO: Update 13%]
[MSG:INFO: Update 14%]
[MSG:INFO: Update 15%]
[MSG:INFO: Update 16%]
[MSG:INFO: Update 17%]
[MSG:INFO: Update 18%]
[MSG:INFO: Update 19%]
[MSG:INFO: Update 20%]
[MSG:INFO: Update 21%]
[MSG:INFO: Update 22%]
[MSG:INFO: Update 23%]
[MSG:INFO: Update 24%]
[MSG:INFO: Update 25%]
[MSG:INFO: Update 26%]
[MSG:INFO: Update 27%]
[MSG:INFO: Update failed]
~~~

bdring, 2024-03-16:
I just tried it and it worked for me. The OTA code is all standard code from Espressif. We have not changed anything. Can you connect FluidTerm via USB and give the startup info? We really can't help unless we can replicate the problem or see your boot messages.

shampeon, 2024-03-16:
I can't connect via USB, the board is connected to 24v power in my controller box (the TinyBee has a jumper to use 24v or enable USB), and I can't fit a USB cable in the box anyway.

It updated correctly the second time. I think the main issue is the failure state causing the board to freeze when triggering the reboot.

Interesting, the log message fails at 27%, but the WebUI showed 36%.

bdring, 2024-03-16:
OTA uses two memory partitions. One for the current firmware and one for the incoming firmware. If the upload fails you will reboot on the old firmware and can try again. If the upload completes and is validated, it boots on the new firmware. If you OTA upload a firmware that does not have WiFi or has a broken OTA, you need to use the USB.

USB uses a bootloader in ROM. You only need one partition. If the upload fails, the bootloader can still load a new firmware.

If you redesign your setup. I suggest making USB accessible, because some day you may need it.

MitchBradley, 2024-03-16:
There's no way to know if your situation was due to low memory unless someone can reproduce the failure.  Even so, it would be hard to say what amount of memory is necessary given the many variables in the wifi stack and the many different wifi environments that are possible.

We don't even know which WiFi mode you are using.  If it is AP mode, all bets are off.  We have reason to suspect that AP mode is unstable due to memory leaks in Espressif's WiFi stack.  That's why we recommend against using it except for initial setup.

shampeon, 2024-03-16:
I was not using AP mode, no. DHCP to my home network.

This is the latest system status post-update.

~~~
Chip ID:  50352
CPU Cores:  2
CPU Frequency:  240Mhz
CPU Temperature:  60.6°C
Free memory:  76.34 KB
SDK:  v4.4.4
Flash Size:  8.00 MB
Sleep mode:  Modem
Available Size for update:  1.88 MB
Available Size for LocalFS:  192.00 KB
Web port:  80
Data port:  23
Hostname:  fluidnc
Current WiFi Mode:  Client Station (08:F9:E0:9E:B0:C4)
Connected to:  mitchell
Signal:  100%
Phy Mode:  11n
Channel:  4
IP Mode:  DHCP
IP:  192.168.0.80
Gateway:  192.168.0.1
Mask:  255.255.0.0
DNS:  192.168.0.15
Disabled Mode:  Access Point (08:F9:E0:9E:B0:C5)
Notifications:  Disabled
FW version:  FluidNC v3.7.15
WebUI version:  github.com/MitchBradley/ESP3D-WEBUI@4c7780c
~~~

And `$ss`

~~~
[MSG:INFO: FluidNC v3.7.15 https://github.com/bdring/FluidNC]]
[MSG:INFO: Compiled with ESP32 SDK:v4.4.4]]
[MSG:INFO: Local filesystem type is littlefs]]
[MSG:INFO: Configuration file:config.yaml]]
[MSG:WARN: gpio.35 does not support :pu attribute]]
[MSG:DBG: Running after-parse tasks]]
[MSG:DBG: Checking configuration]]
[MSG:INFO: Machine ]]
[MSG:INFO: Board MKS TinyBee V1.0 XXYYZ]]
[MSG:INFO: UART2 Tx:gpio.4 Rx:gpio.13 RTS:NO_PIN Baud:9600]]
[MSG:INFO: I2SO BCK:gpio.25 WS:gpio.26 DATA:gpio.27]]
[MSG:INFO: SPI SCK:gpio.18 MOSI:gpio.23 MISO:gpio.19]]
[MSG:INFO: SD Card cs_pin:gpio.5 detect:gpio.34:low freq:8000000]]
[MSG:INFO: Stepping:I2S_static Pulse:5us Dsbl Delay:2us Dir Delay:1us Idle Delay:255ms]]
[MSG:INFO: Axis count 3]]
[MSG:INFO: Axis X (0.000,1300.000)]]
[MSG:INFO:   Motor0]]
[MSG:INFO:     standard_stepper Step:I2SO.1 Dir:I2SO.2 Disable:I2SO.0]]
[MSG:INFO:  X All Limit gpio.33:pd]]
[MSG:INFO:   Motor1]]
[MSG:INFO:     standard_stepper Step:I2SO.10 Dir:I2SO.11 Disable:I2SO.9]]
[MSG:INFO: Axis Y (0.000,1320.000)]]
[MSG:INFO:   Motor0]]
[MSG:INFO:     standard_stepper Step:I2SO.4 Dir:I2SO.5 Disable:I2SO.3]]
[MSG:INFO:  Y All Limit gpio.32:pd]]
[MSG:INFO:   Motor1]]
[MSG:INFO:     standard_stepper Step:I2SO.13 Dir:I2SO.14 Disable:I2SO.12]]
[MSG:INFO: Axis Z (-110.000,0.000)]]
[MSG:INFO:   Motor0]]
[MSG:INFO:     standard_stepper Step:I2SO.7 Dir:I2SO.8 Disable:I2SO.6]]
[MSG:INFO:  Z All Limit gpio.22:pd]]
[MSG:INFO: safety_door_pin gpio.16:low:pu]]
[MSG:INFO: reset_pin gpio.17:low:pu]]
[MSG:INFO: feed_hold_pin gpio.36:low]]
[MSG:INFO: cycle_start_pin gpio.39:low]]
[MSG:INFO: Kinematic system: Cartesian]]
[MSG:INFO: PWM Spindle Ena:NO_PIN Out:gpio.15 Dir:NO_PIN Freq:2500Hz Period:16383]]
[MSG:INFO: Laser Ena:NO_PIN Out:gpio.2:pd Freq:5000Hz Period:8191]]
[MSG:INFO: Huanyang Spindle  Tx:gpio.4 Rx:gpio.13 RTS:NO_PIN Baud:9600]]
[MSG:INFO: Using spindle Huanyang]]
[MSG:INFO: Probe Pin: gpio.35:low]]
[MSG:INFO: Connecting to STA SSID:mitchell]]
[MSG:INFO: Connecting.]]
[MSG:INFO: Connecting..]]
[MSG:INFO: Connected - IP is 192.168.0.80]]
[MSG:INFO: WiFi on]]
[MSG:INFO: Start mDNS with hostname:http://fluidnc.local/]]
[MSG:INFO: SSDP Started]]
[MSG:INFO: HTTP started on port 80]]
[MSG:INFO: Telnet started on port 23]]
~~~

shampeon, 2024-03-16:
And I understand that this is hard to reproduce. The reason I chose FluidNC is that it is standalone, which makes it harder to debug in my setup.

It'd be nice if I could get the serial output without requiring a physical connection, using e.g. https://github.com/ayushsharma82/WebSerial.

MitchBradley, 2024-03-16:
You can get the startup lines by sending $SS from any console, web or otherwise.

shampeon, 2024-03-16:
Not if it freezes....

bdring, 2024-03-16:
You are never going be able to use anything to do with the radios until well after boot. Wifi, WebUI, OTA, anything. 

Nobody can remote debug a broken computer either.

MitchBradley, 2024-03-16:
To get what you want, you would need a separate computer running the web interface and that computer would need to be always-on and super reliable.  You're not going to get everything from one ESP32 that runs the web interface and also the control firmware.

shampeon, 2024-03-16:
Guys, I understand how computers work. Seriously.

If I anticipated encountering errors, I could pre-set up a scenario to hookup a serial output to my board.

But in this case, using the board in normal course and encountering an error that causes a freeze, there's no way for me to backtrack from that, because the serial output isn't available, and no log files survive a restart. I was only able to get the debug messages because the modal timed out and I could navigate to the dashboard and copy from the (dead) web browser.

If, on the other hand, the serial output was always accessible on any page in the Web UI, I could at least copy the output when an error state otherwise froze the board, which would at least give better info for difficult to reproduce issues like this.

MitchBradley, 2024-03-16:
Feel free to modify WebUI to issue $SS on initial connection, save the result, and provide a button to view it.

shampeon, 2024-03-16:
I'm asking this in good faith. What do you guys think I should have done in this scenario?

You clearly think I'm wasting your time.

MitchBradley, 2024-03-16:
You tried again and it worked.  There's not much else that can be done without spending a lot of time that we don't have in order to try and fix a situation that we can't duplicate.  That is where the "waste of time" comes in.  If a lot of people were having the same problem (apparently they aren't), then maybe we could repro it and it would be worth tracking down.  As it is, there is no guarantee or even likelihood that time we spent would every have any form of payoff.  I'm going to lock this thread.  If you or anyone else encounters the problem again and can provide more details, a new issue can be opened.

closed 2024-03-16 by MitchBradley