bdring/FluidNC #1166: Problem: MKS DLC32 v2.1 error rst:0x10 (RTCWDT_RTC_RESET),boot:0x1b (SPI_FAST_FLASH_BOOT)

carlosmorenoaguilera, 2024-03-16:
### Wiki Search Terms

MKS DLC32 v2.1 stuck loading

### Controller Board

MKS DLC32 v2.1

### Machine Description

MKS DLC32 v2.1

### Configuration file

```yaml
first install
```

### Startup Messages

```text
Select a COM port
0: COM3 (\Device\BthModem0)
1: COM4 (\Device\BthModem1)
2: COM5 (\Device\Serial2)
Choice: 2
FluidTerm v1.2.1 using COM5
Exit: Ctrl-C, Ctrl-Q or Ctrl-], Clear screen: CTRL-W
Upload: Ctrl-U, Reset ESP32: Ctrl-R, Send Override: Ctrl-O
ets Jun  8 2016 00:22:57

rst:0x10 (RTCWDT_RTC_RESET),boot:0x1b (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:1184
load:0x40078000,len:13220
ho 0 tail 12 room 4
load:0x40080400,len:3028
entry 0x400805e4
ets Jun  8 2016 00:22:57

rst:0x10 (RTCWDT_RTC_RESET),boot:0x1b (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:1184
load:0x40078000,len:13220
ho 0 tail 12 room 4
load:0x40080400,len:3028
entry 0x400805e4
ets Jun  8 2016 00:22:57

rst:0x10 (RTCWDT_RTC_RESET),boot:0x1b (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:1184
load:0x40078000,len:13220
ho 0 tail 12 room 4
load:0x40080400,len:3028
entry 0x400805e4
ets Jun  8 2016 00:22:57

rst:0x10 (RTCWDT_RTC_RESET),boot:0x1b (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:1184
load:0x40078000,len:13220
ho 0 tail 12 room 4
load:0x40080400,len:3028
entry 0x400805e4
Exited by ^C
ets Jun  8 2016 00:22:57
```

### What happened?

Please Help me, im trying to install FluidNC, im using fluidnc-v3.7.15-win64

in this order>

erase.bat
install-fs.bat
install-wifi.bat

aflter that i open terminal> 

FluidTerm v1.2.1 using COM5
Exit: Ctrl-C, Ctrl-Q or Ctrl-], Clear screen: CTRL-W
Upload: Ctrl-U, Reset ESP32: Ctrl-R, Send Override: Ctrl-O
ets Jun  8 2016 00:22:57

rst:0x10 (RTCWDT_RTC_RESET),boot:0x1b (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:1184
load:0x40078000,len:13220
ho 0 tail 12 room 4
load:0x40080400,len:3028
entry 0x400805e4
ets Jun  8 2016 00:22:57

rst:0x10 (RTCWDT_RTC_RESET),boot:0x1b (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:1184
load:0x40078000,len:13220
ho 0 tail 12 room 4
load:0x40080400,len:3028
entry 0x400805e4
ets Jun  8 2016 00:22:57

in and endless boot, with a orange led blinking, the board that i have is a MakerBase Mks DLC32 v2.1 with a V24 power adapter (Tried with 12v too same result) how i can know if i have a faulty board ? im complety newbie with this please give me your patience

bdring, 2024-03-16:
It looks like there is no firmware on the ESP32. Try loading from the WebUI. You might need to use the boot button. See the Wiki

MitchBradley, 2024-03-16:
I think bdring meant to say [Web Installer ](https://installer.fluidnc.com/)instead of WebUI.  WebUI won't work unless the firmware is working.

carlosmorenoaguilera, 2024-03-17:
Terminal of Web Installer show me same output 

[image]

when select install option 

[image]

MitchBradley, 2024-03-17:
Did you try holding the boot switch while resetting the ESP32?

carlosmorenoaguilera, 2024-03-17:
> Did you try holding the boot switch while resetting the ESP32?

[image]

reset button and this runback button interface are the same?

MitchBradley, 2024-03-17:
ESP32 has a complicated circuit for entering Bootloader mode under control of the USB Serial port.  On some boards, and with some operating systems, it doesn't work.  Most boards have a pushbutton switch labeled "BOOT0" that you can press while powering on to override that circuit. Apparently MKS DLC32 does not have such a switch.

I looked at the schematic and found that the signal line that normally connects to a BOOT0 switch instead goes to the SDA pin on J2.  Therefore, you can probably enter boot mode by putting a jumper between that SDA pin and the adjacent GND pin (labeled G) on that connector.

[image]

The jumper would need to be inserted when the power is off, then turn on the power and try to perform the FluidNC install.  After installation, turn off the power, remove the jumper, and turn the power back on.

It might be possible to insert the jumper then click the Restart button on Web Installer.  If that does not work, try the full power-cycle method suggested above.

We spend a lot of uncompensated time supporting the very low cost DLC32 hardware.  The manufacturer that sells it does not provide much support and does not contribute to our project in any way.  If we have helped you, please consider sponsoring our project.

carlosmorenoaguilera, 2024-03-17:
> ESP32 has a complicated circuit for entering Bootloader mode under control of the USB Serial port. On some boards, and with some operating systems, it doesn't work. Most boards have a pushbutton [...]

i put a jump with a clip and data cable for motors

in FluidNC web installer i see this on terminal 

�c�I!���!����boA�J㰇�ets Jun  8 2016 00:22:57

rst:0x10 (RTCWDT_RTC_RESET),boot:0x3 (DOWNLOAD_BOOT(UART0/UART1/SDIO_REI_REO_V2))
waiting for download

[image]

orange led stop blinking on this step, then i press install:
version: 3.7.15
Processor type: ESP32-WROOM

on firmware variant, i chose WiFi and fresh-install 

[image]

orange led start to blink again and failed, on this page

[image]

if i shutdown and start again the page said lost connection and go to blank page.

thank you for trying to help me i appreciate it

MitchBradley, 2024-03-17:
The fact that it said "waiting for download" means that it entered the correct bootloader mode then.  The problem might be that Web Installer was waiting for that message but the message had already been issued.  Sometimes it helps to install the jumper just before you click on the final button (like "firmware update").

When you have broken hardware like this, it can be tricky to get the timing right.  You may need to experiment with installing the jumper at different times relative to when you start the install operation.

Once the ESP32 is in download mode, it will stay in download mode until you reset it with the jumper out.  This means that you will need to remove the jumper, reset, and then restart the process if you have to do multiple install steps.

That is all the advice I have.  If you still can't get it to work, perhaps you should ask for a refund from the vendor.  Most controllers work correctly.

Willpowerandstring, 2024-04-15:
Run an erase cycle using the MKS Laser tool first then firmware can be uploaded using the batch file scripts.

closed 2024-04-25 by bdring

beta-tester, 2024-11-04:
i had a similar issue with my MKS DLC32 V2.1.
flashing wasn't possible - even with the stock firmware and tools.
i could erase the flash, but uploading the firmware wasn't possible. it turned out, that the stock power supply makes a lot of electric noise and the stock usb cable was picking up that noise. in fact i changed the  power plug and the usb cable. then i was able to install the new filesystem and wifi.

bdring, 2024-11-04:
The github release install method works fine if you are already in waiting for upload (bootloader) mode.

beta-tester, 2024-11-04:
> The github release install method works fine if you are already in waiting for upload (bootloader) mode.

yes, but with a noisy cable connection the upload of the firmware interrupts somewhere in the middle.
anyhow i got it work. just fine tuning the config.yaml i have to finish.

MitchBradley, 2024-11-04:
On one MKS Tinybee, I have observed that firmware upload is more reliable with the external power supply off, powered only from USB.  That could apply to MKS DLC32 as well.  The CH340 USB-Serial chips they both use are less robust than the CP2102 chips that some other ESP32 modules use - but CH340 is very inexpensive.  So you get what you pay for, and we end up having to answer the questions because MKS won't do it.