bdring/FluidNC #1167: Can't implement shared_fault_pin

meakerb, 2024-03-17:
### Wiki Search Terms

shared_fault_pin

### Controller Board

Wemos D1 Mini ESP32

### Machine Description

Two axis machine (XY) using two CL86T closed-loop stepper drivers and two closed-loop NEMA 34 stepper motors (steperonline model 34HS59-6004D-E1000)

### Input Circuits

```yaml
external 4.7K pullup on gpio.26
```

### Configuration file

```yaml
name: "MyMachine"
board: "Wemos D1 Mini ESP32"

stepping:
  engine: RMT
  idle_ms: 255
  dir_delay_us: 3
  pulse_us: 3
  disable_delay_us: 1

axes:
  shared_stepper_disable_pin: NO_PIN
  shared_fault_pin: gpio.26:low:pu

  x:
    steps_per_mm: 8.888889
    max_rate_mm_per_min: 86400
    acceleration_mm_per_sec2: 8000.0
    max_travel_mm: 1000
    soft_limits: false
    homing:
      cycle: -1
      mpos_mm: 0.0
      allow_single_axis: false
    
    motor0:
      standard_stepper:
        step_pin: gpio.18
        direction_pin: gpio.23:low
        disable_pin: gpio.5
    motor1:
      null_motor:

  y:
    steps_per_mm: 8.0
    max_rate_mm_per_min: 86400
    acceleration_mm_per_sec2: 8000.0
    max_travel_mm: 1000
    soft_limits: false
    homing:
      cycle: -1
      mpos_mm: 0.0
      allow_single_axis: false

    motor0:
      standard_stepper:
        step_pin: gpio.22
        direction_pin: gpio.17:low
        disable_pin: gpio.16
    motor1:
      null_motor:

macros:
  startup_line0: $MD

start:
  must_home: false
```

### Startup Messages

```text
Resetting MCU
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:1184
load:0x40078000,len:13220
ho 0 tail 12 room 4
load:0x40080400,len:3028
entry 0x400805e4
[MSG:INFO: uart_channel0 created]
[MSG:RST]
[MSG:INFO: FluidNC 3.7.13 https://github.com/bdring/FluidNC]
[MSG:INFO: Compiled with ESP32 SDK:v4.4.4]
[MSG:INFO: Local filesystem type is spiffs]
[MSG:INFO: Configuration file:config.yaml]
[MSG:WARN: Ignored key shared_fault_pin]
[MSG:INFO: Machine MyMachine]
[MSG:INFO: Board Wemos D1 Mini ESP32]
[MSG:INFO: Stepping:RMT Pulse:3us Dsbl Delay:1us Dir Delay:3us Idle Delay:255ms]
[MSG:INFO: Axis count 3]
[MSG:INFO: Axis X (-1000.000,0.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     standard_stepper Step:gpio.18 Dir:gpio.23:low Disable:gpio.5]
[MSG:INFO:   Motor1]
[MSG:INFO: Axis Y (-1000.000,0.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     standard_stepper Step:gpio.22 Dir:gpio.17:low Disable:gpio.16]
[MSG:INFO:   Motor1]
[MSG:INFO: Axis Z (-1000.000,0.000)]
[MSG:INFO:   Motor0]
[MSG:INFO: Kinematic system: Cartesian]
[MSG:INFO: Using spindle NoSpindle]
[MSG:INFO: BT Started with Shirley]

Grbl 3.7 [FluidNC 3.7.13 (bt) '$' for help]
[MSG:INFO: Running macro startup_line0: $MD]
[MSG:INFO: Disabling all motors]
$CD
board: Wemos D1 Mini ESP32
name: MyMachine
meta:
stepping:
  engine: RMT
  idle_ms: 255
  pulse_us: 3
  dir_delay_us: 3
  disable_delay_us: 1
  segments: 12

spi:
  miso_pin: NO_PIN
  mosi_pin: NO_PIN
  sck_pin: NO_PIN

sdcard:
  cs_pin: NO_PIN
  card_detect_pin: NO_PIN
  frequency_hz: 8000000

kinematics:
  Cartesian:

axes:
  shared_stepper_disable_pin: NO_PIN
  shared_stepper_reset_pin: NO_PIN
  x:
    steps_per_mm: 8.888889
    max_rate_mm_per_min: 86400.000000
    acceleration_mm_per_sec2: 8000.000000
    max_travel_mm: 1000.000000
    soft_limits: false
    homing:
      cycle: -1
      allow_single_axis: false
      positive_direction: true
      mpos_mm: 0.000000
      feed_mm_per_min: 50.000000
      seek_mm_per_min: 200.000000
      settle_ms: 250
      seek_scaler: 1.100000
      feed_scaler: 1.100000

    motor0:
      limit_neg_pin: NO_PIN
      limit_pos_pin: NO_PIN
      limit_all_pin: NO_PIN
      hard_limits: false
      pulloff_mm: 1.000000
      standard_stepper:
        step_pin: gpio.18
        direction_pin: gpio.23:low
        disable_pin: gpio.5

    motor1:
      limit_neg_pin: NO_PIN
      limit_pos_pin: NO_PIN
      limit_all_pin: NO_PIN
      hard_limits: false
      pulloff_mm: 1.000000
      null_motor:

  y:
    steps_per_mm: 8.000000
    max_rate_mm_per_min: 86400.000000
    acceleration_mm_per_sec2: 8000.000000
    max_travel_mm: 1000.000000
    soft_limits: false
    homing:
      cycle: -1
      allow_single_axis: false
      positive_direction: true
      mpos_mm: 0.000000
      feed_mm_per_min: 50.000000
      seek_mm_per_min: 200.000000
      settle_ms: 250
      seek_scaler: 1.100000
      feed_scaler: 1.100000

    motor0:
      limit_neg_pin: NO_PIN
      limit_pos_pin: NO_PIN
      limit_all_pin: NO_PIN
      hard_limits: false
      pulloff_mm: 1.000000
      standard_stepper:
        step_pin: gpio.22
        direction_pin: gpio.17:low
        disable_pin: gpio.16

    motor1:
      limit_neg_pin: NO_PIN
      limit_pos_pin: NO_PIN
      limit_all_pin: NO_PIN
      hard_limits: false
      pulloff_mm: 1.000000
      null_motor:

  z:
    steps_per_mm: 80.000000
    max_rate_mm_per_min: 1000.000000
    acceleration_mm_per_sec2: 25.000000
    max_travel_mm: 1000.000000
    soft_limits: false
    homing:
      cycle: 0
      allow_single_axis: true
      positive_direction: true
      mpos_mm: 0.000000
      feed_mm_per_min: 50.000000
      seek_mm_per_min: 200.000000
      settle_ms: 250
      seek_scaler: 1.100000
      feed_scaler: 1.100000

    motor0:
      limit_neg_pin: NO_PIN
      limit_pos_pin: NO_PIN
      limit_all_pin: NO_PIN
      hard_limits: false
      pulloff_mm: 1.000000
      null_motor:

control:
  safety_door_pin: NO_PIN
  reset_pin: NO_PIN
  feed_hold_pin: NO_PIN
  cycle_start_pin: NO_PIN
  macro0_pin: NO_PIN
  macro1_pin: NO_PIN
  macro2_pin: NO_PIN
  macro3_pin: NO_PIN
  fault_pin: NO_PIN
  estop_pin: NO_PIN

coolant:
  flood_pin: NO_PIN
  mist_pin: NO_PIN
  delay_ms: 0

probe:
  pin: NO_PIN
  toolsetter_pin: NO_PIN
  check_mode_start: true

macros:
  startup_line0: $MD
  startup_line1:
  macro0:
  macro1:
  macro2:
  macro3:
  after_homing:
  after_reset:
  after_unlock:

start:
  must_home: false
  deactivate_parking: false
  check_limits: false

parking:
  enable: false
  axis: Z
  target_mpos_mm: -5.000000
  rate_mm_per_min: 800.000000
  pullout_distance_mm: 5.000000
  pullout_rate_mm_per_min: 250.000000

user_outputs:
  analog0_pin: NO_PIN
  analog1_pin: NO_PIN
  analog2_pin: NO_PIN
  analog3_pin: NO_PIN
  analog0_hz: 5000
  analog1_hz: 5000
  analog2_hz: 5000
  analog3_hz: 5000
  digital0_pin: NO_PIN
  digital1_pin: NO_PIN
  digital2_pin: NO_PIN
  digital3_pin: NO_PIN
  digital4_pin: NO_PIN
  digital5_pin: NO_PIN
  digital6_pin: NO_PIN
  digital7_pin: NO_PIN

NoSpindle:

arc_tolerance_mm: 0.002000
junction_deviation_mm: 0.010000
verbose_errors: false
report_inches: false
enable_parking_override_control: false
use_line_numbers: false
planner_blocks: 16
ok
```

### User Interface Software

FluidTerm

### What happened?

I want to tie the alam outputs of the two CL86Ts together and (with an external pullup resistor) tie them to a pin so that both motors will disable when there is a fault, as described in the wiki page for shared_fault_pin.
I've tried implementing the shared_fault_pin according to the wiki, but it never parses correctly.
I did a $CD (config/dump) to see if it should be in a different group, but shared_fault_pin does not show up.

### Other Information

This is a config.yaml parsing issue as I don't have any connections to the stepper driver at this time.

MitchBradley, 2024-03-17:
The wiki is wrong.  There is no shared_fault_pin in the axes section.  It is now fault_pin in the control: section, per http://wiki.fluidnc.com/en/config/control#fault_pin .

I removed the incorrect information from the wiki.

banana-legs, 2024-03-17:
On some versions of the CL86T, you can also wire the fault outputs (in parallel with each other), in series with the stepper enable lines (wired in parallel too), with the opto-couplers of the stepper enables acting as the pull-up.   That way, if there is a fault, the drivers disable themselves straight away, without relying on the controller software to stop.   I have both a hard-wired stop system and the fault_pin in my setup, and there are times when even though the fault_pin is triggered, the code can progress for some time before the pin is responded to (i.e. long enough for the torque of a Nema 34 to bend something important).    I am happy to be corrected, but I have always assumed that the fault_pin line was intended for 'information purposes' only, and not as an actual safety feature.     Evan

bdring, 2024-03-17:
We have some changes to the fault and limit coming soon. See this branch of the firmware. It primarily deals with startup faults and faults occurring in no running modes. Faults prevent the machine from operating.  

https://github.com/bdring/FluidNC/tree/LimitsOnReset

It would be great to have some help testing it.

In the future we may add separate faults per motor.

banana-legs, 2024-03-17:
Yes, very happy to help :)

MitchBradley, 2024-03-17:
As the person who implemented the fault_pin feature, I can tell you exactly what my intent was.  I have a machine with ihsv57 closed loop steppers.  Sometimes one would stall, issue the fault signal, and stop, while the program continued with the other running.  That caused ruined parts, broken bits, etc.  Now, when there is a stall, the machine stops without causing further damage.  Sometimes I can recover the part.

MitchBradley, 2024-03-17:
I am aware that software stop is not fast enough in all cases.  For my setup it works well enough so it is worth having.

banana-legs, 2024-03-17:
I agree, it is very much worth having and certainly makes restarting/ continuing stopped jobs so much simpler.   I have closed-loop Nema 34 steppers using CL86T's, and an XYYZZ configuration, so have both the Y and Z axes that could get twisted if one of the motor pair stops.    I went for an independent set of logic to shut everything down (all 5 motors, spindle or laser etc.), and then trip the fault line in the end (  I am a coward :)   ).    The machine makes one hell of a bang if it does trip while going at 20m/minute though!

closed 2024-03-17 by MitchBradley

arlimbad222, 2024-03-17:
Hi all 

Recently I make machine with leadshine closed loop stepper motor & driver I use that fault output from stepper driver to power off the fluidnc board i just hook relay at every stepper drive and  make one common relay to operate when any driver is in fault the 12v of fluid nc board will cut of and I also add one alaram red led so whenever any error comes the red led will blink and I have to check panel box restart the drive and it runs again.