bdring/FluidNC #1169: Probe no more working after 3.7.13 Upgrade

cbrassel, 2024-03-19:
### Wiki Search Terms

probe not working, error

### Controller Board

xPRO V5

### Machine Description

Bulkman 3D Ultimate Bee 1500x1500
2xY axis 
H100 VFD
1.5Kw spindle 

### Configuration file

```yaml
board: XPro V5
name: Bulkman 3d 1515
meta: Config 29.11.2023

stepping:
  engine: RMT
  idle_ms: 255
  pulse_us: 4
  dir_delay_us: 1
  disable_delay_us: 0

axes:
  shared_stepper_disable_pin: NO_PIN
  x:
    steps_per_mm: 160.402
    max_rate_mm_per_min: 5000.000
    acceleration_mm_per_sec2: 100.000
    max_travel_mm: 1217.00
    soft_limits: true
    homing:
      cycle: 2
      positive_direction: true
      mpos_mm: 0.000
      feed_mm_per_min: 50.000
      seek_mm_per_min: 2000.000
      settle_ms: 500
      seek_scaler: 1.100
      feed_scaler: 1.100

    motor0:
      limit_neg_pin: gpio.35:high
      limit_pos_pin: NO_PIN
      limit_all_pin: NO_PIN
      hard_limits: true
      pulloff_mm: 2.500
      tmc_5160:
        step_pin: gpio.12
        direction_pin: gpio.14:high
        use_enable: true
        cs_pin: gpio.17
        spi_index: 1
        r_sense_ohms: 0.050
        run_amps: 2.800
        hold_amps: 0.250
        microsteps: 8
        toff_disable: 0
        toff_stealthchop: 5
        run_mode: CoolStep
        homing_mode: CoolStep
        stallguard: 16
        stallguard_debug: false
        toff_coolstep: 3
        tpfd: 4

  y:
    steps_per_mm: 160.42
    max_rate_mm_per_min: 3500.000
    acceleration_mm_per_sec2: 100.000
    max_travel_mm: 1255.000
    soft_limits: true
    homing:
      cycle: 2
      positive_direction: false
      mpos_mm: 0.000
      feed_mm_per_min: 50.000
      seek_mm_per_min: 2000.000
      settle_ms: 500
      seek_scaler: 1.100
      feed_scaler: 1.100

    motor0:
      limit_neg_pin: NO_PIN
      limit_pos_pin: gpio.34:high
      limit_all_pin: NO_PIN
      hard_limits: true
      pulloff_mm: 2.500
      tmc_5160:
        step_pin: gpio.27
        direction_pin: gpio.26:low
        use_enable: true
        cs_pin: NO_PIN
        spi_index: 2
        r_sense_ohms: 0.050
        run_amps: 2.800
        hold_amps: 0.250
        microsteps: 8
        toff_disable: 0
        toff_stealthchop: 5
        run_mode: CoolStep
        homing_mode: CoolStep
        stallguard: 16
        stallguard_debug: false
        toff_coolstep: 3
        tpfd: 4

    motor1:
      limit_neg_pin: NO_PIN
      limit_pos_pin: gpio.36:high
      limit_all_pin: NO_PIN
      hard_limits: true
      pulloff_mm: 2.5000
      tmc_5160:
        step_pin: gpio.33
        direction_pin: gpio.32:low
        use_enable: true
        cs_pin: NO_PIN
        spi_index: 3
        r_sense_ohms: 0.050
        run_amps: 2.800
        hold_amps: 0.250
        microsteps: 8
        toff_disable: 0
        toff_stealthchop: 5
        run_mode: CoolStep
        homing_mode: CoolStep
        stallguard: 16
        stallguard_debug: false
        toff_coolstep: 3
        tpfd: 4

  z:
    steps_per_mm: 400.000
    max_rate_mm_per_min: 1500.000
    acceleration_mm_per_sec2: 50.000
    max_travel_mm: 140.000
    soft_limits: true
    homing:
      cycle: 1
      positive_direction: true
      mpos_mm: 0.000
      feed_mm_per_min: 50.000
      seek_mm_per_min: 500.000
      settle_ms: 500
      seek_scaler: 1.100
      feed_scaler: 1.100

    motor0:
      limit_neg_pin: gpio.39:high
      limit_pos_pin: NO_PIN
      limit_all_pin: NO_PIN
      hard_limits: true
      pulloff_mm: 2.500
      tmc_5160:
        step_pin: gpio.15
        direction_pin: gpio.2:high
        use_enable: true
        cs_pin: NO_PIN
        spi_index: 4
        r_sense_ohms: 0.050
        run_amps: 2.800
        hold_amps: 0.250
        microsteps: 8
        toff_disable: 0
        toff_stealthchop: 5
        run_mode: CoolStep
        homing_mode: CoolStep
        stallguard: 16
        stallguard_debug: false
        toff_coolstep: 3
        tpfd: 4

spi:
  miso_pin: gpio.19
  mosi_pin: gpio.23
  sck_pin: gpio.18

sdcard:
  cs_pin: gpio.5
  card_detect_pin: NO_PIN
  frequency_hz: 10000000

probe:
  pin: gpio.22:low:pu
  check_mode_start: false

control:
  safety_door_pin: NO_PIN
  reset_pin: NO_PIN
  feed_hold_pin: gpio.16:high
  macro0_pin: gpio.13:low
  macro1_pin: gpio.0:low

macros:
  startup_line0:
  startup_line1:
  macro0:
  macro1:
  macro2:
  macro3:

start:
  must_home: false

coolant:
  mist_pin: gpio.21
  delay_ms: 0

uart1:
  txd_pin: gpio.4
  rxd_pin: gpio.25
  rts_pin: NO_PIN
  baud: 19200
  mode: 8N1

H100:
  uart_num: 1
  modbus_id: 1
  tool_num: 0
  #speed_map: 0=0% 6000=25% 12000=50% 24000=100%
  off_on_alarm: false
```

### Startup Messages

```text
ets Jun  8 2016 00:22:57
rst:0xc (SW_CPU_RESET),boot:0x17 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:1184
load:0x40078000,len:13220
ho 0 tail 12 room 4
load:0x40080400,len:3028
entry 0x400805e4
[MSG:INFO: uart_channel0 created]
[MSG:RST]
[MSG:INFO: FluidNC 3.7.13 https://github.com/bdring/FluidNC]
[MSG:INFO: Compiled with ESP32 SDK:v4.4.4]
[MSG:INFO: Local filesystem type is littlefs]
[MSG:INFO: Configuration file:config.yaml]
[MSG:INFO: Machine Bulkman 3d 1515]
[MSG:INFO: Board XPro V5]
[MSG:INFO: UART1 Tx:gpio.4 Rx:gpio.25 RTS:NO_PIN Baud:19200]
[MSG:INFO: SPI SCK:gpio.18 MOSI:gpio.23 MISO:gpio.19]
[MSG:INFO: SD Card cs_pin:gpio.5 detect:NO_PIN freq:10000000]
[MSG:INFO: Stepping:RMT Pulse:4us Dsbl Delay:0us Dir Delay:1us Idle Delay:255ms]
[MSG:INFO: Axis count 3]
[MSG:INFO: Axis X (-1217.000,0.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     tmc_5160 Step:gpio.12 Dir:gpio.14 CS:gpio.17 Disable:NO_PIN Index:1 R:0.050]
[MSG:INFO:  X Neg Limit gpio.35]
[MSG:INFO: Axis Y (0.000,1255.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     tmc_5160 Step:gpio.27 Dir:gpio.26:low CS:NO_PIN Disable:NO_PIN Index:2 R:0.050]
[MSG:INFO:  Y Pos Limit gpio.34]
[MSG:INFO:   Motor1]
[MSG:INFO:     tmc_5160 Step:gpio.33 Dir:gpio.32:low CS:NO_PIN Disable:NO_PIN Index:3 R:0.050]
[MSG:INFO:  Y2 Pos Limit gpio.36]
[MSG:INFO: Axis Z (-140.000,0.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     tmc_5160 Step:gpio.15 Dir:gpio.2 CS:NO_PIN Disable:NO_PIN Index:4 R:0.050]
[MSG:INFO:  Z Neg Limit gpio.39]
[MSG:INFO: X Axis driver test passed]
[MSG:INFO: Y Axis driver test passed]
[MSG:INFO: Y2 Axis driver test passed]
[MSG:INFO: Z Axis driver test passed]
[MSG:INFO: feed_hold_pin gpio.16]
[MSG:INFO: macro0_pin gpio.13:low]
[MSG:INFO: macro1_pin gpio.0:low]
[MSG:INFO: Kinematic system: Cartesian]
[MSG:INFO: H100 Spindle  Tx:gpio.4 Rx:gpio.25 RTS:NO_PIN Baud:19200]
[MSG:INFO: Using spindle H100]
[MSG:INFO: Mist coolant gpio.21]
[MSG:INFO: Probe Pin: gpio.22:low:pu]
[MSG:INFO: VFD: Max speed:24000rpm]
[MSG:INFO: Connecting to STA SSID:wifi.brassel.net]
[MSG:INFO: Connecting.]
[MSG:INFO: VFD: Min speed:0rpm]
[MSG:INFO: VFD: VFD settings read: Freq range(0 , 4000)]]
[MSG:INFO: Connecting..]
[MSG:INFO: Connected - IP is 10.1.3.15]
[MSG:INFO: WiFi on]
[MSG:INFO: Start mDNS with hostname:http://fluidnc.local/]
[MSG:INFO: SSDP Started]
[MSG:INFO: HTTP started on port 80]
[MSG:INFO: Telnet started on port 23]
Grbl 3.7 [FluidNC 3.7.13 (wifi) '$' for help]
```

### User Interface Software

gSender

### What happened?

After Upgrading to latest release 3.7.13 The Z probe did no more working.

The probe pin configuration or the client software did not change.
probe config :
client $probe
/probe:
  pin: NO_PIN
  toolsetter_pin: NO_PIN
  check_mode_start: true
ok

bdring, 2024-03-19:
I do not know why your start messages show a probe defined, but $probe shows no pins. I just tested on 3.7.15 and my probe loads and works fine. We have not made any recent changes to the probe.

Send $SS and $probe again in the same FluidNC session. I want to make sure I see both from the same session.

cbrassel, 2024-03-19:
I have juste restarted the board :

[MSG:INFO: uart_channel0 created]
[MSG:RST]
[MSG:INFO: FluidNC 3.7.13 https://github.com/bdring/FluidNC]
[MSG:INFO: Compiled with ESP32 SDK:v4.4.4]
[MSG:INFO: Local filesystem type is littlefs]
[MSG:INFO: Configuration file:config.yaml]
[MSG:INFO: Machine Bulkman 3d 1515]
[MSG:INFO: Board XPro V5]
[MSG:INFO: UART1 Tx:gpio.4 Rx:gpio.25 RTS:NO_PIN Baud:19200]
[MSG:INFO: SPI SCK:gpio.18 MOSI:gpio.23 MISO:gpio.19]
[MSG:INFO: SD Card cs_pin:gpio.5 detect:NO_PIN freq:10000000]
[MSG:INFO: Stepping:RMT Pulse:4us Dsbl Delay:0us Dir Delay:1us Idle Delay:255ms]
[MSG:INFO: Axis count 3]
[MSG:INFO: Axis X (-1217.000,0.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     tmc_5160 Step:gpio.12 Dir:gpio.14 CS:gpio.17 Disable:NO_PIN Index:1 R:0.050]
[MSG:INFO:  X Neg Limit gpio.35]
[MSG:INFO: Axis Y (0.000,1255.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     tmc_5160 Step:gpio.27 Dir:gpio.26:low CS:NO_PIN Disable:NO_PIN Index:2 R:0.050]
[MSG:INFO:  Y Pos Limit gpio.34]
[MSG:INFO:   Motor1]
[MSG:INFO:     tmc_5160 Step:gpio.33 Dir:gpio.32:low CS:NO_PIN Disable:NO_PIN Index:3 R:0.050]
[MSG:INFO:  Y2 Pos Limit gpio.36]
[MSG:INFO: Axis Z (-140.000,0.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     tmc_5160 Step:gpio.15 Dir:gpio.2 CS:NO_PIN Disable:NO_PIN Index:4 R:0.050]
[MSG:INFO:  Z Neg Limit gpio.39]
[MSG:INFO: X Axis driver test passed]
[MSG:INFO: Y Axis driver test passed]
[MSG:INFO: Y2 Axis driver test passed]
[MSG:INFO: Z Axis driver test passed]
[MSG:INFO: feed_hold_pin gpio.16]
[MSG:INFO: macro0_pin gpio.13:low]
[MSG:INFO: macro1_pin gpio.0:low]
[MSG:INFO: Kinematic system: Cartesian]
[MSG:INFO: H100 Spindle  Tx:gpio.4 Rx:gpio.25 RTS:NO_PIN Baud:19200]
[MSG:INFO: Using spindle H100]
[MSG:INFO: Mist coolant gpio.21]
[MSG:INFO: Probe Pin: gpio.22:low:pu]
[MSG:INFO: VFD: Max speed:24000rpm]
[MSG:INFO: Connecting to STA SSID:wifi.brassel.net]
[MSG:INFO: Connecting.]
[MSG:INFO: VFD: Min speed:0rpm]
[MSG:INFO: VFD: VFD settings read: Freq range(0 , 4000)]]
[MSG:INFO: Connecting..]
[MSG:INFO: Connected - IP is 10.1.3.15]
[MSG:INFO: WiFi on]
[MSG:INFO: Start mDNS with hostname:http://fluidnc.local/]
[MSG:INFO: SSDP Started]
[MSG:INFO: HTTP started on port 80]
[MSG:INFO: Telnet started on port 23]
Grbl 3.7 [FluidNC 3.7.13 (wifi) '$' for help]
client $probe
/probe:
  pin: gpio.22:low:pu
  toolsetter_pin: NO_PIN
  check_mode_start: false

cbrassel, 2024-03-19:
but the probe does not work better

bdring, 2024-03-19:
Something has changed for the better. Send a picture of you probe and the connection to the controller.

Also send "?" with the probe activated, then deactivated. Paste the responses to ? command.

cbrassel, 2024-03-19:
Probe not activated :
<Idle|MPos:0.000,0.000,0.000|FS:0,0|WCO:-355.052,101.995,-66.892>
Probe Activated :
<Idle|MPos:0.000,0.000,0.000|FS:0,0>
[image]
[image]

bdring, 2024-03-19:
Try using a piece or wire to complete the circuit. Then request status with ?

[image]

cbrassel, 2024-03-19:
nope .. it's the same 
The probe's wire seem's to be ok, I have 5v at the probe's end

bdring, 2024-03-19:
I can't help you much more. It appears to me that that something in the config file or you settings changed during your upgrade. That is why you got 2 different responses to the $probe command.

I would revert to the last firmware you had success with to verify you configuration.

cbrassel, 2024-03-19:
ok, I will try it this week-end

cbrassel, 2024-03-20:
Today I had time to test the 3.7.12 firmware, and the probe is working as expected, so it seem's that the problem has been introduced after 3.7.12.

bdring, 2024-03-20:
Post the config file you are currently using.

cbrassel, 2024-03-20:
it's the one posted in this issue

bdring, 2024-03-20:
Ok, I'll try to test it late next week. I don't have access to an xPro right now.

cbrassel, 2024-03-20:
ok, no problem, if I can help giving you access to the PC connected to my xpro we can arrange this.

closed 2025-01-21 by bdring