bdring/FluidNC #1172: Problem:  Issues with uploading *.yaml to FluidNC (latest version)

PST75, 2024-03-20:
### Wiki Search Terms

Root controller, Huanyang VFD, Yaml indentations etc.

### Controller Board

Root Controller ISO. The config should be an exact copy when it comes to sections described in the root controller WIKI (base config, axes (common cathode), rs485 etc. etc.)

### Machine Description

Gantry milling machine (QueenBee Pro) with dual y steppers (X,Y,Y, Z) and Huanyang VFD and 2.2 kW spindle. Controller is Root Controller ISO
[image]
[image]
[image]

### Configuration file

```yaml
board: Root Controller ISO
name: QBCNC v.1.2

stepping:
  engine: I2S_STREAM
  idle_ms: 255
  pulse_us: 4
  dir_delay_us: 1
  disable_delay_us: 3

axes:
  shared_stepper_disable_pin: NO_PIN
  x:
    steps_per_mm: 398.000
    max_rate_mm_per_min: 3000.000
    acceleration_mm_per_sec2: 200.000
    max_travel_mm: 1242.000
    soft_limits: true
    homing:
      cycle: 2
      positive_direction: true
      mpos_mm: 0.00
      feed_mm_per_min: 100.000
      seek_mm_per_min: 800.000
      settle_ms: 500
      seek_scaler: 1.100
      feed_scaler: 1.100

    motor0:
      limit_neg_pin: NO_PIN
      limit_pos_pin: NO_PIN
      limit_all_pin: NO_PIN
      hard_limits: true
      pulloff_mm: 2.500
      standard_stepper:
        step_pin: I2SO.7:low
        direction_pin: I2SO.5:low
        disable_pin: I2SO.3:high

  y:
    steps_per_mm: 400.000
    max_rate_mm_per_min: 3000.000
    acceleration_mm_per_sec2: 200.000
    max_travel_mm: 1277.000
    soft_limits: true
    homing:
      cycle: 2
      positive_direction: true
      mpos_mm: 0.0
      feed_mm_per_min: 100.000
      seek_mm_per_min: 800.000
      settle_ms: 500
      seek_scaler: 1.100
      feed_scaler: 1.100

    motor0:
      limit_neg_pin: NO_PIN
      limit_pos_pin: NO_PIN
      limit_all_pin: NO_PIN
      hard_limits: true
      pulloff_mm: 2.500
      standard_stepper:
        step_pin: I2SO.12:low
        direction_pin: I2SO.10:low
        disable_pin: I2SO.8:high

    motor1:
      limit_neg_pin: NO_PIN
      limit_pos_pin: NO_PIN
      limit_all_pin: NO_PIN
      hard_limits: true
      pulloff_mm: 2.500
      standard_stepper:
        step_pin: I2SO.13:low
        direction_pin: I2SO.11:low
        disable_pin: I2SO.9:high

  z:
    steps_per_mm: 400.000
    max_rate_mm_per_min: 3000.000
    acceleration_mm_per_sec2: 100.000
    max_travel_mm: 80.000
    soft_limits: true
    homing:
      cycle: 1
      positive_direction: true
      mpos_mm: 0.0
      feed_mm_per_min: 100.000
      seek_mm_per_min: 800.000
      settle_ms: 500
      seek_scaler: 1.100
      feed_scaler: 1.100

    motor0:
      limit_neg_pin: NO_PIN
      limit_pos_pin: NO_PIN
      limit_all_pin: NO_PIN
      hard_limits: true
      pulloff_mm: 2.500
      standard_stepper:
        step_pin: I2SO.18:low
        direction_pin: I2SO.16:low
        disable_pin: I2SO.14:high

i2so:
  bck_pin: gpio.22
  data_pin: gpio.12
  ws_pin: gpio.21

spi:
  miso_pin: gpio.19
  mosi_pin: gpio.23
  sck_pin: gpio.18

sdcard:
  card_detect_pin: NO_PIN
  cs_pin: gpio.5

control:
  safety_door_pin: NO_PIN
  reset_pin: NO_PIN
  feed_hold_pin: gpio.34
  cycle_start_pin: gpio.36
  macro0_pin: gpio.39
  macro1_pin: NO_PIN
  macro2_pin: NO_PIN
  macro3_pin: NO_PIN

coolant:
  flood_pin: gpio.33
  mist_pin: gpio.13
  delay_ms: 0

probe:
  pin: gpio.27
  check_mode_start: true

macros:
  startup_line0:
  startup_line1:
  macro0: 
  macro1: 
  macro2: 
  macro3:

user_outputs:
  analog0_pin: NO_PIN
  analog1_pin: NO_PIN
  analog2_pin: NO_PIN
  analog3_pin: NO_PIN
  analog0_hz: 5000
  analog1_hz: 5000
  analog2_hz: 5000
  analog3_hz: 5000
  digital0_pin: NO_PIN
  digital1_pin: NO_PIN
  digital2_pin: NO_PIN
  digital3_pin: i2so.15

start:
  must_home: true

Huanyang:
  uart:
  txd_pin: gpio.17
  rxd_pin: gpio.16
  rts_pin: gpio.4
  baud: 9600
  mode: 8N1
  modbus_id: 1
  tool_num: 0
  speed_map: 0=0% 0=25% 6000=25% 24000=100%

Laser:
  pwm_hz: 5000
  output_pin: gpio.25
  enable_pin: NO_PIN
  disable_with_s0: false
  s0_with_disable: true
  tool_num: 1
  speed_map: 0=0.000% 1000=100.000%
  off_on_alarm: false
```

### Startup Messages

```text
Please see pict shared in Machine Description.

I have also encountered a different error when trying to upload a different config se picture uploaded in machine description.

Also I get some strange characters within the FluidTerm after powering on / of the controller (picture also included above)
```

### User Interface Software

FluidTerm

### What happened?

Nothing. No movement or sound of power to steppers etc.  The steppers are wired to the controller as common cathode and acc. to wiring instructions on the controller.

No comms with the VFD although I cheked the wiring and parameter settings in FluidNc Wiki....

[image]
[image]
[image]

PST75, 2024-03-20:
Added note...the steppers powered on with an green LED and the power led etc. on the root controller was indicating "on", VFD powered as well

bdring, 2024-03-20:
Let's fix the start errors first.

I see an error in your config file. For the Huanyang, the uart parameters are not indented under the uart: line. This causes all the uart parameters to issue warnings.

```yaml
Huanyang:
  uart:
  txd_pin: gpio.17
  rxd_pin: gpio.16
  rts_pin: gpio.4
  baud: 9600
  mode: 8N1
  modbus_id: 1
  tool_num: 0
  speed_map: 0=0% 0=25% 6000=25% 24000=100%
  ```

I suggest using the new style separate uart section as described here.  

http://wiki.fluidnc.com/en/config/config_spindles#huanyang-vfd-with-rs485

```yaml
uart1:
  txd_pin: gpio.17
  rxd_pin: gpio.16
  rts_pin: gpio.4
  baud: 9600
  mode: 8N1

Huanyang:
  uart_num: 1
  modbus_id: 1
  tool_num: 0
  speed_map: 0=0% 0=25% 6000=25% 24000=100%

PST75, 2024-03-20:
Thank you very much. Please see my updated file below and I will test ASAP. What about the error I got when trying to upload a new config (ref in picture incl. in my post). It was written as Returned -2 and if it happens again I will not be able to upload the corrected yaml :(

Also could this "double enter" become an issue for the config?

[image]

Corrected yaml:

board: Root Controller ISO
name: QBCNC v.1.2

stepping:
  engine: I2S_STREAM
  idle_ms: 255
  pulse_us: 4
  dir_delay_us: 1
  disable_delay_us: 3

axes:
  shared_stepper_disable_pin: NO_PIN
  x:
    steps_per_mm: 398.000
    max_rate_mm_per_min: 3000.000
    acceleration_mm_per_sec2: 200.000
    max_travel_mm: 1242.000
    soft_limits: true
    homing:
      cycle: 2
      positive_direction: true
      mpos_mm: 0.00
      feed_mm_per_min: 100.000
      seek_mm_per_min: 800.000
      settle_ms: 500
      seek_scaler: 1.100
      feed_scaler: 1.100

    motor0:
      limit_neg_pin: NO_PIN
      limit_pos_pin: NO_PIN
      limit_all_pin: NO_PIN
      hard_limits: true
      pulloff_mm: 2.500
      standard_stepper:
        step_pin: I2SO.7:low
        direction_pin: I2SO.5:low
        disable_pin: I2SO.3:high

  y:
    steps_per_mm: 400.000
    max_rate_mm_per_min: 3000.000
    acceleration_mm_per_sec2: 200.000
    max_travel_mm: 1277.000
    soft_limits: true
    homing:
      cycle: 2
      positive_direction: true
      mpos_mm: 0.0
      feed_mm_per_min: 100.000
      seek_mm_per_min: 800.000
      settle_ms: 500
      seek_scaler: 1.100
      feed_scaler: 1.100

    motor0:
      limit_neg_pin: NO_PIN
      limit_pos_pin: NO_PIN
      limit_all_pin: NO_PIN
      hard_limits: true
      pulloff_mm: 2.500
      standard_stepper:
        step_pin: I2SO.12:low
        direction_pin: I2SO.10:low
        disable_pin: I2SO.8:high

    motor1:
      limit_neg_pin: NO_PIN
      limit_pos_pin: NO_PIN
      limit_all_pin: NO_PIN
      hard_limits: true
      pulloff_mm: 2.500
      standard_stepper:
        step_pin: I2SO.13:low
        direction_pin: I2SO.11:low
        disable_pin: I2SO.9:high

  z:
    steps_per_mm: 400.000
    max_rate_mm_per_min: 3000.000
    acceleration_mm_per_sec2: 100.000
    max_travel_mm: 80.000
    soft_limits: true
    homing:
      cycle: 1
      positive_direction: true
      mpos_mm: 0.0
      feed_mm_per_min: 100.000
      seek_mm_per_min: 800.000
      settle_ms: 500
      seek_scaler: 1.100
      feed_scaler: 1.100

    motor0:
      limit_neg_pin: NO_PIN
      limit_pos_pin: NO_PIN
      limit_all_pin: NO_PIN
      hard_limits: true
      pulloff_mm: 2.500
      standard_stepper:
        step_pin: I2SO.18:low
        direction_pin: I2SO.16:low
        disable_pin: I2SO.14:high

i2so:
  bck_pin: gpio.22
  data_pin: gpio.12
  ws_pin: gpio.21

spi:
  miso_pin: gpio.19
  mosi_pin: gpio.23
  sck_pin: gpio.18

sdcard:
  card_detect_pin: NO_PIN
  cs_pin: gpio.5

control:
  safety_door_pin: NO_PIN
  reset_pin: NO_PIN
  feed_hold_pin: gpio.34
  cycle_start_pin: gpio.36
  macro0_pin: gpio.39
  macro1_pin: NO_PIN
  macro2_pin: NO_PIN
  macro3_pin: NO_PIN

coolant:
  flood_pin: gpio.33
  mist_pin: gpio.13
  delay_ms: 0

probe:
  pin: gpio.27
  check_mode_start: true

macros:
  startup_line0:
  startup_line1:
  macro0: 
  macro1: 
  macro2: 
  macro3:

user_outputs:
  analog0_pin: NO_PIN
  analog1_pin: NO_PIN
  analog2_pin: NO_PIN
  analog3_pin: NO_PIN
  analog0_hz: 5000
  analog1_hz: 5000
  analog2_hz: 5000
  analog3_hz: 5000
  digital0_pin: NO_PIN
  digital1_pin: NO_PIN
  digital2_pin: NO_PIN
  digital3_pin: i2so.15

start:
  must_home: true

uart1:
  txd_pin: gpio.17
  rxd_pin: gpio.16
  rts_pin: gpio.4
  baud: 9600
  mode: 8N1
  
Huanyang:
  uart_num: 1
  modbus_id: 1
  tool_num: 0
  speed_map: 0=0% 0=25% 6000=25% 24000=100%

Laser:
  pwm_hz: 5000
  output_pin: gpio.25
  enable_pin: NO_PIN
  disable_with_s0: false
  s0_with_disable: true
  tool_num: 1
  speed_map: 0=0.000% 1000=100.000%
  off_on_alarm: false

MitchBradley, 2024-03-20:
The Root Controller people told us [here](https://github.com/bdring/FluidNC/issues/1163#issuecomment-2005120039) that they offer support via email at root@rootcnc.com and Discord at https://discord.gg/93Ue5SwthW

PST75, 2024-03-20:
@MitchBradley 

Thanks...as I'm unsure how to add the three back tics correctly (should it be both in the start and the end of the file) I have renamed the Yaml to a txt file. If possible please have a look if there is something "structually" wrong with the file. and how I have formatted it...most of it is a copy from a config downloaded from the Root Github but maybe I have mad a mess of things in the structure (even if I have not changed much)

Please have a look. Thanks :)

[QBCNCv22.txt](https://github.com/bdring/FluidNC/files/14672549/QBCNCv22.txt)

PST75, 2024-03-20:
```
board: Root Controller ISO
name: QBCNC v.1.2

stepping:
  engine: I2S_STREAM
  idle_ms: 255
  pulse_us: 4
  dir_delay_us: 1
  disable_delay_us: 3

axes:
  shared_stepper_disable_pin: NO_PIN
  x:
    steps_per_mm: 398.000
    max_rate_mm_per_min: 3000.000
    acceleration_mm_per_sec2: 200.000
    max_travel_mm: 1242.000
    soft_limits: true
    homing:
      cycle: 2
      positive_direction: true
      mpos_mm: 0.00
      feed_mm_per_min: 100.000
      seek_mm_per_min: 800.000
      settle_ms: 500
      seek_scaler: 1.100
      feed_scaler: 1.100

    motor0:
      limit_neg_pin: NO_PIN
      limit_pos_pin: NO_PIN
      limit_all_pin: NO_PIN
      hard_limits: true
      pulloff_mm: 2.500
      standard_stepper:
        step_pin: I2SO.7:low
        direction_pin: I2SO.5:low
        disable_pin: I2SO.3:high

  y:
    steps_per_mm: 400.000
    max_rate_mm_per_min: 3000.000
    acceleration_mm_per_sec2: 200.000
    max_travel_mm: 1277.000
    soft_limits: true
    homing:
      cycle: 2
      positive_direction: true
      mpos_mm: 0.0
      feed_mm_per_min: 100.000
      seek_mm_per_min: 800.000
      settle_ms: 500
      seek_scaler: 1.100
      feed_scaler: 1.100

    motor0:
      limit_neg_pin: NO_PIN
      limit_pos_pin: NO_PIN
      limit_all_pin: NO_PIN
      hard_limits: true
      pulloff_mm: 2.500
      standard_stepper:
        step_pin: I2SO.12:low
        direction_pin: I2SO.10:low
        disable_pin: I2SO.8:high

    motor1:
      limit_neg_pin: NO_PIN
      limit_pos_pin: NO_PIN
      limit_all_pin: NO_PIN
      hard_limits: true
      pulloff_mm: 2.500
      standard_stepper:
        step_pin: I2SO.13:low
        direction_pin: I2SO.11:low
        disable_pin: I2SO.9:high

  z:
    steps_per_mm: 400.000
    max_rate_mm_per_min: 3000.000
    acceleration_mm_per_sec2: 100.000
    max_travel_mm: 80.000
    soft_limits: true
    homing:
      cycle: 1
      positive_direction: true
      mpos_mm: 0.0
      feed_mm_per_min: 100.000
      seek_mm_per_min: 800.000
      settle_ms: 500
      seek_scaler: 1.100
      feed_scaler: 1.100

    motor0:
      limit_neg_pin: NO_PIN
      limit_pos_pin: NO_PIN
      limit_all_pin: NO_PIN
      hard_limits: true
      pulloff_mm: 2.500
      standard_stepper:
        step_pin: I2SO.18:low
        direction_pin: I2SO.16:low
        disable_pin: I2SO.14:high

i2so:
  bck_pin: gpio.22
  data_pin: gpio.12
  ws_pin: gpio.21

spi:
  miso_pin: gpio.19
  mosi_pin: gpio.23
  sck_pin: gpio.18

sdcard:
  card_detect_pin: NO_PIN
  cs_pin: gpio.5

control:
  safety_door_pin: NO_PIN
  reset_pin: NO_PIN
  feed_hold_pin: gpio.34
  cycle_start_pin: gpio.36
  macro0_pin: gpio.39
  macro1_pin: NO_PIN
  macro2_pin: NO_PIN
  macro3_pin: NO_PIN

coolant:
  flood_pin: gpio.33
  mist_pin: gpio.13
  delay_ms: 0

probe:
  pin: gpio.27
  check_mode_start: true

macros:
  startup_line0:
  startup_line1:
  macro0: 
  macro1: 
  macro2: 
  macro3:

user_outputs:
  analog0_pin: NO_PIN
  analog1_pin: NO_PIN
  analog2_pin: NO_PIN
  analog3_pin: NO_PIN
  analog0_hz: 5000
  analog1_hz: 5000
  analog2_hz: 5000
  analog3_hz: 5000
  digital0_pin: NO_PIN
  digital1_pin: NO_PIN
  digital2_pin: NO_PIN
  digital3_pin: i2so.15

start:
  must_home: true

uart1:
  txd_pin: gpio.17
  rxd_pin: gpio.16
  rts_pin: gpio.4
  baud: 9600
  mode: 8N1
  
Huanyang:
  uart_num: 1
  modbus_id: 1
  tool_num: 0
  speed_map: 0=0% 0=25% 6000=25% 24000=100%

Laser:
  pwm_hz: 5000
  output_pin: gpio.25
  enable_pin: NO_PIN
  disable_with_s0: false
  s0_with_disable: true
  tool_num: 1
  speed_map: 0=0.000% 1000=100.000%
  off_on_alarm: false
```

MitchBradley, 2024-03-20:
Yes the quoting is correct.  If you select the Preview tab you can see what it looks like before posting.  I will look at the file.  Typically the startup messages will complain if there is a structural problem, usually with a message about indentation or about some item not being recognized.

MitchBradley, 2024-03-20:
Your yaml loads on my controller without complaint.

MitchBradley, 2024-03-20:
Here are the startup message from your yaml.  This is on a midtbot controller that I used for some kinds of testing.  The firmware version is essentially 3.7.15 plus a couple of changes that are unrelated to anything we are discussing here.  The fact that it is not a Root Controller does not affect the ability to parse the yaml file.  It does result in the `[MSG:INFO: VFD RS485 Unresponsive]` message at the end, as I have no VFD on this controller (or any active controller at present).
```
$config/filename=issue1172.flnc
ok
$bye
[MSG:INFO: Restarting]
ok
[MSG:INFO: WiFi Disconnected]
[MSG:INFO: uart_channel0 created]
[MSG:RST]
[MSG:INFO: FluidNC 3.7.13 (SWFlowControl-39938a54-dirty) https://github.com/bdring/FluidNC.git]
[MSG:INFO: Compiled with ESP32 SDK:v4.4.4]
[MSG:INFO: Local filesystem type is spiffs]
[MSG:INFO: Configuration file:issue1172.flnc]
[MSG:INFO: Machine QBCNC v.1.2]
[MSG:INFO: Board Root Controller ISO]
[MSG:INFO: UART1 Tx:gpio.17 Rx:gpio.16 RTS:gpio.4 Baud:9600]
[MSG:INFO: I2SO BCK:gpio.22 WS:gpio.21 DATA:gpio.12]
[MSG:INFO: SPI SCK:gpio.18 MOSI:gpio.23 MISO:gpio.19]
[MSG:INFO: SD Card cs_pin:gpio.5 detect:NO_PIN freq:8000000]
[MSG:INFO: Stepping:I2S_stream Pulse:4us Dsbl Delay:3us Dir Delay:1us Idle Delay:255ms]
[MSG:INFO: User Digital Output:3 on Pin:I2SO.15]
[MSG:INFO: Axis count 3]
[MSG:INFO: Axis X (-1242.000,0.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     standard_stepper Step:I2SO.7:low Dir:I2SO.5:low Disable:I2SO.3]
[MSG:INFO: Axis Y (-1277.000,0.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     standard_stepper Step:I2SO.12:low Dir:I2SO.10:low Disable:I2SO.8]
[MSG:INFO:   Motor1]
[MSG:INFO:     standard_stepper Step:I2SO.13:low Dir:I2SO.11:low Disable:I2SO.9]
[MSG:INFO: Axis Z (-80.000,0.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     standard_stepper Step:I2SO.18:low Dir:I2SO.16:low Disable:I2SO.14]
[MSG:INFO: feed_hold_pin gpio.34]
[MSG:INFO: cycle_start_pin gpio.36]
[MSG:INFO: macro0_pin gpio.39]
[MSG:INFO: Kinematic system: Cartesian]
[MSG:INFO: Huanyang Spindle  Tx:gpio.17 Rx:gpio.16 RTS:gpio.4 Baud:9600]
[MSG:INFO: Laser Ena:NO_PIN Out:gpio.25 Freq:5000Hz Period:8191]
[MSG:INFO: Using spindle Huanyang]
[MSG:INFO: Flood coolant gpio.33]
[MSG:INFO: Mist coolant gpio.13]
[MSG:INFO: Probe Pin: gpio.27]
[MSG:INFO: Connecting to STA SSID:CampOlindaHS]
[MSG:INFO: Connecting.]
[MSG:INFO: Connecting..]
[MSG:INFO: Connected - IP is 192.168.0.67]
[MSG:INFO: WiFi on]
[MSG:INFO: Start mDNS with hostname:http://fluidnc.local/]
[MSG:INFO: SSDP Started]
[MSG:INFO: HTTP started on port 80]
[MSG:INFO: Telnet started on port 23]

Grbl 3.7 [FluidNC 3.7.13 (SWFlowControl-39938a54-dirty) (wifi) '$' for help]
[MSG:INFO: ALARM: Unhomed]
ALARM:14
[MSG:INFO: VFD RS485 Unresponsive]
```

MitchBradley, 2024-03-20:
I compared the attached file with the quoted version and they are identical, thus verifying the quoting correctness.

PST75, 2024-03-20:
Thank you very much Mitch for your support👍😊 Much appreciated.

I will test the config on my machine and report back with my findings/end
result. Thanks!

PST75, 2024-03-24:
Got everything up and working, Both the 2.2.kW spindle and laser is now fully operational and interchangable within the GCode Sender.

Thank you @MitchBradley  & @bdring  for your gr8 assistance and I'm now looking forward on with my journey wit Fluid NC. Appreciate your support on this forum so much! :) 💯 🥇 

[image]

[image]

[image]

closed 2024-03-24 by PST75