bdring/FluidNC #1173: Feature: Folinn BD600 class in "Spindles" for RS485 support

apopanda, 2024-03-22:
### Machine Context

Running a Root controller with the G-Penny spindle + VFD setup. They sent a Folinn BD600. It's not supported via RS485, so I'll write up the support

### Feature Description

I'm attempting to create a working setup for this VFD for my use. After I get it working (and the entire process is followed), it could be merged. PR guidelines state, that an issue needs to be made for discussion. The code will be here:
https://github.com/apopanda/FluidNC
Manual for the VFD is here
https://cononmotor.com.au/wp-content/uploads/2017/09/BD600-Manual.pdf 
And contains most of the necessary addresses, responses, etc. 

### Other Approaches

I really don't want to use PWM or switch cables around, so I'd rather implement it.

### How I Can Help

There's no need for help from the Developers right now.

MitchBradley, 2024-03-22:
I am starting to shy away from including more VFDs into the code base.  Every one increases our support cost.

snoozemoose, 2024-12-03:
Is there any update to this? I also have the BD600 and just bought a board to start using FluidNC.

MitchBradley, 2025-01-08:
I had a look at the code in the github repo above and I have a couple of concerns:
* The commentary does not match the code.  The commentary mentions commands 5, 1 and 4, while the code uses commands 6 and 3.
* It seems to me that the speed will be set to either 0 or max, because of this line
```c++
       uint32_t speed_percentage = (dev_speed / _maxFrequency) * 100 * 100; // percentage, two decimal places
```
It is integer arithmetic, and the phrase `dev_speed / _maxFrequency` will evaluate to 1 if dev_speed==_maxFrequency, otherwise 0.  For integer scaling, you need to do the multiplication before the division.

apopanda, 2025-01-09:
> I had a look at the code in the github repo above and I have a couple of concerns:
>
>     * The commentary does not match the code.  The commentary mentions commands 5, 1 and 4, while the code uses [...]

The code that's uploaded isn't ready. I need to get to my workshop laptop and upload the working version. I have that one running with all the settings setup, frequency working etc.

apopanda, 2025-01-11:
Alright, I uploaded what I had on the laptop. I'll take a look at the commentary soon, though it most likely me fixing the code in debug, and not updating comments. Or copying from the original file and not changing them to the BD600 at all. More likely I filled it out according to what I understood from the manual, but it turned out to be more simple - 06 for write and 03 for read, and I didn't edit it.  

@snoozemoose you will need to setup the VFD as it says in the .cpp file. Basically setting max frequency, command source to RS485, frequency source etc. Make sure to check all of this with the manual, just in case.

snoozemoose, 2025-01-19:
@apopanda, I just wanted to let you know that I will test your code as soon as I can but I still have some things to finish on my machine before I’m ready. Thanks a lot for uploading the fixes!

MitchBradley, 2025-01-21:
It would be great if you could test the BD600 with the ModbusVFD driver.  I wrote itt in hopes of supporting situations like yours but have been disappointed in the lack of feedback from users.

bdring, 2025-01-21:
See this wiki section.

http://wiki.fluidnc.com/en/config/config_spindles#generic-modbusvfd-rs485

apopanda, 2025-01-21:
> [@apopanda](https://github.com/apopanda) There has been so much more done on RS485 in the original/upstream repo that your changes are no longer usable as is but the good news is that it might be possible to use a simple generic `ModbusVFD` configuration instead. Can you please contact me on t w I t t e r? I would like to investigate this option but it would be great to be able to ask you some questions.

Don't have xitter, I'm on Discord as "Maalus" on the fluidnc discord. I'm also able to whatsapp or signal, alternatively skype. Mail adam@glia.org if you want to setup a call, I'd be happy to help however I can.

apopanda, 2025-01-21:
> It would be great if you could test the BD600 with the ModbusVFD driver. I wrote itt in hopes of supporting situations like yours but have been disappointed in the lack of feedback from users.

I'd volunteer if I wasn't in the middle of production. Have a long list of stuff to test - rootCNC with grblHAL for one, since nobody has done that yet, and testing your implementation could be fun too.