bdring/FluidNC #1185: Fluid Dial only ever shows NC when connected to a 6-Pack Board

dustonanthony, 2024-04-10:
### Wiki Search Terms

Fluid Dial, Fluid NC

### Controller Board

6-pack

### Machine Description

XYYZ Router with TMC2130 driver, Fluid Dial Pendent connected to Socket #3

### Input Circuits

```yaml
I followed the Fluid Dial Wiki for connections to the M5Dial and the 6-Pack controller.
Sock #3 uart2:
  txd_pin: gpio.26
  rxd_pin: gpio.4
  rts_pin: NO_PIN
  cts_pin: NO_PIN
  baud: 115200
  mode: 8N1

uart_channel2:
  uart_num: 2
  report_interval_ms: 75
```

### Configuration file

```yaml
board: 6 Pack
name: Desktop CNC Router XYYZ
stepping:
  engine: I2S_STREAM
  idle_ms: 255
  pulse_us: 4
  dir_delay_us: 1
  disable_delay_us: 0
  
  uart2:
  txd_pin: gpio.26
  rxd_pin: gpio.4
  rts_pin: NO_PIN
  cts_pin: NO_PIN
  baud: 115200
  mode: 8N1

uart_channel2:
  uart_num: 2
  report_interval_ms: 75

axes:
  shared_stepper_disable_pin: NO_PIN
  x:
    steps_per_mm: 800.000
    max_rate_mm_per_min: 3000.000
    acceleration_mm_per_sec2: 100.000
    max_travel_mm: 300.000
    soft_limits: true
    homing:
      cycle: 2
      positive_direction: false
      mpos_mm: 150.000
      feed_mm_per_min: 100.000
      seek_mm_per_min: 800.000
      settle_ms: 500
      seek_scaler: 1.100
      feed_scaler: 1.100

    motor0:
      limit_neg_pin: gpio.33:low
      limit_pos_pin: NO_PIN
      limit_all_pin: NO_PIN
      hard_limits: true
      pulloff_mm: 1.000
      tmc_2130:
        cs_pin: i2so.3
        r_sense_ohms: 0.110
        run_amps: 0.750
        hold_amps: 0.750
        microsteps: 8
        stallguard: 0
        stallguard_debug: false
        toff_disable: 0
        toff_stealthchop: 5
        toff_coolstep: 3
        run_mode: CoolStep
        homing_mode: CoolStep
        use_enable: false
        step_pin: I2SO.2
        direction_pin: I2SO.1
        disable_pin: I2SO.0

  y:
    steps_per_mm: 800.000
    max_rate_mm_per_min: 3000.000
    acceleration_mm_per_sec2: 100.000
    max_travel_mm: 300.000
    soft_limits: true
    homing:
      cycle: 2
      positive_direction: false
      mpos_mm: 150.000
      feed_mm_per_min: 100.000
      seek_mm_per_min: 800.000
      settle_ms: 500
      seek_scaler: 1.100
      feed_scaler: 1.100

    motor0:
      limit_neg_pin: gpio.32:low
      limit_pos_pin: NO_PIN
      limit_all_pin: NO_PIN
      hard_limits: true
      pulloff_mm: 1.000
      tmc_2130:
        cs_pin: i2so.6
        r_sense_ohms: 0.110
        run_amps: 0.750
        hold_amps: 0.750
        microsteps: 16
        stallguard: 0
        stallguard_debug: false
        toff_disable: 0
        toff_stealthchop: 5
        toff_coolstep: 3
        run_mode: StealthChop
        homing_mode: StealthChop
        use_enable: false
        step_pin: I2SO.5
        direction_pin: I2SO.4
        disable_pin: I2SO.7
        
        motor1:
      limit_neg_pin: gpio.35:low
      limit_pos_pin: NO_PIN
      limit_all_pin: NO_PIN
      hard_limits: true
      pulloff_mm: 1.000
      tmc_2130:
        cs_pin: i2so.11
        r_sense_ohms: 0.110
        run_amps: 0.750
        hold_amps: 0.750
        microsteps: 16
        stallguard: 0
        stallguard_debug: false
        toff_disable: 0
        toff_stealthchop: 5
        toff_coolstep: 3
        run_mode: StealthChop
        homing_mode: StealthChop
        use_enable: false
        step_pin: I2SO.10
        direction_pin: I2SO.9
        disable_pin: I2SO.8

  z:
    steps_per_mm: 500.000
    max_rate_mm_per_min: 3000.000
    acceleration_mm_per_sec2: 100.000
    max_travel_mm: 56.000
    soft_limits: true
    homing:
      cycle: 1
      positive_direction: true
      mpos_mm: 150.000
      feed_mm_per_min: 100.000
      seek_mm_per_min: 800.000
      settle_ms: 500
      seek_scaler: 1.100
      feed_scaler: 1.100

    motor0:
      limit_neg_pin: NO_PIN
      limit_pos_pin: gpio.34
      limit_all_pin: NO_PIN
      hard_limits: true
      pulloff_mm: 1.000
      tmc_2130:
        cs_pin: i2so.14
        r_sense_ohms: 0.110
        run_amps: 0.750
        hold_amps: 0.750
        microsteps: 16
        stallguard: 0
        stallguard_debug: false
        toff_disable: 0
        toff_stealthchop: 5
        toff_coolstep: 3
        run_mode: StealthChop
        homing_mode: StealthChop
        use_enable: false
        step_pin: I2SO.13
        direction_pin: I2SO.12
        disable_pin: I2SO.15

i2so:
  bck_pin: gpio.22
  data_pin: gpio.21
  ws_pin: gpio.17

spi:
  miso_pin: gpio.19
  mosi_pin: gpio.23
  sck_pin: gpio.18

sdcard:
  card_detect_pin: NO_PIN
  cs_pin: gpio.5

control:
  safety_door_pin: NO_PIN
  reset_pin: NO_PIN
  feed_hold_pin: NO_PIN
  cycle_start_pin: NO_PIN
  macro0_pin: NO_PIN
  macro1_pin: NO_PIN
  macro2_pin: NO_PIN
  macro3_pin: NO_PIN

coolant:
  flood_pin: NO_PIN
  mist_pin: NO_PIN
  delay_ms: 0

probe:
  pin: NO_PIN
  check_mode_start: true

macros:
  startup_line0:
  startup_line1:
  macro0:
  macro1:
  macro2:
  macro3:

user_outputs:
  analog0_pin: NO_PIN
  analog1_pin: NO_PIN
  analog2_pin: NO_PIN
  analog3_pin: NO_PIN
  analog0_hz: 5000
  analog1_hz: 5000
  analog2_hz: 5000
  analog3_hz: 5000
  digital0_pin: NO_PIN
  digital1_pin: NO_PIN
  digital2_pin: NO_PIN
  digital3_pin: NO_PIN

start:
  must_home: false

#PWM:
  #pwm_hz: 5000
  #output_pin: gpio.26
  #enable_pin: gpio.4
  #direction_pin: NO_PIN
  #disable_with_s0: false
  #s0_with_disable: true
  #spinup_ms: 1000
  #spindown_ms: 1000
  #tool_num: 0
  #speed_map: 0=0.000% 1000=100.000%
```

### Startup Messages

```text
$X
ok
$$
$22=1
$21=1
$20=1
$102=500.000
$101=800.000
$100=800.000
$112=3000.000
$111=3000.000
$110=3000.000
$122=100.000
$121=100.000
$120=100.000
$132=56.000
$131=300.000
$130=300.000
$10=1
$30=0
ok
```

### User Interface Software

WebUI, GSender

### What happened?

I have uploaded the most current Code from GitHub for the Fluid Dial and I have set the YAML file for the 6 Pack Controller IAW the Wiki for connecting the pendent to a 6-Pack Controller.  I have check continuity between all cabling and I still get nothing but NC on the Dial Screen.  Any help would be appreciated.

MitchBradley, 2024-04-10:
The latest version of the FluidDial code communicates at 1000000 baud.  Try changing the baud rate in the uart2 section.  Also the indentation is wrong - The line uart2: should not be indented

MitchBradley, 2024-04-10:
To get the startup messages you must type $SS.  $$ gets you the old-style Grbl settings.  You want startup messages, not Grbl settings.  There are probably error messages in those settings that would highlight the indentation problem.  Always look at startup messages when you have problems.

dustonanthony, 2024-04-10:
Thank you for the quick reply.  I fixed the indent on Uart2 and reuploaded the code.  Here is the start up messages.

[MSG:INFO: FluidNC v3.7.16 https://github.com/bdring/FluidNC]]
[MSG:INFO: Compiled with ESP32 SDK:v4.4.4]]
[MSG:INFO: Local filesystem type is littlefs]]
[MSG:INFO: Configuration file:Desktop_CNC_XYYZ.yaml]]
[MSG:WARN: Ignored key uart2]]
[MSG:WARN: Ignored key txd_pin]]
[MSG:WARN: Ignored key rxd_pin]]
[MSG:WARN: Ignored key rts_pin]]
[MSG:WARN: Ignored key cts_pin]]
[MSG:WARN: Ignored key baud]]
[MSG:WARN: Ignored key mode]]
[MSG:WARN: Ignored key motor1]]
[MSG:INFO: Machine Desktop CNC Router XYYZ]]
[MSG:INFO: Board 6 Pack]]
[MSG:ERR: UartChannel: missing uart2]]
[MSG:INFO: I2SO BCK:gpio.22 WS:gpio.17 DATA:gpio.21]]
[MSG:INFO: SPI SCK:gpio.18 MOSI:gpio.23 MISO:gpio.19]]
[MSG:INFO: SD Card cs_pin:gpio.5 detect:NO_PIN freq:8000000]]
[MSG:INFO: Stepping:I2S_stream Pulse:4us Dsbl Delay:0us Dir Delay:1us Idle Delay:255ms]]
[MSG:INFO: Axis count 3]]
[MSG:INFO: Axis X (150.000,450.000)]]
[MSG:INFO:   Motor0]]
[MSG:INFO:     tmc_2130 Step:I2SO.2 Dir:I2SO.1 CS:I2SO.3 Disable:I2SO.0 Index:-1 R:0.110]]
[MSG:INFO:  X Neg Limit gpio.33:low]]
[MSG:INFO: Axis Y (150.000,450.000)]]
[MSG:INFO:   Motor0]]
[MSG:INFO:     tmc_2130 Step:I2SO.10 Dir:I2SO.9 CS:I2SO.11 Disable:I2SO.8 Index:-1 R:0.110]]
[MSG:INFO:  Y Neg Limit gpio.35:low]]
[MSG:INFO: Axis Z (94.000,150.000)]]
[MSG:INFO:   Motor0]]
[MSG:INFO:     tmc_2130 Step:I2SO.13 Dir:I2SO.12 CS:I2SO.14 Disable:I2SO.15 Index:-1 R:0.110]]
[MSG:INFO:  Z Pos Limit gpio.34]]
[MSG:ERR: X Axis TMC driver not detected - expected 0x11 got 0x0]]
[MSG:ERR: Y Axis TMC driver not detected - expected 0x11 got 0x0]]
[MSG:ERR: Z Axis TMC driver not detected - expected 0x11 got 0x0]]
[MSG:INFO: Kinematic system: Cartesian]]
[MSG:INFO: Using spindle NoSpindle]]
[MSG:INFO: Connecting to STA SSID:2565_Cardigan_DR]]
[MSG:INFO: Connecting.]]
[MSG:INFO: Connecting..]]
[MSG:INFO: Connected - IP is 192.168.164.83]]
[MSG:INFO: WiFi on]]
[MSG:INFO: Start mDNS with hostname:http://fluidnc.local/]]
[MSG:INFO: SSDP Started]]
[MSG:INFO: HTTP started on port 80]]
[MSG:INFO: Telnet started on port 23]]
ok

MitchBradley, 2024-04-10:
```
[MSG:WARN: Ignored key uart2]]
```
Looks like you didn't fix the indentation.  uart2: should not be indented

MitchBradley, 2024-04-10:
It should look like this:
```
stepping:
  engine: I2S_STREAM
  idle_ms: 255
  pulse_us: 4
  dir_delay_us: 1
  disable_delay_us: 0
  
uart2:
  txd_pin: gpio.26
  rxd_pin: gpio.4
  rts_pin: NO_PIN
  cts_pin: NO_PIN
  baud: 1000000
  mode: 8N1
```

dustonanthony, 2024-04-10:
Here is the config file code.  Does this look correct now?

board: 6 Pack
name: Desktop CNC Router XYYZ
stepping:
  engine: I2S_STREAM
  idle_ms: 255
  pulse_us: 4
  dir_delay_us: 1
  disable_delay_us: 0
  
 uart2:
  txd_pin: gpio.26
  rxd_pin: gpio.4
  rts_pin: NO_PIN
  cts_pin: NO_PIN
  baud: 1000000
  mode: 8N1
  
  uart_channel2:
  uart_num: 2
  report_interval_ms: 75

axes:
  shared_stepper_disable_pin: NO_PIN
  x:
    steps_per_mm: 800.000
    max_rate_mm_per_min: 3000.000
    acceleration_mm_per_sec2: 100.000
    max_travel_mm: 300.000
    soft_limits: true
    homing:
      cycle: 2
      positive_direction: false
      mpos_mm: 150.000
      feed_mm_per_min: 100.000
      seek_mm_per_min: 800.000
      settle_ms: 500
      seek_scaler: 1.100
      feed_scaler: 1.100

    motor0:
      limit_neg_pin: gpio.33:low
      limit_pos_pin: NO_PIN
      limit_all_pin: NO_PIN
      hard_limits: true
      pulloff_mm: 1.000
      tmc_2130:
        cs_pin: i2so.3
        r_sense_ohms: 0.110
        run_amps: 0.750
        hold_amps: 0.750
        microsteps: 8
        stallguard: 0
        stallguard_debug: false
        toff_disable: 0
        toff_stealthchop: 5
        toff_coolstep: 3
        run_mode: CoolStep
        homing_mode: CoolStep
        use_enable: false
        step_pin: I2SO.2
        direction_pin: I2SO.1
        disable_pin: I2SO.0

  y:
    steps_per_mm: 800.000
    max_rate_mm_per_min: 3000.000
    acceleration_mm_per_sec2: 100.000
    max_travel_mm: 300.000
    soft_limits: true
    homing:
      cycle: 2
      positive_direction: false
      mpos_mm: 150.000
      feed_mm_per_min: 100.000
      seek_mm_per_min: 800.000
      settle_ms: 500
      seek_scaler: 1.100
      feed_scaler: 1.100

    motor0:
      limit_neg_pin: gpio.32:low
      limit_pos_pin: NO_PIN
      limit_all_pin: NO_PIN
      hard_limits: true
      pulloff_mm: 1.000
      tmc_2130:
        cs_pin: i2so.6
        r_sense_ohms: 0.110
        run_amps: 0.750
        hold_amps: 0.750
        microsteps: 16
        stallguard: 0
        stallguard_debug: false
        toff_disable: 0
        toff_stealthchop: 5
        toff_coolstep: 3
        run_mode: StealthChop
        homing_mode: StealthChop
        use_enable: false
        step_pin: I2SO.5
        direction_pin: I2SO.4
        disable_pin: I2SO.7
        
        motor1:
      limit_neg_pin: gpio.35:low
      limit_pos_pin: NO_PIN
      limit_all_pin: NO_PIN
      hard_limits: true
      pulloff_mm: 1.000
      tmc_2130:
        cs_pin: i2so.11
        r_sense_ohms: 0.110
        run_amps: 0.750
        hold_amps: 0.750
        microsteps: 16
        stallguard: 0
        stallguard_debug: false
        toff_disable: 0
        toff_stealthchop: 5
        toff_coolstep: 3
        run_mode: StealthChop
        homing_mode: StealthChop
        use_enable: false
        step_pin: I2SO.10
        direction_pin: I2SO.9
        disable_pin: I2SO.8

  z:
    steps_per_mm: 500.000
    max_rate_mm_per_min: 3000.000
    acceleration_mm_per_sec2: 100.000
    max_travel_mm: 56.000
    soft_limits: true
    homing:
      cycle: 1
      positive_direction: true
      mpos_mm: 150.000
      feed_mm_per_min: 100.000
      seek_mm_per_min: 800.000
      settle_ms: 500
      seek_scaler: 1.100
      feed_scaler: 1.100

    motor0:
      limit_neg_pin: NO_PIN
      limit_pos_pin: gpio.34
      limit_all_pin: NO_PIN
      hard_limits: true
      pulloff_mm: 1.000
      tmc_2130:
        cs_pin: i2so.14
        r_sense_ohms: 0.110
        run_amps: 0.750
        hold_amps: 0.750
        microsteps: 16
        stallguard: 0
        stallguard_debug: false
        toff_disable: 0
        toff_stealthchop: 5
        toff_coolstep: 3
        run_mode: StealthChop
        homing_mode: StealthChop
        use_enable: false
        step_pin: I2SO.13
        direction_pin: I2SO.12
        disable_pin: I2SO.15

i2so:
  bck_pin: gpio.22
  data_pin: gpio.21
  ws_pin: gpio.17

spi:
  miso_pin: gpio.19
  mosi_pin: gpio.23
  sck_pin: gpio.18

sdcard:
  card_detect_pin: NO_PIN
  cs_pin: gpio.5

control:
  safety_door_pin: NO_PIN
  reset_pin: NO_PIN
  feed_hold_pin: NO_PIN
  cycle_start_pin: NO_PIN
  macro0_pin: NO_PIN
  macro1_pin: NO_PIN
  macro2_pin: NO_PIN
  macro3_pin: NO_PIN

coolant:
  flood_pin: NO_PIN
  mist_pin: NO_PIN
  delay_ms: 0

probe:
  pin: NO_PIN
  check_mode_start: true

macros:
  startup_line0:
  startup_line1:
  macro0:
  macro1:
  macro2:
  macro3:

user_outputs:
  analog0_pin: NO_PIN
  analog1_pin: NO_PIN
  analog2_pin: NO_PIN
  analog3_pin: NO_PIN
  analog0_hz: 5000
  analog1_hz: 5000
  analog2_hz: 5000
  analog3_hz: 5000
  digital0_pin: NO_PIN
  digital1_pin: NO_PIN
  digital2_pin: NO_PIN
  digital3_pin: NO_PIN

start:
  must_home: false

#PWM:
  #pwm_hz: 5000
  #output_pin: gpio.26
  #enable_pin: gpio.4
  #direction_pin: NO_PIN
  #disable_with_s0: false
  #s0_with_disable: true
  #spinup_ms: 1000
  #spindown_ms: 1000
  #tool_num: 0
  #speed_map: 0=0.000% 1000=100.000%

dustonanthony, 2024-04-10:
Newest start up messages with the above code.

$SS
[MSG:INFO: FluidNC v3.7.16 https://github.com/bdring/FluidNC]]
[MSG:INFO: Compiled with ESP32 SDK:v4.4.4]]
[MSG:INFO: Local filesystem type is littlefs]]
[MSG:INFO: Configuration file:Desktop_CNC_XYYZ.yaml]]
[MSG:ERR: Skipping key uart2 indent 1 this indent 0]]
[MSG:ERR: Skipping key txd_pin indent 2 this indent 0]]
[MSG:ERR: Skipping key rxd_pin indent 2 this indent 0]]
[MSG:ERR: Skipping key rts_pin indent 2 this indent 0]]
[MSG:ERR: Skipping key cts_pin indent 2 this indent 0]]
[MSG:ERR: Skipping key baud indent 2 this indent 0]]
[MSG:ERR: Skipping key mode indent 2 this indent 0]]
[MSG:WARN: Ignored key motor1]]
[MSG:INFO: Machine Desktop CNC Router XYYZ]]
[MSG:INFO: Board 6 Pack]]

MitchBradley, 2024-04-10:
If you paste a config file without quoting it with triple backticks, GitHub will reformat it so it is impossible to tell if it is correct.

The startup messages will tell you if there was a syntax problem_

dustonanthony, 2024-04-10:
\\\board: 6 Pack
name: Desktop CNC Router XYYZ
stepping:
  engine: I2S_STREAM
  idle_ms: 255
  pulse_us: 4
  dir_delay_us: 1
  disable_delay_us: 0
  
 uart2:
  txd_pin: gpio.26
  rxd_pin: gpio.4
  rts_pin: NO_PIN
  cts_pin: NO_PIN
  baud: 1000000
  mode: 8N1
  
  uart_channel2:
  uart_num: 2
  report_interval_ms: 75

axes:
  shared_stepper_disable_pin: NO_PIN
  x:
    steps_per_mm: 800.000
    max_rate_mm_per_min: 3000.000
    acceleration_mm_per_sec2: 100.000
    max_travel_mm: 300.000
    soft_limits: true
    homing:
      cycle: 2
      positive_direction: false
      mpos_mm: 150.000
      feed_mm_per_min: 100.000
      seek_mm_per_min: 800.000
      settle_ms: 500
      seek_scaler: 1.100
      feed_scaler: 1.100

    motor0:
      limit_neg_pin: gpio.33:low
      limit_pos_pin: NO_PIN
      limit_all_pin: NO_PIN
      hard_limits: true
      pulloff_mm: 1.000
      tmc_2130:
        cs_pin: i2so.3
        r_sense_ohms: 0.110
        run_amps: 0.750
        hold_amps: 0.750
        microsteps: 8
        stallguard: 0
        stallguard_debug: false
        toff_disable: 0
        toff_stealthchop: 5
        toff_coolstep: 3
        run_mode: CoolStep
        homing_mode: CoolStep
        use_enable: false
        step_pin: I2SO.2
        direction_pin: I2SO.1
        disable_pin: I2SO.0

  y:
    steps_per_mm: 800.000
    max_rate_mm_per_min: 3000.000
    acceleration_mm_per_sec2: 100.000
    max_travel_mm: 300.000
    soft_limits: true
    homing:
      cycle: 2
      positive_direction: false
      mpos_mm: 150.000
      feed_mm_per_min: 100.000
      seek_mm_per_min: 800.000
      settle_ms: 500
      seek_scaler: 1.100
      feed_scaler: 1.100

    motor0:
      limit_neg_pin: gpio.32:low
      limit_pos_pin: NO_PIN
      limit_all_pin: NO_PIN
      hard_limits: true
      pulloff_mm: 1.000
      tmc_2130:
        cs_pin: i2so.6
        r_sense_ohms: 0.110
        run_amps: 0.750
        hold_amps: 0.750
        microsteps: 16
        stallguard: 0
        stallguard_debug: false
        toff_disable: 0
        toff_stealthchop: 5
        toff_coolstep: 3
        run_mode: StealthChop
        homing_mode: StealthChop
        use_enable: false
        step_pin: I2SO.5
        direction_pin: I2SO.4
        disable_pin: I2SO.7
        
        motor1:
      limit_neg_pin: gpio.35:low
      limit_pos_pin: NO_PIN
      limit_all_pin: NO_PIN
      hard_limits: true
      pulloff_mm: 1.000
      tmc_2130:
        cs_pin: i2so.11
        r_sense_ohms: 0.110
        run_amps: 0.750
        hold_amps: 0.750
        microsteps: 16
        stallguard: 0
        stallguard_debug: false
        toff_disable: 0
        toff_stealthchop: 5
        toff_coolstep: 3
        run_mode: StealthChop
        homing_mode: StealthChop
        use_enable: false
        step_pin: I2SO.10
        direction_pin: I2SO.9
        disable_pin: I2SO.8

  z:
    steps_per_mm: 500.000
    max_rate_mm_per_min: 3000.000
    acceleration_mm_per_sec2: 100.000
    max_travel_mm: 56.000
    soft_limits: true
    homing:
      cycle: 1
      positive_direction: true
      mpos_mm: 150.000
      feed_mm_per_min: 100.000
      seek_mm_per_min: 800.000
      settle_ms: 500
      seek_scaler: 1.100
      feed_scaler: 1.100

    motor0:
      limit_neg_pin: NO_PIN
      limit_pos_pin: gpio.34
      limit_all_pin: NO_PIN
      hard_limits: true
      pulloff_mm: 1.000
      tmc_2130:
        cs_pin: i2so.14
        r_sense_ohms: 0.110
        run_amps: 0.750
        hold_amps: 0.750
        microsteps: 16
        stallguard: 0
        stallguard_debug: false
        toff_disable: 0
        toff_stealthchop: 5
        toff_coolstep: 3
        run_mode: StealthChop
        homing_mode: StealthChop
        use_enable: false
        step_pin: I2SO.13
        direction_pin: I2SO.12
        disable_pin: I2SO.15

i2so:
  bck_pin: gpio.22
  data_pin: gpio.21
  ws_pin: gpio.17

spi:
  miso_pin: gpio.19
  mosi_pin: gpio.23
  sck_pin: gpio.18

sdcard:
  card_detect_pin: NO_PIN
  cs_pin: gpio.5

control:
  safety_door_pin: NO_PIN
  reset_pin: NO_PIN
  feed_hold_pin: NO_PIN
  cycle_start_pin: NO_PIN
  macro0_pin: NO_PIN
  macro1_pin: NO_PIN
  macro2_pin: NO_PIN
  macro3_pin: NO_PIN

coolant:
  flood_pin: NO_PIN
  mist_pin: NO_PIN
  delay_ms: 0

probe:
  pin: NO_PIN
  check_mode_start: true

macros:
  startup_line0:
  startup_line1:
  macro0:
  macro1:
  macro2:
  macro3:

user_outputs:
  analog0_pin: NO_PIN
  analog1_pin: NO_PIN
  analog2_pin: NO_PIN
  analog3_pin: NO_PIN
  analog0_hz: 5000
  analog1_hz: 5000
  analog2_hz: 5000
  analog3_hz: 5000
  digital0_pin: NO_PIN
  digital1_pin: NO_PIN
  digital2_pin: NO_PIN
  digital3_pin: NO_PIN

start:
  must_home: false

#PWM:
  #pwm_hz: 5000
  #output_pin: gpio.26
  #enable_pin: gpio.4
  #direction_pin: NO_PIN
  #disable_with_s0: false
  #s0_with_disable: true
  #spinup_ms: 1000
  #spindown_ms: 1000
  #tool_num: 0
  #speed_map: 0=0.000% 1000=100.000%
  \\\

MitchBradley, 2024-04-10:
Still not quoted correctly.  Add a .txt extension and attach the file

dustonanthony, 2024-04-10:
[Desktop_CNC_XYYZ.zip](https://github.com/bdring/FluidNC/files/14937175/Desktop_CNC_XYYZ.zip)

dustonanthony, 2024-04-10:
[Desktop_CNC_XYYZ (2).txt](https://github.com/bdring/FluidNC/files/14937189/Desktop_CNC_XYYZ.2.txt)

bdring, 2024-04-10:
Try this. Don't change the indent.

```yaml
 uart2:
  txd_pin: gpio.26
  rxd_pin: gpio.4
  rts_pin: NO_PIN
  cts_pin: NO_PIN
  baud: 1000000
  mode: 8N1
  
uart_channel2:
  uart_num: 2
  report_interval_ms: 75
```

See how easy it is to read when done right?

bdring, 2024-04-10:
http://wiki.fluidnc.com/en/support/requesting_help#pasting-the-config-file-on-github

MitchBradley, 2024-04-10:
The .txt file and the yaml file inside the zip are all one line.  What program are you using to edit it?  You need to use a dumb text editor like notepad, or else a programmer's editor.  Do not use a WYSWIG editor for human documents.

bdring, 2024-04-10:
http://wiki.fluidnc.com/en/support/troubleshooting_config_files#editing-the-file

dustonanthony, 2024-04-10:
[image]
Here is what the YAML file looks like in my NotePad++

bdring, 2024-04-10:
Looks better, but what do the startup messages say?

MitchBradley, 2024-04-10:
There is a space before uart2.  Remove it

dustonanthony, 2024-04-10:
Removed the space and here is what the start up messages are.

[MSG:INFO: FluidNC v3.7.16 https://github.com/bdring/FluidNC]]
[MSG:INFO: Compiled with ESP32 SDK:v4.4.4]]
[MSG:INFO: Local filesystem type is littlefs]]
[MSG:INFO: Configuration file:Desktop_CNC_XYYZ.yaml]]
[MSG:ERR: Skipping key uart2 indent 1 this indent 0]]
[MSG:ERR: Skipping key txd_pin indent 2 this indent 0]]
[MSG:ERR: Skipping key rxd_pin indent 2 this indent 0]]
[MSG:ERR: Skipping key rts_pin indent 2 this indent 0]]
[MSG:ERR: Skipping key cts_pin indent 2 this indent 0]]
[MSG:ERR: Skipping key baud indent 2 this indent 0]]
[MSG:ERR: Skipping key mode indent 2 this indent 0]]
[MSG:WARN: Ignored key motor1]]
[MSG:INFO: Machine Desktop CNC Router XYYZ]]
[MSG:INFO: Board 6 Pack]]
[MSG:ERR: UartChannel: missing uart2]]
[MSG:INFO: I2SO BCK:gpio.22 WS:gpio.17 DATA:gpio.21]]
[MSG:INFO: SPI SCK:gpio.18 MOSI:gpio.23 MISO:gpio.19]]
[MSG:INFO: SD Card cs_pin:gpio.5 detect:NO_PIN freq:8000000]]
[MSG:INFO: Stepping:I2S_stream Pulse:4us Dsbl Delay:0us Dir Delay:1us Idle Delay:255ms]]
[MSG:INFO: Axis count 3]]
[MSG:INFO: Axis X (150.000,450.000)]]
[MSG:INFO:   Motor0]]
[MSG:INFO:     tmc_2130 Step:I2SO.2 Dir:I2SO.1 CS:I2SO.3 Disable:I2SO.0 Index:-1 R:0.110]]
[MSG:INFO:  X Neg Limit gpio.33:low]]
[MSG:INFO: Axis Y (150.000,450.000)]]
[MSG:INFO:   Motor0]]
[MSG:INFO:     tmc_2130 Step:I2SO.10 Dir:I2SO.9 CS:I2SO.11 Disable:I2SO.8 Index:-1 R:0.110]]
[MSG:INFO:  Y Neg Limit gpio.35:low]]
[MSG:INFO: Axis Z (94.000,150.000)]]
[MSG:INFO:   Motor0]]
[MSG:INFO:     tmc_2130 Step:I2SO.13 Dir:I2SO.12 CS:I2SO.14 Disable:I2SO.15 Index:-1 R:0.110]]
[MSG:INFO:  Z Pos Limit gpio.34]]
[MSG:ERR: X Axis TMC driver not detected - expected 0x11 got 0x0]]
[MSG:ERR: Y Axis TMC driver not detected - expected 0x11 got 0x0]]
[MSG:ERR: Z Axis TMC driver not detected - expected 0x11 got 0x0]]
[MSG:INFO: Kinematic system: Cartesian]]
[MSG:INFO: Using spindle NoSpindle]]
[MSG:INFO: Connecting to STA SSID:2565_Cardigan_DR]]
[MSG:INFO: Connecting.]]
[MSG:INFO: Connecting..]]
[MSG:INFO: Connected - IP is 192.168.164.83]]
[MSG:INFO: WiFi on]]
[MSG:INFO: Start mDNS with hostname:http://fluidnc.local/]]
[MSG:INFO: SSDP Started]]
[MSG:INFO: HTTP started on port 80]]
[MSG:INFO: Telnet started on port 23]]

bdring, 2024-04-10:
OK, Stop.

Let me make a file for you.

MitchBradley, 2024-04-10:
```
[MSG:INFO: Configuration file:Desktop_CNC_XYYZ.yaml]]
[MSG:ERR: Skipping key uart2 indent 1 this indent 0]]
```
It still says "indent 1" so I think you didn't save the file, or didn't upload it, or something

bdring, 2024-04-10:
Try this. It loads for me. Be very, very careful when pasting it into your editor to make sure you get all characters and not add some extra spaces.

```yaml
board: 6 Pack
name: Desktop CNC Router XYYZ
stepping:
  engine: I2S_STREAM
  idle_ms: 255
  pulse_us: 4
  dir_delay_us: 1
  disable_delay_us: 0
  
uart2:
  txd_pin: gpio.26
  rxd_pin: gpio.4
  rts_pin: NO_PIN
  cts_pin: NO_PIN
  baud: 1000000
  mode: 8N1
  
uart_channel2:
  uart_num: 2
  report_interval_ms: 75

axes:
  shared_stepper_disable_pin: NO_PIN
  x:
    steps_per_mm: 800.000
    max_rate_mm_per_min: 3000.000
    acceleration_mm_per_sec2: 100.000
    max_travel_mm: 300.000
    soft_limits: true
    homing:
      cycle: 2
      positive_direction: false
      mpos_mm: 150.000
      feed_mm_per_min: 100.000
      seek_mm_per_min: 800.000
      settle_ms: 500
      seek_scaler: 1.100
      feed_scaler: 1.100

    motor0:
      limit_neg_pin: gpio.33:low
      limit_pos_pin: NO_PIN
      limit_all_pin: NO_PIN
      hard_limits: true
      pulloff_mm: 1.000
      tmc_2130:
        cs_pin: i2so.3
        r_sense_ohms: 0.110
        run_amps: 0.750
        hold_amps: 0.750
        microsteps: 8
        stallguard: 0
        stallguard_debug: false
        toff_disable: 0
        toff_stealthchop: 5
        toff_coolstep: 3
        run_mode: CoolStep
        homing_mode: CoolStep
        use_enable: false
        step_pin: I2SO.2
        direction_pin: I2SO.1
        disable_pin: I2SO.0

  y:
    steps_per_mm: 800.000
    max_rate_mm_per_min: 3000.000
    acceleration_mm_per_sec2: 100.000
    max_travel_mm: 300.000
    soft_limits: true
    homing:
      cycle: 2
      positive_direction: false
      mpos_mm: 150.000
      feed_mm_per_min: 100.000
      seek_mm_per_min: 800.000
      settle_ms: 500
      seek_scaler: 1.100
      feed_scaler: 1.100

    motor0:
      limit_neg_pin: gpio.32:low
      limit_pos_pin: NO_PIN
      limit_all_pin: NO_PIN
      hard_limits: true
      pulloff_mm: 1.000
      tmc_2130:
        cs_pin: i2so.6
        r_sense_ohms: 0.110
        run_amps: 0.750
        hold_amps: 0.750
        microsteps: 16
        stallguard: 0
        stallguard_debug: false
        toff_disable: 0
        toff_stealthchop: 5
        toff_coolstep: 3
        run_mode: StealthChop
        homing_mode: StealthChop
        use_enable: false
        step_pin: I2SO.5
        direction_pin: I2SO.4
        disable_pin: I2SO.7
        
    motor1:
      limit_neg_pin: gpio.35:low
      limit_pos_pin: NO_PIN
      limit_all_pin: NO_PIN
      hard_limits: true
      pulloff_mm: 1.000
      tmc_2130:
        cs_pin: i2so.11
        r_sense_ohms: 0.110
        run_amps: 0.750
        hold_amps: 0.750
        microsteps: 16
        stallguard: 0
        stallguard_debug: false
        toff_disable: 0
        toff_stealthchop: 5
        toff_coolstep: 3
        run_mode: StealthChop
        homing_mode: StealthChop
        use_enable: false
        step_pin: I2SO.10
        direction_pin: I2SO.9
        disable_pin: I2SO.8

  z:
    steps_per_mm: 500.000
    max_rate_mm_per_min: 3000.000
    acceleration_mm_per_sec2: 100.000
    max_travel_mm: 56.000
    soft_limits: true
    homing:
      cycle: 1
      positive_direction: true
      mpos_mm: 150.000
      feed_mm_per_min: 100.000
      seek_mm_per_min: 800.000
      settle_ms: 500
      seek_scaler: 1.100
      feed_scaler: 1.100

    motor0:
      limit_neg_pin: NO_PIN
      limit_pos_pin: gpio.34
      limit_all_pin: NO_PIN
      hard_limits: true
      pulloff_mm: 1.000
      tmc_2130:
        cs_pin: i2so.14
        r_sense_ohms: 0.110
        run_amps: 0.750
        hold_amps: 0.750
        microsteps: 16
        stallguard: 0
        stallguard_debug: false
        toff_disable: 0
        toff_stealthchop: 5
        toff_coolstep: 3
        run_mode: StealthChop
        homing_mode: StealthChop
        use_enable: false
        step_pin: I2SO.13
        direction_pin: I2SO.12
        disable_pin: I2SO.15

i2so:
  bck_pin: gpio.22
  data_pin: gpio.21
  ws_pin: gpio.17

spi:
  miso_pin: gpio.19
  mosi_pin: gpio.23
  sck_pin: gpio.18

sdcard:
  card_detect_pin: NO_PIN
  cs_pin: gpio.5

control:
  safety_door_pin: NO_PIN
  reset_pin: NO_PIN
  feed_hold_pin: NO_PIN
  cycle_start_pin: NO_PIN
  macro0_pin: NO_PIN
  macro1_pin: NO_PIN
  macro2_pin: NO_PIN
  macro3_pin: NO_PIN

coolant:
  flood_pin: NO_PIN
  mist_pin: NO_PIN
  delay_ms: 0

probe:
  pin: NO_PIN
  check_mode_start: true

macros:
  startup_line0:
  startup_line1:
  macro0:
  macro1:
  macro2:
  macro3:

user_outputs:
  analog0_pin: NO_PIN
  analog1_pin: NO_PIN
  analog2_pin: NO_PIN
  analog3_pin: NO_PIN
  analog0_hz: 5000
  analog1_hz: 5000
  analog2_hz: 5000
  analog3_hz: 5000
  digital0_pin: NO_PIN
  digital1_pin: NO_PIN
  digital2_pin: NO_PIN
  digital3_pin: NO_PIN

start:
  must_home: false

#PWM:
  #pwm_hz: 5000
  #output_pin: gpio.26
  #enable_pin: gpio.4
  #direction_pin: NO_PIN
  #disable_with_s0: false
  #s0_with_disable: true
  #spinup_ms: 1000
  #spindown_ms: 1000
  #tool_num: 0
  #speed_map: 0=0.000% 1000=100.000%
```

dustonanthony, 2024-04-10:
Uploaded your config file you provided Bart and here is the startup messages.

[MSG:INFO: FluidNC v3.7.16 https://github.com/bdring/FluidNC]]
[MSG:INFO: Compiled with ESP32 SDK:v4.4.4]]
[MSG:INFO: Local filesystem type is littlefs]]
[MSG:INFO: Configuration file:config.yaml]]
[MSG:INFO: Machine Desktop CNC Router XYYZ]]
[MSG:INFO: Board 6 Pack]]
[MSG:INFO: UART2 Tx:gpio.26 Rx:gpio.4 RTS:NO_PIN Baud:1000000]]
[MSG:INFO: uart_channel2 created at report interval: 75]]
[MSG:INFO: I2SO BCK:gpio.22 WS:gpio.17 DATA:gpio.21]]
[MSG:INFO: SPI SCK:gpio.18 MOSI:gpio.23 MISO:gpio.19]]
[MSG:INFO: SD Card cs_pin:gpio.5 detect:NO_PIN freq:8000000]]
[MSG:INFO: Stepping:I2S_stream Pulse:4us Dsbl Delay:0us Dir Delay:1us Idle Delay:255ms]]
[MSG:INFO: Axis count 3]]
[MSG:INFO: Axis X (150.000,450.000)]]
[MSG:INFO:   Motor0]]
[MSG:INFO:     tmc_2130 Step:I2SO.2 Dir:I2SO.1 CS:I2SO.3 Disable:I2SO.0 Index:-1 R:0.110]]
[MSG:INFO:  X Neg Limit gpio.33:low]]
[MSG:INFO: Axis Y (150.000,450.000)]]
[MSG:INFO:   Motor0]]
[MSG:INFO:     tmc_2130 Step:I2SO.5 Dir:I2SO.4 CS:I2SO.6 Disable:I2SO.7 Index:-1 R:0.110]]
[MSG:INFO:  Y Neg Limit gpio.32:low]]
[MSG:INFO:   Motor1]]
[MSG:INFO:     tmc_2130 Step:I2SO.10 Dir:I2SO.9 CS:I2SO.11 Disable:I2SO.8 Index:-1 R:0.110]]
[MSG:INFO:  Y2 Neg Limit gpio.35:low]]
[MSG:INFO: Axis Z (94.000,150.000)]]
[MSG:INFO:   Motor0]]
[MSG:INFO:     tmc_2130 Step:I2SO.13 Dir:I2SO.12 CS:I2SO.14 Disable:I2SO.15 Index:-1 R:0.110]]
[MSG:INFO:  Z Pos Limit gpio.34]]
[MSG:INFO: X Axis driver test passed]]
[MSG:INFO: Y Axis driver test passed]]
[MSG:INFO: Y2 Axis driver test passed]]
[MSG:INFO: Z Axis driver test passed]]
[MSG:INFO: Kinematic system: Cartesian]]
[MSG:INFO: Using spindle NoSpindle]]
[MSG:INFO: Connecting to STA SSID:2565_Cardigan_DR]]
[MSG:INFO: Connecting.]]
[MSG:INFO: Connecting..]]
[MSG:INFO: Connected - IP is 192.168.164.83]]
[MSG:INFO: WiFi on]]
[MSG:INFO: Start mDNS with hostname:http://fluidnc.local/]]
[MSG:INFO: SSDP Started]]
[MSG:INFO: HTTP started on port 80]]
[MSG:INFO: Telnet started on port 23]]

dustonanthony, 2024-04-10:
Main screen looks like this.
[image]

dustonanthony, 2024-04-10:
Internal menus look like this now.
[image]

bdring, 2024-04-10:
You also need to **Upload Filesystem Image**. The easiest way to do it is here.

[image]

dustonanthony, 2024-04-10:
Thank you very much for the Info. I will try that as soon as I get back home.

dustonanthony, 2024-04-10:
All seems to be working now.

closed 2024-04-10 by dustonanthony

kghxkgh, 2024-04-11:
Thankyou. This helped with mine too. I just upgraded the firmware on both ends but neglected to update the baud rate in the config.yaml.