bdring/FluidNC #1186: MKS DLC32, PWM Spindle runs for a second on board reset.

FrankHovis, 2024-04-10:
### Wiki Search Terms

MKS DLC32 spindle, ESP32 reset

### Controller Board

MKS DLC32

### Machine Description

Home made "CNC3018" copy.

### Configuration file

```yaml
board: MKS-DLC32 V2.1
name: Krap CNC
meta: (13.07.2023)

kinematics:
  Cartesian:

stepping:
  engine: I2S_STATIC
  idle_ms: 25
  pulse_us: 4
  dir_delay_us: 1
  disable_delay_us: 0
axes:
  shared_stepper_disable_pin: I2SO.0
  x:
    steps_per_mm: 800
    max_rate_mm_per_min: 1000
    acceleration_mm_per_sec2: 40
    max_travel_mm: 400
    soft_limits: false
    homing:
      cycle: 2
      positive_direction: false
      mpos_mm: 0.000
      feed_mm_per_min: 300.000
      seek_mm_per_min: 5000.000
      settle_ms: 500
      seek_scaler: 1.100
      feed_scaler: 1.100

    motor0:
      limit_neg_pin: gpio.36:low
      hard_limits: false
      pulloff_mm: 2.000
      stepstick:
        step_pin: I2SO.1
        direction_pin: I2SO.2

  y:
    steps_per_mm: 800
    max_rate_mm_per_min: 1000
    acceleration_mm_per_sec2: 40
    max_travel_mm: 200
    soft_limits: false
    homing:
      cycle: 2
      positive_direction: false
      mpos_mm: 0.000
      feed_mm_per_min: 300.000
      seek_mm_per_min: 5000.000
      settle_ms: 500
      seek_scaler: 1.100
      feed_scaler: 1.100

    motor0:
      limit_neg_pin: gpio.35:low
      hard_limits: false
      pulloff_mm: 2.000
      stepstick:
        step_pin: I2SO.5
        direction_pin: I2SO.6:low

  z:
    steps_per_mm: 800
    max_rate_mm_per_min: 800
    acceleration_mm_per_sec2: 20
    max_travel_mm: 80.000
    soft_limits: false
    homing:
      cycle: 1
      positive_direction: true
      mpos_mm: 0.000
      feed_mm_per_min: 150.000
      seek_mm_per_min: 500.000
      settle_ms: 500
      seek_scaler: 1.100
      feed_scaler: 1.100

    motor0:
      limit_neg_pin: gpio.34:low
      hard_limits: false
      pulloff_mm: 1.000
      stepstick:
        step_pin: I2SO.3
        direction_pin: I2SO.4

i2so:
  bck_pin: gpio.16
  data_pin: gpio.21
  ws_pin: gpio.17

spi:
  miso_pin: gpio.12
  mosi_pin: gpio.13
  sck_pin: gpio.14

sdcard:
  cs_pin: gpio.15
  card_detect_pin: NO_PIN

control:
  safety_door_pin: NO_PIN
  reset_pin: NO_PIN
  feed_hold_pin: NO_PIN
  cycle_start_pin: NO_PIN
  macro0_pin: gpio.33:low:pu
  macro1_pin: NO_PIN
  macro2_pin: NO_PIN
  macro3_pin: NO_PIN

macros:
  startup_line0:
  startup_line1:
  macro0: 
  macro1: 
  macro2:
  macro3:

coolant:
  flood_pin: NO_PIN
  mist_pin: NO_PIN
  delay_ms: 0

probe:
  pin: gpio.22:low:pu
  check_mode_start: true

pwm:
  pwm_hz: 500
  #L on Beeper / IN on TTL
  output_pin: gpio.32
  enable_pin: I2SO.7
  disable_with_s0: false
  s0_with_disable: true
  tool_num: 0
  spinup_ms: 1000
  spindown_ms: 1000
  speed_map: 0=0.000% 10000=100.000%
# 135=0mA 270=5mA 400=10mA 700=16mA

user_outputs:
  analog0_pin: NO_PIN
  analog1_pin: NO_PIN
  analog2_pin: NO_PIN
  analog3_pin: NO_PIN
  analog0_hz: 5000
  analog1_hz: 5000
  analog2_hz: 5000
  analog3_hz: 5000
  digital0_pin: NO_PIN
  digital1_pin: NO_PIN
  digital2_pin: NO_PIN
  digital3_pin: NO_PIN

start:
  must_home: false

# 5,18,19,22,23,25,26,27,32,33,39,I2SO.7
# SDA 0 / SCL 4
```

### Startup Messages

```text
[MSG:INFO: FluidNC v3.7.16 https://github.com/bdring/FluidNC]
[MSG:INFO: Compiled with ESP32 SDK:v4.4.4]
[MSG:INFO: Local filesystem type is littlefs]
[MSG:INFO: Configuration file:config.yaml]
[MSG:INFO: Machine Krap CNC]
[MSG:INFO: Board MKS-DLC32 V2.1]
[MSG:INFO: I2SO BCK:gpio.16 WS:gpio.17 DATA:gpio.21]
[MSG:INFO: SPI SCK:gpio.14 MOSI:gpio.13 MISO:gpio.12]
[MSG:INFO: SD Card cs_pin:gpio.15 detect:NO_PIN freq:8000000]
[MSG:INFO: Stepping:I2S_static Pulse:4us Dsbl Delay:0us Dir Delay:1us Idle Delay:25ms]
[MSG:INFO: Axis count 3]
[MSG:INFO: Shared stepper disable I2SO.0]
[MSG:INFO: Axis X (0.000,400.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     stepstick Step:I2SO.1 Dir:I2SO.2 Disable:NO_PIN]
[MSG:INFO:  X Neg Limit gpio.36:low]
[MSG:INFO: Axis Y (0.000,200.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     stepstick Step:I2SO.5 Dir:I2SO.6:low Disable:NO_PIN]
[MSG:INFO:  Y Neg Limit gpio.35:low]
[MSG:INFO: Axis Z (-80.000,0.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     stepstick Step:I2SO.3 Dir:I2SO.4 Disable:NO_PIN]
[MSG:INFO:  Z Neg Limit gpio.34:low]
[MSG:INFO: macro0_pin gpio.33:low:pu]
[MSG:INFO: Kinematic system: Cartesian]
[MSG:INFO: PWM Spindle Ena:I2SO.7 Out:gpio.32 Dir:NO_PIN Freq:500Hz Period:131071]
[MSG:INFO: Using spindle PWM]
```

### What happened?

When the board is reset, either by hardware reset pin or CTRL-R in Fluidterm, the spindle runs for a second or so while the ESP32 is resetting. It seems to start before the ESP32 gives its reset diagnostic message.  This doesn't happen on a power-on reset.

MitchBradley, 2024-04-10:
It is very difficult to control the state of I2SO pins during reset.  Bart's controllers have some tricky hardware to try and keep them at 0 then, but it doesn't always work.  I doubt that MKS  DLC32 even tries.  You might need to use a different pin for the spindle enable.

bdring, 2024-04-10:
http://wiki.fluidnc.com/en/config/config_IO#i2so-section

http://wiki.fluidnc.com/en/hardware/controller_design_guidelines#is2o-chips