bdring/FluidNC #1187: Probe not working on 6x board or Wemos D1

lytke, 2024-04-11:
### Wiki Search Terms

fluidnc probe not working 
See also issue #1180

### Controller Board

Issue exist with both 6x and Wemos D1 boards

### Machine Description

QueenAnt Pro w/HBS57H closed loop controllers and FluidNC 6x board.
Tried with Wemos D1 controller not connected to any machine to isolate issue.
- different ports tried without luck, same result - probe inresponsive.

### Configuration file

```yaml
6x config
----------------
board: 6x
name: 6x Default
stepping:
  engine: I2S_STATIC
  idle_ms: 255
  pulse_us: 4
  dir_delay_us: 5
  disable_delay_us: 5
axes:
  shared_stepper_disable_pin: i2so.0
  x:
    steps_per_mm: 162
    max_rate_mm_per_min: 5000
    acceleration_mm_per_sec2: 15
    max_travel_mm: 700
    soft_limits: false
    homing:
      cycle: 2
      positive_direction: false
      mpos_mm: 1
      feed_mm_per_min: 100
      seek_mm_per_min: 800
      settle_ms: 500
      seek_scaler: 1.1
      feed_scaler: 1.1
    motor0:
      limit_neg_pin: gpio.32:pu:low
      limit_pos_pin: gpio.2:pu:low
      # limit_all_pin: gpio.2:pu:low
      hard_limits: true
      pulloff_mm: 1
      standard_stepper:
        step_pin: i2so.2
        direction_pin: i2so.1
        # disable_pin: i2so.0
  y:
    steps_per_mm: 162
    max_rate_mm_per_min: 5000
    acceleration_mm_per_sec2: 15
    max_travel_mm: 800
    soft_limits: false
    homing:
      cycle: 2
      positive_direction: false
      mpos_mm: 1
      feed_mm_per_min: 100
      seek_mm_per_min: 800
      settle_ms: 500
      seek_scaler: 1.1
      feed_scaler: 1.1
    motor0:
      limit_neg_pin: NO_PIN
      limit_pos_pin: NO_PIN
      limit_all_pin: gpio.26:pu:low
      hard_limits: true
      pulloff_mm: 1
      standard_stepper:
        step_pin: i2so.5
        direction_pin: i2so.4
        disable_pin: i2so.7
    motor1:
      limit_all_pin: NO_PIN
      limit_neg_pin: NO_PIN
      limit_pos_pin: NO_PIN
      hard_limits: true
      pulloff_mm: 1
      standard_stepper:
        step_pin: i2so.13
        direction_pin: i2so.12
        disable_pin: i2so.15
  z:
    steps_per_mm: 320
    max_rate_mm_per_min: 3000
    acceleration_mm_per_sec2: 10
    max_travel_mm: 150
    soft_limits: false
    homing:
      cycle: 1
      positive_direction: true
      mpos_mm: 150
      feed_mm_per_min: 100
      seek_mm_per_min: 800
      settle_ms: 500
      seek_scaler: 1.1
      feed_scaler: 1.1
    motor0:
      limit_neg_pin: NO_PIN
      limit_pos_pin: NO_PIN
      limit_all_pin: gpio.33:pu:low
      hard_limits: true
      pulloff_mm: 1
      standard_stepper:
        step_pin: i2so.10
        direction_pin: i2so.9
        disable_pin: i2so.8
i2so:
  bck_pin: gpio.22
  data_pin: gpio.21
  ws_pin: gpio.17
spi:
  miso_pin: gpio.19
  mosi_pin: gpio.23
  sck_pin: gpio.18
sdcard:
  card_detect_pin: NO_PIN
  cs_pin: gpio.5
control:
  safety_door_pin: NO_PIN
  reset_pin: gpio.35:low
  feed_hold_pin: gpio.36:low
probe:
  pin: gpio.34:low
  toolsetter_pin: NO_PIN
  check_mode_start: true
10V:
  forward_pin: gpio.15
  reverse_pin: gpio.14
  output_pin: gpio.13
  pwm_hz: 5000
  spinup_ms: 0
  spindown_ms: 0
  enable_pin: NO_PIN
  direction_pin: NO_PIN
  disable_with_s0: false
  s0_with_disable: true
  # speed_map: 0=0% 1000=100%
  speed_map: 0=0.000% 1000=0.000% 24000=100.000%
  off_on_alarm: false
  tool_num: 0
start:
  must_home: false
  deactivate_parking: false
coolant:
  flood_pin: gpio.12
  mist_pin: gpio.4
  delay_ms: 0
Laser:
  speed_map: 0=0% 1000=100%
  output_pin: gpio.25
  enable_pin: gpio.27
  tool_num: 100

Wemos D1 config:
-----------------
name: Wemos D1
stepping:
  engine: RMT
  idle_ms: 250
  pulse_us: 4
  dir_delay_us: 0
  disable_delay_us: 0
probe:
  pin: gpio.4:pu:low
  toolsetter_pin: NO_PIN
  check_mode_start: true
control:
  reset_pin: NO_PIN
  feed_hold_pin: NO_PIN
  fault_pin: NO_PIN
  estop_pin: NO_PIN
  safety_door_pin: NO_PIN
  cycle_start_pin: NO_PIN
  macro0_pin: NO_PIN
  macro1_pin: NO_PIN
  macro2_pin: NO_PIN
  macro3_pin: NO_PIN
```

### Startup Messages

```text
[MSG:RST]
[MSG:INFO: FluidNC v3.7.15 https://github.com/bdring/FluidNC]
[MSG:INFO: Compiled with ESP32 SDK:v4.4.4]
[MSG:INFO: Local filesystem type is littlefs]
[MSG:INFO: Configuration file:config.yaml]
[MSG:INFO: Machine 6x Default]
[MSG:INFO: Board 6x]
[MSG:INFO: I2SO BCK:gpio.22 WS:gpio.17 DATA:gpio.21]
[MSG:INFO: SPI SCK:gpio.18 MOSI:gpio.23 MISO:gpio.19]
[MSG:INFO: SD Card cs_pin:gpio.5 detect:NO_PIN freq:8000000]
[MSG:INFO: Stepping:I2S_static Pulse:4us Dsbl Delay:5us Dir Delay:5us Idle Delay:255ms]
[MSG:INFO: Axis count 3]
[MSG:INFO: Shared stepper disable I2SO.0]
[MSG:INFO: Axis X (1.000,701.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     standard_stepper Step:I2SO.2 Dir:I2SO.1 Disable:NO_PIN]
[MSG:INFO:  X Neg Limit gpio.32:low:pu]
[MSG:INFO:  X Pos Limit gpio.2:low:pu]
[MSG:INFO: Axis Y (1.000,801.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     standard_stepper Step:I2SO.5 Dir:I2SO.4 Disable:I2SO.7]
[MSG:INFO:  Y All Limit gpio.26:low:pu]
[MSG:INFO:   Motor1]
[MSG:INFO:     standard_stepper Step:I2SO.13 Dir:I2SO.12 Disable:I2SO.15]
[MSG:INFO: Axis Z (0.000,150.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     standard_stepper Step:I2SO.10 Dir:I2SO.9 Disable:I2SO.8]
[MSG:INFO:  Z All Limit gpio.33:low:pu]
[MSG:INFO: reset_pin gpio.35:low]
[MSG:INFO: feed_hold_pin gpio.36:low]
[MSG:INFO: Kinematic system: Cartesian]
[MSG:INFO: 10V Spindle Ena:NO_PIN Out:gpio.13 Dir:NO_PIN Fwd:gpio.15 Rev:gpio.14 Freq:5000Hz Period:8191]
[MSG:INFO: Laser Ena:gpio.27 Out:gpio.25 Freq:5000Hz Period:8191]
[MSG:INFO: Using spindle 10V]
[MSG:INFO: Flood coolant gpio.12]
[MSG:INFO: Mist coolant gpio.4]
[MSG:INFO: Probe Pin: gpio.34:low]
[MSG:INFO: Connecting to STA SSID:xxxxxx]
[MSG:INFO: Connecting.]
[MSG:INFO: Connecting..]
[MSG:INFO: Connected - IP is 192.168.1.65]
[MSG:INFO: WiFi on]
[MSG:INFO: Start mDNS with hostname:http://fluidnc.local/]
[MSG:INFO: SSDP Started]
[MSG:INFO: HTTP started on port 80]

Grbl 3.7 [FluidNC v3.7.15 (wifi) '$' for help]
ok
[VER:3.7 FluidNC v3.7.15:]
[OPT:MPHS]
[MSG:Machine: 6x Default]
[MSG:Mode=STA:SSID=xxxxxx:Status=Connected:IP=192.168.1.65:MAC=D8-13-2A-75-7A-E4]
ok

Grbl 3.7 [FluidNC v3.7.15 (wifi) '$' for help]
```

### User Interface Software

gsender, openbuilds control, UGS

### What happened?

Probe does simply not work, not reporting connected or not. Its a simple aligator clip attached to router bit and an aliminium plate.

When testing with wemos, I just used a simple jumper wire. Using same port as reset input works excelent but when changing same port to probe and restarting it turns unresponsive.

lytke, 2024-04-11:
[image]
[image]

lytke, 2024-04-11:
@bdring let me know if more details or testing is required.

bdring, 2024-04-11:
I wanted to do some testing with your controller type and your config.

Some recent change broke the reading of the probe pins. It worked a little differently from other switches since the original Grbl days so the issues was not initially noticed. We have a pull request that should fix it.

https://github.com/bdring/FluidNC/pull/1181

You can compile the code from that PR. It works for me with your config. The PR should be in the next release. It changed a few other things, so some additional testing needs to be dome before the release.

closed 2024-04-25 by bdring

lytke, 2024-04-29:
Upgraded to https://github.com/bdring/FluidNC/releases/tag/v3.7.17 today and found that issue with probe was solved. Thank you to @bdring and @MitchBradley.