bdring/FluidNC #1191: Problem: XPRO v5 with Laser Head. Relay not switch on and PWM always 4.96V

slvv26, 2024-04-20:
### Wiki Search Terms

I have learned sections - Quick Start, Spindle, Toolhead, Relay on https://github.com/Spark-Concepts/xPro-V5/wiki
Not sure they considering FluidNC version of firmware. Unfortunately, didn't find any notice about.

Also, I have learned section - Spindles on http://wiki.fluidnc.com/en/config/config_spindles

### Controller Board

XPRO v5

### Machine Description

UltimateBee 1500x1500, TMC5160, dual Y motors, Huanyang VFD 2.2kW Spindle, LT-K40 LaserTree Head.

### Input Circuits

```yaml
I'm using that schema to connect laser head because it not have "enable" pin.

https://github.com/Spark-Concepts/xPro-V5/raw/main/images/LasPWM_3.jpg
```

### Configuration file

```yaml
board: XPro V5
name: xPro-V5 UltimateBee 1500x1500
stepping:
  engine: RMT
  idle_ms: 255
  pulse_us: 4
  dir_delay_us: 1
  disable_delay_us: 0

axes:
  shared_stepper_disable_pin: NO_PIN
  x:
    steps_per_mm: 320.095
    max_rate_mm_per_min: 2500.000
    acceleration_mm_per_sec2: 100.000
    max_travel_mm: 1200.00
    soft_limits: false
    homing:
      cycle: 2
      positive_direction: true
      mpos_mm: 0.000
      feed_mm_per_min: 100.000
      seek_mm_per_min: 2000.000
      settle_ms: 500
      seek_scaler: 1.100
      feed_scaler: 1.100

    motor0:
      limit_neg_pin: gpio.35:low
      limit_pos_pin: NO_PIN
      limit_all_pin: NO_PIN
      hard_limits: false
      pulloff_mm: 2.500
      tmc_5160:
        step_pin: gpio.12
        direction_pin: gpio.14:low
        use_enable: true
        cs_pin: gpio.17
        spi_index: 1
        r_sense_ohms: 0.050
        run_amps: 2.470
        hold_amps: 0.250
        microsteps: 16
        toff_disable: 0
        toff_stealthchop: 5
        run_mode: CoolStep
        homing_mode: CoolStep
        stallguard: 16
        stallguard_debug: false
        toff_coolstep: 3
        tpfd: 4

  y:
    steps_per_mm: 320
    max_rate_mm_per_min: 2500.000
    acceleration_mm_per_sec2: 100.000
    max_travel_mm: 1150.000
    soft_limits: false
    homing:
      cycle: 2
      positive_direction: true
      mpos_mm: 0.000
      feed_mm_per_min: 100.000
      seek_mm_per_min: 2000.000
      settle_ms: 500
      seek_scaler: 1.100
      feed_scaler: 1.100

    motor0:
      limit_neg_pin: gpio.34:low
      limit_pos_pin: NO_PIN
      limit_all_pin: NO_PIN
      hard_limits: false
      pulloff_mm: 2.500
      tmc_5160:
        step_pin: gpio.27
        direction_pin: gpio.26
        use_enable: true
        cs_pin: NO_PIN
        spi_index: 2
        r_sense_ohms: 0.050
        run_amps: 2.470
        hold_amps: 0.250
        microsteps: 16
        toff_disable: 0
        toff_stealthchop: 5
        run_mode: CoolStep
        homing_mode: CoolStep
        stallguard: 16
        stallguard_debug: false
        toff_coolstep: 3
        tpfd: 4

    motor1:
      limit_neg_pin: NO_PIN
      limit_pos_pin: NO_PIN
      limit_all_pin: NO_PIN
      hard_limits: false
      pulloff_mm: 2.5000
      tmc_5160:
        step_pin: gpio.33
        direction_pin: gpio.32
        use_enable: true
        cs_pin: NO_PIN
        spi_index: 3
        r_sense_ohms: 0.050
        run_amps: 2.470
        hold_amps: 0.250
        microsteps: 16
        toff_disable: 0
        toff_stealthchop: 5
        run_mode: CoolStep
        homing_mode: CoolStep
        stallguard: 16
        stallguard_debug: false
        toff_coolstep: 3
        tpfd: 4

  z:
    steps_per_mm: 803.213
    max_rate_mm_per_min: 1500.000
    acceleration_mm_per_sec2: 50.000
    max_travel_mm: 160.000
    soft_limits: false
    homing:
      cycle: 1
      positive_direction: true
      mpos_mm: 0.000
      feed_mm_per_min: 100.000
      seek_mm_per_min: 500.000
      settle_ms: 500
      seek_scaler: 1.100
      feed_scaler: 1.100

    motor0:
      limit_neg_pin: gpio.39:low
      limit_pos_pin: NO_PIN
      limit_all_pin: NO_PIN
      hard_limits: false
      pulloff_mm: 2.500
      tmc_5160:
        step_pin: gpio.15
        direction_pin: gpio.2:low
        use_enable: true
        cs_pin: NO_PIN
        spi_index: 4
        r_sense_ohms: 0.050
        run_amps: 2.470
        hold_amps: 0.250
        microsteps: 16
        toff_disable: 0
        toff_stealthchop: 5
        run_mode: CoolStep
        homing_mode: CoolStep
        stallguard: 16
        stallguard_debug: false
        toff_coolstep: 3
        tpfd: 4

spi:
  miso_pin: gpio.19
  mosi_pin: gpio.23
  sck_pin: gpio.18

sdcard:
  card_detect_pin: NO_PIN
  cs_pin: gpio.5
  frequency_hz: 10000000

probe:
  pin: gpio.22:low
  check_mode_start: true

control:
  safety_door_pin: gpio.16:high
#  macro0_pin: gpio.13:low
#  macro1_pin: gpio.0:low

macros:
  startup_line0:
  startup_line1:
  macro0:
  macro1:
  macro2:
  macro3:

start:
  must_home: false

#coolant:
#  mist_pin: gpio.21
#  delay_ms: 0

# For spindle configuration remove the "#" preceding your spindle type BELOW...

Laser:
  pwm_hz: 5000
  output_pin: gpio.4
  enable_pin: NO_PIN
  disable_with_s0: false
  s0_with_disable: true
  tool_num: 0
  speed_map: 0=0.000% 255=100.000%
  off_on_alarm: true
```

### Startup Messages

```text
[MSG:INFO: uart_channel0 created]
[MSG:RST]
[MSG:INFO: FluidNC v3.7.16 https://github.com/bdring/FluidNC]
[MSG:INFO: Compiled with ESP32 SDK:v4.4.4]
[MSG:INFO: Local filesystem type is littlefs]
[MSG:INFO: Configuration file:config.yaml]
[MSG:DBG: Running after-parse tasks]
[MSG:DBG: Checking configuration]
[MSG:INFO: Machine xPro-V5 UltimateBee 1500x1500 ]
[MSG:INFO: Board XPro V5]
[MSG:INFO: SPI SCK:gpio.18 MOSI:gpio.23 MISO:gpio.19]
[MSG:INFO: SD Card cs_pin:gpio.5 detect:NO_PIN freq:10000000]
[MSG:INFO: Stepping:RMT Pulse:4us Dsbl Delay:0us Dir Delay:1us Idle Delay:255ms]
[MSG:INFO: Axis count 3]
[MSG:INFO: Axis X (-1200.000,0.000)]
[MSG:INFO:   Motor0]
[MSG:DBG: TMCStepper Library Ver. 0x703]
[MSG:INFO:     tmc_5160 Step:gpio.12 Dir:gpio.14:low CS:gpio.17 Disable:NO_PIN Index:1 R:0.050]
[MSG:INFO:  X Neg Limit gpio.35:low]
[MSG:INFO: Axis Y (-1150.000,0.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     tmc_5160 Step:gpio.27 Dir:gpio.26 CS:NO_PIN Disable:NO_PIN Index:2 R:0.050]
[MSG:INFO:  Y Neg Limit gpio.34:low]
[MSG:INFO:   Motor1]
[MSG:INFO:     tmc_5160 Step:gpio.33 Dir:gpio.32 CS:NO_PIN Disable:NO_PIN Index:3 R:0.050]
[MSG:INFO: Axis Z (-160.000,0.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     tmc_5160 Step:gpio.15 Dir:gpio.2:low CS:NO_PIN Disable:NO_PIN Index:4 R:0.050]
[MSG:INFO:  Z Neg Limit gpio.39:low]
[MSG:INFO: X Axis driver test passed]
[MSG:DBG: X Axis Coolstep]
[MSG:INFO: Y Axis driver test passed]
[MSG:DBG: Y Axis Coolstep]
[MSG:INFO: Y2 Axis driver test passed]
[MSG:DBG: Y2 Axis Coolstep]
[MSG:INFO: Z Axis driver test passed]
[MSG:DBG: Z Axis Coolstep]
[MSG:INFO: safety_door_pin gpio.16]
[MSG:INFO: Kinematic system: Cartesian]
[MSG:INFO: Laser Ena:NO_PIN Out:gpio.4 Freq:5000Hz Period:8191]
[MSG:INFO: Using spindle Laser]
[MSG:INFO: Probe Pin: gpio.22:low]
[MSG:INFO: Connecting to STA SSID:linksys]
[MSG:INFO: Connecting.]
[MSG:INFO: Connecting..]
[MSG:INFO: Connected - IP is 192.168.13.248]
[MSG:INFO: WiFi on]
[MSG:INFO: Start mDNS with hostname:http://fluidnc.local/]
[MSG:INFO: SSDP Started]
[MSG:INFO: HTTP started on port 80]
[MSG:INFO: Telnet started on port 23]

Grbl 3.7 [FluidNC v3.7.16 (wifi) '$' for help]
[MSG:DBG:  Z Neg Limit 0]
[MSG:DBG:  X Neg Limit 0]
[MSG:DBG:  Y Neg Limit 0]
[MSG:DBG: safety_door_pin 0]
```

### User Interface Software

WebUI, FluidTerm, CNCjs

### What happened?

My CNC router with Huanyang VFD 2.2KW spindle works well.
Now I want to install a laser on it. LT-K40 LaserTree only has PWM and GND pins.
I connected everything as described in the Spark-Concepts wiki:
- PWM via a relay to the laser board
- GND - to the laser board
- the switcher to “EN/PWM”
- the jumper to “Relay SP EN”
- changed "Spindle Section" in the configuration file from Huanyang to Laser (as on FluidNC wiki)
- rebooted the controller
- gave $32=1

As a result I get:

- The relay is not activated.
- The PWM output is always 4.96V.

Since the Spark-Concepts wiki doesn't have any indication of which firmware version their method is, so after no success I tried that too.

$Spindle/Type=Laser
$Gcode/LaserMode=ON
$31 = 1

I sent the first two commands to the console in CNCjs. They cause ERROR:3
As far as I understand, they are not supported.

If I leave the configuration file unchanged with the Huanyang settings, after rebooting XPRO v5 turns on the relay at startup and also sets the PWM to 4.96 V.

In both cases the PWM is at 4.96 V.
The only difference is whether the relay is activated or not.

Please, help me understand what is wrong.

MitchBradley, 2024-04-20:
You don't need the $ commands.  The presence of a laser: section in the config file implies laser mode and laser spindle type. 

I think you need to use the pin definitions given in the commented-out "pwm" section in https://github.com/Spark-Concepts/xPro-V5/blob/main/FluidNC/config.yaml , with the section name "laser".  Apparently gpio25 is the PWM output pin and gpio4 is the enable pin.  You have gpio4 as the pwm output and nothing as enable.

slvv26, 2024-04-21:
Thank you very much for your quick response.
A simple solution and it work. I assumed so, but “enable” confused me and misled me.
Apparently “enable” here means activating the entire TOOLHEAD-port.
And this gets confused with the “enable” PIN in the connector itself. Which is indirectly confirmed by the settings from Spark-Concepts.

I guess it will be useful to have description about it in the wiki.

Now my settings for xPro-V5:

_Laser:
  pwm_hz: 5000
  output_pin: gpio.25
  enable_pin: gpio.4
  disable_with_s0: false
  s0_with_disable: true
  tool_num: 0
  speed_map: 0=0.000% 255=100.000%
  off_on_alarm: true_

Thanks a lot again.

MitchBradley, 2024-04-21:
> I guess it will be useful to have description about it in the wiki.

You are welcome to edit the wiki if you think it will help others.  You could also take up the matter with Spark Concepts, who, unlike us, received money for your purchase of their product.

slvv26, 2024-04-23:
When registering on the wiki, it gives an error with SSL.

140709849348912:error:1408F10B:SSL routines:ssl3_get_record:wrong version number:../deps/openssl/openssl/ssl/record/ssl3_record.c:331:

Sorry to hear about Spark. This is not entirely fair, given that they abandoned their application and directly indicate the use of FluidNC. Some time ago, it was common practice to support free software developers by the companies that develop the hardware that uses them.

bdring, 2024-04-23:
http://wiki.fluidnc.com/en/wiki_contributions

What user name did you register with?

slvv26, 2024-04-23:
***@***.***

closed 2024-04-25 by bdring

Kebirom, 2024-11-26:
Good morning,
I use my cnc with xpro v5 with problem in rs485 and the VFD.
However I cannot use the PWM for using a laser
I have installed the firmware for Laser
Config Grbl Ok
S32 a 1
S31= 1 (or 0)
Spindle/type = laser
Slaser = ON

But I always have 5 volts between Pwm and Gnd even when varying the speed (M3S500 for example)

I think it's a card problem...
Can you give me a firmware that works in Pmw laser for you and help me find a solution?
Thank you for any possible help

MitchBradley, 2024-11-26:
Start a new issue. This one is closed

Kebirom, 2024-11-27:
> Commencez un nouveau sujet. Celui-ci est fermé

Sorry thanks